cyring/CoreFreq #129: Does not build on Ubuntu 18.04 LTS 64-bit with latest kernel

olejon, 2019-06-22:
* `make` fails on `Linux 4.15.0-52-generic`, current latest kernel for Ubuntu 18.04 LTS 64-bit
* Straight forward SkyLake Workstation CPU: `Intel(R) Core(TM) i5-6600K CPU @ 3.50GHz`
* Worked like a charm first time I installed CoreFreq
* Logically did not work after next kernel update, since CF kernel module it isn't compiled for it
* Compiled again and worked fine again
* But now with `Linux 4.15.0-52-generic`, it fails with the error below
* Installed `libelf-dev` since `make` suggested to do so, and *that warning* goes away, but still build fails
* Has not bothered with adding it to `dkms` since `make` takes like 5 seconds on my system (seems like that was a good thing not to do)
* Overclocked simply enabling Intel XMP in UEFI, which enables all cores to go to max TurboBoost (temperature fine) + automatically detects my XMP compatible 3200 MHz DDR4 RAM, and such sets correct values

```
$ make
...
make -j1 -C /lib/modules/4.15.0-52-generic/build M=/home/username/Downloads/CoreFreq modules
make[1]: Entering directory '/usr/src/linux-headers-4.15.0-52-generic'
  CC [M]  /home/username/Downloads/CoreFreq/corefreqk.o
/home/username/Downloads/CoreFreq/corefreqk.c: In function ‘SMBIOS_Collect’:
/home/username/Downloads/CoreFreq/corefreqk.c:9107:5: error: ‘DMI_PRODUCT_SKU’ undeclared (first use in this function); did you mean ‘DMI_PRODUCT_UUID’?
   { DMI_PRODUCT_SKU, Proc->SMB.Product.SKU },
     ^~~~~~~~~~~~~~~
     DMI_PRODUCT_UUID
/home/username/Downloads/CoreFreq/corefreqk.c:9107:5: note: each undeclared identifier is reported only once for each function it appears in
scripts/Makefile.build:337: recipe for target '/home/username/Downloads/CoreFreq/corefreqk.o' failed
make[2]: *** [/home/username/Downloads/CoreFreq/corefreqk.o] Error 1
Makefile:1552: recipe for target '_module_/home/username/Downloads/CoreFreq' failed
make[1]: *** [_module_/home/username/Downloads/CoreFreq] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-52-generic'
Makefile:70: recipe for target 'all' failed
make: *** [all] Error 2
```

Any ideas, or is it simply the code? Haven't installed or changed anything that should have anything to do with building from source code.

olejon, 2019-06-22:
BTW: `intel-microcode` package was also updated recently.

cyring, 2019-06-22:
Will fix asap.
Some dmi might be available with latest kernel 5.x

olejon, 2019-06-22:
Great :-)

cyring, 2019-06-22:
Just checked (w/ Smartphone) the minimum Kernel version is 4.18.x for those missing definitions (DMI_PRODUCT_SKU,...)

I will refactor code using conditions to let previous versions build through.

Also thinking about writing myself parts of the DMI parsing to offer the remaining strings

olejon, 2019-06-22:
I usually install the Ubuntu LTS Enablement Stacks once they are out (newer kernel and Xorg/Wayland/Mesa/etc for desktops and just newer kernel for servers).

Pretty sure the one's in the repo now is > 4.18.

https://wiki.ubuntu.com/Kernel/LTSEnablementStack

> The 18.04.2 and newer point releases will ship with an updated kernel and X stack by default for the desktop. Server installations will default to the GA kernel and provide the enablement kernel as optional. The 18.04 HWE Stacks will follow a new Rolling Update Model as documented at the following location: https://wiki.ubuntu.com/Kernel/RollingLTSEnablementStack Installing the HWE stack is simple: DESKTOP sudo apt-get install --install-recommends linux-generic-hwe-18.04 xserver-xorg-hwe-18.04

olejon, 2019-06-22:
BTW will install the above and check if I get to it before you do :) Don't worry, no rush for me.

Just gotta say CoreFreq, although praised in the article I found it in, was way above my expectations! You should try to get it into the Debian repos!

cyring, 2019-06-22:
Hello, for your tests, fix is committed.

olejon, 2019-06-22:
Thanks for responding so fast and fixing. I will test soon, with the kernel mentioned in the issue.

olejon, 2019-06-23:
Confirmed building as before and working on `Linux 4.15.0-52-generic` without installing the LTS Enablement Stacks.

* *Ubuntu 18.04 LTS 64-bit with latest default kernel and `intel-microcode` package*

Thanks again!

I have a suggestion. Basically deactivating NMI Watchdog - *it is there for a reason* - only when CF is running, not permanently.

I have created a script like this as `/usr/local/bin/cf`:

```
#!/bin/bash

### CF SCRIPT IS RUN

bash -c "echo 0 > /proc/sys/kernel/nmi_watchdog"

sleep 1

insmod /opt/corefreq/corefreqk.ko

sleep 1

/opt/corefreq/corefreqd &

sleep 2

su -l -c "/opt/corefreq/corefreq-cli" MYUSERNAME

### CTRL+C IS PRESSED TO STOP THE CLI

sleep 4

rmmod corefreqk.ko

sleep 1

bash -c "echo 1 > /proc/sys/kernel/nmi_watchdog"
```

The sleeps are necessary to give some time before starting and stopping. I have added a little extra because you never know. Sometimes it may fail to `rmmod` the module if only 2 seconds after quitting the cli, for instance.

Clue is as you can see to modify NMI Watchdog on-demand, like:

When starting, first run:
`bash -c "echo 0 > /proc/sys/kernel/nmi_watchdog"`

When everything is stopped/unloaded, run as last command:
`bash -c "echo 1 > /proc/sys/kernel/nmi_watchdog"`

I have added `/usr/local/bin/cf` to the whitelist using `visudo` so it does not require a password to run. I simply type `sudo cf` to run it.

Pretty sure you can do this in `corefreqd` on start/stop?

At least it can be an option :-)

cyring, 2019-06-23:
Thanks, I will process your returns.  
Meanwhile, nmi_watchdog is not the source of the incompatibility issue; it just rewrites periodically the fixed counter register. And that biases the readings. Evidence can be found in the Kernel source code.  

As a workaround you can also build CoreFreq with  directives to use other registers: APERF and MPERF  
Just type `make help`

You can then keep the nmi_watchdog while `corefreqk.ko` reads `MSR_IA32_APERF`  and `MSR_IA32_MPERF`

olejon, 2019-06-23:
Thanks. Just adding a little information about this to `README.md` would seem enough. It's a tool for power users anyway. Maybe a link to what the NMI Watchdog does. You can maybe add my script to the repo as an example if you want to. My solution works just fine so I won't touch the build process. I don't use CF *that often*. It does replace a lot of other BASH aliases I've created for frequencies and temperature, updated every X seconds. I use them still to verify (frequencies polled from the kernel and temperature from `sensors`). Also having the GUI PSensor program is nice. Lots of flexibility. Uses lm-sensors, AMD/NVIDIA APIs if found and udisks or udev for disks temperature, no need for the hd monitor daemon package. One nice thing is that you can set alarms if a certain part, say the CPU package or individual core goes above a certain temperature. A secure way to avoid parts being destroyed if HW and kernel modules built in protection fails and a heavy process has freezed and causes 100 % usage, especially if over clocked and basic protection like voltage regulator is disabled in UEFI. That heats up the CPU extremely fast under `stress -c numcores`. Just set command to run, if above X °C, run `bash -c reboot` for instance. Could be something for this tool, the daemon at least. Well AFAIK CF doesn't write any configuration, so would need that. `htop` does. Nice since I always change the settings. Basically the only basic monitoring tool CF can't replace, but CF isn't really meant to either I suppose.

BTW I can probably just fine host CF as a deb in my repo (only for packages that never are in the Debian/Ubuntu repos). Build it from latest source, (tested by me first on Debian and Ubuntu) on first install and add to dkms, or simply ask user to `apt install --reinstall corefreq` after kernel update. I saw a PPA it seems you made but it's totally out of date, and no link to it from here. But could use it as a base for my deb. Of course only if you're interested, like having a link to an up to date CF repo for debs.

cyring, 2019-06-23:
Getting your support for distributions repo will be a great help. Indeed I want to focus on R&D.

_CoreFreq_ is monitoring thermal thresholds, those which are Processor  internal such as thermal throatling. For instance if temp reaches tjmax, `HOT` will light on. Press the `H` key to clear the associated event register bit.

I don't think about programming what other monitoring software are doing much better. The goal of _CoreFreq_ is more to offer features which are not pretty **centralized** in the Linux space: Processor + BIOS + OS + Performance 
Although I'm noticing that _CoreFreq_ is used in areas I was not expecting to see it: Trading, Production,...; _CoreFreq_ remains an experimental open source subject and I have tons of idea not developed yet.

cyring, 2019-06-24:
1. In your script I don't see when the daemon is requested to close. It may explain why the driver has shutdown issue. Shared memory is not cleanly released. 
 
Please give a look into `corefreqd.service` : the daemon is listening to signals.
Daemon is then sending a shutdown signal to the client ==> It's a cascade procedure. 

2. Disabling the nmi_watchdog from the user-space won't guarantee the counter register is freed by kernel. That's why I recommend to blacklist the counter usage straight from the Kernel boot command line. However, as stated previously, you can build _CoreFreq_ with APERF/MPERF counters and let nmi_watchdog runs.

3. Give a look into the other monitoring GUI tools documentation and source code for hardware probing incompatibilities. I have started a list of _CoreFreq_ conflicts into the Wiki - Software incompatibility

Regards
Cyril

olejon, 2019-06-24:
First of all, with this script, the daemon, for some reason, shuts itself down like immediately after pressing Ctrl+C, BUT only when the commands are run in a script like that, *even when the CLI program is run as the desktop user*.

HOWEVER, when the commands are run separately one by one, *all commands as root except the CLI program which is run by the desktop user either using `su -l -c` or simply in a desktop terminal*, it does NOT shut itself down. Interesting. But run as this the daemon always shuts itself down, otherwise it refuses to `rmmod` the module, but that *never* happens when run as a script, only when the commands are run one by one in 2 terminals, one for the root commands and one for the CLI program.

Easy to fix to be 100 % sure. Before `rmmod`, add command:

`pkill corefreqd` (pkill does not give any output if the wanted-to-kill command is not running either so no need to send output to `/dev/null` or anything).

**This I discovered and have now confirmed** by running `ps aux | grep coref` and also `lsmod`, which gives that *both the CLI program and the daemon is not running anymore when closing the CLI using Ctrl+C, NONE of them*, SO fast that I can not even switch to the other terminal using Alt+Tab to run `ps` and see it - it is already shut down. The module as per the script not unloaded before 4 seconds, *so it is a "safety buffer" to 100 % avoid any error*.

I can not explain why, but that is why I have not added a command to shut down the daemon because it does it itself on my system. The sleep time of 4 seconds was because a few times just got the message that the module was still in use with a sleep time of just 1 second for instance. But 4 seconds is a sure "safety buffer", more than enough seconds, just added to NEVER ever get that the module is still in use.

**Now my script is like this and seems to work just fine:**

```
#!/bin/bash

bash -c "echo 0 > /proc/sys/kernel/nmi_watchdog"

insmod /opt/corefreq/corefreqk.ko

/opt/corefreq/corefreqd &

sleep 2

su -l -c "/opt/corefreq/corefreq-cli" MYDESKTOPUSERNAME

pkill corefreqd

sleep 2

rmmod corefreqk.ko

bash -c "echo 1 > /proc/sys/kernel/nmi_watchdog"
```

Both `ps` and `lsmod` shows everything is killed and unloaded fine. No errors from kernel or terminal.

Output from terminal when run:

```
CoreFreq Daemon 1.57.0  Copyright (C) 2015-2019 CYRIL INGENIERIE
 ...terminated.
```

Output from dmesg:

```
CoreFreq(2:-1): Processor [ 06_5E] Architecture [Skylake/S] CPU [4/4]
CoreFreq: Unload
NMI watchdog: Enabled [again]. Permanently consumes one hw-PMU counter.
```

**Note 1:** Kernel does not report when NMI Watchdog is disabled, only when enabled, but can be verified disabled by checking value of `/proc/sys/kernel/nmi_watchdog` of course, which is also 0 when using the kernel parameter.

**Note 2:** As you can see I have removed some unnecessary sleep commands (on my system at least), and shortened sleep time from CLI exit to 2 seconds. Now, this should be tested on a slow system (in every way), *but since CoreFreq is not made for VMs, it is hard for me to check*. Basically having both more sleep commands and sleep seconds makes sure it should work on slow systems as well.

Does it work correctly on 32-bit? If so I may test on a really old laptop... Maybe CoreFreq does not support that CPU even, like not enough "info and toggles" in the kernel for CoreFreq to use, or CoreFreq is aimed at newer CPUs than that one.

**Regarding NMI Watchdog:** I was not aware of that, hmm... Many tutorials, blog posts, wikis etc uses on-demand toggle, or simply disabling NMI Watchdog in sysctl or something that makes it reversable without a reboot, not as a kernel boot parameter. In MY case I do not see any difference if it is enabled or disabled. I will double check. Especially interesting then to compare CoreFreq values when NMI Watchdog is disabled as a kernel boot parameter compared to *not disabled in any way*.

**Regarding deb package:** Have a pretty good idea. Instead of in a repo, you simply host it here on GitHub.

* Basic `control` file, we only need 4 small dependencies on Debian/Ubuntu derivatives
* The package only puts 2 files into right place (script to run CoreFreq in `/usr/bin` or `/usr/sbin`, and the copyright file standard for debs)
* The `postinst` script does all the magic, *including acting as an upgrade script if CoreFreq already found to be installed* (easy check), cloning latest from GitHub and compiling into suitable folder like `/opt/corefreq`
* If package is deleted, the `prerm` script simply deletes `/opt/corefreq`. The 2 other files are automatically deleted as they are part of the package.
* The script to run CoreFreq will check if NMI Watchdog is enabled, and if so, give a warning and a link to https://github.com/cyring/CoreFreq
* It will also check if the kernel has changed since last run, give a warning, and offer [Y/n] to download the deb from this repo and reinstall it, and such it will compile it again against the new kernel headers/source

**Note:** This is EXACTLY what the VirtualBox guest additions install script for VMs under VirtualBox does, puts files in `/opt`, compiles guest modules, I think now adds them to DKMS, but when a new version of the guest additions are available, which it is always when a new version of VirtualBox is out, and you must reinstall the guest additions. I always use the CLI way on Linux VMs, run the script directly as root and reboot. IDK if running the software you are asked to run under certain Desktop Environments, when inserting the virtual CD with the guest additions, actually work, and if so, I am sure it just opens the default terminal and runs the script, not like GUI "Next > Next > Compiling drivers > Windows reacts to new drivers > Allow > Next > Reboot" on Windows, or something in that standard order. I do not use VirtualBox anymore in Linux, use KVM+QEMU+libvirt, which is way better of course. Just on macOS.

olejon, 2019-06-24:
EDIT: Some changes, typos, formatting etc.

olejon, 2019-06-25:
* My bad, regarding askin about 32-bit, didn't even read the main description - that it is made for 64-bit CPUs.

* I have now ran `make` so that NMI Watchdog can be on. At a glance it seems to show the same information as before. "SMBIOS data" shows "To be filled by O.E.M." but sure it was like that before, and if I understood correctly could show the info if kernel is >= 4.18, which can be done just by installing LTS Enablement Stacks (one command).

Q: That data does not really matter for me anyway, *but does it make CoreFreq more accurate if available?*

cyring, 2019-06-26:
SMBIOS data is just informative in _CoreFreq_ 
It's a work in progress as the main idea is to backport some code of my previous project CoreMod (see my repo)
Because DMI data are not fully fulfilled by motherboard manufacturers, I don't rely on them to compute the various indicators. 
As I understand you have a Ryzen processor and I still don't produce DRAM information yet: search of the registers, specifications, and so on

olejon, 2019-06-26:
Ok. No, as written in first post:

* Straight forward unlocked SkyLake Workstation CPU: *Intel Core i5-6600K*

cyring, 2019-06-26:
> * Straight forward unlocked SkyLake Workstation CPU: _Intel Core i5-6600K_

Great, have you been able to overclock or downclock your i5-6600K as you wish ?  

Can you also post screenshots showing at least : 
* When stressing a single Core (press [F3])
* With the Processor window (from OSPM to TDP ratios) (press [p])
* With the measured Vcore (press [V])
* Two cases: idle and high load to show : C-States (view [f]) and Package C-States (view [g]) 
* A complete text of output `corefreq-cli -s -m -M -k -B`

Can you please post these screenshots and outputs for a new section in [CPU support](https://github.com/cyring/CoreFreq/wiki/CPU-support)

Thanks for helping _CoreFreq_
Cyril

olejon, 2019-07-01:
Yes I can do that for you, and a MacBook Air Mid-Summer 2013 13" top specced CPU and RAM, running Fedora 30 and CoreFreq works.

**BUT**, to do what you ask, I need your help on something I wanted to ask you *for your expertise* anyway. I am a pro-user of Linux, *but do not have the deep understanding of CPUs + Linux Kernel as you do*. I will divide into 2 sections, one for "Strange Phenomenon", and one for "What I have Done and Tried".

**Strange Phenomenon:**

* Yes, I can basically overclock my CPU as I want to. UEFI allows it and CPU allows it.
* Activating `Intel XMP` activates an overclock that always works (CPU & RAM).
* When I overclock the CPU even more, the system, after a totally random time, decides to revert my overclock to standard `Intel XMP`, *even though Intel XMP is disabled in UEFI*...
* For example this CPU is 3.5 GHz with a Turbo Boost to 3.9 GHz, that of course goes for only 1/4 cores at a time. Stressing 2 cores gives then the half and so on.
* When `Intel XMP` is activated, it sets "Sync All Cores", meaning all cores can Turbo Boost at the same time, and have no problem keeping it up constantly when running `stress -c 4`, *because the temperature is well below 80 Celsius, actually it keeps itself around 60-69, way below what it can do before CPU throttling happens, which happens on Intel CPUs when reaching 100 Celsius, seen very well using CoreFreq on my MacBook Air, it throttles immediately when 1 core reaches 100 Celsius, which it does super fast just maxing out 1 core with `stress -c 1`.
* I can without any problems run for instance in UEFI the ASUS tool `OC Tuner II`, which checks my cooling and suggests a max overclock. For me it sets "Sync All Cores" at 44, meaning all cores can Turbo Boost to 4.4 GHz, and this works fine *with temperatures between 75-83 Celsius* (checked with `sensors` and CoreFreq etc), WELL within safe zone, nowhere near 100 Celsius that should cause throttling, but I have never managed to reach 100 Celsius, even with the CPU SVID disabled...
* Having `Intel XMP` off or on with manual overclock, same result, on then means it just adjusts UEFI automatically to my correct RAM Voltage and MHz, 1.35 V and 3200 MHz DDR4, it does not adjust the CPU settings
* **BIG PROBLEM:** Until after a random time, cane be 1-3 hours, AND system can have been *totally idle, regular use or stress, does NOT matter, no pattern found*, **the system resets my manual overclock ratio**, even when `Intel XMP` is not enabled, **it resets to** `Intel XMP` "standards", meaning **STILL CPU overclocked** for ALL cores to Turbo Boost to 3.9 GHz, but no more to 4.4 GHz, Sync All Cores to Value = 44, or whatever much lower value I choose manually, for example it also happens it set to Value = 41 (4.1 GHz).
* **Only solution:** REBOOT, that simple, works again immediately, no shutdown needed first. No trace anywhere of why (logs, kernel, tools).
* Tools like CoreFreq, `turbostat`, `i7z` etc all confirms this, also looking in `/sys` and `/proc` values, they all show (tools as well) **everywhere** that Max Freqency is 4.4 GHz, it just does not get there, **but stays dead stable on 3.9 GHz**, with temperatures so low it does not makes sense for any system or BIOS component to do that (stays at 60-70 Celsius, fans barely make noise).
* CoreFreq for instance shows in the Frequency view on the X line that max is 44, but usage goes only to 39...

Why the heck does this happen after a random time when temperatures are all good, and the same, according to all reliable tools available?

**What I have Done and Tried:**

* Uninstalled `thermald` of course. Stupid thing to have on a PC without a battery anyway. CPUs and the kernel has *plenty* of safeguards against overheating, like my MacBook throttling all cores immediately when just 1 core reaches 100 Celsius. AFAIK Intel allows 1 CPU core to stay a little while at 100 or over, and the "relax it a little bit extra" and send the job to another core, but this happens so quickly one can not see it without changing the time in UEFI.
* **ABSOLUTELY NOTHING shows any change when this happens**, like logs, `dmesg` and no values in `/sys` or `/proc` related to the CPU changes when this happens. It still says it can go to 4.4 GHz (or whatever manually set ratio).
* Again, temperatures fine and system totally stable at both 4.4 GHz, which is the max I have tried (under full stress and idle), according to CpuFreq, `turbostat` and my alias `freq` which runs: `watch -n 2 cat /sys/devices/system/cpu/cpu*/cpufreq/scaling_cur_freq`, can run for a longer time than I have bothered at 4.4 GHz, never throttling
* So it is NOT throttling, but rather *"underclocking from overclocking to Intel XMP standards, even it disabled or enabled"*, **WHICH LEAVES IT STILL OVERCLOCKED**, but obviously every time at the Intel XMP standards (enabling Intel XMP is overclocking and gives a warning that enough cooling is required, but as said enabling just that which leaves "Sync All Cores" = "Auto" instead of a ratio number, means 3.9 GHz Turbo Boost on all cores.
* Tried basically any UEFI setting that can have to do with this. Only turning off CPU SVID (Voltage Regulator), which basically the UEFI recommends if overclocking, *but does not turn off nor ask when choosing to overclock*, makes it heat if very, very fast on even just value 41.
[... cut ...]
* BTW: I have seen some toggles in CpuFreq, those between `< toggle >` and things like "Hardware-Controlled Performance States" makes me interested, it says ON. I guess these are C-states? Anyway, maybe there is a toggle you can recommend to try in CpuFreq, that simple?

**Remember:** The thing is to reproduce this I must reboot, which immediately makes manual overclock to ANY ratio work again, then basically just wait and do a test now and then, using `stress -c 4` and checking one of `turbostat` my alias `freq` or CpuFreq. They ALL show the same when it is working and later when underclocked **without trace or reason**.

cyring, 2019-07-02:
Many things to read in your post !

First I don't have an XMP to face your issue but I'll experiment as soon as I put my hands on it...
Meanwhile what you describe can be linked to the HWP management.

If my Cli says HWP is ON then change its profile to low (zero value) and stress a single Core again (press F3)

In Processor window, try to play with HWP min, HWP max and HWP target.  
Intel says to set Min=Max=Highest HWP 

HWP can be enabled once, it may explain why you are recovering after reboot

Til HWP is disabled, the legacy OSPM, its profile (from 0 to 15), and its target ratio are in effect.

olejon, 2019-07-02:
I have now disabled intel_pstate completely at boot as a kernel parameter. One can also just disable HWP. And set that the CPU can not go below C1. If that works, then I'll play more with, if just say disabling HWP in the intel_pstate module is enough, or that + active vs passive mode (latter requires HWP off as well), basically play with intel_pstate kernel parameters. I don't want it completely disabled if not necessary. It has its benefits (like now there's no info from it in /sys)

Because now it uses acpifreq as cpufreq driver (IIRC, I'm on my phone, but another driver), not Intel cpufreq, hence the "old standard governors". 

intel_idle is set to disabled (confirmed by kernel at boot). This enables all usual governors, like it now defaults to "ondemand", while a modern Intel CPU system has just "powersave" and "performance", and will default to "powersave", which basically works as ondemand but actually better as I can read, more optimized, at least where power is a concern. Only other option by default is "performance". I doubt changing to that will help. Must be changed for each core as well. There is another thing when intel_pstate is enabled regarding "energy efficiency". Those toggles are not available for me now with intel_pstate disabled.

Keeping C-state to never go below 1 is no concern for me. Yes, it uses more power, but it's not a laptop and Norway is 100 % hydro so... speeds up the system response as going from deeper C to C0 takes more time (it's actually noticable). Pages for low latency servers, like Red Hat docs etc recommend it. It does allow the CPU to idle at minimum speed as well (800 MHz), although somewhat more likely to spike to ~1,000 MHz+, probably the less optimized ondemand governor. Doesn't matter for me though.

cyring, 2019-07-02:
Please see my previous answer, I've edited the HWP instructions

olejon, 2019-07-02:
BTW: Don't bother with Intel XMP. It's just a "Intel certified overclock and detect faster than standard DDR4 RAM chips" as far as I understand it. My RAM chips, which supports 3200 MHz, way above the default DDR4, says "Intel XMP certified", which was what led me to enable XMP in UEFI in the first place. Nice way to see the UEFI then adjusting voltage and MHz to exactly what it says on the chips themselves. What was unknown to me was that it overclocks by default the CPU as well (with the warning about sufficient cooling it's logical), but as said this set Sync All Cores = Auto. So basically leaves to the UEFI and what it knows about the CPU I guess to set a supported overclock, which at least in my case defaults to that all cores can Turbo Boost instead of just one, to the max advertised TB speed

olejon, 2019-07-02:
Will do after seeing if this "total disabling" works. See my post above regarding XMP.

Anyway if it's HWP, as said I'll just disable it as one can with intel_pstate kernel parameter. Although cool to play with at runtime, according to the Intel docs in the kernel docs, if HWP is enabled at boot it can not be fully "played with" at runtime. But I'll check, would be cool if so, and to know what kernel toggles it CpuFreq uses then.

cyring, 2019-07-02:
HWP is a processor feature that your UEFI BIOS can leave it disabled. Be sure no other driver will enable this register afterwards. Then, the Processor is in OSPM where you can set the BIAS HINT for Energy Policy and the Target P-State for desired frequency.  

Next:
* Enable SpeedStep EIST  
* Enable Turbo  
* Set BIAS HINT  
* Set the highest Target  
* Set the Turbo ratios 
* Activate C-States  
* Disable Clock Modulation 

Test:
* Stress a single CPU 
* Monitor CPU frequency and ratio for Turbo values 
* All other CPUs are in idle C-States 
* Monitor Vcore for highest voltage 
* Prefer a simple Linux console and lightweight GUI

_CoreFreq_ brings you all these features

olejon, 2019-07-02:
Hooray! After an uptime of over 8 hours, running `stress -c 4` (still the tool that best stresses my cores to the max), shows that it gets to 4.4 GHz without any problems!

I'm pretty sure there's no setting for HWP in UEFI, (which on a decent ASUS MB has a lot of settings, all you need, in "Advanced Mode"), although it seems like it should be a setting, and if it's actually there *it's so hiding under a weird name with a bad description at a weird place in the settings*.

As with C-states that's no problem, Linux would surely ignore the UEFI setting there as well. Since `intel_pstate` has a parameter for exactly just that, disabling HWP, no problem then...

I've, as I would anyway, gone through all UEFI Advanced options, and the of the ones having to do with the CPU, which are in various places, no one mentions HWP in the name of description. I would probably recognize something similar (different name for it), but can't think of any.

SpeedStep and Turbo is if course on. As long as HWP is set to off as a kernel parameter, I doubt the rest you write is necessary at all. I read the docs for intel_pstate and yes it uses BIAS hints etc (very complicated docs), but as I understand it these are just my OC settings/values read from what set in UEFI/BIOS anyway (where it is of course set more intuitive with logical names and good descriptions).

With "Sync All Cores" one can't set "Turbo ratios" in plural, just one that goes for all cores. Everything above 35 multiplier by default become Turbo, as it should since default core frequency without is 3.5 GHz (35).

So, *if running totally stock Intel Turbo Boost, a tool like `turbostat`, shows it's in 39/38/37/36 multipliers mode*, meaning if all cores are stressed it can go to 3.6 GHz max, and if just 1 core is stressed it can go to 3.9 GHz, which is what Intel etc advertise as their "Turbo Boost speed", although it's really only possible for a single thread at a time. In real life use, unless doing single threaded work, since most applications take advantage of all the cores, these turbo spikes are shared between cores, so no core never reach full Turbo, but they "share between them the extra 0.4 GHz".

This of course all changes when Sync All Cores is set, or Per Core is set and the multiplier the same for all, or there I can set say 44/43/42/41 instead. But syncing all cores to 44 (same as 44/44/44/44), is of course better when the CPU can handle it with the cooling I have, which is nothing spectacular, but nicely designed big tower chassis + the pretty massive CoolerMaster heatsink+ fan that came with the CPU.

There is no Intel stock cooling solution for i5 and i7 for this CPU at least, just the i3 model, which I have laying around, both the CPU and the Intel stock cooling solution. It is a 2-core with hyperthreading to make it seem like 4. It runs by default at a higher clock speed as well, but that is normal for an i3 compared to an i5 or even i7 some times. As you know hyperthreading heats up the CPU a lot, but no problems when only 2 cores. There's a bigger change if one goes from an i5 typically without hyperthreading to an i7, usually not more cores, and just a little higher max speed and max turbo speed, but it also has hyperthreading so say for such a family it would have 4 cores + 4 HT, and those heat up pretty quickly if say running `stress -c 8` while that does not stress an i5 without HT any more than when running the same command with `-c 4`.

These multipliers I can change individually, which is "Per Core" (default for Stock Default Turbo Boost also) instead of "Sync All Cores" (which I prefer when it can OC so much without any problems.

Sync All Cores mean greys out the other 3 values and when I set 44 as the (first) multiplier for instance, they all change graphically in UEFI, and a tool like `turbostat` confirms on start that all are on 44 multiplier.

 where they can all be the same value or the first value is max, then the next ones must be either the same or a lower than the previous going down. For example the ASUS built in OC Tuners (I mentioned OC Tuner II suggesting Sync All Cores = 44 multiplier), the OC Tuner I suggests a "safer" overclock, with typically 43/42/41/41.

In real life usage one wouldn't notice that, the TB by default is shared among the cores, and when necessary one or more can spike. 

But maybe it's not there since just as with C-states, the OS can ignore the BIOS on that, which Linux does, always. Several pages says, and they are correct, that for example disabling C-states or setting the CPU to not be able to go deeper than C1 *will not work for Linux* when set in UEFI, although all the settings are there. They must using kernel the different parameters for `intel_idle` (and possibly some for `intel_cstate`, which is a module, blacklisting it does not make Linux honor UEFI settings since it's really up to `intel_idle` parameters) which do just the same.

In UEFI I can fine tune to things like disable a specific C-states, say C6, in UEFI, or C-states support all together, or max deep level, but as said the Linux kernel ignores them Windows folow them. According to one page *"Changing C-state settings in BIOS/UEFI only has effect on Windows"*, which is exactly what I'm seeing, not that it's a problem. Even changing all the C-state settings in UEFI to "do something ridiculous" (something that if they take effect it should be clearly visible in tools and even noticable in usage). But yeah, page is right. Linux does not honor these, but according to the page Windows does.

NO PROBLEM again since these for Linux can be adjusted in/by the kernel.

Well how I want my C-states are up to my no? I like it staying at C1 when idle TBH. Everything even snappier when I thought it couldn't become. Booted already in 2 seconds etc, but it is noticable in tiny operations when otherwise the CPU would have to go all the way from C7 to C0.

**What do you mean by disabling clock modulation?**

cyring, 2019-07-02:
> What do you mean by disabling clock modulation?

This ODCM setting: if enabled in conjunction with DutyCycle percent will reduce performances
[image]

Btw: do you really want to help _CoreFreq_ ?

olejon, 2019-07-02:
Yes I do. You asked for if I can overclock my CPU freely, and for screenshots, and now I can give you such screenshots, after finding the sinner, HWP. IDK why I should touch Clock Modulation if overclock works as it should? On my phone but will check what it says.

I also want to make that deb package for you. Should be pretty easy. I'm used to packaging those + Arch Linux packages (but that you already have in AUR).

But year first I needed some help here around intel_pstate, and although it's completely disabled now I'm pretty sure I can enable it again, also intel_idle and default C-states, and it will keep my overclock. That will be my next test. Well actually first removing all kernel parameters and see if using CoreFreq to disable HWP at runtime, even though enabled at boot, fixes it, which it seemed you meant it should be able to? Or did I not understand that one? At least it can be toggled there so... Kind of contradicts what I've read in the docs, but might have misunderstood them.

olejon, 2019-07-02:
So to add, thanks for explanations! Why this is not on any page, and googling "right", which I'm pretty good at, usually finding an answer with the first search, although technical (my previous Googlers over at Silicon Valley would laugh if I wasn't :P).

The point is, *I can't be the only one experiencing this on such a popular motherboard and CPU*, and no search gives that any other sees the same and a solution to it. Weird! Maybe because it happens in silence and people don't check their actual CPU performance after it has been checked after boot, since this occurs X hours after without any trace/hints in logs, kernel etc

olejon, 2019-07-02:
> This ODCM setting: if enabled in conjunction with DutyCycle percent will reduce performances
> [image]

* `Clock Modulation` already says: `ODCM <Disable>`. IDK if because of `intel_pstate` being disabled, or if that is a result of a UEFI setting. Will revert back to defaults, no kernel parameters, meaning enabling `intel_pstate` and `intel_idle` and all C-states available, and see what the relevant values then are.
* Under "Performance Monitoring", `Hardware-Controlled Performance States` now says: `HWP <OFF>`. This I know said `<ON>` before so disabling intel_pstate sure did that.

Of course going back to defaults will cause the problem again, but will see if can be fixed just using the toggles in CpuFreq. If not, simply see if it is enough to boot with kernel parameter `intel_pstate=no_hwp`.

cyring, 2019-07-02:
Hope these settings to reach Turbo (even w/ HWP) will help.
[image]
```
Processor                              [Intel(R) Core(TM) i7-6700 CPU @ 3.40GHz]
|- Architecture                                                      [Skylake/S]
|- Vendor ID                                                      [GenuineIntel]
|- Microcode                                                        [       186]
|- Signature                                                            [ 06_5E]
|- Stepping                                                             [     3]
|- Online CPU                                                           [  8/8 ]
|- Base Clock                                                           [100.25]
|- Frequency            (MHz)                      Ratio                        
                 Min    802.00                    [   8 ]                       
                 Max   3408.50                    [  34 ]                       
|- Factory                                                              [100.00]
                       3400                       [  34 ]                       
|- Performance                                                                  
   |- OSPM                                                                      
                 TGT   4010.00                    <  40 >                       
   |- HWP                                                                       
                 Min    100.25                    <   1 >                       
                 Max   4010.00                    <  40 >                       
                 TGT   4010.00                    <  40 >                       
|- Turbo Boost                                                          [UNLOCK]
                  1C   4010.00                    <  40 >                       
                  2C   3909.75                    <  39 >                       
                  3C   3809.50                    <  38 >                       
                  4C   3709.25                    <  37 >                       
|- Uncore                                                               [UNLOCK]
                 Min    802.00                    <   8 >                       
                 Max   4010.00                    <  40 >                       
|- TDP                                                            Level [  0:3 ]
   |- Programmable                                                      [UNLOCK]
   |- Configuration                                                     [  LOCK]
   |- Turbo Activation                                                  [UNLOCK]
             Nominal   3408.50                    [  34 ]                       
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N,N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AVX-512      [N]  BMI1/BMI2 [Y/Y]        CLFSH [Y]        CMOV [Y] 
|- CMPXCH8      [Y]     CMPXCH16 [Y]         F16C [Y]         FPU [Y] 
|- FXSR         [Y]    LAHF/SAHF [Y]    MMX/Ext [Y/N]     MONITOR [Y] 
|- MOVBE        [Y]          MPX [Y]     PCLMULDQ [Y]      POPCNT [Y] 
|- RDRAND       [Y]       RDSEED [Y]       RDTSCP [Y]         SEP [Y] 
|- SGX          [Y]          SSE [Y]         SSE2 [Y]        SSE3 [Y] 
|- SSSE3        [Y]  SSE4.1/4A [Y/N]       SSE4.2 [Y]     SYSCALL [Y] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Present]
|- Advanced Configuration & Power Interface                     ACPI   [Present]
|- Advanced Programmable Interrupt Controller                   APIC   [Present]
|- Core Multi-Processing                                  CMP Legacy   [Missing]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Present]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Present]
|- CPL Qualified Debug Store                                  DS-CPL   [Present]
|- 64-Bit Debug Store                                         DTES64   [Present]
|- Fast-String Operation                                Fast-Strings   [Present]
|- Fused Multiply Add                                       FMA|FMA4   [Present]
|- Hardware Lock Elision                                         HLE   [Present]
|- Long Mode 64 bits                                         IA64|LM   [Present]
|- LightWeight Profiling                                         LWP   [Missing]
|- Machine-Check Architecture                                    MCA   [Present]
|- Model Specific Registers                                      MSR   [Present]
|- Memory Type Range Registers                                  MTRR   [Present]
|- OS-Enabled Ext. State Management                          OSXSAVE   [Present]
|- Physical Address Extension                                    PAE   [Present]
|- Page Attribute Table                                          PAT   [Present]
|- Pending Break Enable                                          PBE   [Present]
[... 55 lines cut ...]
[... cut ...]
|- Package Thermal Management                                    PTM   [Present]
|- Thermal Monitor 1                                         TM1|TTP   [ Enable]
|- Thermal Monitor 2                                         TM2|HTC   [Present]
|- Units                                                                        
   |- Power                                               watt   [  0.125000000]
   |- Energy                                             joule   [  0.000061035]
   |- Window                                            second   [  0.000976562]
```

olejon, 2019-07-02:
I have taken the screenshots and captured the output you asked me to. I was *a little unsure* since you wrote:

> Package C-States (view [g])

... while view [g] is named Package cycles. I guess you meant that anyway since you can already see C-States on the others.

**Can you please verify them for me? Found here:**

https://www.olejon.net/files/CF/

If you need more, or something changed please tell me.

**Conditions:**

* NMI Watchdog disabled as kernel parameter
* HWP disabled as kernel parameter (stops the downclocking)

`GRUB_CMDLINE_LINUX_DEFAULT="quiet intel_pstate=no_hwp nmi_watchdog=0"`

olejon, 2019-07-02:
**EDIT:** Added info regarding which F3-button-stresstest I used. Just saw your comment. Will check out later. Although is there any disadvantage of HWP = OFF for me? I saw you have it OFF in your Skyake screenshot too, so.

**Regarding the deb package:**

* Compiling like you suggested with `make MSR_CORE_PERF_UCC=MSR_IA32_APERF MSR_CORE_PERF_URC=MSR_IA32_MPERF` *seems to work fine with NMI Watchdog ON here*. Should I make it the default or is it too uncertain to say it will work correctly on supported systems?
* I can always add a warning to your `README.md` if the script finds that NMI Watchdog is on (basically it will always be so maybe add the warning anyway)
* Can ask the user if he/she wants prefers to compile CoreFreq, building like above (APERF/MPERF), and warn that it might not be accurate on all systems (if that's true?)

Right now I have NMI Watchdog disabled as a kernel parameter for your screenshots, and built it normally, just `make`, but before I have built it as above. Haven't spotted any difference on my system, but CoreFreq really seems to work fine with normal build and NMI Watchdog on also on my system, at least when using the on-demand toggles as in my script in the earlier comment, but after your comment on that I changed to `make` with APERF/MPERF.

cyring, 2019-07-02:
> Can you please verify them for me? Found here:

**Thanks** for your screenshots ! 
The Memory Controller output shows that I've to improve the decoding. Indeed a CAS latency decoded as 5 seems wrong. Also the DRAM geometry and size don't show up which means that decoding is incomplete.

> Regarding the deb package:

I presume that APERF/MPERF is working with any x86 architecture (especially w/ AMD).
I've not found an accuracy difference w/ the UCC + URC counters.  
It just that I had started the project using those counters, long time a go
APERF/MPERF should be safe in all cases; I just miss hardware for non regression tests.

> Can ask the user if he/she prefers to build CoreFreq "with NMI Watchdog support"

Following the ArchLinux PKGBUILD guidelines: I avoided to request user for actions.  
Arch package has to be fully automated and has to care about dependencies.  
I can't really tell about Debian/Ubuntu package requirements then I would suggest an auto build based on APERF/MPERF registers

olejon, 2019-07-02:
> The Memory Controller output shows that I've to improve the decoding. Indeed a CAS latency decoded as 5 seems wrong. Also the DRAM geometry and size don't show up which means that decoding is incomplete.

Well, a little technical for me :P Just remember that I have the DDR4 RAM over the standard 2166 MHz. Enabling Intel XMP does it automatically (as it says "Intel XMP Certified" it was logical to enable that), BUT it does no more magic than detect and *correctly* set in UEFI settings:

* DRAM to 1.35V
* DRAM to DDR4-3200MHz
* DRAM to 16-18-18-36

> APERF/MPERF should be safe in all cases

OK, then I'll just use that and not ask the user.

> Following the ArchLinux PKGBUILD guidelines: I avoided to request user for actions.

deb packages totally supports advanced user interaction upon install/upgrade/uninstall, pretty advanced too, with input fields, multiple choice, yes/no etc. There is a reason why `dpkg-reconfigure` exists ;-) It can set up heck let's say phpMyAdmin for you, and if no web server or MySQL/MariaDB installed, will install those and do the basic setup for a working secure system of those, then configure phpMyAdmin to connect to the created root DB user, and which web server alias to use (which web path should point to install path).

**BUT**: I would not do it that way in a small package like this one. Actually I don't know how to use that interface. Surely a lot of docs to read and testing and failing. In this case a simple BASH readline is enough for input, with (yes/no)... Since you say I can just compile with APERF/MPERF by default, no need for asking. Although a text with a link to your GitHub repo for documentation should show IMO when the installation has finished, maybe with the most important reminders. BUT when CoreFreq notices the kernel has changed, then it must tell the user it must be recompiled, the user will be warned and must agree that to work it will recompile itself, and when doing so first downloading the latest version (if not accepting, user gets same warning next time trying to run CoreFreq), and simply run the `postinst` script again which does it all (download, compile, take wanted files if compile success and place them where we want them, then delete the source files since no longer needed).

Of course not many packages use this advanced setup for packages, but some must, like accepting a license from a package in official Ubuntu repos that say, downloads MS fonts, some really makes it easier to set up the basics (various server services), and some just say e.g. when updating timezone data package, says something like "Timezone set to: "Europe/Oslo". To change it run: sudo dpkg-reconfigure tzdata". Usually an upgrade will never trigger "action required", as it is already configured, but an info text like this is shown when it's a good idea.

I know Arch Linux "guidelines", and how the hostile the community can be even though one tries his best, from the forum to packages. Say a person upload the first ever package to AUR *that this person knows only a few people will use, and receives very good feedback from those* (they even bother to comment on how much they like it), but THEN comes some stranger that I'm 99 % sure does not use the package - nor intend to - but just checks the PKGBUILD and download the package and looks at it just to find something to nitpick on.

Erhm which is stupid because according to the Arch Wiki, AUR is not and will not be supported in any way by Arch directly, meaning unless a package make its way into their main (not community) repos, and `pacman` will not support it natively, and the wiki tells people to use e.g. `yaourt` if they want a package manager for AUR. I guess some just have a hobby of hunting packages with 5-20 votes, to completely trash the user who have tried to contribute for the first time ever (like a little tool/app that is great but has a limited user base), even when there is NO way installing/upgrading/uninstalling the package can harm the system in ANY way. The problem for these guys? It's not done 100 % the, and everyone seems to have their own version of, "The Arch Way".

Seen the most stupid comments from such people on their high horse just complaining, with no suggestion for a "better way", even though people comment back that it is their first upload to AUR and are wide open for suggestions to fix/improve the package, like a link to a relevant section on the wiki, but answer is always "read the wiki stupid", which most people has already done, *but not all guidelines and best practices are actually explained with examples*. Basically you must download tens of in-repo packages doing something similar as yours, and try to understand them, they can be complicated or simple, but often they do things slightly different, so you must just go for the common things and then for the different approaches choose the one that seems cleanest or you understand what does at least. But NO NO, on AUR & the forum such knowledge is no guarantee that you're not 2 seconds away from being called "stupid", "ugly code" or "messy package" etc.

Those people are just there to keep your spirit down, **when you try to contribute**... I mean what the heck? Arch is all about breaking your system doing something stupid or simply because it's bleeding edge so even the main packages breaks stuff sometimes. So why a little package with like 5 votes on AUR gets bashed by senior Arch people when they constantly tell people "Arch is a r [... cut ...] no one dares say against if he is "kind of right" so...).

Don't get me wrong, *I LOVE* the Arch Wiki, since it is *the place* to find stuff, examples, tweaks etc you *can* find other places but without any good explanation what it does or why it works, like a random blog post saying I found "adding/changing X made Y better!". Then google it and usually the Arch Wiki comes up and has an explanation. Just know *your* distro and you'll know where to apply stuff that are in different locations etc.

olejon, 2019-07-02:
TLDR: Will create a deb package. I will make it as stable as possible, all deps required and set to support only x86_64, with checks in place, and make sure it works on Ubuntu 18.04 LTS (both GA kernel and HWE kernel). If you have a Debian running on HW (not VM) it would be nice. I guess I can always create a VM with Debian and check that the package installs and compiles correctly. Running CoreFreq may at least launch it, although incorrect values. IDK. As long as it builds in a VM it should work.

cyring, 2019-07-02:
Thanks a lot ! 

To face any kernel upgrade I use dkms to rebuild `corefreqk.ko`

About the IMC, do you have a screenshot of `corefreq-cli` showing the Memory Controller window ?

cyring, 2019-07-03:
Hello,

New entry for the i5-6600K added into the Wiki [CPU support](https://github.com/cyring/CoreFreq/wiki/CPU-support)
Thanks a lot

> Although is there any disadvantage of HWP = OFF for me?

Look rather this [screenshot](https://user-images.githubusercontent.com/11563789/60512633-993dd480-9cc4-11e9-872a-eac3a9e1b187.png), you will notice that HWP is ON  

To reach the Turbo frequency with HWP:

1- Activate HWP
```
|- Hardware-Controlled Performance States                        HWP       < ON>
   |- Capabilities      (MHz)                      Ratio                        
              Lowest    100.25                    [   1 ]                       
           Efficient   1303.25                    [  13 ]                       
          Guaranteed   3408.50                    [  34 ]                       
             Highest   4010.00                    [  40 ]                       
```
2- Set the HWP ratios accordingly to the capabilities
```
|- Performance                                                                  
   |- OSPM                                                                      
                 TGT   4010.00                    <  40 >                       
   |- HWP                                                                       
                 Min    100.25                    <   1 >                       
                 Max   4010.00                    <  40 >                       
                 TGT   4010.00                    <  40 >                       
```
3- Set the EPP policy to zero
```
|- Power Management                                         PWR MGMT   [   LOCK]
   |- Energy Policy                                        Bias Hint   <      0>
   |- Energy Policy                                          HWP EPP   <      0>
```
4- Stress a single Core

olejon, 2019-07-03:
Thanks for making the Gist and host it here so I can delete them later from that server. You know better than me what structure you prefer, so I hoped you'd do it, thanks again.

*Are you also interested in the same screenshots and output from my MacBook Air?*

**Here is the screenshot you asked for** - although incorrect values like the clock speed seems to just default to standard DDR4 - don't know if that's something you can fix/control:

https://www.olejon.net/files/CF/CF-MC-Window.png

* Regarding HWP I see no reason to enable it again. Things working as they should now. As long as that is the only thing I need to do to get avoid the downclock to "default XMP overclock" I'm happy. The entire `intel_pstat` interface is there in `/sys/devices/system/cpu/intel_pstate`, only seems to be missing one "point" that has the name "hwp" in it when booting with HWP enabled. I doubt that points is writeable. I see you also use `corefreqk` as both CPU-Freq and CPU-Idle driver. I'm not very interested in that (and wonder what happens if I set that and then I update the kernel forgetting to compile CoreFreq against it and add the module before starting the new kernel).
* I think I will also turn remove NMI Watchdog = 0 in the kernel parameters now, and compile CoreFreq again with APERF/MPERF, and set `intel_idle` to not let the system go below C1. It still allows idle at Min MHz (800), while only consuming a little more power when idle (according to CoreFreq), but the system is snappier.
* Also the docs from Intel regardin `intel_pstate` added to the Kernel docs are right. If booting with HWP *disabled*, it can be activated at runtime, with CoreFreq for instance (shows both "enable" and "disable"), but as soon as you do that, you can *not* disable it again (shows just "enable", *which it already is*). If booting with HWP *enabled*, you can *not* disable it in any way on runtime. CoreFreq then just shows "enable", which it already is. Now I may be ahead of you, that if using `corefreqk` as driver it is possible(?), but TBH I don't want that, but probably others, we all have different usecases for CoreFreq. For me it's the best monitoring tool with all the stuff I otherwise use 2-3 commands at the same time to check, and CoreFreq adds more details as well in the standard Frequency View, which is always nice!

**Possible bug?**

* CoreFreq randomly (no pattern found except maybe time gone, although CF totally unloaded in the meanwhile, like just happened now when opening and module loaded again etc) suddenly indicates that TURBO = OFF, or saying the TURBO text is grey, not green. And indeed, going to the "Technologies" window it says `TURBO BOST    TURBO    <OFF>` and I can turn it on again, **BUT**, both other tools and stresstests, even CF's own, contradicts that, and the value of `cat /sys/devices/system/cpu/intel_pstate/no_turbo` is `0` always, and that means that `TURBO = ON`, because if this says `1` Turbo is then  *actually off*, as stresstests and other tools confirm. Also the Intel-made `turbostat` *shows that Turbo is on when CoreFreq indicates it is not*. ALSO stresstests, *even in CoreFreq* shows that, when CoreFreq indicates Turbo is off, it goes to 4.4 GHz as usualy, AND CoreFreq says in the Frequency View on the X axis before the C-states: `Turbo: 125%`, and of course `C0: 100%, meaning it also shows Turbo is going on right(?). A little "hmm" for me.

olejon, 2019-07-03:
BTW: The "Energy Policy" thing makes sense. I mentioned it in a previous comment, that it is in the `intel_pstate` (and thought could be a possible solution to allow HWP on without it downclocking after X time):

https://www.kernel.org/doc/html/latest/admin-guide/pm/intel_pstate.html#energy-vs-performance-hints

**Note:** Every change requires reboot, X hours to see if solution works, meaning to see if overclocks holds up over time, which it now does, so basically testing and failing can take up an entire day to check... That's why I'm not "jumping in and testing straight away" (since now HWP is off and well, turning on, if your suggestion does not work abd underclocking happs again, will require reboot). **As said, PC runs server services for others as well**. Not so ideal for testing that may require several reboots (which it will here if it comes to testing and failing).

> If intel_pstate works in the active mode with the HWP feature enabled in the processor, additional attributes are present in every CPUFreq policy directory in sysfs. They are intended to allow user space to help intel_pstate to adjust the processor’s internal P-state selection logic by focusing it on performance or on energy-efficiency, or somewhere between the two extremes

Anyway, if I try this through CoreFreq and it works, of course I want an official way to do it. Meaning like through `sysfs` as said in the docs, or something, to set on boot, rather than opening CoreFreq after every boot :P Though Pretty happy with HWP off and `intel_idle.max_cstate=1`. Never underclocks and darn snappy.

Meanwhile read my above comment, more important.

cyring, 2019-07-03:
Thanks for your returns.

* That will interesting to test with a MacBook: should be x86 compatible, isn't it. However, be prepared for a processor crash; save your files before launching the driver.

* HWP is a technology. Hard to tell if it's better than the legacy PM across the capable architectures. I have however found that the **Vcore is lower when CPU is idling**.
HWP is managed by hardware and not controlled by OS as w/ OSPM: I believe transitions are "smoother", and you can still give a Hint and Target frequencies (incl. Min, Max) ; also other settings such as the Activity Windows (not implemented yet).

To quote the SDM:
>  14.4 ... Hardware-Controlled Performance States (HWP), which autonomously
selects performance states while utilizing OS supplied performance guidance hints.  ... When HWP is enabled, the processor autonomously selects performance states as deemed appropriate for the
applied workload and with consideration of constraining hints that are programmed by the OS. ... preference towards energy efficiency or performance, and the specification of a relevant workload history observation time window. ...

> 14.4.2 Enabling HWP 
IA32_PM_ENABLE MSR (bit 0, R/W1Once) ...  **can only be enabled once** from the default value. Once set, writes to the HWP_ENABLE bit are ignored. **Only RESET will clear** this bit. Default = zero (0). 

That's why I only offer an Enable option. But you can escape the Pop-up ;-)

* Memory Controller will be a long road to debug. I have access to the same architecture, but different Processor model. The DDR speed is based on bits decoding (from the datasheet) : standards values, no XMP involved.

* CoreFreq's Idle and Freq sub-drivers are a mean to take the control over P-States and the C-States. My idea is to offer deterministic tunings which are controlled by users.
Imagine a system which wants full perf or full savings : you can dynamically ask the kernel to maintain C-States as bellow
[image]
Changes don't require a reboot. Remember, _CoreFreq_ aims to be a BIOS straight from the Linux session.

Many other use cases you will make your own. Such as 
1- disabling all Cores beside one for overclocking purpose 
2- select the coldest Cores on which to run intensive algorithms
[image]

> For me it's the best monitoring tool with all the stuff
Appreciate it a lot; thank you

cyring, 2019-07-03:
> rather than opening CoreFreq after every boot 

I try to understand your need but what you ask is one my goal : 

change -> tune -> test -> stress -> decide (if happy or not with results)

And repeat the cycle without leaving the UI

For those "control" reasons, CoreFreq implements its own drivers: `corefreqk-perf` and `corefreqk-idle` try however to be compatible with the /sys interface. At most, to query the current values, but changes have to go through the UI (the spirit of the software)

olejon, 2019-07-03:
Yes, I totally understand that. Well I don't exactly use it for seeing my RAM info. With the incorrect, in the Dashboard View it shows then 2133 MHz and of course then not `16-18-18-36`, which XMP, as said sets automatically in UEFI when enabling it. When enabling XMP in UEFI, upon "Save changes and exit", it shows a confirmation dialogue of values changed. You see what it's done. After that you can basically disable XMP back to "Auto" (off), which will reset these values, and set them manually, which will give the same results. So XMP is just a quick way to detect certified overclock for RAM and CPU and set those values based on HW found. The CPU is always AFAIK set then to Sync All Cores = Auto (on first of X cores multiplier), which in my case enables the full CPU spec Turbo Boost, but on all cores, instead of just one at the time (in my case 3.9 GHz, or a multiplier of 39, although it says "Auto", confirmed by `turbostat`, so I could overclock the CPU manually in a certified way by just setting Sync All Cores = 39). Same goes for memory, the 3 values of `16-18-18-36`, (actually that the UEFI seems to detect/do anyway, without XMP enabled), but setting DRAM voltage to 1.35 and DRAM speed to DDR4-3200 MHz must be done then manually. I've done this with XMP off, same result (one of the things I tried when I saw the downclocking was to keep XMP disabled, as per default, but set those values manually).

The MacBook Air (mid-summer 2013) 13", with top specced CPU (which gives a better iGPU) and RAM, of course chosen on purchase from apple.no (with student discount), **I've already run CoreFreq on several times, no crashes or anything**.

It's very standard x86 HW, just needs on install USB tethering through an Android phone, instantly recognized by Fedora/NetworkManagrr as Ethernet connection > Wi-Fi through my phone, install RPM Fusion and then one can install the driver for the cheap Broadcom chip Apple still uses instead of using Intel's which they could've as it can be part of the SoC. So install the `wl` driver (a package in RPM Fusion) and `akmods` (like dkms) and it works kaboom, disconnect phone and you'll have Wi-Fi in the GNOME Control Center. Actually set to boot as primary OS too, macOS needs Alt-pressed down to show Apple's multiboot menu heh (this runs straight on the metal, not Bootcamp or anything, thats only supports Windows).

CoreFrew builds fine as soon as kernel-devel and necessary build tools and libs are installed on Fedora (although I think I only needed to install kernel-devel, Fedora now even includes git and wget etc, not before, just curl, but not nano still ehh).

You saw the weird TURBO bug, if it is. Seems so, since it's the only tool reporting it off, more official tools and sysfs reports it on.

cyring, 2019-07-03:
* Perhaps booting using the standard timings and speed of the DRAM (no XMP and other auto-tunings), you could confirm if CoreFreq's Memory Controller output is matching the BIOS ?

* MacBook Air: may be you could boot it with the [CoreFreq Live](https://github.com/cyring/CoreFreq/wiki/Live-CD) image for a quick test

> I've already run CoreFreq on several times, no crashes or anything.

I love to read this. Thanks

> You saw the weird TURBO bug

I'm not sure to have notice a Turbo bug . Can you please point it ?

olejon, 2019-07-03:
It's in my comment a few comments above under bold as "Possible bug?". In case you just read the emails, I do tend to edit the comments afterwards with more info, if you haven't responded yet, so please read the comment here and not just the email. Although the part about possible turbo bug was not edited in. Doesn't matter as you can see. It's a false reading as you can see as well. Other tools, sysfs and stresstests, even in CoreFreq shows it goes full turbo but CoreFreq reports turbo as off, like randomly.

cyring, 2019-07-03:
OK I see. 
But maybe not a bug. 

CoreFreq reacts to some bits changed by other software.

W/ your architecture , the Turbo conditions are:
* EIST enable
* IDA capable
* Turbo msr not disable
* Target ratio above Max non-turbo ratio (if not HWP)
* Turbo ratios 1C, 2C, 3C, 4C above Max non-turbo 
* C-States activated may also offer TDP

If the combination is not set, Turbo is marked OFF

Thus if Turbo toggles, it means that another logic is changing those bits on behalf of CoreFreq 

And that's another reason, I now programming the CPU Freq and Idle sub drivers

Give a try, you will understand. 

Can't wait too see your results but it's late in Paris.

Regards
Cyril

olejon, 2019-07-03:
**BUT:** How do you explain this then?

**Run conditions:**
* CoreFreq is run with X seconds apart (from 2-10 seconds)
* Absolutely *NO* software is run between the runs of CoreFreq
* CoreFreq is run like this: `Load Module > Launch Daemon > Launch CLI (as desktop user) > Ctrl+C > Stop Daemon > Unload Module`
* All of `ps`, `lsmod` and `dmesg` confirms this. Nothing is staying running/loaded afterwards

**Results:**
* First run of CoreFreq can show for example TURBO = ON (GREEN)
* Second run can *also* show TURBO = ON (GREEN), or OFF, *you never know*
* Third run typically shows TURBO = OFF (GREY)
* Running stresstest within CoreFreq maxes out  CPU to 4.4 GHz no matter if TURBO is marked as GREEN or GREY showing Turbo 125% and C0 100% (or running `stress -c 4` *which does not change any CPU settings*)
* Running `turbostat` without `--quiet` shows *a lot* of info about the CPU before starting monitoring, *many with the keyword "MSR"*. I can check tomorrow if I see any difference when CoreFreq has TURBO = ON/OFF, or paste the 2 outputs here so you can see (you may spot something I don't).

**PS1:** Not sure if it was *this* "jumping around" when NMI-W was off as a kernel parameter and CoreFreq only compiled with `make`, because now NMI-W is back on and CoreFreq compiled with APERF/MPERF. Can test, but can not reboot this often as lately :-)

**PS2:** Remember every test you suggest (big thanks!) to me involves turning on HWP, which then can not be undone. If your suggestions *does not fix the underclocking happening after X hours*, I will have to reboot again, and both people on LAN and WAN relies on my server services having as much uptime as possible. But big thanks, will always check some time in the near future. But can't check like all the time. Although I only risk it goes down to 3.9 GHz even if saying 4.4 GHz is max, *it is still dead stable* and fast enough - but I am a perfectionist... I simply don't like when my system does stuff I don't want it to, especially because I have tried something, and it didn't work and I have to test and fail God knows how many times, if it at all works. *EVERY test and fail requires reboot > make changes in CoreFreq > wait X hours and see if has been underclocked or not*. That's the problem. Time consuming, and I see no disadvantage with my system now... But rest sure, I *will try as I am way too curious to not do it :P*

**BTW:** Can also give you screenshots and output from my Intel NUC running Ubuntu Server 18.04 LTS 64-bit (with LTS HWE stack installed, meaning `kernel 4.18`). It does run VMs (KVM + QEMU using Libvirt), but I can shut them down without problems if that messes up results.

**Regarding this approach to the deb package and its advantages:**
* My idea is a package that does not use DKMS, but instead, when running `corefreq` (or whatever you want the main command to run it named), say installed to `/usr/sbin` by the deb package (that BASH script + copyright file will be the only installed files by the actual package), *will automatically check when run, if CoreFreq is already installed fully* (downloaded latest code + built)
* If not already installed, will tell the user to press any key to download and build the latest version, or Ctrl+C to abort
* If already installed, if the kernel module fails to load, "warn" the user in a *not-scary manner*, a sensible "warning", basically that it must be rebuilt, but also that the latest version will be installed in the process - advantage - since module failing to load is probably due to running another kernel than before (usually a newer because of a system update) and if so, tell the user to press any key to download the latest version and build against the running kernel, *then having it working without any reboot and also the latest and greatest code*
* Putting, as non-repo packages, with their own "files/system/services/libs/modules" really should (like Spotify, Google Chrome and Skype for instance), the only necessary files to run CoreFreq after build in `/opt` (really just `corefreqk.ko, corefreqd, corefreq-cli`)
* An advantage over DKMS is since then *users will be running pretty recent if not latest code*, since kernel updates happens pretty often, although most are minor updates
* No new deb package must be created every time you commit new code here*. The main BASH script will use those files to run CoreFreq
* The advantage of this is that the deb package only needs to be made once*, the rest is done. If uninstalled the `prerm` script will remove the files in `/opt` and the 2 other files (copyright + run script) will be removed automatically as they are part of the actual package
* A flag can be added to the main BASH script, say running `corefreq -u` can download and build the latest package even though the kernel module loads, just so that users can *easily run the latest version*
* Only 3 dependencies really needed, and these are *base packages* so Debian and Ubuntu share the same names for them, 2 packages that in the history of using Debian/Ubuntu *never has changed names*, the 3rd may have (`libpthread-stubs0-dev`), but that I can easily check in packages.ubuntu.com for all Ubuntu versions supported today, and add *alternative names* if I see the 3rd having changed name (or for some reason is different on Debian). BUT, I will add, to be 100% sure, dependencies for `linux-headers`, `wget` and `unzip`. The other 2 are `build-essential` and `libelf-dev` (which `make` recommends because wit [... cut ...] te April as it is with October, next scheduled HWE Stack according to the chart then is in August (when I say 3 months after, you may think July, but since it's always late-April, almost May, it becomes August, unless they have serious problems, but they offer an edge-package so users can try the HWE stack for 19.04 already now, so both the devs and users make sure it is bulletproof when August comes, and of course any bug found afterwards is as with all packages fixed with a minor update ASAP).

cyring, 2019-07-04:
* Debugging why the Turbo indicator is flipping is required:

Start the daemon in debug mode and watch its output   
`corefreqd -d`

I'm interested by traces happening when Turbo flips. 

Also be aware that ACPI can modify the Target frequency. One ACPI kernel module may run in your system 

* Yes, a NUC run, w/ screenshots, will help me to cover this architecture.

* My non regression build environment test is made of virtualized Ubuntu, CentOS, OpenSuse. All of them in console mode only.

Whereas ArchLinux is my day to day developer environment. Updated every week. Stable since 12 years.
A bare metal boot without unused drivers and no virtualisation

Xen w/ same Arch Kernel as dom0 is occasionally booted however. 

awesome is my WM.

I don't make use of Ubuntu : too much CPU overhead

olejon, 2019-07-04:
**Here comes screenshots from the MacBook Air:**

https://www.olejon.net/files/CF-MacBook-Air/

* Memory Controller window can not be opened (in the menu the text is in bold)
* Stress tests ran with "Conic Compute" when stressing all cores
* Screenshots taken with a 5 second delay

**DEB:**

* There are NO file called "corefreq" in any Ubuntu package according to packages.ubuntu.com, in fact not even a file with the text "corefreq" *in its name exists*, as an example: "x-corefreq-y". None of those exists (then shouldn't be in any Debian package either)
* Putting `*.sh` in a bin-directory is wrong. If you still have a problem with it, better to choose another name, like `corefreq-cli` or `cf-cli` - none of those are taken either. Whatever you like. The user will be told how to run it after successful installation

**NUC:**

* *Will get you some screenshots for that as well soon!* Will shut down VMs so the tests runs truly idle etc
* Well those are clean, standard and generic Intel x86_64 PCs, but with a special small MB to fit in it with some special slots for that reason (guess you can call it an "architecture")
* They come with one of the mobile-line CPUs of that year, but in the higher end I think, like those made more for stationary small PCs like the NUC
* Memory and storage must be bought apart. Wi-Fi + Bluetooth module if needed as well (little slot on the MB). I don't need that so. Comes with a great variety of ports. So included are chassis, MB and CPU, and in my case a taller chassis with a SATA fit-kit (which can't be changed on mine but maybe on newer ones, but one can always just buy the SoC without the chassis and install it if wanting a CPU and general HW upgrade)
* Therefore I like to think of it as a "The previous modular Mac Minis, without paying the Apple or Microsoft tax" (as it comes with no OS, of course, since doesn't come with storage by default)
* Mine was bought in 2014, so it's quite old now. I'm sure newer NUCs are great, and Ubuntu has special Ubuntu Core images for them, just as for other IoTs.
* M.2 storage was so expensive then, was only like 2 chips available for it, don't know if what is on the MB is actually a standard M.2 connector (NVMe or PCIe), and the 2 chips had bad user reviews, so I bough the one with a slighly taller chassis, which costs a little more, but comes with a complete fit-kit for a SATA drive, and bought a much cheaper (price vs GB) Kingston SSD that IIRC according to the specs performed just as well as the chips, just a matter of the NUC size
* I'm sure today's NUCs have proper NVMe M.2 and maybe not chassis more for a SATA drive
* It has an Intel Core i3, while a cheaper Celeron-based model was available, and would've worked perfectly fine for what I used it for then (HTPC/Kodi + some services), bought the i3 for future-proofing
* In fact runs so well still, that even with "Powersave" set in UEFI for the HW, that the minimal Ubuntu Server host can, with one command, download and get up and running any version of Ubuntu available as a Cloud Image, selecting a specific "Ubuntu Cloud Image"
* In fact those are so small that by default the command downloads all Ubuntu versions available as Cloud Images*. May sound "scary" but those are just barebone Ubuntu Servers made for running in a VM, since by default it comes with a `linux-virtual` kernel, and the Cloud Package make installs a random private key and gives you the public key, which is added to the Libvirt XML so one can SSH in with a special command using that VMs' name, IIRC from anywhere in the world (maybe that requires an extra step which is shown the first time you SSH in from the host, don't remember. Needed because the default user has no password so you can not use `virsh console vmname`
* So for complete barebone local VM just SSH in, set password for default user and uninstall the Cloud Package if you don't want it. Runs on standard Libvirtd anyway, using standard XML, no hidden magic, controlled by `virsh` as all others
* Especially when configured manually with `virsh edit vmname` and setting the VMs CPU to "host-passthrough" there's absolutely no difference in the performance of the VM and the Host, although haven't tried with the default which is simply "Copy Host CPU Topology". `virt-manager` (GUI) does not have that option, just to avoid that it won't work when people do things people do with VMs, well, the reason is in the excellent Libvirt docs, which covers all the XML options and values, but when running on the same Host always, never a problem. And can always just switch to "Copy Host CPU Topology", which is default in `virt-manager`, which is kind of excellent too now. VirtualBox long gone. Like when adding/editing/removing HW to the VM there are way more stuff available

cyring, 2019-07-04:
First time I can see _CoreFreq_ driving an Haswell ULT. **Thank you very much**

Indeed, what needs to be completed for this architecture:
* Uncore clock
* Uncore ratios over/down-clocking if platform capable
* Memory controller

I'll put your screenshots in Wiki during WE.

Usage advises:
* Prefer xterm (or a compatible color palette) to display the dark characters
* Press [B] key to change the SMBIOS name.  
  You can also start `corefreq-cli -J <num>` where `<num>` is the zero based index of your selected string
* Press [H} key to show and clear the thermal events. This explains why temperatures are in red.

Can't wait to see the NUC in actions ...

Thanks again

olejon, 2019-07-04:
First of all, why would I want to change the SMBIOS name, for what purpose? Does it do anything permanent? Interesting that my MacBook basically shows exactly what an iPhone/iPad shows in `dmesg` and `lsusb` when plugged in. Manufacturer, Specific Model - but what looks like being the serial number is not. It begins with the same characters but then totally different.

Didn't know about the H shortcut, guess it's on the Shortcuts menu option, should've known. Tried stopping the daemon and unloading the module, but didn't cleared the "HOT"/red colors. Suppose it writes something to registers or similar. Anyway people can see current temperature.

If I were a new user, it would just be nice to see that CoreFreq has this feature, storing max temperatures and storing if "hot" has been triggered, *which any, including newer, MacBooks and iMacs will show anyway if something heavy has run just for a few seconds, as they are infamous for their throttling*, and Apple's lies (legally correct but fooling users into expecting something more), since on their website they're just using that Intel's CPU specs and not expected frequencies to see in real life on that Mac.

If I had left the stresstests run longer you'd seen way more throttling. As you can see, starting one then setting screenshot timer for 5 seconds immediately afterwards, even that shows it reaches 100 Celsius. When run longer the throttling is very visible, frequencies go down so much that color of the bars change to yellow/orange and stays a little above the middle of max turbo frequency, even just stressing one core. Guess that's not important for these screenshots.

Regarding colors, I had this one actually set to the  "white on black" custom palette (IDK if it changes other colors than those two) which I used to prefer. CoreFreq ignores this anyway, say "black on white" doesn't change CoreFreq's colors.

Changed to system default palette to match the others, *but really because*, with newer versions of GNOME Terminal, I find the default dark theme with its palette more esthetically pleasing, but more importantly, easier on the eye.

**If you refer to the missing values between the _commas_ to the right of "TURBO", they aren't there no matter what color palette or terminal program**.

I've used CoreFreq a lot with SSH-ing in with Termius on Android, which is set (by default) to xterm-256color and "white on black". Shows the same colors. And nothing between those commas at least.

I find the darker colors that *do show*, absolutely visible enough, and my guess is GNOME Terminal (which uses dark theme by default) is the most used terminal, so my screenshots will reflect what most users will see.

If the output suggests those values between the commas should show(?), they simply DON'T on the MacBook. That CoreFreq ignores like "black on white" is just a good idea for such a feature rich program which needs many colors, similar in all terminals, including mobile apps. **GNOME Terminal use xterm-256color AFAIK**.

Only thing changed is font. On Fedora used the default Source Code Pro Regular 12, which Termius on Android/iOS uses as well. On the workstation with Ubuntu, it's changed to Google's Noto Monospace, since it goes very well (also one of the recommend IIRC) with the fantastic (!) "Materia" GTK3+ theme, which you definitely should check out if you like Google's Material Design - or anyway - just those beautiful ripple animations everywhere, from window buttons to toggles to clickable sections like in the GNOME Control Center. 

Animations, toggles, radio and check buttons and icons just like on Android in other words. Blew my mind, and I've like tried all kinds of themes from gnome/kde-looks.org since like year 1999 (when I was 11). Now included in the official Ubuntu repos, which says something (not in the Fedora ones, just one of the two shitty ones that says it's based on Material Design, but so ugly, and in Ubuntu's repos there are all three). Not many themes are included in any distro's official repos. I use the official PPA to always have the latest version, but no update yet IIRC, it's dead stable anyway. IIRC Materia is in the Arch Community repos (if not AUR). My main desktop VM runs Arch and XFCE. Looks nice there too. But really shines on newer GNOME versions.

Maintained (PPA, GitHub etc), not as boring and static like most themes. It includes a GNOME Shell theme as well, which uses Material Design Icons. Think it changes basic grayscale icons like in the GNOME Control Center, Nautilus etc as well, adapting to light/dark. Just FYI/tip.

Don't expect much from the NUC I think. On my phone now Copied straight from Termius, this is from `/proc/cpuinfo`:

```
processor: 0
vendor_id: GenuineIntel
cpu family: 6
model: 69
model name: Intel(R) Core(TM) i3-4010U CPU @ 1.70GHz
```

Must get back to you regarding checking that debug on TURBO indicator turning off, checking if any diff from debug info `turbostat` shows when it's on and off (I never turn it off myself, "I let CoreFreq do that" :P I do turn it on if I see it off, before exiting CoreFreq, just in case it actually does something, which nothing indicates).

olejon, 2019-07-04:
Edited a little. BTW I know there is a new BIOS version for my NUC. Best to install that first, it's time anyway, *and set the performance setting to default as it's now in Powersave mode*, but must connect to a monitor (uses mini-HDMI to HDMI, or mini-DP) so have just a short cable. And BIOS updates must be done by copying to a USB stick with FAT32. Weird since the UEFI absolutely supports networking. My workstation checks straight online.

cyring, 2019-07-04:
If HOT or Temps are RED, just press the [H] key then select the colored events and press [Enter]
This will send an order to the driver to clear bits of the Processor thermal registers (event, throttling and so on).
You don't need to stop or restart _CoreFreq_ to clear those events.
[image]

cyring, 2019-07-04:
You are free to change the name displayed in the footer because the data populated in the DMI area is not consistent among motherboards (issue #127)
[image]

olejon, 2019-07-04:
Freakin' what? The values between the commas actually shows using `xterm` from the Fedora repo... Really weird when they show on my workstation also using GNOME Terminal, in "default mode". I tried changing GNOME Terminal in every way.

* Palette to "Xterm"
* Not using colors from system theme
* Switching to Light theme
* Switching to "Black on white" (like xterm)
* Changing font and font size

No success... Argh, don't know if I have energy to repeat the screenshots any soon. It is a time consuming task with all the tests you ask for, and those values are shown on several of them, and anyway if I change to taking screenshots of xterm I want all of them to be from xterm...

Must check by SSH-ing in from my workstation, that's also running GNOME Terminal. If it shows there then could be because Fedora has a newer version. Or I'll change to the exact same config on both (need to install fonts on Fedora then). < 5% battery left now so then the firmware of a MacBook always throttle heavily down.

cyring, 2019-07-04:
> Must get back to you regarding checking that debug on TURBO indicator turning off, checking if any diff from debug info turbostat shows when it's on and off ...

Be aware that _CoreFreq_ and turbostat may use the same Processor registers. Turbo indicator could be the result of this software conflict.

Regarding the i3-4010U, it is the [Haswell](https://ark.intel.com/content/www/us/en/ark/products/75107/intel-core-i3-4010u-processor-3m-cache-1-70-ghz.html) architecture, I expect to match the features

_CoreFreq_ has been conceived for console (80x25) and X11 xterm

True, need improvement: I will add the H shortcut in the help window. The UI still looks a bounty treasures ;-)

olejon, 2019-07-04:
Indeed I know it's Haswell.

As said in my test I didn't run any software between the runs of CoreFreq. But I can always try a reboot when I can and just run CoreFreq, again several times, to see. I never run them at the same time. You're saying even if CoreFreq is totally exited and unloaded, running turbostat, which doesn't start any daemon or anything, may ruin CoreFreq readings afterwards? Meaning I can NEVER run turbostat if I want to use CoreFreq unless I reboot first?

Well I certainly run more than 80x25, but as long as it's bigger, shouldn't be a problem... Even in portrait mode on my Android phone it's just fine. Considering GNOME Terminal with default settings shows everything correctly on my workstation, I'll try SSH to the MacBook and see if they show up exactly as on the workstation. If not it's weird.

I wouldn't develop such a tool today with only xterm in mind. How well will xterm hold up over time in Wayland? And when XWayland one day is severely limited and finally not used anymore? Well if it's the newer version of GNOME Terminal in Fedora's fault, then not your problem. I kind of suspect it. It should be totally xterm-256 compliant. But testing against the probably most used terminal program (considering it's the default in Ubuntu, Fedora, probably the default or most used and documented choice for Debian now, other siblings of Debian/Ubuntu like Mint, and openSUSE also has GNOME as default desktop checked during install if booting the full ISO last I checked, think they even rewrote YaST to GTK... CMIIW.

cyring, 2019-07-04:
It's more how are left some msr registers after running a performance software.  
Off course rebooting will reset them, but it's fastidious. On its side, CoreFreq is reserving a save area to restore registers at exit.  
I try to document the prerequisites for a good usage, although there is still a lot of improvements to do

* Why 80x25, why 8 colors, why xterm ?  

At beginning of the project, for those reasons:  

+ **low latency**: the UI must have the lowest CPU overhead. (no curses, everything is written from scratch...)  

_CoreFreq_ is a Processor monitor, not another top. 

The effort focus on lightweight loops. 
Perhaps have you notice that the CoreFreq usage percent can go down to 1%  (for a session without all the fancies of a DE, and with a plain console)  

 + Most compatibility : Users can be seated in data center with just a console... 

+ Remote terminal (ssh, telnet, etc) 

+ Full compatibility down to kernel 3.2

+ Other customer requirements : trading, high performance computing, Research,..

UTF is on the roadmap. Here is Europe. I've to face languages...

* Going from console to xterm was the clever path

cyring, 2019-07-05:
Screenshots of CoreFreq for i7-4650U updated into [WikI](https://github.com/cyring/CoreFreq/wiki/CPU-support)
Thank you

[352 more replies]

closed 2020-03-18 by cyring