cyring/CoreFreq #131: Segmentation fault 💪💪💪

dEROZA, 2019-07-03:
**luckyman@irk-1:~/CoreFreq$ ./corefreq-cli**
_Segmentation fault (core dumped)_

# apt install make gcc
$ git clone https://github.com/cyring/CoreFreq.git
$ cd CoreFreq
$ make

# insmod corefreqk.ko
# ./️⃣corefreqd
$ ./corefreq-cli -> sg taken 🥇

cyring, 2019-07-03:
kernel log about CoreFreq plz

dEROZA, 2019-07-04:
> CoreFreq

Yep. nmi_watchdog was already disabled:
```
$  cat /proc/cmdline
BOOT_IMAGE=/vmlinuz-5.0.0-20-generic root=UUID=6cd0085e-fa69-4d84-b1a8-19a6d39c97cd ro nmi_watchdog=0
```

Kern log
```
4601:Jul  3 15:29:47 irk-1 kernel: [ 1102.120009] corefreqk: loading out-of-tree module taints kernel.
4602:Jul  3 15:29:47 irk-1 kernel: [ 1102.120108] corefreqk: module verification failed: signature and/or required key missing - tainting kernel
4603:Jul  3 15:29:47 irk-1 kernel: [ 1102.174017] CoreFreq(1:-1): Processor [ 6F_02] Architecture [Bulldozer/Piledriver] CPU [8/8]
4605:Jul  3 15:30:16 irk-1 kernel: [ 1131.887200] corefreq-cli[6010]: segfault at a0 ip 000055c230b74a2e sp 00007fff92608920 error 4 in corefreq-cli[55c230b3a000+51000]
4607:Jul  3 15:30:37 irk-1 kernel: [ 1152.166972] corefreq-cli[6055]: segfault at a0 ip 000056058eaa1a2e sp 00007fff0994ee20 error 4 in corefreq-cli[56058ea67000+51000]
4609:Jul  3 15:31:59 irk-1 kernel: [ 1234.485087] corefreq-cli[6182]: segfault at a0 ip 00005558e8597a2e sp 00007ffe10fe6a60 error 4 in corefreq-cli[5558e855d000+51000]
4611:Jul  3 15:34:39 irk-1 kernel: [ 1394.905438] corefreq-cli[6185]: segfault at a0 ip 0000559f28d31a2e sp 00007fff0637bca0 error 4 in corefreq-cli[559f28cf7000+51000]
4613:Jul  3 15:46:05 irk-1 kernel: [ 2080.589951] CoreFreq: Unload
```

cyring, 2019-07-04:
Bulldozer/Piledriver

Can you print these outputs:
1- `corefreqd -d`
2- `corefreq-cli -s`
3- `corefreq-cli -m`
4- `corefreq-cli -B`
5- `corefreq-cli -c`
6- `corefreq-cli -u`

dEROZA, 2019-07-04:
```
Driver connection error code 2
/dev/corefreqk: 'No such file or directory' @ line 4127
```

```
Daemon connection error code 2
corefreq-shm: 'No such file or directory' @ line 9264
```

and other too..

cyring, 2019-07-04:
Thank you very much for using _CoreFreq_

B/C I don't have a Bulldozer, can you please post the output of 
1. `corefreq-cli -R -u -m -s`
2. `corefreq-cli -c # idle and high load`

dEROZA, 2019-07-04:
> 
> 
> Thank you very much for using _CoreFreq_
> 
> B/C I don't have a Bulldozer, can you please post the output of
> 
>     1. `corefreq-cli -R -u -m -s`
> 
>     2. `corefreq-cli -c # idle and high load`
> 
> 
> The Base Clock has not be estimated: this is an issue.

https://github.com/moonsiber/Buldozer-log--FX-9590-

cyring, 2019-07-04:
Excellent, thank you.
I'll process those data. 
Apparently the Base Clock is showing good. 
Regards
Cyril

dEROZA, 2019-07-04:
> 
> 
> Excellent, thank you.
> I'll process those data.
> Apparently the Base Clock is showing good.
> Regards
> Cyril

I configuired Sabertooths' BIOS settings. Its not default.

cyring, 2019-08-27:
Hello,
Is the latest version stable with the FX-9590 ?

h1z1, 2019-09-15:
I get a segfault with corefreq-cli which bisected to 2b9c64b35871d772c331874354763e973f69dd4c

Running with ```corefreq-cl -d``` works, not sure if this is the same.

cyring, 2019-09-15:
Standard output appears messy.
Does it run better in a plain XTerm ?

h1z1, 2019-09-15:
No different in xterm.  Console semi works but crashes more often (can't change views at all, doesn't appear to scale).

cyring, 2019-09-16:
I believe some strings collected by driver are not fulfilled. Here, the Threadripper brand.
Can you post the output of
```
corefreq-cli -s
```

cyring, 2019-09-16:
Found the issue, fix committed.
[image]

h1z1, 2019-09-16:
There's a whole lot of compiler warnings now but the segfault is gone :)

```
  [RSC_BOX_RECORDER_TITLE]= LDT( RSC_BOX_RECORDER_TITLE_CODE_EN,
  ^
corefreq-cli-rsc.c:1511:2: warning: (near initialization for ‘Resource[525].Size[0]’) [enabled by default]
corefreq-cli-rsc.c:1511:2: warning: initializer element is not a constant expression [enabled by default]
corefreq-cli-rsc.c:1511:2: warning: (near initialization for ‘Resource[525].Size[1]’) [enabled by default]
corefreq-cli-rsc.c:1513:2: warning: initializer element is not a constant expression [enabled by default]
  [RSC_SMBIOS_TITLE] = LDT( RSC_SMBIOS_TITLE_CODE_EN,
  ^
```
etc

cyring, 2019-09-16:
Thanks for your return, please let me know about your compiling flags, I can track the warnings with.

I'm a bit surprise by the Conic stress functions: except the TSC and the PMC measurements (an assembly code which wraps all stress threads), Conics are made in plain C code, involving however the FPU ...

h1z1, 2019-09-16:
No special flags, using the provided makefile.

Not sure re the stresstests, atomic burn and crc32 work as does core turbo

cyring, 2019-09-16:
Thks, I will check my gcc system flags.

About other stress tests, I'm gonna ask other Zen users...

cyring, 2019-09-16:
* Threadripper 1950X [test](https://github.com/cyring/CoreFreq/issues/54#issuecomment-531725721): no crash

h1z1, 2019-09-19:
@cyring  Compiler warnings are indeed gone now, thanks.

closed 2019-11-04 by cyring