cyring/CoreFreq #133: corefreq-cli seg faults on Arch Linux + Ryzen 3000

logan2611, 2019-07-10:
As the title says, corefreq-cli seg faults on my AMD Ryzen 3000 setup.

CPU: AMD Ryzen 7 3700X
Mobo: ASUS ROG B450-F
OS: Arch Linux
Kernel: 5.2.0

GDB:
`(gdb) run
Starting program: /usr/bin/corefreq-cli 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x0000555555599846 in ?? ()`

dmesg:
`[ 1565.404946] corefreq-cli[14359]: segfault at a0 ip 0000565555598846 sp 00007ffbffffb260 error 4 in corefreq-cli[56555555e000+51000]`

journalctl:
`Jul 09 22:12:19 EVA-L systemd-coredump[14361]: Process 14359 (corefreq-cli) of user 1000 dumped core.
                                               
                                               Stack trace of thread 14359:
                                               #0  0x0000565555598846 n/a (corefreq-cli)
                                               #1  0x0000565555598c6e n/a (corefreq-cli)
                                               #2  0x000056555559c316 n/a (corefreq-cli)
                                               #3  0x000056555559f49f n/a (corefreq-cli)
                                               #4  0x00005655555a0210 n/a (corefreq-cli)
                                               #5  0x00007efbff3b2ee3 __libc_start_main (libc.so.6)
                                               #6  0x000056555555e36e n/a (corefreq-cli)`

cyring, 2019-07-10:
can you try to run Cli with option:
```
corefreq-cli -s
```
and posts the output (even if partial)

logan2611, 2019-07-10:
Sure

CLI output:
`Processor                                  [AMD Ryzen 7 3700X 8-Core Processor ]
|- Architecture                                               [Zen/Summit Ridge]
|- Vendor ID                                                      [AuthenticAMD]
|- Microcode                                                        [ 141561873]
|- Signature                                                            [ 8F_71]
|- Stepping                                                             [     0]
|- Online CPU                                                           [ 16/16]
|- Base Clock                                                           [ 99.81]
|- Frequency            (MHz)                      Ratio                        
                 Min    399.25                    [   4 ]                       
                 Max   3593.23                    [  36 ]                       
|- Factory                                                              [100.00]
                       3600                       [  36 ]                       
|- Performance                                                                  
   |- OSPM                                                                      
                 TGT   2195.86                    <  22 >                       
|- Turbo Boost                                                          [  LOCK]
                 XFR   3593.23                    [  36 ]                       
                 CPB   3593.23                    [  36 ]                       
                  1C   2794.74                    [  28 ]                       
                  2C   2195.86                    [  22 ]                       
|- Uncore                                                               [  LOCK]
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N,N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AVX-512      [N]  BMI1/BMI2 [Y/Y]        CLFSH [Y]        CMOV [Y] 
|- CMPXCH8      [Y]     CMPXCH16 [Y]         F16C [Y]         FPU [Y] 
|- FXSR         [Y]    LAHF/SAHF [Y]    MMX/Ext [Y/Y]     MONITOR [Y] 
|- MOVBE        [Y]          MPX [N]     PCLMULDQ [Y]      POPCNT [Y] 
|- RDRAND       [Y]       RDSEED [Y]       RDTSCP [Y]         SEP [Y] 
|- SGX          [N]          SSE [Y]         SSE2 [Y]        SSE3 [Y] 
|- SSSE3        [Y]  SSE4.1/4A [Y/Y]       SSE4.2 [Y]     SYSCALL [Y] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Present]
|- 100 MHz multiplier Control                            100MHzSteps   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Present]
|- Advanced Programmable Interrupt Controller                   APIC   [Present]
|- Core Multi-Processing                                  CMP Legacy   [Present]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Present]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Missing]
|- CPL Qualified Debug Store                                  DS-CPL   [Missing]
|- 64-Bit Debug Store                                         DTES64   [Missing]
|- Fast-String Operation                                Fast-Strings   [Missing]
|- Fused Multiply Add                                       FMA|FMA4   [Present]
|- Hardware Lock Elision                                         HLE   [Missing]
|- Long Mode 64 bits                                         IA64|LM   [Present]
|- LightWeight Profiling                                         LWP   [Missing]
|- Machine-Check Architecture                                    MCA   [Present]
|- Model Specific Registers                                      MSR   [Present]
|- Memory Type Range Registers                                  MTRR   [Present]
|- No-Execute Page Protection                                     NX   [Present]
|- OS-Enabled Ext. State Management                          OSXSAVE   [Present]
|- Physical Address Extension                                    PAE   [Present]
|- Page Attribute Table                                          PAT   [Present]
|- Pending Break Enable                                          PBE   [Missing]
|- Process Context Identifiers                                  PCID   [Missing]
|- Perfmon and Debug Capability                                 PDCM   [Missing]
|- Page Global Enable                                            PGE   [Present]
|- Page Size Extension                                           PSE   [Present]
|- 36-bit Page Size Extension                                  PSE36   [Present]
|- Processor Serial Number                                       PSN   [Missing]
|- Restricted Transactional Memory                               RTM   [Missing]
|- Safer Mode Extensions                                         SMX   [Missing]
|- Self-Snoop                                                     SS   [Missing]
[... 103 lines cut ...]
    Thread [7efbfab35700] Init CYCLE 008
    Thread [7efbfdb3b700] Init CHILD 002
[... cut ...]
The only other information I was able to find in the kernel logs was this (I might be missing something):
`Jul 10 00:07:05 EVA-L kernel: corefreq-cli[2376]: segfault at a0 ip 0000565555598846 sp 00007ffbffffb260 error 4 in corefreq-cli[56555555e000+51000]
Jul 10 00:07:05 EVA-L kernel: Code: c7 b8 08 00 00 00 e8 c9 5a fc ff 48 98 48 83 c4 10 5b 41 5c 5d c3 55 48 89 e5 48 83 ec 10 48 89 7d f8 48 89 75 f0 48 8b 45 f0 <8b> b8 a0 00 00 00 48 8b 45 f8 8b 88 b4 00 00 00 48 8b 45 f0 8b 90`

I would like to note though that as far as I can tell the corefreq daemon seems to be working fine.

cyring, 2019-07-11:
Hello,

Can you post the output of `corefreq-cli -c`

logan2611, 2019-07-11:
Output:
```
CPU Freq(MHz) Ratio  Turbo  C0(%)  C1(%)  C3(%)  C6(%)  C7(%)  Min TMP:TS  Max
#00   73.45 ( 0.74)   2.04   2.42  97.58   0.00   0.00   0.00  0  /  0:0  /  0
#01   53.65 ( 0.54)   1.49   2.42  97.58   0.00   0.00   0.00  0  /  0:0  /  0
#02   76.44 ( 0.77)   2.13   3.62  96.38   0.00   0.00   0.00  0  /  0:0  /  0
#03   31.39 ( 0.31)   0.87   1.28  98.72   0.00   0.00   0.00  0  /  0:0  /  0
#04   23.12 ( 0.23)   0.64   1.07  98.93   0.00   0.00   0.00  0  /  0:0  /  0
#05   61.15 ( 0.61)   1.70   2.81  97.19   0.00   0.00   0.00  0  /  0:0  /  0
#06   43.26 ( 0.43)   1.20   2.04  97.96   0.00   0.00   0.00  35 / 45:367/ 47
#07   58.36 ( 0.58)   1.62   2.63  97.37   0.00   0.00   0.00  0  /  0:0  /  0
#08   77.61 ( 0.78)   2.16   3.07  96.93   0.00   0.00   0.00  0  /  0:0  /  0
#09   42.95 ( 0.43)   1.20   1.91  98.09   0.00   0.00   0.00  0  /  0:0  /  0
#10    8.94 ( 0.09)   0.25   0.44  99.56   0.00   0.00   0.00  0  /  0:0  /  0
#11   51.76 ( 0.52)   1.44   2.12  97.88   0.00   0.00   0.00  0  /  0:0  /  0
#12   30.84 ( 0.31)   0.86   1.40  98.60   0.00   0.00   0.00  0  /  0:0  /  0
#13   59.92 ( 0.60)   1.67   2.80  97.20   0.00   0.00   0.00  0  /  0:0  /  0
#14   65.87 ( 0.66)   1.83   3.13  96.87   0.00   0.00   0.00  0  /  0:0  /  0
#15   32.96 ( 0.33)   0.92   1.48  98.52   0.00   0.00   0.00  0  /  0:0  /  0

    Averages:        Turbo  C0(%)  C1(%)  C3(%)  C6(%)  C7(%)    TjMax:    Pkg:
                      1.36   2.13  97.87   0.00   0.00   0.00       0 C    35 C
```
It does keep printing after this.

cyring, 2019-07-11:
Thanks, the issue is that the temperature offset (aka TjMax) is not implemented yet for this Ryzen version.

cyring, 2019-07-13:
Matisse parameters added.
Can you please try the last commit, thanks.

logan2611, 2019-07-13:
Still seg faults, same error as before.

corefreq-cli does now report a TjMax though (albeit considerably lower than it should be)

```
CPU Freq(MHz) Ratio  Turbo  C0(%)  C1(%)  C3(%)  C6(%)  C7(%)  Min TMP:TS  Max
#00  124.92 ( 1.25)   3.48   3.68  96.32   0.00   0.00   0.00  28 / 34:357/ 41
#01   52.36 ( 0.52)   1.46   2.16  97.84   0.00   0.00   0.00  0  /  0:0  /  0
#02   27.57 ( 0.28)   0.77   1.32  98.68   0.00   0.00   0.00  0  /  0:0  /  0
#03   21.99 ( 0.22)   0.61   1.03  98.97   0.00   0.00   0.00  0  /  0:0  /  0
#04   31.84 ( 0.32)   0.89   1.47  98.53   0.00   0.00   0.00  0  /  0:0  /  0
#05  114.69 ( 1.15)   3.19   5.22  94.78   0.00   0.00   0.00  0  /  0:0  /  0
#06   90.50 ( 0.91)   2.52   4.29  95.71   0.00   0.00   0.00  0  /  0:0  /  0
#07   19.98 ( 0.20)   0.56   0.92  99.08   0.00   0.00   0.00  0  /  0:0  /  0
#08  121.95 ( 1.22)   3.39   3.72  96.28   0.00   0.00   0.00  0  /  0:0  /  0
#09   62.71 ( 0.63)   1.75   2.61  97.39   0.00   0.00   0.00  0  /  0:0  /  0
#10   65.49 ( 0.66)   1.82   3.15  96.85   0.00   0.00   0.00  0  /  0:0  /  0
#11  150.91 ( 1.51)   4.20   7.25  92.75   0.00   0.00   0.00  0  /  0:0  /  0
#12   10.88 ( 0.11)   0.30   0.50  99.50   0.00   0.00   0.00  0  /  0:0  /  0
#13   52.41 ( 0.53)   1.46   2.37  97.63   0.00   0.00   0.00  0  /  0:0  /  0
#14   87.51 ( 0.88)   2.44   4.06  95.94   0.00   0.00   0.00  0  /  0:0  /  0
#15   12.95 ( 0.13)   0.36   0.61  99.39   0.00   0.00   0.00  0  /  0:0  /  0

    Averages:        Turbo  C0(%)  C1(%)  C3(%)  C6(%)  C7(%)    TjMax:    Pkg:
                      1.82   2.77  97.23   0.00   0.00   0.00      10 C    36 C

^CCPU Freq(MHz) Ratio  Turbo  C0(%)  C1(%)  C3(%)  C6(%)  C7(%)  Min TMP:TS  Max

    Averages:        Turbo  C0(%)  C1(%)  C3(%)  C6(%)  C7(%)    TjMax:    Pkg:
                      1.82   2.77  97.23   0.00   0.00   0.00      10 C    36 C
```

Also temps seem to be 10 degrees off from what lm-sensors reports.
[image]

cyring, 2019-07-13:
I've asked for a 2700X test above.

Probably some data is not present w/ the 3000's + Motherboard

Can you test each Cli option for a bug ?

Probably missing data from the SMBIOS: try with `corefreq-cli -J 0 -t`

cyring, 2019-07-13:
In complement to the above request: it will be helpful to debug the Cli with the Valgring memchecker.  
* Install the Valgring package  
* Rebuild _CoreFreq_ for symbols  
```
make CC='cc -g' clean all
```
* Post the output of the Cli startup execution
```
valgrind -v corefreq-cli
```

Thank U

cyring, 2019-07-13:
Forget all requests above. Found the bug.
Please update to the last commit.

About temperature: it's hard to tell without clear specification.  
Some users are saying that no offset is required.  
But looking at temps on some reddit's screenshots confirms that the max value is around 60°C which is what _CoreFreq_ shows in your screenshot.

lm_sensors, the k10temp module, seems not to have change its [code](https://elixir.bootlin.com/linux/latest/source/drivers/hwmon/k10temp.c#L78) with the introduction of the Ryzen 3000
That's why you see the 10°C difference

I think that only the manufacturer tool can be the reference to compare with. Either the BIOS or the AMD Windows tool.

logan2611, 2019-07-13:
Rebuilt CoreFreq, seems to work almost perfectly!
[image]

I decided to test the temperature by having the fan go full blast (around 3200RPM on the stock cooler) once it hit 70C, and comparing the temperatures of lm-sensors and corefreq.

(Pardon the low quality image, I didn't feel like waiting for my desktop to start)
[image]

Comparing lm-sensors to HWInfo and AI Suite III in Windows, lm-sensors seems to be pretty close to what the Windows tools report (forgot to screenshot, but in both Linux and Windows the fan was at 3200RPM when lm-sensors/AI Suite reported 70C)

To get temperature readings in lm-sensors I am using [a fork or groeck's it87 kernel module](https://github.com/a1wong/it87).

I am curious though what other people are getting for CPU temperature, although that might be best suited for another issue report, as the original problem in this one has been solved.

cyring, 2019-07-14:
* Nice fork, its 8665 code should be part of the mainstream kernel.
I'm not sure we can get the temperature offset from the `sensors -u` command: can you `printk` the value read from register at line [1239](https://github.com/a1wong/it87/blob/master/it87.c#L1239) ?
Some modified code like:
```C
			if (i < data->num_temp_offset) {
				data->temp[i][3] =
				  data->read(data, data->REG_TEMP_OFFSET[i]);

				printk("REG_TEMP_OFFSET[%d]\n",data->temp[i][3]);
			}
```
* _CoreFreq_ voltage core formula appears also wrong. Can't be `1.10 V` full load
Can you screenshot the view `Power & Voltage` in 2 cases: idle then full load

* CPB and XFR ratios seems too high. 
  Can you stress a single Core. Go to `Menu > Tools > Turbo Select CPU... > 00`

logan2611, 2019-07-14:
CPU at idle:
[image]
[image]

1 thread load:
[image]
[image]

All thread load using MPrime:
[image]
[image]

I'll get back to you with the offset in a little bit.

cyring, 2019-07-14:
Thanks, I'm also processing those values ...

* but I'm wondering how the `it87` can show `in0` at `1.43V` when CPU is idling

* I'm noticing the RAPL values computed by _CoreFreq_ : at full load, this can't be `10 W` 
  can you please press `w` to show me the Units

* Fyi, in last commit the DMI string should now be centered in footer.

cyring, 2019-07-14:
Sorry, take into account this trace code. It will help to locate the index of sensors.
```C
			if (i < data->num_temp_offset) {
				data->temp[i][3] =
				  data->read(data, data->REG_TEMP_OFFSET[i]);

				printk("REG_TEMP_OFFSET[%d]=%d\n",i,data->temp[i][3]);
			}
```

logan2611, 2019-07-14:
Output of modified driver:
```
[11215.102253] REG_TEMP_OFFSET[0]=0
[11215.102337] REG_TEMP_OFFSET[1]=0
[11215.102419] REG_TEMP_OFFSET[2]=0
[11215.102502] REG_TEMP_OFFSET[3]=0
[11215.102605] REG_TEMP_OFFSET[4]=0
[11215.102699] REG_TEMP_OFFSET[5]=0
```

Also here are the units:
[image]

Also not sure why it sometimes says 1.4V, probably brief periods of load.

cyring, 2019-07-14:
So driver say no temperature offset, will remove the 10°C 

If you have short periods of load, you should see some spikes in the UI. The measurement interval can be narrowed down to 100 ms in the Settings menu; and this idle session can be recorded in ASCII screens by pressing `ALT`+`p` 
Use `less -r` command to replay the output file left in current directory.

3700X RAPL units differ from the 2700X units : I have to verify the formula...

cyring, 2019-07-14:
Can you please comment those lines of code:
https://github.com/cyring/CoreFreq/blob/f2108e8445cd0fb8e5de72ec91bab363cca196c2/corefreqd.c#L493
like below:
```C
/*	if (maxCoreCount != 0) {
		Shm->Proc.Power.Unit.Watts  /= maxCoreCount;
		Shm->Proc.Power.Unit.Joules /= maxCoreCount;
	}	*/
```
Rebuild and check the consumed power facing the 3700X specs

G2G2G2G, 2019-07-14:
@cyring  want teamviewer access to my 3700X? (with root access) it's Mint 19 (ubuntu 18.04) cuz nothing else will boot right now lol.

cyring, 2019-07-14:
> @cyring want teamviewer access to my 3700X? (with root access) it's Mint 19 (ubuntu 18.04) cuz nothing else will boot right now lol.

yes, how do we proceed ?

G2G2G2G, 2019-07-14:
> > @cyring want teamviewer access to my 3700X? (with root access) it's Mint 19 (ubuntu 18.04) cuz nothing else will boot right now lol.
> 
> yes, how do we proceed ?

Do you have a way to contact you? Email? I was looking didn't see any

cyring, 2019-07-14:
> > > @cyring want teamviewer access to my 3700X? (with root access) it's Mint 19 (ubuntu 18.04) cuz nothing else will boot right now lol.
> > 
> > 
> > yes, how do we proceed ?
> 
> Do you have a way to contact you? Email? I was looking didn't see any

labs@cyring.fr

G2G2G2G, 2019-07-14:
Did you receive it?

cyring, 2019-07-14:
> Did you receive it?

Yes, replying

cyring, 2019-07-14:
To build and run driver w/o caring the NMI watchdog
```
make MSR_CORE_PERF_UCC=MSR_IA32_APERF MSR_CORE_PERF_URC=MSR_IA32_MPERF clean all
```

cyring, 2019-07-14:
OK Guys,
Here is the deal with Ryzen (I should have read closer the specs) :  
the AMD counter is per Physical Core (not event Thread Core) whereas the Intel counter is the sum of all Cores.

I have to refactor parts of driver and daemon code to handle this difference.
Teamspeak has unfortunately ends out but I have every changes in my head.

cyring, 2019-07-14:
Hello,

Can you test the Power measurements of the latest version ?
1. Press `V`
2. For each case { idle; single Core load; full load }
    * Monitor the `Cores` power against your 3700X specs
    * Check if the `Package` power has no regression
3. screenshots welcomed

logan2611, 2019-07-16:
Idle: 
[image]

Single Core Load (Using Turbo Select CPU):
[image]

All Core Load (Using Atomic Burn):
[image]

All Core Load (Using MPrime):
[image]

cyring, 2019-07-16:
* Thanks for the screenshots: full load, I was not expecting such low Joule values
For 2700X, in the issue #54, I'm providing few source code change [here](https://github.com/cyring/CoreFreq/issues/54#issuecomment-511413376) to fix the Power formula

* But there's another linked issue: despite the SMT is enabled, I don't see the two light on CPU numbers (first column). This could either come from the terminal palette - or - one physical core and its sibling thread core are not defined. Last case is a sign of a topology issue.
Can you post `corefreq-cli -m`

logan2611, 2019-07-17:
Sure
```
CPU Pkg  Apic  Core Thread  Caches      (w)rite-Back (i)nclusive              
 #   ID   ID    ID     ID  L1-Inst Way  L1-Data Way      L2  Way      L3  Way 
00: BSP     0     0      0      32  8        32  8       512  8     32768  9  
01:   0     2     1      0      32  8        32  8       512  8     32768  9  
02:   0     4     2      0      32  8        32  8       512  8     32768  9  
03:   0     6     3      0      32  8        32  8       512  8     32768  9  
04:   0     8     4      0      32  8        32  8       512  8     32768  9  
05:   0    10     5      0      32  8        32  8       512  8     32768  9  
06:   0    12     6      0      32  8        32  8       512  8     32768  9  
07:   0    14     7      0      32  8        32  8       512  8     32768  9  
08:   0     1     0      1      32  8        32  8       512  8     32768  9  
09:   0     3     1      1      32  8        32  8       512  8     32768  9  
10:   0     5     2      1      32  8        32  8       512  8     32768  9  
11:   0     7     3      1      32  8        32  8       512  8     32768  9  
12:   0     9     4      1      32  8        32  8       512  8     32768  9  
13:   0    11     5      1      32  8        32  8       512  8     32768  9  
14:   0    13     6      1      32  8        32  8       512  8     32768  9  
15:   0    15     7      1      32  8        32  8       512  8     32768  9
```

cyring, 2019-07-17:
Thanks, topology looks OK. Thread layout different than 2700X if I remember.
Make sure you are testing with the latest repo; and source code fully rebuilt.

cyring, 2019-07-21:
Hello,

To debug the RAPL [issue](https://github.com/cyring/CoreFreq/issues/57#issuecomment-513529506), can you please provide the screenshot of the view `Power & Voltage` when all cores are fully stressed. Press `F3 > Power & Voltage`

Use _only_ the integrated algorithm `F2 > Tools > Conic > Hyperboloid of two sheets` to stress the processor.

I will also need `F4 > Power & Thermal`

Please update to the latest _CoreFreq_ version, Thanks for helping.

logan2611, 2019-07-26:
Just updated CoreFreq to the latest commit.

F3 > Power & Voltage:
[image]

F4 > Power & Thermal:
[image]

cyring, 2019-07-26:
Sum of Cores is higher than the 105W TDP

Facing the 2700X [measurement] (https://user-images.githubusercontent.com/9773655/61589142-fca87d00-ab73-11e9-99e9-8420eadd51e2.png) the Package power is an issue. Although both processors are following the same code path, and have same units, I don't find why such gap.

cyring, 2019-07-26:
One idea could be the CCX topology. May be, one CCX is not counting for the other CCX.
`corefreqk.ko` can be launched with a CPU binding to the _Service Processor_ which is dedicated to Package counters reading.
Will you get very different power measurements depending on the CCX ?

G2G2G2G, 2019-07-27:
@logan2611 he's been on my machine for probably like 10 hours total last time & today
CPU: AMD Ryzen 7 3700X
Mobo: ASUS ROG B450-F

my system is similar unfortunately:
CPU: AMD Ryzen 7 3700X
Mobo: ASUS ROG CROSSHAIR VI HERO X370

Know anyone that has other motherboards? I have a 3600X with a B450 MSI motherboard at my office, I'll try it but idk.

cyring, 2019-07-28:
Hello,
Last version brings the AMD Core Complex ID in the topology, could you please add your screenshot in this [issue](https://github.com/cyring/CoreFreq/issues/1#issuecomment-515750513)

cyring, 2019-09-02:
X570 is apparently the only chipset providing good power measure; I believe no more issue. 
Feel free to re-open this issue in any case.

closed 2019-09-02 by cyring