cyring/CoreFreq #134: How to get the values of TCC activation temperature(IA32_Temperature_Target - T_offset ) in Intel SandyBridge platform?

NRevathi, 2019-07-20:


cyring, 2019-07-20:
Hello,
If you're asking for details, try the Client in json output. 

Please provide more information about what you can not get ?

NRevathi, 2019-07-20:
./corefreq-cli -j
./corefreq-cli -s
gives the Target/Junction Temperature as 100C of the processor. I wanted to get the value of T_offset i.e. lower thermal threshold limit of the processor at which the Thermal Monitoring mechanism is triggered.
Thanks

cyring, 2019-07-20:
Can you put a finger on the right MSR register ?
Those specs look like the closest subjects which feat your need:
* The TM2 Transition Target
* On of the Threshold Value fields
* Other ?

[image]
[image]
[image]

cyring, 2019-07-22:
Or is it this one you are interested in ?

https://github.com/cyring/CoreFreq/blob/ee915c4cb803208cf3326cfb1e8e0fa754ef391e/intelmsr.h#L883

From specs: 
> MSR_TEMPERATURE_TARGET (1A2H) Package
> TCC Activation Offset (R/W) [27:24]
> Specifies a temperature offset in degrees C from the temperature target (bits 23:16). PROCHOT# will assert at the offset target temperature. Write is permitted only if MSR_PLATFORM_INFO.[30] is set.

cyring, 2020-03-18:
Postponed to the Roadmap #169

closed 2020-03-18 by cyring