cyring/CoreFreq #137: DKMS config uses current kernel rather than the one passed to it

jonjonw, 2019-08-06:
There seems to be an issue with the dkms config when using a different kernel. corefreq attempts to build for the current kernel instead of the one passed to it by dkms. Here is what I'm getting, note it is passed 5.2.**6**-1-ck-skylake but uses 5.2.**5**-1-ck-skylake:

```
$ sudo pacman -Syu
...
==> dkms install corefreqk/1.61 -k 5.2.6-1-ck-skylake
Error! Bad return status for module build on kernel: 5.2.6-1-ck-skylake (x86_64)
Consult /var/lib/dkms/corefreqk/1.61/build/make.log for more information.
```
```
$ cat /var/lib/dkms/corefreqk/1.61/build/make.log
...
/lib/modules/5.2.5-1-ck-skylake/build: No such file or directory.
...
```

cyring, 2019-08-06:
I presume dkms is not ck kernel ready ?

Thus clone directly source code form GitHub, change to the CoreFreq working directory then build and install with:
```
make clean all
make install
make module-install
```

jonjonw, 2019-08-06:
I'll try again on the next kermel upgrade since I've already booted into the new kernel. Once I'm in the new kermel, ```dkms autoinstall``` works fine.

I have another dkms module (vboxhost) installed and it doesn't have the same issue. Here is a bit more from when I ran the upgrade command:

```
$ sudo pacman -Syu
...
==> dkms install corefreqk/1.61 -k 5.2.6-1-ck-skylake
Error! Bad return status for module build on kernel: 5.2.6-1-ck-skylake (x86_64)
Consult /var/lib/dkms/corefreqk/1.61/build/make.log for more information.
...
==> dkms install vboxhost/6.0.10_OSE -k 5.2.6-1-ck-skylake
[ No error... installs for the new kernel ]
...
```
I was thinking it might be a packaging issue but since dkms.conf is from this project I figured that likely where this issue is. I'll dig in more on my next kernel upgrade.

jonjonw, 2019-08-06:
I looked at dkms.conf and see what you mean for ```$source_tree``` now. Looks like an issue with the package not passing it correctly so I'll get in touch with the package maintainer. Thanks!

jonjonw, 2019-08-06:
I dug around dkms in archlinux a bit and I'm going to try setting ```KERNELDIR="$kernel_source_dir"``` in dkms.conf and see if it works for my next kernel upgrade.

I"m not sure if this variable is unique to archlinux but I'm guessing not so I'll re-open this and report back on my next kernel upgrade.

cyring, 2019-08-06:
Thanks for your detailed results. I'll also reinstall the CK kernel to debug the issue.

jonjonw, 2019-08-08:
```KERNELDIR="$kernel_source_dir"``` didn't seem to do the trick. It sill tries to use the currently running kernel.

I tried manually setting ```KERNELDIR="/lib/modules/5.2.7-1-ck-skylake/build"``` in */usr/src/corefreqk-1.61/dkms.conf* but it seems that is being ignored and it still uses the directory for the currently running kernel:

```
$ uname -r
5.2.6-1-ck-skylake

$ ls /lib/modules/
5.2.7-1-ck-skylake  extramodules-ck-skylake

$ sudo dkms install corefreqk/1.61 -k 5.2.7-1-ck-skylake
...
make -j4 KERNELRELEASE=5.2.7-1-ck-skylake -C /usr/src/corefreqk-1.61...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.2.7-1-ck-skylake (x86_64)
Consult /var/lib/dkms/corefreqk/1.61/build/make.log for more information.
```

```
$ cat /var/lib/dkms/corefreqk/1.61/build/make.log
...
make[1]: *** /lib/modules/5.2.6-1-ck-skylake/build: No such file or directory.
...
```

```
grep KERNELDIR /usr/src/corefreqk-1.61/dkms.conf
KERNELDIR="/lib/modules/5.2.7-1-ck-skylake/build"
```

Unless I'm reading the Makefile wrong, it looks like it should only use the default kernel if KERNELDIR isn't set?
```
grep KERNELDIR /usr/src/corefreqk-1.61/Makefile
KERNELDIR ?= /lib/modules/$(shell uname -r)/build
...
```
I'll do some more digging later.

cyring, 2019-08-27:
Hello,
Did you manage to build as you wish ? Closing the issue ?

cyring, 2019-09-01:
KERNELDIR looks OK.  
Below, building _CoreFreq_ for kernel 5.2.11 from current running 5.2.10
```
$ (uname -a ; echo ; make KERNELDIR=/lib/modules/5.2.11-arch1-1-ARCH/build/ clean all ; echo ; modinfo corefreqk.ko|grep "vermagic")>/tmp/build.log 2>&1
$ cat /tmp/build.log
```
```
Linux RAM 5.2.10-arch1-1-ARCH #1 SMP PREEMPT Thu Aug 29 19:38:41 CEST 2019 x86_64 GNU/Linux

rm -f corefreqd corefreq-cli
make -j1 -C /lib/modules/5.2.11-arch1-1-ARCH/build/ M=/home/cyril/src/CoreFreq clean
make[1]: Entering directory '/usr/lib/modules/5.2.11-arch1-1-ARCH/build'
  CLEAN   /home/cyril/src/CoreFreq/.tmp_versions
  CLEAN   /home/cyril/src/CoreFreq/Module.symvers
make[1]: Leaving directory '/usr/lib/modules/5.2.11-arch1-1-ARCH/build'
cc  -Wall -pthread -c corefreqd.c \
	-D FEAT_DBG=1 -o corefreqd.o
cc  -Wall -c corefreqm.c \
	-D FEAT_DBG=1 -o corefreqm.o
cc  -Wall corefreqd.c corefreqm.c \
	-D FEAT_DBG=1 -o corefreqd -lpthread -lm -lrt
cc  -Wall -c corefreq-cli.c \
	-D FEAT_DBG=1 -o corefreq-cli.o
cc  -Wall -c corefreq-ui.c \
	-D FEAT_DBG=1 -o corefreq-ui.o
cc  -Wall -c corefreq-cli-rsc.c \
	-D FEAT_DBG=1 -o corefreq-cli-rsc.o
cc  -Wall -c corefreq-cli-json.c \
	-D FEAT_DBG=1 -o corefreq-cli-json.o
cc  -Wall -c corefreq-cli-extra.c \
	-D FEAT_DBG=1 -o corefreq-cli-extra.o
cc  -Wall \
	corefreq-cli.c corefreq-ui.c corefreq-cli-rsc.c \
	corefreq-cli-json.c corefreq-cli-extra.c \
	-D FEAT_DBG=1 -o corefreq-cli -lm -lrt
make -j1 -C /lib/modules/5.2.11-arch1-1-ARCH/build/ M=/home/cyril/src/CoreFreq modules
make[1]: Entering directory '/usr/lib/modules/5.2.11-arch1-1-ARCH/build'
  CC [M]  /home/cyril/src/CoreFreq/corefreqk.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/cyril/src/CoreFreq/corefreqk.mod.o
  LD [M]  /home/cyril/src/CoreFreq/corefreqk.ko
make[1]: Leaving directory '/usr/lib/modules/5.2.11-arch1-1-ARCH/build'

vermagic:       5.2.11-arch1-1-ARCH SMP preempt mod_unload 
```

cyring, 2020-01-21:
With the latest Makefile and dkms instructions to build against a Kernel directory, do you close the issue ?

jonjonw, 2020-01-21:
Thanks. I'll check back when I do my next kernel upgrade and let you know if it's working.

jonjonw, 2020-01-23:
Looks like it's working great now. Thanks!

closed 2020-01-23 by jonjonw