cyring/CoreFreq #141: CPU data missing on AMD Epyc 7451

munorc, 2019-08-14:
Running on a dual CPU AMD Epyc 7451 machine, with SMT enabled, 96 cores.   Data is shown for cores 0-63.  Cores 64-96 show no data.

cyring, 2019-08-14:
Hello, thank you for trying _CoreFreq_

How long time do you give me to enhance the source code to handle more than 64 CPUs ?
This will require several tests from your side. 
Regards
Cyril

munorc, 2019-08-14:
# ./corefreq-cli -s -m
Processor                                     [AMD EPYC 7451 24-Core Processor ]
|- Architecture                                                    [EPYC/Naples]
|- Vendor ID                                                      [AuthenticAMD]
|- Microcode                                                        [ 134222375]
|- Signature                                                            [ 8F_01]
|- Stepping                                                             [     2]
|- Online CPU                                                           [ 96/96]
|- Base Clock                                                           [ 99.81]
|- Frequency            (MHz)                      Ratio                        
                 Min    399.25                    <   4 >                       
                 Max   2295.70                    <  23 >                       
|- Factory                                                              [100.00]
                       2300                       [  23 ]                       
|- Performance                                                                  
   |- OSPM                                                                      
                 TGT   2295.70                    <  23 >                       
|- Turbo Boost                                                          [UNLOCK]
                 XFR   3194.01                    [  32 ]                       
                 CPB   2894.58                    [  29 ]                       
                  1C   1796.63                    <  18 >                       
                  2C   1197.76                    <  12 >                       
|- Uncore                                                               [  LOCK]
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N,N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AVX-512      [N]  BMI1/BMI2 [Y/Y]        CLFSH [Y]        CMOV [Y] 
|- CMPXCH8      [Y]     CMPXCH16 [Y]         F16C [Y]         FPU [Y] 
|- FXSR         [Y]    LAHF/SAHF [Y]    MMX/Ext [Y/Y]     MONITOR [Y] 
|- MOVBE        [Y]          MPX [N]     PCLMULDQ [Y]      POPCNT [Y] 
|- RDRAND       [Y]       RDSEED [Y]       RDTSCP [Y]         SEP [Y] 
|- SGX          [N]          SSE [Y]         SSE2 [Y]        SSE3 [Y] 
|- SSSE3        [Y]  SSE4.1/4A [Y/Y]       SSE4.2 [Y]     SYSCALL [Y] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Present]
|- 100 MHz multiplier Control                            100MHzSteps   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Present]
|- Advanced Programmable Interrupt Controller                   APIC   [Present]
|- Core Multi-Processing                                  CMP Legacy   [Present]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Present]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Missing]
|- CPL Qualified Debug Store                                  DS-CPL   [Missing]
|- 64-Bit Debug Store                                         DTES64   [Missing]
|- Fast-String Operation                                Fast-Strings   [Missing]
|- Fused Multiply Add                                       FMA|FMA4   [Present]
|- Hardware Lock Elision                                         HLE   [Missing]
|- Long Mode 64 bits                                         IA64|LM   [Present]
|- LightWeight Profiling                                         LWP   [Missing]
|- Machine-Check Architecture                                    MCA   [Present]
|- Model Specific Registers                                      MSR   [Present]
|- Memory Type Range Registers                                  MTRR   [Present]
|- No-Execute Page Protection                                     NX   [Present]
|- OS-Enabled Ext. State Management                          OSXSAVE   [Present]
|- Physical Address Extension                                    PAE   [Present]
|- Page Attribute Table                                          PAT   [Present]
|- Pending Break Enable                                          PBE   [Missing]
|- Process Context Identifiers                                  PCID   [Missing]
|- Perfmon and Debug Capability                                 PDCM   [Missing]
|- Page Global Enable                                            PGE   [Present]
|- Page Size Extension                                           PSE   [Present]
|- 36-bit Page Size Extension                                  PSE36   [Present]
|- Processor Serial Number                                       PSN   [Missing]
|- Restricted Transactional Memory                               RTM   [Missing]
|- Safer Mode Extensions                                         SMX   [Missing]
|- Self-Snoop                                                     SS   [Missing]
|- Time Stamp Counter                                            TSC [Invariant]
[... cut ...]
89:   6   109  1  22     1      64  4        32  8       512  8 i   32768 64w 
90:   7   113  0  24     1      64  4        32  8       512  8 i   32768 64w 
91:   7   115  0  25     1      64  4        32  8       512  8 i   32768 64w 
92:   7   117  0  26     1      64  4        32  8       512  8 i   32768 64w 
93:   7   121  1  28     1      64  4        32  8       512  8 i   32768 64w 
94:   7   123  1  29     1      64  4        32  8       512  8 i   32768 64w 
95:   7   125  1  30     1      64  4        32  8       512  8 i   32768 64w

munorc, 2019-08-14:
The place where data is missing is in the various windows that are updated e.g. Dashboard.

cyring, 2019-08-14:
Thank you.

* The Processor data appear OK, please confirm if they match BIOS ?
* The topology offers enough distinct values per CPU.  
  Fyi, the Package ID is not the Socket as expected with Intel, but the Node ID of Zen
https://github.com/cyring/CoreFreq/blob/e21a6d5480fff6a095edec7b6ea01310ad25c0e9/corefreqk.c#L1352

Before I jump into code refactoring I have to clarify this questions:
What's the max Core count per Socket in the whole x86 architectures range ?
1. if less or equal 64 then I can _instantiate_ the API structures per Socket
2. if higher than 64 then I preserve the API but I have to refactor any asm atomic operations

cyring, 2019-08-14:
Meanwhile, _CoreFreq_ will run fine with your EPYC if you disable SMT to lmit to 48 CPUs

munorc, 2019-08-14:
EPYC (Naples has a max of 32 cores/64 threads per package.   The 7451 is 24 core, 48 threads). 
EPYC 2 (Rome) has a max of 64 cores per package. We should have those in place soon as well (256 threads per server node with SMT enabled).

cyring, 2019-08-14:
Huge impacts, please contact me on labs[at]cyring[dot]fr

cyring, 2019-08-16:
Hello,
Does this code returns 1 with your EPYC ?
```
/*
	CMPXCHG16B.c

	-- Build & Run --
	cc CMPXCHG16B.c -o CMPXCHG16B
	./CMPXCHG16B; echo $?
*/

#define CMPXCHG16B( _dest, _src )					\
({									\
	register unsigned char _ret = 0;				\
									\
	__asm__ volatile						\
	(								\
		"movq		%[losrc], %%rax"	"\n\t"		\
		"movq		%[hisrc], %%rdx"	"\n\t"		\
	"rex "	"cmpxchg16b	%[dest]"		"\n\t"		\
		"setz		%[ret]" 				\
		: [dest] "=m" (_dest), [ret] "+r" (_ret)		\
		: [losrc] "m" (_src[0]), [hisrc] "m" (_src[1])		\
		: "cc", "memory", "%rax", "%rbx", "%rcx", "%rdx"	\
	);								\
									\
	_ret;								\
})

int main(int argc, char *argv[])
{
	unsigned long long src[2] __attribute__ ((aligned (128))) = {
	0b1111101010101010101010101010101010101010101010101010101010101010,
	0b1111111111111111111111111111111100000000000000000000000000000001
	}, dest[2] __attribute__ ((aligned (128))) = {
	0b1111101010101010101010101010101010101010101010101010101010101010,
	0b1111111111111111111111111111111100000000000000000000000000000001
	};

	unsigned char ret = CMPXCHG16B(src, dest);

	return((int) ret);
}
```

munorc, 2019-08-16:
It appears to.

$ cc CMPXCHG16B.c -o CMPXCHG16B

$ ./CMPXCHG16B; echo $?
1

cyring, 2019-08-16:
Great, we have our first 128 bits primitive for a 128 CPUs bitmap.

Unfortunately, I still have found no instruction to handle 256 CPUs.
And any FPU extension that can do with 256 bits registers, such as AVX2, is not supported by the Kernel.

cyring, 2019-08-17:
I made [mt128.c](https://github.com/cyring/Tips/blob/master/C/mt128.c) to launch 128 threads, one per SMT Core. The main thread will join them using a 128 bits room bitmap. After 1000 loops, main returns the remaining counter. It has to echo zero.

I have try to test it in a 128 x CPU VM but it is too slow and couldn't complete. You should have a better result with your hardware.

Btw, the VM shows that CoreFreq can currently display the tasks of 128 CPUs.
[image]

munorc, 2019-08-17:
mt128 returns 0

cyring, 2019-08-17:
Hello,

Can you try this version. It focuses on driver and daemon back-ends only to handle up to 128 Cores.
UI has not been refactored yet, thus you won't read good CPU numbers.

Please provides screenshot of the Client with the last 72 Cores. I would check if we read frequencies for all CPUs.

[CoreFreq.tar.gz](https://github.com/cyring/CoreFreq/files/3512268/CoreFreq.tar.gz)

cyring, 2019-08-18:
New optimized development version for testings.
[CoreFreq.tar.gz](https://github.com/cyring/CoreFreq/files/3512591/CoreFreq.tar.gz)

munorc, 2019-08-18:
Which screen are ypu lookig for a a screenshot of?  Most data is being displayed as "inf"

cyring, 2019-08-18:
default startup screen

munorc, 2019-08-18:
[image]

cyring, 2019-08-18:
Is it virtualized or bare-metal ?

munorc, 2019-08-18:
bare metal

cyring, 2019-08-18:
Has the module been unload b/c between the 2 versions the API has changed.

cyring, 2019-08-18:
# Development revision 99
* Source code:
[CoreFreq.tar.gz](https://github.com/cyring/CoreFreq/files/3513339/CoreFreq.tar.gz)

* Driver & Daemon resized to 256 CPUs
* Client, beside Dashboard, UI impacts fixed

* What it may look like
[image]

adatum, 2019-08-18:
[image]

cyring, 2019-08-18:
@adatum :1st_place_medal: Thank you. Apparently no regression

Now waiting for EPYC results and the high Core Count properly support ?

cyring, 2019-08-23:
# Development revision 97
(_remark: development revisions in descending order_)  

* Source code:
[CoreFreq_163_97.tar.gz](https://github.com/cyring/CoreFreq/files/3532380/CoreFreq_163_97.tar.gz)

  * Try to improve the atomic operations to support 256 Cores  
  * Fixed several memory alignements 
  * UI: fixed CPUs scrolling

cyring, 2019-08-23:
# Development revision 96
* Source code:
[CoreFreq_163_96.tar.gz](https://github.com/cyring/CoreFreq/files/3534354/CoreFreq_163_96.tar.gz)
  * Daemon: restored bit synchronization of threads
  * Driver: fixed OnLine CPU count

cyring, 2019-08-24:
Hello,
The 256 SMT-Cores Scaling version is released.
Thank you for your tests.

munorc, 2019-08-24:
Just tested on the 2x AMD Epyc (Naples). I see data for all 96 cores.

cyring, 2019-09-16:
Hello,
Can you please try the stress tests of the last CoreFreq version.
If Conics, especially, crash your EPYC processor ?
In the issue #131 , Threadripper hard locks with Conics

closed 2019-10-22 by cyring

cyring, 2019-10-22:
Hello,
I've a request to fix the topology with dual EPYC. Can you help ?

cyring, 2020-07-06:
Hello,
Among the last changes in issue #195 , an I/O ASM access to the FCH have been added: can you please check if the last _CoreFreq_ version is starting fine ?
Thank you

cyring, 2020-10-29:
Hello,

Can you help with an issue about the inconsistency of the instructions counter with Zen 1st generation architecture.

Based on the latest _CoreFreq_ version, show this view while EPYC is idling:
[image]

Regards
CyrIng