cyring/CoreFreq #142: Support of dual Xeon E5-2680 v2

louwrentius, 2019-08-15:
I have not decoded the geometry and thus DIMM size

Per CPU it has indeed 4 channels. The DRAM speed in MHz is wrong according to my purchase slip 

> RAM Geheugen DDR3: 128GB 10600R 1333MHz / 16x 8GB DDR3

and what HP iLO reports: 

>Processor 2 	12 	64 GB 	1333 MHz 	1.35 V 
>Processor 1   	12 	64 GB 	1333 MHz 	1.35 V 

dmidecode --type 17

``
Memory Device
	Array Handle: 0x1001
	Error Information Handle: Not Provided
	Total Width: 72 bits
	Data Width: 64 bits
	Size: 8192 MB
	Form Factor: DIMM
	Set: 23
	Locator: PROC  2 DIMM 12 
	Bank Locator: Not Specified
	Type: DDR3
	Type Detail: Synchronous Registered (Buffered)
	Speed: 1333 MT/s
	Manufacturer: UNKNOWN
	Serial Number: Not Specified
	Asset Tag: Not Specified
	Part Number: NOT AVAILABLE       
	Rank: 2
	Configured Clock Speed: 1333 MT/s
	Minimum Voltage: 1.35 V
	Maximum Voltage: 1.5 V
	Configured Voltage: 1.35 V
``

cyring, 2019-08-15:
Thank you
I believe your DIMM specs in DMI are somewhat the SPD data. But what CoreFreq is decoding @ 1866 MHz, through the PCI registers comes the IMC controller, thus the current timings.
I would suggest instead to compare with the BIOS screen, if available ?

louwrentius, 2019-08-15:
I've rebooted the machine. Unfortunately, the BIOS screen does not show memory speed.

cyring, 2019-08-15:
> I've rebooted the machine. Unfortunately, the BIOS screen does not show memory speed.

It happens with server BIOS, I will suggest to boot [MemTest86](https://www.memtest86.com/)

louwrentius, 2019-08-15:
[image]

cyring, 2019-08-15:
Unfortunately the MemTest report seems not a great help.
How the DDR3 can have a CAS latency of 19 !

louwrentius, 2019-08-15:
This is beyond my wheelhouse honestly, but if there is a tool I can run that should give the definitive answer I’m ok with that.

cyring, 2019-08-16:
Found those datasheets. I'm not sure if it explains why I have decode the IMC speed @ 1866

* [Configuring and using DDR3 memory with HP ProLiant Gen8 Servers](https://community.hpe.com/hpeb/attachments/hpeb/itrc-264/147773/1/MemoryGen8.pdf)
> HP engineers have designed the ProLiant Gen8 platform architecture to run at memory speeds up to 1866 MT/s once processor chipsets and DIMMs that support this speed are available.

* [Memory options](http://h17007.www1.hpe.com/docs/iss/DL380pGen8/setup_install/advanced/Content/172025.htm)
> Quad-rank PC3L-14900 (DDR3-1866) LRDIMMs operating as dual-rank DIMMs at up to 1866 MT/s

cyring, 2019-08-16:
About DIMM geometry, I have not implemented MP controller yet but only single socket IVB-EP.
Can you plz post `lspci -nn` ?

louwrentius, 2019-08-16:
`00:00.0 Host bridge [0600]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 DMI2 [8086:0e00] (rev 04)
00:01.0 PCI bridge [0604]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 PCI Express Root Port 1a [8086:0e02] (rev 04)
00:01.1 PCI bridge [0604]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 PCI Express Root Port 1b [8086:0e03] (rev 04)
00:02.0 PCI bridge [0604]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 PCI Express Root Port 2a [8086:0e04] (rev 04)
00:02.1 PCI bridge [0604]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 PCI Express Root Port 2b [8086:0e05] (rev 04)
00:02.2 PCI bridge [0604]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 PCI Express Root Port 2c [8086:0e06] (rev 04)
00:02.3 PCI bridge [0604]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 PCI Express Root Port 2d [8086:0e07] (rev 04)
00:03.0 PCI bridge [0604]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 PCI Express Root Port 3a [8086:0e08] (rev 04)
00:03.1 PCI bridge [0604]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 PCI Express Root Port 3b [8086:0e09] (rev 04)
00:03.2 PCI bridge [0604]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 PCI Express Root Port 3c [8086:0e0a] (rev 04)
00:03.3 PCI bridge [0604]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 PCI Express Root Port 3d [8086:0e0b] (rev 04)
00:04.0 System peripheral [0880]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Crystal Beach DMA Channel 0 [8086:0e20] (rev 04)
00:04.1 System peripheral [0880]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Crystal Beach DMA Channel 1 [8086:0e21] (rev 04)
00:04.2 System peripheral [0880]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Crystal Beach DMA Channel 2 [8086:0e22] (rev 04)
00:04.3 System peripheral [0880]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Crystal Beach DMA Channel 3 [8086:0e23] (rev 04)
00:04.4 System peripheral [0880]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Crystal Beach DMA Channel 4 [8086:0e24] (rev 04)
00:04.5 System peripheral [0880]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Crystal Beach DMA Channel 5 [8086:0e25] (rev 04)
00:04.6 System peripheral [0880]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Crystal Beach DMA Channel 6 [8086:0e26] (rev 04)
00:04.7 System peripheral [0880]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Crystal Beach DMA Channel 7 [8086:0e27] (rev 04)
00:05.0 System peripheral [0880]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 VTd/Memory Map/Misc [8086:0e28] (rev 04)
00:05.2 System peripheral [0880]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 IIO RAS [8086:0e2a] (rev 04)
00:05.4 PIC [0800]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 IOAPIC [8086:0e2c] (rev 04)
00:11.0 PCI bridge [0604]: Intel Corporation C600/X79 series chipset PCI Express Virtual Root Port [8086:1d3e] (rev 05)
00:1a.0 USB controller [0c03]: Intel Corporation C600/X79 series chipset USB2 Enhanced Host Controller #2 [8086:1d2d] (rev 05)
00:1c.0 PCI bridge [0604]: Intel Corporation C600/X79 series chipset PCI Express Root Port 1 [8086:1d10] (rev b5)
00:1c.7 PCI bridge [0604]: Intel Corporation C600/X79 series chipset PCI Express Root Port 8 [8086:1d1e] (rev b5)
00:1d.0 USB controller [0c03]: Intel Corporation C600/X79 series chipset USB2 Enhanced Host Controller #1 [8086:1d26] (rev 05)
00:1e.0 PCI bridge [0604]: Intel Corporation 82801 PCI Bridge [8086:244e] (rev a5)
00:1f.0 ISA bridge [0601]: Intel Corporation C600/X79 series chipset LPC Controller [8086:1d41] (rev 05)
00:1f.2 SATA controller [0106]: Intel Corporation C600/X79 series chipset 6-Port SATA AHCI Controller [8086:1d02] (rev 05)
01:00.0 System peripheral [0880]: Hewlett-Packard Company Integrated Lights-Out Standard Slave Instrumentation & System Support [103c:3306] (rev 05)
01:00.1 VGA compatible controller [0300]: Matrox Electronics Systems Ltd. MGA G200EH [102b:0533]
01:00.2 System peripheral [0880]: Hewlett-Packard Company Integrated Lights-Out Standard Management Processor Support and Messaging [103c:3307] (rev 05)
01:00.4 USB controller [0c03]: Hewlett-Packard Company Integrated Lights-Out Standard Virtual USB Controller [103c:3300] (rev 02)
02:00.0 RAID bus controller [0104]: Hewlett-Packard Company Smart Array Gen8 Controllers [103c:323b] (rev 01)
03:00.0 Ethernet controller [0200]: Broadcom Inc. and subsidiaries NetXtreme BCM5719 Gigabit Ethernet PCIe [14e4:1657] (rev 01)
03:00.1 Ethernet controller [0200]: Broadcom Inc. and subsidiaries NetXtreme BCM5719 Gigabit Ethernet PCIe [14e4:1657] (rev 01)
03:00.2 Ethernet controller [0200]: Broadcom Inc. and subsidiaries NetXtreme BCM5719 Gigabit Ethernet PCIe [14e4:1657] (rev 01)
03:00.3 Ethernet controller [0200]: Broadcom Inc. and subsidiaries NetXtreme BCM5719 Gigabit Ethernet PCIe [14e4:1657] (rev 01)
1f:08.0 System peripheral [0880]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 QPI Link 0 [8086:0e80] (rev 04)
1f:09.0 System peripheral [0880]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 QPI Link 1 [8086:0e90] (rev 04)
1f:0a.0 System peripheral [0880]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Power Control Unit 0 [8086:0ec0] (rev 04)
1f:0a.1 System peripheral [0880]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Power Control Unit 1 [8086:0ec1] (rev 04)
1f:0a.2 System peripheral [0880]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Power Control Unit 2 [8086:0ec2] (rev 04)
1f:0a.3 System peripheral [0880]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Power Control Unit 3 [8086:0ec3] (rev 04)
[... cut ...]
3f:13.5 Performance counters [1101]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 QPI Ring Performance Ring Monitoring [8086:0e36] (rev 04)
3f:16.0 System peripheral [0880]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 System Address Decoder [8086:0ec8] (rev 04)
3f:16.1 System peripheral [0880]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Broadcast Registers [8086:0ec9] (rev 04)
3f:16.2 System peripheral [0880]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Broadcast Registers [8086:0eca] (rev 04)`

louwrentius, 2019-08-16:
[lspci.txt](https://github.com/cyring/CoreFreq/files/3508443/lspci.txt)

cyring, 2019-08-17:
Hello,
Can you try this version. It focuses on back-end only to handle up to 128 Cores. UI has been refactored yet.
Please provides screenshot of last 72 Cores if possible

[CoreFreq.tar.gz](https://github.com/cyring/CoreFreq/files/3512266/CoreFreq.tar.gz)

louwrentius, 2019-08-17:
Attached screenshots, I only have 40 logical cores.

cyring, 2019-08-18:
Hello,
Where are the screenshots , I don't find them.
New optimized development version : hope it will cover the 128 Cores ?
[CoreFreq.tar.gz](https://github.com/cyring/CoreFreq/files/3512588/CoreFreq.tar.gz)

cyring, 2019-08-30:
* Based on the latest CoreFreq version, at lines of this source file:
https://github.com/cyring/CoreFreq/blob/97618a3b18848e7a41ec55c3f6d3673f50e8b74f/corefreqk.c#L2925

* Can you change the 2 functions below.
```c
static PCI_CALLBACK SNB_EP_CTRL0(struct pci_dev *dev)
{	printk("SNB_EP_CTRL0(BUS[%d])\n", dev->bus->number);
	if (Proc->Uncore.CtrlCount < 1) {
		Proc->Uncore.CtrlCount = 1;
	}
	SNB_EP_CTRL(dev, 0);

	return(0);
}

static PCI_CALLBACK SNB_EP_CTRL1(struct pci_dev *dev)
{	printk("SNB_EP_CTRL1(BUS[%d])\n", dev->bus->number);
	if (Proc->Uncore.CtrlCount < 2) {
		Proc->Uncore.CtrlCount = 2;
	}
	SNB_EP_CTRL(dev, 1);

	return(0);
}
```

1. Rebuild
2. Load driver
3. Print the kernel log `dmesg`
4. Can you please post traces like `SNB_EP_CTRL#(BUS[#])`

cyring, 2019-09-03:
Up

cyring, 2020-03-18:
Do you have better results with the latest version ?
Or feel free to close the issue.

closed 2020-03-28 by cyring