cyring/CoreFreq #143: Haswell/Desktop - Energy Policy - Bias Hint

Waydjinn, 2019-08-27:
Bias Hint can be successfully changed on Haswell/Desktop, signature 06_3C in Experimental mode.

Please add it to the white-list.
[image]

cyring, 2019-08-28:
Thanks for your report.

On Clock Modulation is also depending on the Experimental mode: can you  please test several ODCM percents with this Haswell processor ?  

Remarks:  
* You'll have to stress CPU to check if the frequency ratio is indeed limited by ODCM.  
* Tools menu provides CPU stressing functions.

Regards.

Waydjinn, 2019-08-29:
ODCM is working fine.
See various DutyCycle stressed tests:
[image]
[image]
[image]
[image]

cyring, 2019-08-29:
Commit released.

cyring, 2019-08-30:
Thanks for your star.  
Enjoy _CoreFreq_ and stay tuned, more to come...

closed 2019-08-30 by cyring