cyring/CoreFreq #146: CPU 9980xe and 6850k support.

TDP-9000w, 2019-09-30:
I have a 9980xe and a 6850k that don't show the voltages on all cpus.
I saw another issue thread where some code changes were made to support.
What should I change to support these cpus?

Also, is there any way to make the corefreq cli run multi-instance?

TDP-9000w, 2019-09-30:
`./corefreq-cli -s -m
Processor                            [Intel(R) Core(TM) i9-9980XE CPU @ 3.00GHz]
|- Architecture                                                      [Skylake/X]
|- Vendor ID                                                      [GenuineIntel]
|- Microcode                                                        [  33554526]
|- Signature                                                           [  06_55]
|- Stepping                                                            [      4]
|- Online CPU                                                          [ 18/ 18]
|- Base Clock                                                          [ 99.997]
|- Frequency            (MHz)                      Ratio                        
                 Min   1199.97                    [  12 ]                       
                 Max   2999.92                    [  30 ]                       
|- Factory                                                             [100.000]
                       3000                       [  30 ]                       
|- Performance                                                                  
   |- OSPM                                                                      
                 TGT   4199.89                    <  42 >                       
|- Turbo Boost                                                         [ UNLOCK]
                  1C   4199.89                    <  42 >                       
                  2C   4199.89                    <  42 >                       
                  3C   4199.89                    <  42 >                       
                  4C   4199.89                    <  42 >                       
                  5C   4199.89                    <  42 >                       
                  6C   4199.89                    <  42 >                       
                  7C   4199.89                    <  42 >                       
                  8C   4199.89                    <  42 >                       
                  9C    199.99                    <   2 >                       
                 10C    399.99                    <   4 >                       
                 11C    799.98                    <   8 >                       
                 12C   1199.97                    <  12 >                       
                 13C   1599.96                    <  16 >                       
                 14C   1799.95                    <  18 >                       
                 15C   2399.94                    <  24 >                       
                 16C   2799.93                    <  28 >                       
|- Uncore                                                              [ UNLOCK]
                 Min   2699.93                    <  27 >                       
                 Max   2699.93                    <  27 >                       
|- TDP                                                           Level [  0:2  ]
   |- Programmable                                                     [ UNLOCK]
   |- Configuration                                                    [   LOCK]
   |- Turbo Activation                                                 [ UNLOCK]
             Nominal   2999.92                    [  30 ]                       
              Level1   2099.94                    [  21 ]                       
              Level2   2099.94                    [  21 ]                       
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AVX-512      [Y]  BMI1/BMI2 [Y/Y]      CLFLUSH [Y]        CMOV [Y] 
|- CMPXCHG8B    [Y]   CMPXCHG16B [Y]         F16C [Y]         FPU [Y] 
|- FXSR         [Y]    LAHF/SAHF [Y]    MMX/Ext [Y/N]  MONITOR/X[Y/N] 
|- MOVBE        [Y]          MPX [Y]    PCLMULQDQ [Y]      POPCNT [Y] 
|- RDRAND       [Y]       RDSEED [Y]       RDTSCP [Y]         SEP [Y] 
|- SGX          [N]          SSE [Y]         SSE2 [Y]        SSE3 [Y] 
|- SSSE3        [Y]  SSE4.1/4A [Y/N]       SSE4.2 [Y]     SYSCALL [Y] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Present]
|- Advanced Configuration & Power Interface                     ACPI   [Present]
|- Advanced Programmable Interrupt Controller                   APIC   [Present]
|- Core Multi-Processing                                  CMP Legacy   [Missing]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Direct Cache Access                                           DCA   [Present]
|- Debugging Extension                                            DE   [Present]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Present]
|- CPL Qualified Debug Store                                  DS-CPL   [Present]
|- 64-Bit Debug Store                                         DTES64   [Present]
|- Fast-String Operation                                Fast-Strings   [Present]
|- Fused Multiply Add                                       FMA|FMA4   [Present]
|- Hardware Lock Elision                                         HLE   [Present]
[... cut ...]
011:  0    36    18      0   32768  8     32768  8   1048576 16  25952256 11  
012:  0    38    19      0   32768  8     32768  8   1048576 16  25952256 11  
013:  0    40    20      0   32768  8     32768  8   1048576 16  25952256 11  
014:  0    48    24      0   32768  8     32768  8   1048576 16  25952256 11  
015:  0    50    25      0   32768  8     32768  8   1048576 16  25952256 11  
016:  0    52    26      0   32768  8     32768  8   1048576 16  25952256 11  
017:  0    54    27      0   32768  8     32768  8   1048576 16  25952256 11 `

TDP-9000w, 2019-09-30:
Full load with Prime95    4.1ghz  Stable vcore 1.135 (bios)   180degF  NH-D15S air cooled. (ugh)
[image]

cyring, 2019-10-01:
Nice processor!

* For all CPU voltage, move this switch case...  

https://github.com/cyring/CoreFreq/blob/a8c5ef0c36c815d59fd009e7b79e85c8e4e39616/corefreq-cli.c#L8071

... just bellow...  

https://github.com/cyring/CoreFreq/blob/a8c5ef0c36c815d59fd009e7b79e85c8e4e39616/corefreq-cli.c#L8058

* About multi-client support : it will impact Daemon. What's your use case? 

* Skylake/X Uncore is not fully implemented b/c lack of hardware to work/test with. Same for IMC.

* Screenshots, I see wrong power measurements of 4W full Cores stressed. To debug

TDP-9000w, 2019-10-01:
What's your use case?
Just local use...
I just want to run the regular corefreq-cli command in one terminal 
ALONG with a corefreq-cli -V in another terminal window.
When I do both, one of them stops.
Yea hardware should be cheaper!   Luck has it that I have access to this!
I can test stuff/screenshot/debug if you want to code for this platform.
Thanks, i will give the voltage fix a try.

TDP-9000w, 2019-10-01:
Good,  it shows the voltages on all cores...  since i have all cores vlocked right now they are the same.
[image]

Now I need to change the vcore to auto and watch the clocks, voltages, etc and write them down to get an idea of what each core can do.

cyring, 2019-10-01:
* Fix released in latest version 1.66.5

* About 4.7 GHz, try one of my stress tools: go to `[F2] Menu > Tools` or shortcut `[O]`
`Conics` variants involve the FPU; others stress the ALU  
Press `[F10]` at any time to stop stressing loops.

cyring, 2019-10-01:
Not sure if it is the right product [sheet] (https://ark.intel.com/content/www/us/en/ark/products/189126/intel-core-i9-9980xe-extreme-edition-processor-24-75m-cache-up-to-4-50-ghz.html), but I'm noticing some gaps:

* _CoreFreq_ got a TjMax of 110 whereas spec says 84°C
Please check if temperature in Celsius is right ?
* Turbo Boost or Turbo Max is version 3. 
In the Cli, do you read this version number beside TURBO in the "Technologies" window ?

I've noticed C-States are disabled in your BIOS. Could you activate them and check that they are well measured by _CoreFreq_
1. Core C-States
2. Package C-States 

Also can you enable in BIOS and verify the state of SpeedStep, HyperThreading and VT is well reported

About Power, this needs some debug code but we need a reference : not sure about a max of 165W

cyring, 2019-10-01:
* On-Demand Clock Modulation 
Could you check if this legacy feature works with Skylake/X

1. Activate the `Experimental` Mode in Settings
[image]
2. Enable `Clock Modulation`
[image]
3. Once `ODCM` enabled, change the `DutyCycle` to 25%
[image]
4. Starts a stress `Tool` on a selected CPU and check if the Core frequency is limited to 25% of its max
[image]
5. Restore the CPU nominal frequency by deactivating `Clock Modulation`. Frequency should go back its top value.
[image]

* Remarks:
1. _this is an unstested feature with SKL/X, please save, sync your files before testing_
2. If test is OK, I will remove the Experimental requirement for SKL/X in next _CoreFreq_ release

Thank you

cyring, 2019-10-01:
>  Some cores like to run 4.7ghz, and others hate it.

In [datasheet vol 1](https://www.intel.com/content/dam/www/public/us/en/documents/datasheets/6th-gen-x-series-datasheet-vol-1.pdf)
1. Turbo Boost Max 3.0, one of the Core should be set in BIOS to realize the highest performance.
This CPU is the one to stress in the `Tools > Turbo Select CPU...`
2. With my Nehalem, Turbo Boost works only with SpeedStep enabled.  
The `Technologies window` can let you enable them at any time.
3. Overclock the One Core Ratio in `Turbo Boost` section, item `1C`
[image]

$ 5.2.5: The Target Ratio has to be set to the max value.
1. set the `Bias Hint` to the lowest energy policy in `Power & Thermal` window  
[image]

2. in `Processor` window, the `OSPM TGT` can be set to a fixed ratio.  
  This requires however to let CoreFreq driver be registered as the OS CPU-Freq driver (thus blacklist any other drivers)
[image]

* In your previous output, HWP appears OFF.  
Not sure if the BIOS can present this feature. If feasible then:
1. enable HWP in `Performance` window  
2. set `HWP-EPP` to the lowest energy profile in `Power & Thermal` window  
3. set the `HWP TGT` Target and `HWP Max` Ratio to the highest value in `Processor` window.

* If the _CoreFreq_ kernel module is registered as the OS Idle driver then you can set one idle limit in the `Kernel data window` to reduce the C-State latency to C1 during the CPU stress.
[image]

cyring, 2019-10-01:
[Datasheet volume 2](https://www.intel.com/content/dam/www/public/us/en/documents/datasheets/7th-gen-core-family-desktop-s-processor-lines-datasheet-vol-2.pdf) Table 2-3: does the `lspci -nn` command show the HOST and DRAM Controller Device ID `906E9` ?

TDP-9000w, 2019-10-01:
Ok, lots of stuff to try.

The datasheets you linked are not the correct ones.

I found two datasheets here.
https://www.intel.com/content/www/us/en/products/processors/core/x-series/i9-9980xe.html#relatedresourcesblad

9980XE datasheet.
https://www.intel.com/content/www/us/en/products/processors/core/6th-gen-x-series-datasheet-vol-1.html

Spec update.
https://www.intel.com/content/www/us/en/products/docs/processors/core/6th-gen-x-series-spec-update.html

TDP-9000w, 2019-10-01:
So, Running two instances of corefreq-cli would allow me to monitor all 3 things at once, 
core voltages, core temperatures, and core frequency.
Switching the window to see the voltages causes a delay, so screenshots of all 3 are not possible for analysis.

Or....  It would be useful for the voltage page to show core temps also.
[image]
[image]

Does your stress tools verify errors and stop only the core that failed?
I use prime95 because it only stops the core that fails the test, but keeps others going.
This allows one to determine the weakest cores versus the strongest in relation to initial thermal stress package hot spots.

cyring, 2019-10-02:
> So, Running two instances of corefreq-cli would allow me to monitor all 3 things at once,
> core voltages, core temperatures, and core frequency.
> Switching the window to see the voltages causes a [...]

This is definitively an enhancement to add. In addition to Energy/Power counters per Core for capable Processors such as Zen. Planed for next days ...

> Does your stress tools verify errors and stop only the core that failed?
> I use prime95 because it only stops the core that fails the test, but keeps others going.
> This allows one to determine the weakest cores versus the strongest in relation to initial thermal stress package hot spots.

Nop, but it's a good use case to add.
So far, _CoreFreq_ can claim the OS to disable a Core : press shortcut `[#]` then select CPU
[image]

TDP-9000w, 2019-10-02:
So I  think my memory controller is NOT in the coretypes.h  you mentioned...
I saw numbers similar, but not exact.

Controller Device ID  of   :   0580      or   0880     or     8086,   
I don't know which one,   WTF is with intel and their datasheet not mentioning it.../ missing vol 2?
Not something we expect from such a company...  
 Whats next, intel cpus inside our crackerjack boxes?

 I would just add it to the list somewhere formatting the entry like the others, but I do not know if that is all that is needed, nor do I know which ID should be used from the output of lspci above, I hope i included the proper info from that output.

cyring, 2019-10-02:
1. Location where you will store IDs for probing
https://github.com/cyring/CoreFreq/blob/734baac66a23bdfa4e7bb192f92119bca96db154/corefreqk.h#L1353

* This list contains device id to probe and the associated function which query registers
```c
static struct pci_device_id PCI_Skylake_X_ids[] = {
	{
	PCI_DEVICE(PCI_VENDOR_ID_INTEL, 0x1234),
		.driver_data = (kernel_ulong_t) SKL_IMC
	},
	{0, }
};
```
* you just have to replace `0x1234` with the IMC id. Datasheet mentions it HA or SA Agent
I would start with `0x2066` from your lspci output
* _list must be terminated by `{0, }`_

2. Rebuild and test
* if lucky `SKL_IMC` function may decode bits
  _if your IMC platform works the same as Skylake/Desktop_
* High chance of CPU crash, save your files.

cyring, 2019-10-02:
3. Next add a case switch in the Daemon bellow this stattement
https://github.com/cyring/CoreFreq/blob/734baac66a23bdfa4e7bb192f92119bca96db154/corefreqd.c#L2805
```c
	case 0x1234:	/* Skylake/X */
		SKL_CAP(Shm, Proc, Core);
		SKL_IMC(Shm, Proc);
		SET_CHIPSET(IC_SUNRISEPOINT);
		break;
```
* replace `0x1234` with the working IMC id you found

TDP-9000w, 2019-10-02:
I tried 0x2066 for both.
The daemon, and cli both start, but the client does nothing when selecting the memory controller menu.   No errors under syslog, just statement of 
kernel: corefreq: unload 
kernel: corefreq: (6:-1): Processor [ 06_55] Architecture [Skylake/X] CPU [18/18]

cyring, 2019-10-02:
Thank you for trying.  
It means that all IDs tested have been probed, execution had followed the expected code path where bus speed is hard corded. Unfortunately registers failed to be queried. So we need find a specification of the architecture to decode the right registers. Especially the MCHBAR and the relative offsets where to query the IMC registers, plus the layout of their bits.

TDP-9000w, 2019-10-02:
Only thing i can see related is https://www.intel.com/content/dam/www/public/us/en/documents/manuals/6th-gen-core-family-uncore-performance-monitoring-manual.pdf
Page 17
```
3.3
IMC Events
The integrated memory controller un
it of the 6th Generation Intel
®
 Processor contains 
five model specific, fixed counters that allow for monitoring the number of requests to 
DRAM.
The fixed counters residing in the memory controller monitor transaction requests 
coming from various sources, e.g. the processor cores, the graphic engine, or other I/O 
agents. Unlike the MSR based performance counter registers in the CBo and ARB, the 
IMC fixed counter interface uses memory-ma
pped I/O reads from physical memory at 
the offsets specified in the IMC event table.
This set of counters are free-running and always-running. Software can read the value, 
wait for a desired internal, read again, then
 subtract the first sample from the second 
to determine how many times the event 
incremented in the sample interval.
The IMC counters below are model specific, meaning they may change or not be 
supported in the future.
To obtain the BAR address, read the value (in PCI configuration space) at Bus 0; Device 
0; Function 0; Offset 48H and mask with the value 0x0007FFFFF8000.
§
Table 3-3.
IMC Counters
Name
Address
Description
DRAM_GT_REQUESTS
BAR+0x5040
Counts every read/write
 request entering the Memory Controller 
to DRAM (sum of all channels) from the GT engine. Each partial 
write request counts as a request incrementing this counter. 
However same-cache-line partial wr
ite requests are combined to 
a single 64-byte data transfers from DRAM. Therefore 
multiplying the number of reques
ts by 64-bytes will lead to 
inaccurate GT memory bandwidth. The inaccuracy is proportional 
to the number of same-cache-
line partial writes combined.
DRAM_IA_REQUESTS
BAR+0x5044
Counts every read/w
rite request (demand and HW prefetch) 
entering the Memory Controller to DRAM (sum of all channels) 
from IA. Each partial write request counts as a request 
incrementing this counter. However same-cache-line partial write 
requests are combined to a sing
le 64-byte data transfers from 
DRAM. Therefore multiplying th
e number of requests by 64-
bytes will lead to inaccurate
 IA memory bandwidth. The 
inaccuracy is proportional to the number of same-cache-line 
partial writes combined.
DRAM_IO_REQUESTS
BAR+0x5048
Counts 
every read/write request entering the Memory Controller 
to DRAM (sum of all channels) from all IO sources (e.g. PCIe, 
Display Engine, USB audio, etc.). Each partial write request 
counts as a request incrementing this counter. However same-
cache-line partial write requests
 are combined to a single 64-
byte data transfers from DR
AM. Therefore mu
ltiplying the 
number of requests by 64-bytes will lead to inaccurate IO 
memory bandwidth. The inaccuracy is proportional to the 
number of same-cache-line partial writes combined.
DRAM_DATA_READS
BAR+0x5050
Counts every read (R
dCAS) issued by the 
Memory Controller to 
DRAM (sum of all channels). All 
requests result in 64-byte data 
transfers from DRAM. Use for accurate memory bandwidth 
calculations.
DRAM_DATA_WRITES
BAR+0x5054
Counts every write (WrC
AS) issued by the Memory Controller to 
DRAM (sum of all channels). All 
requests result in 64-byte data 
transfers from DRAM. Use for accurate memory bandwidth 
calculations
```

cyring, 2019-10-02:
Does lspci give an ID facing BDF 0:0:0 ?

cyring, 2019-10-02:
Kernel got something:
https://github.com/torvalds/linux/blob/0f1a7b3fac0583083ca19d4de47403511ced3521/drivers/edac/skx_base.c#L65

TDP-9000w, 2019-10-03:
You mean this ?
```
lspci
00:00.0 Host bridge [0600]: Intel Corporation Sky Lake-E DMI3 Registers [8086:2020] (rev 04)
```

cyring, 2019-10-03:
That's enough IDs and I bet this edac driver is running in your system ?
Enabling the kernel debug level should reveal its `edac_dbg` traces, and let us read the IMC properties.
Clearly this driver code shows registers and decoding protocol I never work with; and I wonder where those specs are coming from...

TDP-9000w, 2019-10-03:
I tried turning on kernel debug but was confused at what string i should use.
Too many options...
```
        acpi.debug_layer=       [HW,ACPI,ACPI_DEBUG]
        acpi.debug_level=       [HW,ACPI,ACPI_DEBUG]
                        Format: <int>
                        CONFIG_ACPI_DEBUG must be enabled to produce any ACPI
                        debug output.  Bits in debug_layer correspond to a
                        _COMPONENT in an ACPI source file, e.g.,
                            #define _COMPONENT ACPI_PCI_COMPONENT
                        Bits in debug_level correspond to a level in
                        ACPI_DEBUG_PRINT statements, e.g.,
                            ACPI_DEBUG_PRINT((ACPI_DB_INFO, ...
                        The debug_level mask defaults to "info".  See
                        Documentation/acpi/debug.txt for more information about
                        debug layers and levels.

                        Enable processor driver info messages:
                            acpi.debug_layer=0x20000000
                        Enable PCI/PCI interrupt routing info messages:
                            acpi.debug_layer=0x400000
                        Enable AML "Debug" output, i.e., stores to the Debug
                        object while interpreting AML:
                            acpi.debug_layer=0xffffffff acpi.debug_level=0x2
                        Enable all messages related to ACPI hardware:
                            acpi.debug_layer=0x2 acpi.debug_level=0xffffffff

                        Some values produce so much output that the system is
                        unusable.  The "log_buf_len" parameter may be useful
                        if you need to capture more output.
```
https://www.kernel.org/doc/html/v4.14/admin-guide/kernel-parameters.html

I'll give these a try.
```
                        Enable all messages related to ACPI hardware:
                            acpi.debug_layer=0x2 acpi.debug_level=0xffffffff

                        Some values produce so much output that the system is
                        unusable.  The "log_buf_len" parameter may be useful
                        if you need to capture more output.
```

cyring, 2019-10-04:
`skx_edac` is the driver to track.  
Do a `lsmod` first to check if it is running or not in your system.  

In its source code, the `edac_dbg` prints are source of interest, such as DIMM geometry (here)[https://elixir.bootlin.com/linux/latest/source/drivers/edac/skx_base.c#L526] 

In code, most traces seem at debug level 2; you can change level just for the time of restarting this driver, to find them among the last lines of kernel log. 
```
modprobe -r skx_edac ; modprobe skx_edac
```

TDP-9000w, 2019-10-04:
my lsmod lists nothing about skx_edac

cyring, 2019-10-12:
Hello,

Version 1.67, per Core Temperature and RAPL counters added to the view `Power & Voltage`

Thanks for your tests

Regards

TDP-9000w, 2019-10-18:
Ok, will get to those tests asap, thanks for the change works great.

So it turns out my logs show this:
```EDAC sbridge: Failed to register device with error -19.
EDAC sbridge: Couldn't find mci handler
EDAC sbridge: CPU SrcID #0, Ha #0, Channel #0 has DIMMs, but ECC is disabled
```

cyring, 2019-10-18:
Thanks.

So one EDAC static driver may run, not a module. I believe, the log can be dynamically increased to get further details.

I'm still refactoring the Voltage output which is now a 3^2 dimension (Voltage, Temp, Power), each matrix cell of 4 case values (None, Per SMT, Per Core, Per Package). Maths and code optimizations are still in progress.
I've released the intermediate version 1.67.4 which should conditionally display or not the relevant information depending on hardware. Feel free to post your View to let me evaluate any issues.

cyring, 2019-10-19:
Refactoring of the view is completed in the latest version. Thank you for your returns.

cyring, 2019-10-25:
Hello,
Did you have a chance to test the new Voltage view ? What about the per Core results facing total power ?  
Do you also have results from previous requests, starting from ODCM ?
Thanks

closed 2019-11-04 by cyring