cyring/CoreFreq #147: CoreFreq doens't found the corefreqk module on Solus

bkdwt, 2019-10-03:
Hi! The build worked normally, but when trying to load the module I get the error message below:

bkdwt@nsa ~/CoreFreq $ make MSR_CORE_PERF_UC=MSR_CORE_PERF_FIXED_CTR1 MSR_CORE_PERF_URC=MSR_CORE_PERF_FIXED_CTR2
cc  -Wall -pthread -c corefreqd.c \
	-D FEAT_DBG=0 -D UBENCH=0 \
	-o corefreqd.o
cc  -Wall -c corefreqm.c \
	-D FEAT_DBG=0 -D UBENCH=0 \
	-o corefreqm.o
cc  -Wall corefreqd.c corefreqm.c \
	-D FEAT_DBG=0 -D UBENCH=0 \
	-o corefreqd -lpthread -lm -lrt
cc  -Wall -c corefreq-cli.c \
	-D FEAT_DBG=0 -D UBENCH=0 \
	-o corefreq-cli.o
cc  -Wall -c corefreq-ui.c \
	-D FEAT_DBG=0 -D UBENCH=0 \
	-o corefreq-ui.o
cc  -Wall -c corefreq-cli-rsc.c \
	-D FEAT_DBG=0 -D UBENCH=0 \
	-o corefreq-cli-rsc.o
cc  -Wall -c corefreq-cli-json.c \
	-D FEAT_DBG=0 -D UBENCH=0 \
	-o corefreq-cli-json.o
cc  -Wall -c corefreq-cli-extra.c \
	-D FEAT_DBG=0 -D UBENCH=0 \
	-o corefreq-cli-extra.o
cc  -Wall \
	corefreq-cli.c corefreq-ui.c corefreq-cli-rsc.c \
	corefreq-cli-json.c corefreq-cli-extra.c \
	-D FEAT_DBG=0 -D UBENCH=0 \
	-o corefreq-cli -lm -lrt
make -j1 -C /lib/modules/5.2.13-126.current/build M=/home/bkdwt/CoreFreq modules
make[1]: Entering directory '/usr/src/linux-headers-5.2.13-126.current'
  CC [M]  /home/bkdwt/CoreFreq/corefreqk.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/bkdwt/CoreFreq/corefreqk.mod.o
  LD [M]  /home/bkdwt/CoreFreq/corefreqk.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.2.13-126.current'
bkdwt@nsa ~/CoreFreq $ sudo modprobe corefreqk
modprobe: FATAL: Module corefreqk not found in directory /lib/modules/5.2.13-126.current

Is the compiler installing the module in a different directory?

bkdwt, 2019-10-03:
Tried insmod:

bkdwt@nsa ~/CoreFreq $ sudo modprobe corefreqk
Password: 
modprobe: FATAL: Module corefreqk not found in directory /lib/modules/5.2.13-126.current
bkdwt@nsa ~/CoreFreq $ sudo insmod corefreqk.ko
bkdwt@nsa ~/CoreFreq $ sudo corefreqd
sudo: corefreqd: command not found
bkdwt@nsa ~/CoreFreq $ sudo systemctl start corefreqd
Failed to start corefreqd.service: Unit corefreqd.service not found.
bkdwt@nsa ~/CoreFreq $ corefreq-cli
bash: corefreq-cli: command not found

cyring, 2019-10-03:
Hello,

I would recommend you use two terminals:

* the first, as root
```
insmod /working_directory/corefreq.ko

/working_directory/corefreqd
```
 * the second terminal as standard user
```
/working_directory/corefreq-cli
```

where /working_directory is the full path to the location you built _CoreFreq_

bkdwt, 2019-10-03:
> Hello,
> 
> I would recommend you use two terminals:
> 
>     * the first, as root
> 
> 
> ```
> insmod /working_directory/corefreqk.ko
> 
> /working_directory/corefreqd
> ```
> 
>     * the second terminal as standard user
> 
> 
> ```
> /working_directory/corefreq-cli
> ```
> 
> where /working_directory is the full path to the location you built _CoreFreq_

>root@nsa /home/bkdwt/CoreFreq # insmod corefreqk.ko
>insmod: ERROR: could not insert module corefreqk.ko: File exists
>root@nsa /home/bkdwt/CoreFreq # corefreqd
>bash: corefreqd: command not found

>bkdwt@nsa ~/CoreFreq $ corefreq-cli
>bash: corefreq-cli: command not found

Still not working :/

closed 2019-10-03 by bkdwt

cyring, 2019-10-03:
Your welcome.

Please run `corefreq-cli` in Xterm (or Console) and you will notice the dark gray characters missing in your screenshot.

cyring, 2019-10-03:
I just notice that this Xeon is wrongly reported as Haswell / Desktop. It should be Haswell / Server. I need to fix this.
Can you please post the output of:
* System Information 
```
corefreq-cli -s
```
* Memory Controller 
```
corefreq-cli -M
```

Thank you

bkdwt, 2019-10-03:
@crying I'll post on home. 👍🏻

bkdwt, 2019-10-04:
@cyring here we go: https://pastebin.com/6RXakZ9Q

cyring, 2019-10-04:
very helpful, thank you

(Haswell)[http://www.cpu-world.com/Related_CPUs/Socket%201150_Haswell.html] 

those CPUs share the same CPUID signature.

I will release a fix

cyring, 2019-10-06:
# HASWELL CPUID `06_3C`
```
{.ExtFamily=0x0, .Family=0x6, .ExtModel=0x3, .Model=0xC}
```
Architecture     |Code Name        |Brand string discriminant
--------------------|-----------------------|---------------------------------
Devil's Canyon|`Haswell/Canyon`|`Intel(R) Core(TM) i7-47`
Devil's Canyon|`Haswell/Canyon`|`Intel(R) Core(TM) i5-46`

cyring, 2019-10-06:
# HASWELL CPUID `06_3C`
```
{.ExtFamily=0x0, .Family=0x6, .ExtModel=0x3, .Model=0xC}
```
## Ordered Reduction
Architecture     |Code Name        |Brand string discriminant
--------------------|-----------------------|---------------------------------
Haswell Mobile eXtreme|`Haswell/Mobile/eXtreme`|`Intel(R) Core(TM) i7-4940MX`
Haswell Mobile|`Haswell/Crystal Well`|`Intel(R) Core(TM) i7-4800MQ`
Haswell Mobile|`Haswell/Mobile`|`Intel(R) Core(TM) i7-4710HQ`
Devil's Canyon|`Haswell/Canyon`|`Intel(R) Core(TM) i7-47`
Devil's Canyon|`Haswell/Canyon`|`Intel(R) Core(TM) i5-46`
Haswell Mobile|`Haswell/Mobile`|`Intel(R) Core(TM) i5-4300M`
Haswell Mobile|`Haswell/Mobile`|`Intel(R) Core(TM) i5-4200H`
Haswell|`Haswell/Desktop`|`Intel(R) Core(TM) i5-45`
Haswell|`Haswell/Desktop`|`Intel(R) Core(TM) i5-44`
Haswell Mobile|`Haswell/Mobile`|`Intel(R) Core(TM) i3-4000M`
Haswell|`Haswell/Desktop`|`Intel(R) Core(TM) i3-43`
Haswell|`Haswell/Desktop`|`Intel(R) Core(TM) i3-41`
Haswell|`Haswell/Desktop`|`Intel(R) Pentium(R) CPU G`
Haswell|`Haswell/Desktop`|`Intel(R) Celeron(R) CPU G`
Denlow|`Haswell/Denlow`|`Intel(R) Xeon(R) CPU E3-12`
Haswell Embedded|`Haswell/Embedded`|`Intel® 4th Gen Core™ i7-4700EQ`
Haswell Mobile|`Haswell/Mobile`|`Intel(R) Pentium(R) CPU 3`
Haswell Mobile|`Haswell/Mobile`|`Intel(R) Celeron(R) CPU 2`

cyring, 2019-10-06:
Hello,

Please update to the latest version, you should read `Haswell/Denlow` as the architecture name ?

_screenshot welcomed_

bkdwt, 2019-10-07:
> Hello,
> 
> Please update to the latest version, you should read `Haswell/Denlow` as the architecture name ?
> 
> _screenshot welcomed_

Yep, now shows > Haswell/Denlow
[image]

cyring, 2019-10-07:
> Yep, now shows Haswell/Denlow

Thank you. 

> I think corefreq is not reporting frequencies correctly. Just see that it is reporting a frequency of 15MHz with the CPU in IDLE.

That's the expecting behavior of _CoreFreq_ which measures the performance counters of the Processor over an interval (default, 1 second) .
See the [Hardware performance counter](https://en.m.wikipedia.org/wiki/Hardware_performance_counter)

At the GHz speed, processors are going too fast to be measured in real-time. 
Continuous monitoring instructions would saturate the flow, and you may just monitor the monitoring loop during the idle time (which will not be entered btw) 

So what you get is the Intel recommended algorithm based on PMC counters.
In the Client, you will however get various _absolute_ frequencies such as those in the Processor window.

bkdwt, 2019-10-07:
> Thank you.
>
>
> That's the expecting behavior of _CoreFreq_ which measures the performance counters of the Processor over an interval (default, 1 second) .
> See the [Hardware performance [...]

Thanks for the explanation! :)