cyring/CoreFreq #153: Build error on linux 5.4.1

terencode, 2019-12-03:
DKMS reports the following: 
```
DKMS make.log for corefreqk-r833.29cf53b for kernel 5.4.1-3-tkg-bmq (x86_64)
mar. déc.  3 15:19:57 CET 2019
make: Entering directory '/usr/src/corefreqk-r833.29cf53b'
cc  -Wall corefreqd.c corefreqm.c \
	-D FEAT_DBG=0 -D UBENCH=0 \
	-o corefreqd -lpthread -lm -lrt
cc  -Wall \
	corefreq-cli.c corefreq-ui.c corefreq-cli-rsc.c \
	corefreq-cli-json.c corefreq-cli-extra.c \
	-D FEAT_DBG=0 -D UBENCH=0 \
	-o corefreq-cli -lm -lrt
make -j1 -C /lib/modules/5.4.1-4-tkg-pds/build M=/var/lib/dkms/corefreqk/r833.29cf53b/build modules
make[1]: Entering directory '/usr/src/corefreqk-r833.29cf53b'
make[1]: warning: -jN forced in submake: disabling jobserver mode.
  CC [M]  /var/lib/dkms/corefreqk/r833.29cf53b/build/corefreqk.o
/var/lib/dkms/corefreqk/r833.29cf53b/build/corefreqk.c: In function ‘Sys_DumpTask’:
/var/lib/dkms/corefreqk/r833.29cf53b/build/corefreqk.c:5668:43: error: ‘struct task_struct’ has no member named ‘se’
 5668 |   SysGate->taskList[cnt].runtime  = thread->se.sum_exec_runtime;
      |                                           ^~
make[2]: *** [scripts/Makefile.build:266: /var/lib/dkms/corefreqk/r833.29cf53b/build/corefreqk.o] Error 1
make[1]: *** [Makefile:1656: /var/lib/dkms/corefreqk/r833.29cf53b/build] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.4.1-4-tkg-pds/build'
make: *** [Makefile:29: all] Error 2
make: Leaving directory '/usr/src/corefreqk-r833.29cf53b'
```

cyring, 2019-12-03:
Thanks for reporting this. Yesterday 5.4 was still not official in Arch repos; but I'll update from 5.3

Meanwhile according to to the Makefile output, the `task_struct` has been impacted, and apparently the scheduler properties...

cyring, 2019-12-03:
Edit: nothing in relation with 5.4

Do you understand the impacts of the tkg flavor has on the mainstream task_struct ?

https://github.com/Tk-Glitch/PKGBUILDS/blob/master/linux54-tkg/linux54-tkg-patches/0004-5.4-ck1.patch

Why your patch don't report as `CONFIG_SCHED_MUQSS` or `CONFIG_SCHED_BMQ` ? 

Here it is what we have already worked on:
https://github.com/cyring/CoreFreq/blob/29cf53b70ffd4f3c5dfe7926216c7ea11ed2dfdb/corefreqk.c#L5665

terencode, 2019-12-03:
It is a problem with 5.4.1 because it also fails for the official arch kernel. The error log gets overwritten each time so it just had the last failed build.
here is the log for the official kernel:
```
DKMS make.log for corefreqk-r834.9d9495a for kernel 5.4.1-arch1-1 (x86_64)
mar. déc.  3 23:58:46 CET 2019
make: Entering directory '/usr/src/corefreqk-r834.9d9495a'
cc  -Wall corefreqd.c corefreqm.c \
        -D FEAT_DBG=0 -D UBENCH=0 \
        -o corefreqd -lpthread -lm -lrt
cc  -Wall \
        corefreq-cli.c corefreq-ui.c corefreq-cli-rsc.c \
        corefreq-cli-json.c corefreq-cli-extra.c \
        -D FEAT_DBG=0 -D UBENCH=0 \
        -o corefreq-cli -lm -lrt
make -j1 -C /lib/modules/5.4.1-4-tkg-pds/build M=/var/lib/dkms/corefreqk/r834.9d9495a/build modules
make[1]: Entering directory '/usr/src/corefreqk-r834.9d9495a'
make[1]: warning: -jN forced in submake: disabling jobserver mode.
  CC [M]  /var/lib/dkms/corefreqk/r834.9d9495a/build/corefreqk.o
/var/lib/dkms/corefreqk/r834.9d9495a/build/corefreqk.c: In function ‘Sys_DumpTask’:
/var/lib/dkms/corefreqk/r834.9d9495a/build/corefreqk.c:5668:43: error: ‘struct task_struct’ has no member named ‘se’
 5668 |   SysGate->taskList[cnt].runtime  = thread->se.sum_exec_runtime;
      |                                           ^~
make[2]: *** [scripts/Makefile.build:266: /var/lib/dkms/corefreqk/r834.9d9495a/build/corefreqk.o] Error 1
make[1]: *** [Makefile:1656: /var/lib/dkms/corefreqk/r834.9d9495a/build] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.4.1-4-tkg-pds/build'
make: *** [Makefile:29: all] Error 2
make: Leaving directory '/usr/src/corefreqk-r834.9d9495a
```

cyring, 2019-12-04:
Probably a Package issue with DKMS. Does it fail, it you clone source code from GitHub then build straight from the working directory ?
```
make clean all
```

terencode, 2019-12-04:
Yes, sorry, forgot to I had to do a clean build... Fixed.

closed 2019-12-04 by terencode

cyring, 2019-12-04:
Glade you fixed it.
Feel free to come back for any help; and I will appreciate any screenshots of CoreFreq driving your Processor, such as those in my Wiki - CPU support 

Enjoy CoreFreq

terencode, 2019-12-04:
Sure, it's a 3600 but I see there already exists one for the 3600X.

cyring, 2019-12-04:
Indeed, they are closed in specs.

At least, what I would like to see/check:
* Ratio, frequency view of a single Core stressed
* Power & Voltage view for single then full Cores stressed
* Rest can be listed with just `corefreq-cli -s`

The thing is that most CoreFreq algorithms detect registers to provide values, but AMD has not published all bits yet for the Zen architecture... 
For example, the Core Performance Boost frequency ratios, I've to maintain a table of bins, and I need check the _calibration_

John-Gee, 2020-09-06:
> Edit: nothing in relation with 5.4
> 
> Do you understand the impacts of the _tkg-pds_ flavor has on the mainstream `task_struct` ?
> 
> [https://github.com/Tk-Glitch/PKGBUILDS/blob/master/linux54-tkg/linux54-tkg-patches/0004-5.4-ck1.patch#L7333](https://github.com/Tk-Glitch/PKGBUILDS/blob/master/linux54-tkg/linux54-tkg-patches/0004-5.4-ck1.patch?rgh-link-date=2019-12-03T18%3A39%3A36Z#L7333)
> 
> Why your patch don't report as `CONFIG_SCHED_MUQSS` or `CONFIG_SCHED_BMQ` ?
> 
> Here it is what we have already worked on:
> https://github.com/cyring/CoreFreq/blob/29cf53b70ffd4f3c5dfe7926216c7ea11ed2dfdb/corefreqk.c#L5665

Would you mind adding the same check for CONFIG_SCHED_PDS? It builds and runs fine with that.

cyring, 2020-09-06:
> Would you mind adding the same check for CONFIG_SCHED_PDS? It builds and runs fine with that.

Sure. It is now available in the `develop` branch for your tests.
Regards

John-Gee, 2020-09-06:
It's pretty much the same thing I did so it should work the same, but just in case I tried the develop branch and it works fine.

Thank you for the prompt reply!

cyring, 2020-09-06:
> ... but just in case I tried the develop branch and it works fine.

Thanks a lot for your code and your tests.
I want to be sure of the changes I'm releasing for other users.

John-Gee, 2020-09-06:
Sure thing!