cyring/CoreFreq #154: Strange voltage readings

horus-dot, 2019-12-06:
First, thank you so much for this excellent software. The following issue may not be a real problem, however it seemed sufficiently suspicious to warrant checking.

I'm having strange voltage readings on a fresh Arch Linux install with a Ryzen 3700X CPU in an ASRock 570M Pro motherboard.  With no CPU load, corefreq-cli -V reads every core at 0.9 V. At max CPU load (confirmed by both htop and corefreq, achieved by running "stress -c 16 -t 5m"), the voltages rise to 1.1 and go no further.

This seems quite strange in light of the fact that the normal operating range for the 3700X is [0.2 to 1.5 V](https://www.reddit.com/r/Amd/comments/cbls9g/the_final_word_on_idle_voltages_for_3rd_gen_ryzen/). Additionally, the BIOS shows a Vcore of 1.456. (Indeed, it was this abnormally high voltage that made me seek out corefreq to figure out whether the reading was legitimate.)

I have disabled Precision Boost Overdrive in the BIOS, but besides that and changing fan settings, the BIOS is at stock settings. 

Additionally, the "in0" reading from the sensors command fluctuates between 96 mV at idle and 680 mV under load. My impression is that this measurement is usually the Vcore, however it is not consistent with the others above or the CPU specifications. No other voltage measurements from sensors fluctuate under load.

Is corefreq measuring voltage accurately here, and if not, is there a way I can fix it?

cyring, 2019-12-06:
> voltages rise to 1.1

Indeed I need to improve the Vcore: what I'm showing is the Core Voltage ID, the VID included in each P-State register which is then converted to a Vcore value.

These VID, Vcore should match with what you read/set in BIOS

That's unfortunately all published info I've found in the Zen specs.

I'm seeing that other Windows software are showing the real voltage, apparently from an external IC or the SMU but I have not found their registers definition. 

FYI, here is the code to: 
* query the VID
https://github.com/cyring/CoreFreq/blob/b15f927df74ddd9ae15fec36ff0aee4cde18382d/corefreqk.c#L8308
* then, convert to Vcore
https://github.com/cyring/CoreFreq/blob/b15f927df74ddd9ae15fec36ff0aee4cde18382d/corefreqd.c#L170

cyring, 2019-12-10:
Thanks to this dump:
https://github.com/InstLatx64/InstLatx64/blob/77fd92f7a4e3e0e127a1dd805e5cd04c8fa6d7da/AuthenticAMD/AuthenticAMD0830F10_K17_CastlePeak_CPUID2.txt#L5611

We could measure the Vcore from register `0xC0010293` which apparently reads at `1.45V`

Please dump your Processor with this MSR in 3 cases: Idle, Single Core stressed, All Cores stressed; Turbo CPB enabled off course.

With your results, I'll then convert VID to Voltage Core

cyring, 2019-12-10:
**Edit** please try the followings
1. Replace this line:
https://github.com/cyring/CoreFreq/blob/69c2cd77a9ae35b88cf0897dfc255812bb382b6f/corefreqk.c#L8308
With this code:
```c
                RDMSR(PstateDef, MSR_AMD_PSTATE_F17H_BOOST);
```
2. Rebuild all
```
make clean all
```
3. Reload all, especially the driver
```
# rmmod corefreqk.ko
# insmod corefreqk.ko
# corefreqd
```
_switch to another term_
```
$ corefreq-cli
```
4. Test the 3 cases (idle, single core load, all cores all)
For each, please screenshot the view Power & Voltage

horus-dot, 2019-12-10:
Thank you for this. It is really wonderful work. 

I followed the instructions in your second post and got sensible results. They also match the readings from lm_sensors. 

Please see here for screenshots: https://imgur.com/a/MpF8NBP

cyring, 2019-12-11:
Your welcome.
I can barely read digits in screenshots but it looks coherent with other Ryzen in issue #54 
I now miss Threadripper, and EPYC results, especially, per cluster results.
Regards

cyring, 2019-12-11:
Version `1.70.2` released with the Vcore fixed.

cyring, 2019-12-19:
Closing the fixed issue.

closed 2019-12-19 by cyring

cyring, 2020-07-06:
Hello,
Among the last changes in issue #195 , an I/O ASM access to the FCH have been added: can you please check if the last _CoreFreq_ version is starting fine ?
Thank you

horus-dot, 2020-07-08:
Yes, I built from source code and it works fine for me. No crashes.

cyring, 2020-07-08:
> Yes, I built from source code and it works fine for me. No crashes.

Thank you very much.