cyring/CoreFreq #160: Memory Controller data not available on Intel Core i7-4650U

olejon, 2020-01-15:
### Problem
* The Memory Controller window is greyed out in the CLI
* Running `corefreq-cli -M` gives no output
* This also affects my Intel NUC with a Haswell CPU (`Intel Core i3-4010U`)
* Both are both "Mobile" & "Ultra Low TDP" if that matters, vs other Haswells
* Same problem on Fedora 31 and Ubuntu 18.04 HWE stack
* Verbose output from various commands is listed below as hopefully helpful data, separated by horizontal lines

### Output from various commands from the i7-4650U
```
# corefreq-cli -B -k -u -M

[ 0] Apple Inc.                                                                 
[ 1] 118.0.0.0.0                                                                
[ 2] 08/06/2019                                                                 
[ 3] Apple Inc.                                                                 
[ 4] MacBookAir6,2                                                              
[ 5] 1.0                                                                        
[ 6] C---5---F---                                                               
[ 7] System SKU#                                                                
[ 8] MacBook Air                                                                
[ 9] Mac-7DF21CB3ED6977E5                                                       
[10] MacBookAir6,2                                                              
[11] C---3---7---H---Z                                                          
Linux:                                                                          
|- Release                                               [5.4.8-200.fc31.x86_64]
|- Version                                  [#1 SMP Mon Jan 6 16:44:18 UTC 2020]
|- Machine                                                              [x86_64]
Memory:                                                                         
|- Total RAM                                                          8086020 KB
|- Shared RAM                                                          172272 KB
|- Free RAM                                                           5837440 KB
|- Buffer RAM                                                           43028 KB
|- Total High                                                               0 KB
|- Free High                                                                0 KB
CPU-Freq driver                                               [    intel_pstate]
Governor                                                      [         Missing]
CPU-Idle driver                                               [      intel_idle]
|- Idle Limit                                                 [             C10]
   |- State       POLL      C1     C1E      C3      C6     C7s      C8      C9  
   |-          CPUIDLE MWAIT 0 MWAIT 0 MWAIT 0 MWAIT 0 MWAIT 0 MWAIT 0 MWAIT 0 M
   |- Power         -1       0       0       0       0       0       0       0  
   |- Latency        0       2      10      33     133     166     300     600  
   |- Residency      0       2      20     100     400     500     900    1800  
CPU #0   function         EAX          EBX          ECX          EDX            
|- 00000000:00000000    0000000d     756e6547     6c65746e     49656e69         
   |- Largest Standard Function=0000000d                                        
|- 80000000:00000000    80000008     00000000     00000000     00000000         
   |- Largest Extended Function=80000008                                        
|- 00000001:00000000    00040651     00100800     7ffafbff     bfebfbff         
|- 00000002:00000000    76036301     00f0b5ff     00000000     00c10000         
|- 00000003:00000000    00000000     00000000     00000000     00000000         
|- 00000004:00000000    1c004121     01c0003f     0000003f     00000000         
|- 00000004:00000001    1c004122     01c0003f     0000003f     00000000         
|- 00000004:00000002    1c004143     01c0003f     000001ff     00000000         
|- 00000004:00000003    1c03c163     03c0003f     00000fff     00000006         
|- 00000005:00000000    00000040     00000040     00000003     11142120         
|- 00000006:00000000    00000077     00000002     00000009     00000000         
|- 00000007:00000000    00000000     000027ab     00000000     9c000400         
|- 00000009:00000000    00000000     00000000     00000000     00000000         
|- 0000000a:00000000    07300403     00000000     00000000     00000603         
|- 0000000b:00000000    00000001     00000002     00000100     00000000         
|- 0000000d:00000000    00000007     00000340     00000340     00000000         
|- 0000000d:00000001    00000001     00000000     00000000     00000000         
|- 0000000d:00000002    00000100     00000240     00000000     00000000         
|- 0000000d:00000003    00000000     00000000     00000000     00000000         
|- 0000000d:00000004    00000000     00000000     00000000     00000000         
|- 0000000d:0000003e    00000000     00000000     00000000     00000000         
|- 0000000f:00000000    00000007     00000340     00000340     00000000         
|- 0000000f:00000001    00000001     00000000     00000000     00000000         
[... 310 lines cut ...]
00:02.0 VGA compatible controller [0300]: Intel Corporation Haswell-ULT Integrated Graphics Controller [8086:0a26] (rev 09)
00:03.0 Audio device [0403]: Intel Corporation Haswell-ULT HD Audio Controller [8086:0a0c] (rev 09)
00:14.0 USB controller [0c03]: Intel Corporation 8 Series USB xHCI HC [8086:9c31] (rev 04)
00:15.0 DMA controller [0801]: Intel Corporation 8 Series Low Power Sub-System DMA [8086:9c60] (rev 04)
00:15.4 Serial bus controller [0c80]: Intel Corporation 8 Series SPI Controller #1 [8086:9c66] (rev 04)
00:16.0 Communication controller [0780]: Intel Corporation 8 Series HECI #0 [8086:9c3a] (rev 04)
00:1b.0 Audio device [0403]: Intel Corporation 8 Series HD Audio Controller [8086:9c20] (rev 04)
00:1c.0 PCI bridge [0604]: Intel Corporation 8 Series PCI Express Root Port 1 [8086:9c10] (rev e4)
00:1c.1 PCI bridge [0604]: Intel Corporation 8 Series PCI Express Root Port 2 [8086:9c12] (rev e4)
00:1c.2 PCI bridge [0604]: Intel Corporation 8 Series PCI Express Root Port 3 [8086:9c14] (rev e4)
00:1c.4 PCI bridge [0604]: Intel Corporation 8 Series PCI Express Root Port 5 [8086:9c18] (rev e4)
00:1c.5 PCI bridge [0604]: Intel Corporation 8 Series PCI Express Root Port 6 [8086:9c1a] (rev e4)
00:1f.0 ISA bridge [0601]: Intel Corporation 8 Series LPC Controller [8086:9c43] (rev 04)
00:1f.3 SMBus [0c05]: Intel Corporation 8 Series SMBus Controller [8086:9c22] (rev 04)
02:00.0 Multimedia controller [0480]: Broadcom Inc. and subsidiaries 720p FaceTime HD Camera [14e4:1570]
03:00.0 Network controller [0280]: Broadcom Inc. and subsidiaries BCM4360 802.11ac Wireless Network Adapter [14e4:43a0] (rev 03)
04:00.0 SATA controller [0106]: Marvell Technology Group Ltd. 88SS9183 PCIe SSD Controller [1b4b:9183] (rev 14)
05:00.0 PCI bridge [0604]: Intel Corporation DSL3510 Thunderbolt Controller [Cactus Ridge 4C 2012] [8086:1547] (rev 03)
06:00.0 PCI bridge [0604]: Intel Corporation DSL3510 Thunderbolt Controller [Cactus Ridge 4C 2012] [8086:1547] (rev 03)
06:03.0 PCI bridge [0604]: Intel Corporation DSL3510 Thunderbolt Controller [Cactus Ridge 4C 2012] [8086:1547] (rev 03)
06:04.0 PCI bridge [0604]: Intel Corporation DSL3510 Thunderbolt Controller [Cactus Ridge 4C 2012] [8086:1547] (rev 03)
06:05.0 PCI bridge [0604]: Intel Corporation DSL3510 Thunderbolt Controller [Cactus Ridge 4C 2012] [8086:1547] (rev 03)
06:06.0 PCI bridge [0604]: Intel Corporation DSL3510 Thunderbolt Controller [Cactus Ridge 4C 2012] [8086:1547] (rev 03)
07:00.0 System peripheral [0880]: Intel Corporation DSL3510 Thunderbolt Controller [Cactus Ridge 4C 2012] [8086:1547] (rev 03)
```

olejon, 2020-01-15:
BTW: The Intel NUC has the exact same DRAM controller:

```
lspci -nn

00:00.0 Host bridge [0600]: Intel Corporation Haswell-ULT DRAM Controller [8086:0a04] (rev 09)
...
```

cyring, 2020-01-15:
Thanks, I also need the output of `corefreq-cli -s` for each processor

olejon, 2020-01-16:
Here it is. No point of saying which PC is which as the output has the model name.

---

```
Processor                             [Intel(R) Core(TM) i7-4650U CPU @ 1.70GHz]
|- Architecture                                          [Haswell/Ultra Low TDP]
|- Vendor ID                                                      [GenuineIntel]
|- Microcode                                                        [        37]
|- Signature                                                           [  06_45]
|- Stepping                                                            [      1]
|- Online CPU                                                          [  4/  4]
|- Base Clock                                                          [ 99.995]
|- Frequency            (MHz)                      Ratio                        
                 Min    799.96                    [   8 ]                       
                 Max   2299.88                    [  23 ]                       
|- Factory                                                             [100.000]
                       1700                       [  17 ]                       
|- Performance                                                                  
   |- OSPM                                                                      
                 TGT    799.96                    <   8 >                       
|- Turbo Boost                                                         [ UNLOCK]
                  1C   3299.83                    <  33 >                       
                  2C   2899.85                    <  29 >                       
                  3C   2899.85                    <  29 >                       
                  4C   2899.85                    <  29 >                       
|- Uncore                                                              [   LOCK]
                 Min    799.96                    [   8 ]                       
                 Max   2299.88                    [  23 ]                       
|- TDP                                                           Level [  0:2  ]
   |- Programmable                                                     [ UNLOCK]
   |- Configuration                                                    [ UNLOCK]
   |- Turbo Activation                                                 [ UNLOCK]
             Nominal   1699.91                    [  17 ]                       
              Level1    799.96                    [   8 ]                       
              Level2   2299.88                    [  23 ]                       
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [N]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AVX-512      [N]  BMI1/BMI2 [Y/Y]      CLFLUSH [Y]        CMOV [Y] 
|- CMPXCHG8B    [Y]   CMPXCHG16B [Y]         F16C [Y]         FPU [Y] 
|- FXSR         [Y]    LAHF/SAHF [Y]    MMX/Ext [Y/N]  MONITOR/X[Y/N] 
|- MOVBE        [Y]          MPX [N]    PCLMULQDQ [Y]      POPCNT [Y] 
|- RDRAND       [Y]       RDSEED [N]       RDTSCP [Y]         SEP [Y] 
|- SGX          [N]          SSE [Y]         SSE2 [Y]        SSE3 [Y] 
|- SSSE3        [Y]  SSE4.1/4A [Y/N]       SSE4.2 [Y]     SYSCALL [Y] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- Core Multi-Processing                                  CMP Legacy   [Missing]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Capable]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Capable]
|- CPL Qualified Debug Store                                  DS-CPL   [Capable]
|- 64-Bit Debug Store                                         DTES64   [Capable]
|- Fast-String Operation                                Fast-Strings   [Capable]
|- Fused Multiply Add                                     FMA | FMA4   [Capable]
|- Hardware Lock Elision                                         HLE   [Missing]
|- Long Mode 64 bits                                       IA64 | LM   [Capable]
|- LightWeight Profiling                                         LWP   [Missing]
|- Machine-Check Architecture                                    MCA   [Capable]
|- Model Specific Registers                                      MSR   [Capable]
|- Memory Type Range Registers                                  MTRR   [Capable]
|- OS-Enabled Ext. State Management                          OSXSAVE   [Capable]
|- Physical Address Extension                                    PAE   [Capable]
|- Page Attribute Table                                          PAT   [Capable]
|- Pending Break Enable                                          PBE   [Capable]
[... 230 lines cut ...]
|- Last Level Cache References                                         [Capable]
|- Last Level Cache Misses                                             [Capable]
[... cut ...]
|- Thermal Design Power                                          TDP   [     15]
   |- Minimum Power                                              Min   [Missing]
   |- Maximum Power                                              Max   [Missing]
|- Units                                                                        
   |- Power                                               watt   [  0.125000000]
   |- Energy                                             joule   [  0.000061035]
   |- Window                                            second   [  0.000976562]
```

cyring, 2020-01-18:
* [Mobile 4th Generation Intel® Core™ Processor Family, Mobile Intel® Pentium® Processor Family, and Mobile Intel® Celeron® Processor Family Datasheet – Volume 2 of 2](https://www.intel.com/content/dam/www/public/us/en/documents/datasheets/4th-gen-core-family-mobile-u-y-processor-lines-vol-2-datasheet.pdf)

cyring, 2020-01-18:
* Attached is the IMC test version for Haswell Mobile U/Y with Host Agent  `0x0a04`
[CoreFreq.tar.gz](https://github.com/cyring/CoreFreq/files/4080471/CoreFreq.tar.gz)
* Plz unload any previous CoreFreq
* Untar, build and load this one
* Supply the output of `corefreq-cli -M`

regards
Cyril

olejon, 2020-01-18:
Thanks!

### Core i7 (MacBook)

* This is after a Cold Boot. FYI compiled without any flags because `nmi_watchdog=0`:

```
$ ./corefreq-cli -M

                         Lynx Point-M  [ A04]                         
Controller #0                                           Dual Channel  
 Bus Rate  5000 MT/s      Bus Speed 5000 MT/s     DRAM Speed 1600 MHz 
                                                                      
 Cha    CL  RCD   RP  RAS  RRD  RFC   WR RTPr WTPr  FAW  B2B  CWL Rate
  #0    12    0    0    0    0  104    0    0    0    0    0    6   1N
  #1     0    0    0    0    0    0    0    0    0    0    0    0   1N
      ddWR drWR srWR ddRW drRW srRW ddRR drRR srRR ddWW drWW srWW  ECC
  #0     8    8   19   16   16   16    7    7    4    7    7    4   0 
  #1     0    0   16    0   12    0   15    4    0    3   13    0   0 
                                                                      
 DIMM Geometry for channel #0                                         
      Slot Bank Rank     Rows   Columns    Memory Size (MB)           
       #0                                                             
       #1                                                             
 DIMM Geometry for channel #1                                         
      Slot Bank Rank     Rows   Columns    Memory Size (MB)           
       #0                                                             
       #1
```
---

### Core i3 (Intel NUC)

* This *not from a Cold Boot* as it's a Server with VMs. FYI compiled with flags `APERF/MPERF` as NMI Watchdog is on:

```
$ ./corefreq-cli

                         Lynx Point-M  [ A04]                         
Controller #0                                           Dual Channel  
 Bus Rate  5000 MT/s      Bus Speed 4988 MT/s     DRAM Speed 1600 MHz 
                                                                      
 Cha    CL  RCD   RP  RAS  RRD  RFC   WR RTPr WTPr  FAW  B2B  CWL Rate
  #0    11    0    0    0    0  208    0    0    0    0    0    8   1N
  #1     0    0    0    0    0    0    0    0    0    0    0    0   1N
      ddWR drWR srWR ddRW drRW srRW ddRR drRR srRR ddWW drWW srWW  ECC
  #0     4    3   20   12   12   12    6    6    4    7    7    4   0 
  #1     0    0   16    0   12    0   12    4    0    3    1    0   0 
                                                                      
 DIMM Geometry for channel #0                                         
      Slot Bank Rank     Rows   Columns    Memory Size (MB)           
       #0                                                             
       #1                                                             
 DIMM Geometry for channel #1                                         
      Slot Bank Rank     Rows   Columns    Memory Size (MB)           
       #0                                                             
       #1
```

* Memory Controller window also working
* But no information about the `DIMM Geometry` and a lot of `0` values
* FYI The Intel NUC has 2 different RAM chips (1 x 4 GB + 1 x 8 GB) with different timings, so according to UEFI it logically defaults to the slowest one's timings/values, in case the numbers surprise you
* But definitely seems you're on the right path :+1:

cyring, 2020-01-18:
Thanks, could you print for both the SPD data and geometry with 
```
dmidecode -t memory
```

olejon, 2020-01-18:
### Core i3 (Intel NUC)

```
# dmidecode 3.1
Getting SMBIOS data from sysfs.
SMBIOS 2.8 present.

Handle 0x003E, DMI type 16, 23 bytes
Physical Memory Array
	Location: System Board Or Motherboard
	Use: System Memory
	Error Correction Type: None
	Maximum Capacity: 16 GB
	Error Information Handle: Not Provided
	Number Of Devices: 2

Handle 0x003F, DMI type 17, 40 bytes
Memory Device
	Array Handle: 0x003E
	Error Information Handle: Not Provided
	Total Width: 64 bits
	Data Width: 64 bits
	Size: 8192 MB
	Form Factor: SODIMM
	Set: None
	Locator: DIMM 1
	Bank Locator: Channel A Slot 0
	Type: DDR3
	Type Detail: Synchronous
	Speed: 1600 MT/s
	Manufacturer: Kingston
	Serial Number: 075015113072
	Asset Tag: 9876543210
	Part Number: KHX1600C9S3L/8G   
	Rank: 2
	Configured Clock Speed: 1600 MT/s
	Minimum Voltage: 1.35 V
	Maximum Voltage: 1.5 V
	Configured Voltage: 1.35 V

Handle 0x0041, DMI type 17, 40 bytes
Memory Device
	Array Handle: 0x003E
	Error Information Handle: Not Provided
	Total Width: 64 bits
	Data Width: 64 bits
	Size: 4096 MB
	Form Factor: SODIMM
	Set: None
	Locator: DIMM 2
	Bank Locator: Channel B Slot 0
	Type: DDR3
	Type Detail: Synchronous
	Speed: 1600 MT/s
	Manufacturer: 1315
	Serial Number: 255255255255
	Asset Tag: 9876543210
	Part Number:                   
	Rank: 2
	Configured Clock Speed: 1600 MT/s
	Minimum Voltage: 1.35 V
	Maximum Voltage: 1.5 V
	Configured Voltage: 1.35 V
```

### Core i7 (MacBook)

* **Whoops:** An errors below, *BUT not for the i3 at least*. Can try with a Ubuntu USB stick later, just in case it's Fedora. As you can see `dmidecode` version is newer on Fedora (`SELinux == disabled` FYI). If not it's... Apple I guess...

``` 
# dmidecode 3.2
Getting SMBIOS data from sysfs.
SMBIOS 2.4 present.

Wrong DMI structures length: 2628 bytes announced, only 2410 bytes available.
Handle 0x0005, DMI type 16, 15 bytes
Physical Memory Array
	Location: System Board Or Motherboard
	Use: System Memory
	Error Correction Type: None
	Maximum Capacity: 8 GB
	Error Information Handle: Not Provided
	Number Of Devices: 2

Handle 0x0007, DMI type 17, 27 bytes
Memory Device
	Array Handle: 0x0005
	Error Information Handle: No Error
	Total Width: Unknown
	Data Width: Unknown
	Size: 4096 MB
	Form Factor: SODIMM
	Set: None
	Locator: DIMM0
	Bank Locator: BANK 0
	Type: DDR3
	Type Detail: Synchronous
	Speed: 1600 MT/s
	Manufacturer: 0x80AD
	Serial Number: 0x00000000
	Asset Tag: Unknown
	Part Number: 0x483943434E4E4E384A544D4C41522D4E544D

Handle 0x0009, DMI type 17, 27 bytes
Memory Device
	Array Handle: 0x0005
	Error Information Handle: No Error
	Total Width: Unknown
	Data Width: Unknown
	Size: 4096 MB
	Form Factor: SODIMM
	Set: None
	Locator: DIMM0
	Bank Locator: BANK 1
	Type: DDR3
	Type Detail: Synchronous
	Speed: 1600 MT/s
	Manufacturer: 0x80AD
	Serial Number: 0x00000000
	Asset Tag: Unknown
	Part Number: 0x483943434E4E4E384A544D4C41522D4E544D

Invalid entry length (0). DMI table is broken! Stop.
```

---

Don't get why the **i7** MacBook has a max of 8 GB RAM (it has, I ordered the top-specced), while the **i3** NUC has a limit of 16 GB. Must be the motherboard.

olejon, 2020-01-18:
Some edits...

olejon, 2020-01-18:
Since the Apple hides the name of the Manufacturer behind a code and never says in specs, must open of possible and boid warranty, [here they are](https://github.com/hewigovens/hewigovens.github.com/wiki/Memory-vendor-code) (darn thought I had Samsung ones). They don't want you to know, just as some iPhones suddenly had either Samsung or Sony cameras (before only Samsung), but people noticed. People found ways to take photos of certain shapes to determine, and then a guy managed to read it and published an app that Apple banned, LOL.

cyring, 2020-01-18:
[CoreFreq.tar.gz](https://github.com/cyring/CoreFreq/files/4081668/CoreFreq.tar.gz)

This archive will try to decode **only** the number of Channels and the geometry of DIMMs

olejon, 2020-01-19:
Compiled, checked and written on my phone.

Seems to work! *Only tried on the Intel NUC* as the MacBook is busy downloading stuff from my file server to watch on the express boat back to Bergen (my apartment), then will be packed, as I'm leaving soon. So you'll get the output from the MacBook later.

### Intel NUC (Core i3)

```
$ ./corefreq-cli -M

                         Lynx Point-M  [ A04]                         
Controller #0                                           Dual Channel  
 Bus Rate  5000 MT/s      Bus Speed 4988 MT/s     DRAM Speed 1600 MHz 
                                                                      
 Cha    CL  RCD   RP  RAS  RRD  RFC   WR RTPr WTPr  FAW  B2B  CWL Rate
  #0    11    0    0    0    0  208    0    0    0    0    0    8   1N
  #1     0    0    0    0    0    0    0    0    0    0    0    0   1N
      ddWR drWR srWR ddRW drRW srRW ddRR drRR srRR ddWW drWW srWW  ECC
  #0     4    3   20   12   12   12    6    6    4    7    7    4   0 
  #1     0    0   16    0   12    0   12    4    0    3    1    0   0 
                                                                      
 DIMM Geometry for channel #0                                         
      Slot Bank Rank     Rows   Columns    Memory Size (MB)           
       #0     8    2     16384      1024           8192               
 DIMM Geometry for channel #1                                         
      Slot Bank Rank     Rows   Columns    Memory Size (MB)           
       #0     4    2     16384      1024           4096
```

BTW: Thanks for introducing me to the `dmidecode` command. Wasn't aware of it. Nice way to check that say my chips in my workstation are actually running at 3200 MHz (XMP). No other tool, not CoreFreq either, show that.

Such nice commands which are very unlikely to be included with Windows or macOS (latter probably has it locked down for third party apps too), but installed with the system on any distro it seems, server and desktop.

Also a Google search made me aware there's a `decode-dimms` command if you install a package, but doesn't work on MacBook/Fedora or NUC/Ubuntu, trying both modules it suggests, but doesn't manage to read anything and probably not supported on the running system/HW.

cyring, 2020-01-19:
> Compiled, checked and written on my phone.
> 
> Seems to work! 

Super !

* Now I'm looking at the timings of channel 1

cyring, 2020-01-19:
Released

olejon, 2020-01-19:
Nice. Luckily pretty good 4G in the deep fjords now. At my MacBook now, and get this from a cold boot with latest code:

```
$ corefreq-cli -M

                         Lynx Point-M  [ A04]                         
Controller #0                                           Dual Channel  
 Bus Rate  5000 MT/s      Bus Speed 4999 MT/s     DRAM Speed 1600 MHz 
                                                                      
 Cha    CL  RCD   RP  RAS  RRD  RFC   WR RTPr WTPr  FAW  B2B  CWL Rate
  #0    12    0    0    0    0  104    0    0    0    0    0    6   1N
  #1    12    0    0    0    0  104    0    0    0    0    0    6   1N
      ddWR drWR srWR ddRW drRW srRW ddRR drRR srRR ddWW drWW srWW  ECC
  #0     8    8   19   16   16   16    7    7    4    7    7    4   0 
  #1     8    8   19   16   16   16    7    7    4    7    7    4   0 
                                                                      
 DIMM Geometry for channel #0                                         
      Slot Bank Rank     Rows   Columns    Memory Size (MB)           
       #0     2    2     32768      1024           4096               
 DIMM Geometry for channel #1                                         
      Slot Bank Rank     Rows   Columns    Memory Size (MB)           
       #0     2    2     32768      1024           4096
```

:+1:

olejon, 2020-01-19:
Edit: Added NUC

cyring, 2020-01-19:
Can be the same data timings but not sourced from the same registers.
When the IMC _boots_ it trains DIMM latencies, speed, geometry from SPD, if available. But it has its own internal logic to set a runtime configuration.
That's the all point of _CoreFreq_, giving you the usage, rather than the capability of the Memory settings.
I'm not sure DMI will supply this, but it's nice to have both. Perhaps, in the future, _CoreFreq_ will be able with SPD...
However, the datasheets of all Haswell variants I've gone through didn't reveal some registers. That's why you still read zero values.

Previous AMD processors have a BIOS & Kernel Developer Guide (see BKDG in my CoreFreq wiki) which specify the Memory Controller registers.
Zen architecture BKDG has not been made publicly available. Just a preview can be downloaded. That's a reason, most open source initiatives provide less information than a Ryzen Master or any UEFI BIOS.
ACPI tables may reveal some IMC data...

cyring, 2020-03-18:
No more work.

closed 2020-03-18 by cyring