cyring/CoreFreq #166: Vcore zero (V[0.00]) and too low idle frequencies - Xeon X5650, X58 (Asus P6T)

thebucc, 2020-02-06:
Hi, I've just installed CoreFreq and it's great, everything seems to show the correct values except for:
- CPU voltage is stuck at 0.00 (it's readable with lm-sensors)
- the frequencies when idle are not credible, way too low (we are talking 5MHz). They are ok under load. The CPU is just a bit overclocked at 155MHz, and the value is shown correctly.

A similar bug is #68 , but it concerns the json output (how do I get it BTW?)

System: Ubuntu 16.04, Xeon X5650, X58 (Asus P6T)

[image]

cyring, 2020-02-07:
> * CPU voltage is stuck at 0.00 (it's readable with lm-sensors)

go to the directory where the _CoreFreq_ source code is; rebuild as bellow, then run programs (driver, daemon, client) from this location 
```
make HWM_CHIPSET=W83627 clean all
```
if you have the mentioned super I/O Chipset, you should read the Vcore. If not, please return to me the precise reference name of yours

> * the frequencies when idle are not credible, way too low (we are talking 5MHz). They are ok under load. The CPU is just a bit overclocked at 155MHz, and the value is shown correctly.

This is the expected behavior, _CoreFreq_ is providing the relative frequency among a time interval (default 1 sec) 

https://software.intel.com/en-us/vtune-help-average-cpu-frequency

cyring, 2020-02-08:
Hello,

Do you get the Vcore with `HWM_CHIPSET=W83627` ?

thebucc, 2020-02-08:
Hi, sorry for the delay, I was doing something under Windows.
Yes, it works with the new chip, now it shows the same values as lm-sensors.

thebucc, 2020-02-08:
About the "frequency" shown, it's an interesting metric, but unusual for me, I'm used to see the _real_ operating frequency of the CPU, even at idle, essentially _multiplier x BCLK_. What is shown now is the product of it and the CPU load. It would be useful to see the active multiplier, is there a way to show it?

cyring, 2020-02-08:
At best, you can read the frequencies in the Processor window. Press `p` key in the UI

That's something you will read in manufacturer specs, but at the GHz speed of modern processors, it becomes a challenge to monitor with accuracy; meaning you'll miss data _events_ over a low sampling interval. If you reduce this last one, the monitoring instructions will occur a CPU overhead by itself. You endly will monitor the monitor.
That's why APERF, MPERF and other performance counters are made for.

thebucc, 2020-02-11:
@cyring: absolutely, the sampling time should stay relatively long, we don't want to make the monitoring the main activity of the CPU! Even if the frequency is not constantly correct, it is still interesting to sample the state of (voltage, multiplier) every second or even more, to see how speedstep is working. This is what CPU-Z and most other monitoring applications under windows do.
Anyway, from my point of view the issue is closed: do I close it or you?

cyring, 2020-02-11:
I add to the _CoreFreq_ roadmap, the every sec update of the current p-state and thus the absolute frequency: I'll have to handle the architecture cases among AMD and Intel.
Stay tuned

thebucc, 2020-02-17:
Hi @cyring , I just looked up an error that always shows up when I boot into linux, and it is in fact about the IO chip:
```
Feb 17 12:29:23 i7 systemd[1]: Starting Load Kernel Modules...`
Feb 17 12:29:23 i7 systemd-modules-load[860]: Failed to insert 'w83627ehf': Device or resource busy
Feb 17 12:29:24 i7 systemd-modules-load[860]: Failed to insert 'w83627ehf': Device or resource busy
Feb 17 12:29:24 i7 systemd[1]: systemd-modules-load.service: Main process exited, code=exited, status=1/FAILURE
Feb 17 12:29:24 i7 systemd[1]: Failed to start Load Kernel Modules.
Feb 17 12:29:24 i7 systemd[1]: systemd-modules-load.service: Unit entered failed state.
Feb 17 12:29:24 i7 systemd[1]: systemd-modules-load.service: Failed with result 'exit-code'.
```
I assume this is the reason why CoreFreq didn't read the voltage without your fix. Is it possible to fix the problem at the kernel level? Should I open a bug report to the kernel people (where?)
I'm searching online but I haven't an answer yet, I'm sure you know a lot more than me about this kind of errors.

cyring, 2020-02-17:
* _CoreFreq_ reads the Vcore of W83627  _iff_
1. The mainboard has a Winbond W83627 chipset
2. The processor belongs to the Nehalem or Westmere of Intel families
https://github.com/cyring/CoreFreq/blob/37f8c5fe1562fa1d66c544cf3502333df3ebfefb/corefreqk.c#L6998

* _CoreFreq_ is using those assembly instructions to read  the Super I/O [ Chipset W83627]
https://github.com/cyring/CoreFreq/blob/37f8c5fe1562fa1d66c544cf3502333df3ebfefb/corefreqk.h#L495

thus it has no kernel dependencies **but** if another driver is reading the W83627 , in parallel of _CoreFreq_, then it will lead to a registers conflict.  --> Make sure any lm_sensors drivers are not running/loaded.

* If you still have an issue, boot the _CoreFreq_ live [image](https://github.com/cyring/CoreFreq/wiki/Live-CD) :
1. abort the compilation when prompted
2. change to the directory ```/opt/CoreFreq```
3. build with ```make HWM_CHIPSET=W83627 clean all```
4. load driver & daemon ```insmod corefreqk.ko ; ./corefreqd &```
5. start the Cli ```./corefreq-cli```

I'm reopening this issue until your problem is fixed.

thebucc, 2020-02-17:
Thanks for the explanation. 
There is no need to reopen the bug, I don't have any problems with CoreFreq now, it worked straight away after I built with `make HWM_CHIPSET=W83627 clean all` (see my second comment in the thread).
My last comment was just me wondering if the initial problem was connected to the boot error (and you clearly say it isn't).

cyring, 2020-02-17:
Referring to:
```
Feb 17 12:29:23 i7 systemd[1]: Starting Load Kernel Modules...`
Feb 17 12:29:23 i7 systemd-modules-load[860]: Failed to insert 'w83627ehf': Device or resource busy
```
what is your chipset detected by ```sensors-detect```

thebucc, 2020-02-17:
sensor-detect finds it as 
```
Driver `w83627ehf':
  * ISA bus, address 0x290
    Chip `Winbond W83667HG Super IO Sensors' (confidence: 9)
```

But I also found that I had this same block 3 times in in /etc/modules:
```
# Chip drivers
coretemp
i5500_temp
w83627ehf
```
I've remove the 2 extra blocks, maybe the warning was from trying to load the module multiple times. I'll reboot and report if the error message changes.

cyring, 2020-02-17:
OK, so closing the issue.
However a well made driver won't reopen multiple times. It usually uses a lock during its startup to allow a unique load instance.
You should try one of I/O kernel command line argument, such as
```
acpi_enforce_resources={ strict | lax | no} 
```
I'm using lax to reach Super I/O

closed 2020-02-17 by cyring

cyring, 2020-02-18:
BTW, do you mind to show me what you get in the Memory Controller window; are the values accurate ?

thebucc, 2020-02-18:
Do you mean this window (pressing 'm')? The L3 shows 12MB correctly, the other values I can't remember off the top of my head what they are supposed to be, I'd have to go and check the CPU specs
[image]

thebucc, 2020-02-18:
ok, I tried ./corefreq-cli --help and of course it is all there. What you asked should be this, right?
```
$ ./corefreq-cli -M
                         X58/Tylersburg  [2D98]                       
Controller #0                                          Triple Channel 
 Bus Rate  4800 MT/s      Bus Speed 5579 MT/s     DRAM Speed 1549 MHz 
                                                                      
 Cha    CL  RCD   RP  RAS  RRD  RFC   WR RTPr WTPr  FAW  B2B  CWL Rate
  #0     9    9    9   24    4  174   10    7   21   20    0    7   1N
  #1     9    9    9   24    4  107   10    7   21   20    0    7   1N
  #2     9    9    9   24    4  107   10    7   21   20    0    7   1N
      ddWR drWR srWR ddRW drRW srRW ddRR drRR srRR ddWW drWW srWW  ECC
  #0     5    5   16   10   10   11    7    6    4    7    7    4   0 
  #1     5    5   16   10   10   11    7    6    4    7    7    4   0 
  #2     5    5   16   10   10   11    7    6    4    7    7    4   0 
                                                                      
 DIMM Geometry for channel #0                                         
      Slot Bank Rank     Rows   Columns    Memory Size (MB)           
       #0     8    1     65536      1024           4096               
 DIMM Geometry for channel #1                                         
      Slot Bank Rank     Rows   Columns    Memory Size (MB)           
       #0     8    2     32768      1024           4096               
 DIMM Geometry for channel #2                                         
      Slot Bank Rank     Rows   Columns    Memory Size (MB)           
       #0     8    2     32768      1024           4096
```
I don't know the actual timings, but these seem ok, and DIMM info and frequency are certainly correct.

cyring, 2020-02-18:
Thanks, IMC is what I want to see, and it's nice to check the topology also.
In the UI, you press the same key than the help. 
It's case sensitive, so `M` for the IMC window
At any time, press `h` and `F1` for help and shortcuts, which are also recalled in the menu.

Your P6T BIOS should have the DIMM features you can compare _CoreFreq_ with
I've got every timings, size, geometry, speed, QPI clock, ECC,... equalled with my Xeon W3690 + Asus Rampage II gene; also w/ i7-920

I can also notice your OCed BaseClock, is the _CoreFreq_  ~155 MHz  estimation accurate ?

thebucc, 2020-02-18:
First of all, can I ask you whether your W3690 is multiplier unlocked? There is conflicting information on the web.

> I can also notice your OCed BaseClock, is the _CoreFreq_ ~155 MHz estimation accurate ?

Yes, my Xeon is multiplier locked and that's the max overclock it takes without increasing voltage. Unfortunately this motherboard disables Speedstep as soon as I change any voltage, which is unacceptable for me. It's not like it's impossible: I had other motherboards that let you just add 0.xV and still keep speedstep, but not this stupid motherboard...

cyring, 2020-02-19:
Yes, my W3690 is fully multiplier unlocked.
From 1C to 6C Turbo ratios freely [programmable](https://gist.github.com/cyring/d90bf2ae2fe31cb3b7921891103ad5ac)

thebucc, 2020-02-19:
here it is:
[image]

cyring, 2020-02-19:
Thanks. 
So both our processors are sharing the same issue: Package C-States counters remain at zero.
For a short time, I have tested this [W3530](https://gist.github.com/cyring/ab59b664892bec6a20ca9965f8b8c989), which reveals values.
I can't tell if it is due to the Lenovo motherboard capability or this Nehalem processor itself

thebucc, 2020-02-21:
This was just to clarify whether your CPU or motherboard had an issue, right? Is this something bad, how exactly? I hope it helps you to improve the application, and feel free to ask more.
About your W3690 (and other Xeons...), I have a couple more questions:
- the page you link for the W3690 seems to suggest that it is possible to "program" some parameters of the processor, multiplier for sure (on the W3690), but what else? Voltage at given multiplier?
- is anything "programmable" in the X5650? Where can I find information that can help to achieve better overclocking? Not the countless youtube videos that essentially say "crank the voltage to a silly value and you can set the BCLK to 200+MHz": as I said before, I want a CPU that runs quickly on demand, but consumes little energy at idle.
- some people online say that W3690 and W3680 are unlocked, but the W3670 isn't, do you have any sure info about this?
Thanks.

cyring, 2020-02-21:
# Voltage Core
All the materials I'm aware of:
* [Intel® 64 and IA-32 Architectures Software Developer’s Manual](https://software.intel.com/sites/default/files/managed/39/c5/325462-sdm-vol-1-2abcd-3abcd.pdf)
Table 2-2. IA-32 Architectural MSRs - Register `IA32_PERF_CTL (0x199)`
* [Intel® Xeon® Processor 5600 Series Datasheet, Volume 1](https://www.intel.com/content/dam/www/public/us/en/documents/datasheets/xeon-5600-vol-1-datasheet.pdf)  
Paragraph 2.1.7.3 - Processor Vcc VID
* [Nuveton/Winbond W83627](https://www.nuvoton.com/resource-files/DA00-WW83627DHG-P.pdf)

I have gathered the register bit layout and its architectural variation but found nothing about writing the VID with Nehalem/Westmere processors
https://github.com/cyring/CoreFreq/blob/91e8b8a47110cd5e35210fe274f11bddc87bfeb7/intelmsr.h#L442
`NHM.TargetRatio` spreads to the first 16-bits whereas `CORE.TargetVoltage` is specified in first 8-bits

No answer how to program the voltage core

# Westmere
* [Westmere forum](https://forum.hardware.fr/hfr/Hardware/Processeur/processeurs-nehalem-westmere-sujet_794002_1.htm) (french only) where I found that W3690 and W3680 are unlocked Xeon
* Got mine from this [store](https://fr.aliexpress.com/item/32851811023.html?spm=a2g0s.9042311.0.0.12aa6c37wqsxpw)

thebucc, 2020-02-24:
Thanks for the links. I thought that maybe the app could actually change some parameters of the CPU, because you said : "From 1C to 6C Turbo ratios freely **_programmable_**", but I can see now that you are not trying to do that: it's strictly for monitoring.

cyring, 2020-02-24:
> Thanks for the links. I thought that maybe the app could actually change some parameters of the CPU, because you said : "From 1C to 6C Turbo ratios freely **_programmable_**", but I can see now that you are not trying to do that: it's strictly for monitoring.

Yes, the W3690 is programmable and CoreFreq let you change the {ratio; multiplier; coefficient} of the turbo frequencies. Call it frequency overclocking.
But the voltage identifier is not writable: it's read-only register (according to specification)