cyring/CoreFreq #167: .ko won't install on Ub 18.04, 4.15.0-88

dlehenky, 2020-02-19:
Make install throws this error: root@hero:/usr/local/src/CoreFreq# make install
make -C /lib/modules/4.15.0-88-generic/build M=/usr/local/src/CoreFreq modules_install
make[1]: Entering directory '/usr/src/linux-headers-4.15.0-88-generic'
  INSTALL /usr/local/src/CoreFreq/corefreqk.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:72
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:79
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  4.15.0-88-generic
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-88-generic'
install -Dm 0755 corefreq-cli corefreqd -t /usr/bin
install -Dm 0644 corefreqd.service \
	/usr/lib/systemd/system/corefreqd.service

A number of other people with different app installs have experience similar errors. Something about the kernel module needs to be signed with a cert before it will load. Tried the suggestions to disable module verification (mokutil --disable-validation;reboot), but got the same result. Any help would be much appreciated!!

dlehenky, 2020-02-20:
Well, I did manage to get the kernel module to install with 'insmod' from the source build directory; it won't load with 'make install' or 'modprobe', even though the corefreqk.ko is in the modules "extra" directory.

dlehenky, 2020-02-20:
One more note: the systemd service fails to start because it expects corefreqd to be installed in /bin, but it's installed in /usr/bin by "make install". I sym-linked it to /bin and the service now starts successfully. Also, the kernel module is a dependency of the daemon, so the service will fail if the module isn't already installed. Just saying.

dlehenky, 2020-02-20:
OK, so after some more hacking, the kernel module loads on boot. Had to put it's name in /etc/modules, without the .ko, and then run 'depmod' and reboot. With that working, the corefreq.service starts successfully on boot, as well. No worries; it's all good. I hope this little dialog with myself helps someone else :)

cyring, 2020-02-20:
Hello

I have not find a generic & easy way to sign the module for any Linux distribution case.  
You may sign [manually](https://www.kernel.org/doc/html/latest/admin-guide/module-signing.html) `corefreqk.ko`

Something like:
```
scripts/sign-file sha512 kernel-signkey.priv \
        kernel-signkey.x509 corefreqk.ko
```

dlehenky, 2020-02-20:
> Hello
>
> I have not find a generic & easy way to sign the module for any Linux distribution case.
> You may sign [manually](https://www.kernel.org/doc/html/latest/admin-guide/module-signing.html) [...]

Thanks cry. I don't mind the whole "tainted kernel" business, so I doubt I'll bother with trying to sign it. I know it's good, so really, for my uses, that's all that counts ;) I surely hope you continue with this work! It seems you are about the only dev in the entire Linux community that sees a serious need for hardware system monitoring tools. With today's tech, you *must* be able to monitor component temps and associated cooling devices. I'd also say, based on my recent experience with Zen2/Ryzen 3000, that AMD is coming on strong, and I expect there's going to be a lot more AMD kit in the data centers soon.
-Best Care

chrisleup, 2020-02-20:
I do the following and things work well --
```
root@bawks:/usr/src/linux-headers-5.3.0-23/certs# cat x509.genkey
[ req ]
default_bits = 4096
distinguished_name = req_distinguished_name
prompt = no
string_mask = utf8only
x509_extensions = myexts

[ req_distinguished_name ]
CN = Modules

[ myexts ]
basicConstraints=critical,CA:FALSE
keyUsage=digitalSignature
subjectKeyIdentifier=hash
authorityKeyIdentifier=keyid

root@bawks# openssl req -new -nodes -utf8 -sha512 -days 36500 -batch -x509 -config x509.genkey -outform DER -out signing_key.x509 -keyout signing_key.pem
```
make yourself your modules then =)

dlehenky, 2020-02-20:
@chrisleup Thank you very much for that! Have a good evening.
-David

chrisleup, 2020-02-20:
@dlehenky 
Absolutely, friend, glad to help and post that up for posterity, it happens in various distros.. You may also have to do this for /usr/src/linux-headers-5.3.0-23-generic directory as well as linking/copying /boot/System.map-versionstringhere into the directories just as System.map ... 

```
ln -s /boot/System.map-version- /usr/src/linux-versionstring-generic/System.map
lrwxrwxrwx  1 root root   33 Jan 28 10:17 System.map -> /boot/System.map-5.3.0-29-generic
root@bawks:/usr/src/linux-headers-5.3.0-29-generic#
```
 ^^^ - beware, this will leave some leftover breadcrumbs if you use automated removal  (apt purge/autoremove)

cyring, 2020-02-20:
> I do the following and things work well --
> 
> ```
> root@bawks:/usr/src/linux-headers-5.3.0-23/certs# cat x509.genkey
> [ req ]
> default_bits = 4096
> distinguished_name = req_distinguished_name
> prompt = no
> string_mask = utf8only
> x509_extensions = myexts
> 
> [ req_distinguished_name ]
> CN = Modules
> 
> [ myexts ]
> basicConstraints=critical,CA:FALSE
> keyUsage=digitalSignature
> subjectKeyIdentifier=hash
> authorityKeyIdentifier=keyid
> 
> 
> root@bawks:/usr/src/linux-headers-5.3.0-23/certs# openssl req -new -nodes -utf8 -sha512 -days 36500 -batch -x509 -config x509.genkey -outform DER -out signing_key.x509 -keyout signing_key.pem
> ```
> 
> make yourself your modules then =)

Thank you 
Do you think of making a shell script to sign the corefreqk.ko passed as an argument ? 
Taking part of the project, I will then call it from the Makefile when user enters, for example :
```
make sign
```
Currently for Ubuntu, this script would be completed later by distro specific procedures if required.

What do you think about this ?

dlehenky, 2020-02-20:
I recently tried an AMD-specific driver called “zenpower” (on GitHub) that seems to do all this cert stuff in his build/install process. You might see something useful there. HTH.

-David

cyring, 2020-02-21:
OK, will think about that.
No remaining install problems; closing the issue.

closed 2020-02-21 by cyring