cyring/CoreFreq #169: Roadmap

cyring, 2020-03-04:
## Tools
* AVX stress loops per CPU
* Idle loops if Processor is user mwait capable

## Build
* User defined CORE_COUNT to scale programs up to 8 x sockets x 64 Cores x 2 threads = 1024 CPUs
* `man` manuel & pdf guide
* make menuconfig à la buildroot

## Architecture
* AMD EPYC & Threadripper : to provide the temperature sensor registers per CCD based on msr `0x00059954` `0x00059958`
* Refactor the topology to include cluster identiers (AMD CCD, AMD CCX, Intel cluster)
* Make one service processor per Cluster (CPU hot-plugging safe) 
* Decode the remaining memory controllers I have chance to work with. Scale to 8-12 channels
* Probably decode the DIMM SPD timings. The base of ICS is quiet large
* Add Super-I/O registers for voltage and fans; when datasheet is available

## UI
* A view or window to dump the usual MSR, and _some_ PCI space registers. Allow user defined addresses to be queried 
* Choose what to do about the GUI which won't be the Cli. Can be BIOS dashboard; Desktop widgets
* GUI framework : the Xlib has the lowest CPU overhead but it's too _primitive_ to program with. GL is even faster and claims a GPU. Beside that, high level graphic toolkits have a significant CPU overhead

## Algorithms
* Synchronization of the monitoring loops in the kernel module with the backend threads in the user-space daemon: need to find a way to relinquish the waiting time to the  kernel like the condition wait but without involving the (deadly) mutexes
* Create a kind of unified RCU shared by driver and the user-space part. Task list as a use case.
* A set of CPU features getters/setters shared by Cli and GUI

gdiazlo, 2020-03-13:
Thanks for this great tool! 

I guess you already give it a though, but, Is it too complicated to support the sensors interface? that way other tools can handle the UI (ksysguard, gnome monitor, etc.)

Again, thanks!

cyring, 2020-03-13:
> Thanks for this great tool!

Thank you for using _CoreFreq_

> 
> I guess you already give it a though, but, Is it too complicated to support the sensors interface? that way other tools can handle the UI (ksysguard, gnome monitor, etc.)
> 
> Again, thanks!

Not so complicated when looking at (coretemp)[https://elixir.bootlin.com/linux/latest/source/drivers/hwmon/coretemp.c] as an example. 
_CoreFreq_ is, so far, covering much less sensors than the available lm_sensors drivers ...

cyring, 2020-03-18:
## Power limits 
( if registers available ) : https://github.com/cyring/CoreFreq/issues/60#issuecomment-517957899
Intel: TVB, PL1, PL2, TAU
AMD Zen: PPT, TDC, EDC

## Thermal Velocity Boost
 https://github.com/cyring/CoreFreq/issues/60#issuecomment-449668251
[Thermal Velocity Boost](https://en.wikichip.org/wiki/intel/thermal_velocity_boost)
Maximus X/XI & Strix Motherboards (Z370/Z390) > [MCE explanations and others](https://rog.asus.com/forum/showthread.php?106375-MCE-explanations-and-others)
> MSR 0x150 ... set bit 63 to 1 and [39:32] to 18h
> Bit 3 shows you if TVB is enabled or disabled (0=disabled).

## Some Registers
Raised in issue #60

cyring, 2020-09-03:
# Refactoring the `Topology` window to display the AMD Zen CCD

* _The tests bellow are also including a disabled CPU_
[image]
* Intel non regression
[image]

cyring, 2021-02-27:
# AMD APM Master Mode
* [what-does-amd-application-power](https://www.ronwoods.us/2014/02/what-does-amd-application-power.html)
* coreboot/latest/source/src/vendorcode/amd/agesa/f16kb/Proc/CPU/Family/0x16/KB/F16KbPowerMgmt.h
```
/* Core Performance Boost Control Register D18F4x15C */

/// Core Performance Boost Control Register
typedef struct {
  UINT32 BoostSrc:2;                 ///< Boost source
  UINT32 NumBoostStates:3;           ///< Number of boosted states
  UINT32 :2;                         ///< Reserved
  UINT32 ApmMasterEn:1;              ///< APM master enable
  UINT32 CstatePowerEn:1;            ///< C-state Power Enable
  UINT32 :22;                        ///< Reserved
  UINT32 BoostLock:1;                ///<
} CPB_CTRL_REGISTER;
```
* AMD Ryzen 9 3950X 16-Core Processor
```
zencli pci bus:0x0-dev:0x18-fn:0x4-reg:0x15c
[0x8000c45c] READ(pci) = 0x001b0189 (1769865)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 1011 0000 0001 1000 1001
```

closed 2021-09-25 by cyring