cyring/CoreFreq #170: Conky support

livello, 2020-03-24:
Is it possible to add a switch to corefreq-cli, that will make it compatible with conky?
${color1}${exec corefreq-cli --one-time}

cyring, 2020-03-24:
That's a challenge. 
So far à JSON output is provided by the Cli but it lacks 2 main things:
* a well defined data output for Conky
* synchronization with the CoreFreq internal timer. 

In CoreFreq, data are _pushed_ from the backend driver up to the Cli. Which means that a Conky polling will block until CoreFreq signals the data availability. Conky, as a consumer will have a fraction of the Interval to collect before the next data being swapped to the alternate buffer.

cyring, 2020-04-04:
The lowest level of integration is based on the Nvidia sensors:
https://github.com/brndnmtthws/conky/blob/9517ab1e6c8112ba7348bdc5aa55c2957cf8431c/src/nvidia.cc

I have also analyse the tolua source code. 

Both cases, I don't believe we will get in final a **low latency** monitoring, which can be governed by my daemon and the driver timers. 

Feel free to close the issue. 

Regards

livello, 2020-04-04:
Thank you very much for your answer and your great work!

closed 2020-04-04 by livello

cyring, 2020-04-06:
There's a way by reading values through the `/sys` file-system like:
* this idle case:
```
# cat /sys/devices/system/cpu/cpu*/cpufreq/cpuinfo_cur_freq 
6418
597
978
1675
2379
16730
745
4458
3619
886
2465
8391
```
* and this Turbo case:
```
# cat /sys/devices/system/cpu/cpu*/cpufreq/cpuinfo_cur_freq 
3603847
3603847
3603847
3603847
3603847
3603847
3603847
3603848
3603847
3603848
3603847
3603847
```
* You will have to setup _CoreFreq_ to govern your Kernel CPU-Freq:
```
corefreq-cli -k
Linux:                                                                          
...
CPU-Freq driver                                               [  corefreqk-perf]
Governor                                                      [ corefreq-policy]
CPU-Idle driver                                               [  corefreqk-idle]
|- Idle Limit                                                 <              C6>
...
```
* First, I'm black-listing the mainstream drivers in the boot command line:
```
modprobe.blacklist=pcspkr,iTCO_wdt,acpi_cpufreq,pcc_cpufreq,intel_cstate,intel_uncore,intel_powerclamp,i7core_edac,i5500_temp,coretemp,asus_atk0110 idle=halt intel_pstate=disable intel_idle.max_cstate=0
```
* I tell _CoreFreq_ to register itself as the CPU-Freq and Governor driver, optionally CPU-Idle :
```
# insmod corefreqk.ko Register_CPU_Idle=1 Register_CPU_Freq=1 Register_Governor=1

cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_driver 
corefreqk-perf

cat /sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_min_freq 
1602053

cat /sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_max_freq 
3471115

# cat /sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_cur_freq 
7541

cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_setspeed 
3604508
# echo '2100000' > /sys/devices/system/cpu/cpu0/cpufreq/scaling_setspeed
cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_setspeed 
2002665
```
* Things you change through `/sys` will be reflected in the UI monitoring;
  like the Target P-States frequencies, the C-States idle deepest level

What do you think about this solution ?