cyring/CoreFreq #171: support for Intel i7-9750H ??

Arthav24, 2020-03-26:


cyring, 2020-03-26:
Hello !

First, can you start _CoreFreq_ ?
* If yes, can you post the output of the followings:
`corefreq-cli -s`
`corefreq-cli -m`
`corefreq-cli -u`
`corefreq-cli -M`
`corefreq-cli -k`
* If no, can you describe what is happening after loading its kernel module (dmesg, crash, ...)

Arthav24, 2020-03-27:
No , after succesfullly building it using 
make MSR_CORE_PERF_UC=MSR_CORE_PERF_FIXED_CTR1 MSR_CORE_PERF_URC=MSR_CORE_PERF_FIXED_CTR2
and then insmod corefreqk.ko 
thows this 
root@arthav-G7:/home/arthav/CoreFreq# insmod corefreqk.ko
insmod: ERROR: could not insert module corefreqk.ko: Operation not permitted
root@arthav-G7:/home/arthav/CoreFreq#

cyring, 2020-03-27:
> ...
> root@arthav-G7:/home/arthav/CoreFreq# insmod corefreqk.ko
> insmod: ERROR: could not insert module corefreqk.ko: Operation not permitted
> root@arthav-G7:/home/arthav/CoreFreq#
> 
> dmesg shows
> [ 1138.472081] Lockdown: insmod: Loading of unsigned module is restricted; see man kernel_lockdown.7

Solved @ https://github.com/cyring/CoreFreq/issues/167#issuecomment-588567507

You should have 2 options: sign the kernel module or disable the system signature enforcement.

regards

Arthav24, 2020-03-27:
after signing the module, 
root@arthav-G7:/home/arthav/CoreFreq# corefreqd
Driver connection error code 13
Version 0.0.0: 'Permission denied' @ line 5526

cyring, 2020-03-27:
Something behaves wrongly : Version is `0.0.0`
* Plz, check if errors appear in `dmesg` ?
* Do you see the driver loaded ?
`lsmod | grep corefreqk`
* Be sure to download / build and run only one source code
```sh
## as user
git clone https://github.com/cyring/CoreFreq.git

cd CoreFreq
make clean all

## as root, sign the module before loading it
insmod corefreqk.ko
./corefreqd

## as user, in another term
cd CoreFreq
./corefreq-cli
```

* If no results, then plz show me the followings:
```sh
cat /proc/cpuinfo 
lspci -nn
```
* Are these the [specs](http://www.cpu-world.com/cgi-bin/CPUID.pl?CPUID=68427) of your Processor ?

Especially the CPUID signature of Family = 06 and Model = 9e (in hexadecimal values)

If those match, then Processor belongs to `Kabylake` and it is supported here in the source code:
https://github.com/cyring/CoreFreq/blob/2f4cac48406c1e3e0bc3c7970c405c9bb94026f0/corefreqk.h#L5083

Arthav24, 2020-03-27:
- output of dmesg when I ran rmmod corefreqk and then insmod corefreqk.ko

[12248.902047] CoreFreq: Unload
[12271.990794] CoreFreq(2:8): Processor [ 06_9E] Architecture [Kaby Lake] SMT [12/12]

- arthav@arthav-G7:~$ lsmod | grep corefreqk
corefreqk             172032  0

- did that still same error on corefreqd
root@arthav-G7:/home/arthav/CoreFreq# corefreqd
Driver connection error code 13
Version 0.0.0: 'Permission denied' @ line 5526

- Outputs
**arthav@arthav-G7:~$ cat /proc/cpuinfo** 
processor	: 0
vendor_id	: GenuineIntel
cpu family	: 6
model		: 158
model name	: Intel(R) Core(TM) i7-9750H CPU @ 2.60GHz
stepping	: 10
microcode	: 0xca
cpu MHz		: 833.066
cache size	: 12288 KB
physical id	: 0
siblings	: 12
core id		: 0
cpu cores	: 6
apicid		: 0
initial apicid	: 0
fpu		: yes
fpu_exception	: yes
cpuid level	: 22
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid mpx rdseed adx smap clflushopt intel_pt xsaveopt xsavec xgetbv1 xsaves dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp md_clear flush_l1d
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs itlb_multihit
bogomips	: 5199.98
clflush size	: 64
cache_alignment	: 64
address sizes	: 39 bits physical, 48 bits virtual
power management:

processor	: 1
vendor_id	: GenuineIntel
cpu family	: 6
model		: 158
model name	: Intel(R) Core(TM) i7-9750H CPU @ 2.60GHz
stepping	: 10
microcode	: 0xca
cpu MHz		: 818.473
cache size	: 12288 KB
physical id	: 0
siblings	: 12
core id		: 1
cpu cores	: 6
apicid		: 2
initial apicid	: 2
fpu		: yes
fpu_exception	: yes
cpuid level	: 22
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid mpx rdseed adx smap clflushopt intel_pt xsaveopt xsavec xgetbv1 xsaves dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp md_clear flush_l1d
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs itlb_multihit
bogomips	: 5199.98
clflush size	: 64
cache_alignment	: 64
address sizes	: 39 bits physical, 48 bits virtual
power management:

processor	: 2
[... 279 lines cut ...]
00:16.0 Communication controller [0780]: Intel Corporation Cannon Lake PCH HECI Controller [8086:a360] (rev 10)
00:17.0 RAID bus controller [0104]: Intel Corporation 82801 Mobile SATA Controller [RAID mode] [8086:282a] (rev 10)
00:1b.0 PCI bridge [0604]: Intel Corporation Cannon Lake PCH PCI Express Root Port 21 [8086:a32c] (rev f0)
00:1d.0 PCI bridge [0604]: Intel Corporation Device [8086:a335] (rev f0)
00:1d.6 PCI bridge [0604]: Intel Corporation Device [8086:a336] (rev f0)
00:1f.0 ISA bridge [0601]: Intel Corporation Device [8086:a30d] (rev 10)
00:1f.3 Audio device [0403]: Intel Corporation Cannon Lake PCH cAVS [8086:a348] (rev 10)
00:1f.4 SMBus [0c05]: Intel Corporation Cannon Lake PCH SMBus Controller [8086:a323] (rev 10)
00:1f.5 Serial bus controller [0c80]: Intel Corporation Cannon Lake PCH SPI Controller [8086:a324] (rev 10)
01:00.0 VGA compatible controller [0300]: NVIDIA Corporation Device [10de:1f11] (rev a1)
01:00.1 Audio device [0403]: NVIDIA Corporation Device [10de:10f9] (rev a1)
01:00.2 USB controller [0c03]: NVIDIA Corporation Device [10de:1ada] (rev a1)
01:00.3 Serial bus controller [0c80]: NVIDIA Corporation Device [10de:1adb] (rev a1)
02:00.0 PCI bridge [0604]: Intel Corporation JHL6340 Thunderbolt 3 Bridge (C step) [Alpine Ridge 2C 2016] [8086:15da] (rev 02)
03:00.0 PCI bridge [0604]: Intel Corporation JHL6340 Thunderbolt 3 Bridge (C step) [Alpine Ridge 2C 2016] [8086:15da] (rev 02)
03:01.0 PCI bridge [0604]: Intel Corporation JHL6340 Thunderbolt 3 Bridge (C step) [Alpine Ridge 2C 2016] [8086:15da] (rev 02)
03:02.0 PCI bridge [0604]: Intel Corporation JHL6340 Thunderbolt 3 Bridge (C step) [Alpine Ridge 2C 2016] [8086:15da] (rev 02)
04:00.0 System peripheral [0880]: Intel Corporation JHL6340 Thunderbolt 3 NHI (C step) [Alpine Ridge 2C 2016] [8086:15d9] (rev 02)
3a:00.0 USB controller [0c03]: Intel Corporation Device [8086:15db] (rev 02)
3b:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. Device [10ec:2502] (rev 1f)
3c:00.0 Network controller [0280]: Intel Corporation Wireless-AC 9260 [8086:2526] (rev 29)

Yes, the CPUID signature of Family = 06 and Model = 9e 
https://ark.intel.com/content/www/us/en/ark/products/191045/intel-core-i7-9750h-processor-12m-cache-up-to-4-50-ghz.html
ark.intel says its coffee lake !

cyring, 2020-03-27:
Great !
Please let me know about your _CoreFreq_ experience.
Feel free to post  screenshots that I'll gather into the Wiki such as:
https://github.com/cyring/CoreFreq/wiki/CPU-support

Arthav24, 2020-03-27:
yeah sure .
**arthav@arthav-G7:~/CoreFreq$ ./corefreq-cli -s**
Processor                             [Intel(R) Core(TM) i7-9750H CPU @ 2.60GHz]
|- Architecture                                                      [Kaby Lake]
|- Vendor ID                                                      [GenuineIntel]
|- Microcode                                                        [       202]
|- Signature                                                           [  06_9E]
|- Stepping                                                            [     10]
|- Online CPU                                                          [ 12/ 12]
|- Base Clock                                                          [ 99.691]
|- Frequency            (MHz)                      Ratio                        
                 Min    797.53                    [   8 ]                       
                 Max   2591.97                    [  26 ]                       
|- Factory                                                             [100.000]
                       2600                       [  26 ]                       
|- Performance                                                                  
   |- OSPM                                                                      
                 TGT   4486.10                    <  45 >                       
   |- HWP                                                                       
                 Min    797.53                    <   8 >                       
                 Max   4486.10                    <  45 >                       
                 TGT       N/A                    <   0 >                       
|- Turbo Boost                                                         [ UNLOCK]
                  1C   4486.10                    <  45 >                       
                  2C   4386.41                    <  44 >                       
                  3C   4286.71                    <  43 >                       
                  4C   4187.02                    <  42 >                       
                  5C   4087.33                    <  41 >                       
                  6C   3987.64                    <  40 >                       
|- Uncore                                                              [ UNLOCK]
                 Min    797.53                    <   8 >                       
                 Max   4187.02                    <  42 >                       
|- TDP                                                           Level [  0:3  ]
   |- Programmable                                                     [ UNLOCK]
   |- Configuration                                                    [ UNLOCK]
   |- Turbo Activation                                                 [ UNLOCK]
             Nominal   2591.97                    [  26 ]                       
              Level1   2193.20                    [  22 ]                       
               Turbo   2492.28                    [  25 ]                       
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AVX-512      [N]  BMI1/BMI2 [Y/Y]      CLFLUSH [Y]        CMOV [Y] 
|- CMPXCHG8B    [Y]   CMPXCHG16B [Y]         F16C [Y]         FPU [Y] 
|- FXSR         [Y]    LAHF/SAHF [Y]    MMX/Ext [Y/N]  MONITOR/X[Y/N] 
|- MOVBE        [Y]          MPX [Y]    PCLMULQDQ [Y]      POPCNT [Y] 
|- RDRAND       [Y]       RDSEED [Y]       RDTSCP [Y]         SEP [Y] 
|- SGX          [Y]          SSE [Y]         SSE2 [Y]        SSE3 [Y] 
|- SSSE3        [Y]  SSE4.1/4A [Y/N]       SSE4.2 [Y]     SYSCALL [Y] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- Core Multi-Processing                                  CMP Legacy   [Missing]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Capable]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Capable]
|- CPL Qualified Debug Store                                  DS-CPL   [Capable]
|- 64-Bit Debug Store                                         DTES64   [Capable]
|- Fast-String Operation                                Fast-Strings   [Capable]
|- Fused Multiply Add                                     FMA | FMA4   [Capable]
|- Hardware Lock Elision                                         HLE   [Missing]
|- Long Mode 64 bits                                       IA64 | LM   [Capable]
|- LightWeight Profiling                                         LWP   [Missing]
|- Machine-Check Architecture                                    MCA   [Capable]
|- Model Specific Registers                                      MSR   [Capable]
|- Memory Type Range Registers                                  MTRR   [Capable]
|- OS-Enabled Ext. State Management                          OSXSAVE   [Capable]
[... cut ...]
    Averages:        Turbo  C0(%)  C1(%)  C3(%)  C6(%)  C7(%)    TjMax:    Pkg:
                      2.96   6.44  14.28   1.56   4.00  73.72     100 C    55 C

**arthav@arthav-G7:~/CoreFreq$ ./corefreq-cli -g**
		Cycles		State(%)
PC02	        1610194352	  93.19
PC03	                 0	   0.00
PC06	                 0	   0.00
PC07	                 0	   0.00
PC08	                 0	   0.00
PC09	                 0	   0.00
PC10	                 0	   0.00
PTSC	        2592003108
UNCORE	            355178

[image]
[image]
[image]

Arthav24, 2020-03-27:
arthav@arthav-G7:~/CoreFreq$ ./corefreq-cli -s
Processor                             [Intel(R) Core(TM) i7-9750H CPU @ 2.60GHz]
|- Architecture                                                      [Kaby Lake]
|- Vendor ID                                                      [GenuineIntel]
|- Microcode                                                        [       202]
|- Signature                                                           [  06_9E]
|- Stepping                                                            [     10]
|- Online CPU                                                          [ 12/ 12]
|- Base Clock                                                          [ 99.691]
|- Frequency            (MHz)                      Ratio                        
                 Min    797.53                    [   8 ]                       
                 Max   2591.96                    [  26 ]                       
|- Factory                                                             [100.000]
                       2600                       [  26 ]                       
|- Performance                                                                  
   |- OSPM                                                                      
                 TGT   4486.09                    <  45 >                       
   |- HWP                                                                       
                 Min    797.53                    <   8 >                       
                 Max   4486.09                    <  45 >                       
                 TGT       N/A                    <   0 >                       
|- Turbo Boost                                                         [ UNLOCK]
                  1C   4486.09                    <  45 >                       
                  2C   4386.40                    <  44 >                       
                  3C   4286.71                    <  43 >                       
                  4C   4187.02                    <  42 >                       
                  5C   4087.33                    <  41 >                       
                  6C   3987.64                    <  40 >                       
|- Uncore                                                              [ UNLOCK]
                 Min    797.53                    <   8 >                       
                 Max   4187.02                    <  42 >                       
|- TDP                                                           Level [  0:3  ]
   |- Programmable                                                     [ UNLOCK]
   |- Configuration                                                    [ UNLOCK]
   |- Turbo Activation                                                 [ UNLOCK]
             Nominal   2591.96                    [  26 ]                       
              Level1   2193.20                    [  22 ]                       
               Turbo   2492.27                    [  25 ]                       
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AVX-512      [N]  BMI1/BMI2 [Y/Y]      CLFLUSH [Y]        CMOV [Y] 
|- CMPXCHG8B    [Y]   CMPXCHG16B [Y]         F16C [Y]         FPU [Y] 
|- FXSR         [Y]    LAHF/SAHF [Y]    MMX/Ext [Y/N]  MONITOR/X[Y/N] 
|- MOVBE        [Y]          MPX [Y]    PCLMULQDQ [Y]      POPCNT [Y] 
|- RDRAND       [Y]       RDSEED [Y]       RDTSCP [Y]         SEP [Y] 
|- SGX          [Y]          SSE [Y]         SSE2 [Y]        SSE3 [Y] 
|- SSSE3        [Y]  SSE4.1/4A [Y/N]       SSE4.2 [Y]     SYSCALL [Y] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- Core Multi-Processing                                  CMP Legacy   [Missing]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Capable]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Capable]
|- CPL Qualified Debug Store                                  DS-CPL   [Capable]
|- 64-Bit Debug Store                                         DTES64   [Capable]
|- Fast-String Operation                                Fast-Strings   [Capable]
|- Fused Multiply Add                                     FMA | FMA4   [Capable]
|- Hardware Lock Elision                                         HLE   [Missing]
|- Long Mode 64 bits                                       IA64 | LM   [Capable]
|- LightWeight Profiling                                         LWP   [Missing]
|- Machine-Check Architecture                                    MCA   [Capable]
|- Model Specific Registers                                      MSR   [Capable]
|- Memory Type Range Registers                                  MTRR   [Capable]
|- OS-Enabled Ext. State Management                          OSXSAVE   [Capable]
[... cut ...]
|- Thermal Design Power                                          TDP   [     45]
   |- Minimum Power                                              Min   [Missing]
   |- Maximum Power                                              Max   [Missing]
|- Units                                                                        
   |- Power                                               watt   [  0.125000000]
   |- Energy                                             joule   [  0.000061035]
   |- Window                                            second   [  0.000976562]

Arthav24, 2020-03-27:
please suggest me some undervolting utility for 9750h. I wish to undervolt it by 0.125mV

cyring, 2020-03-27:
> please suggest me some undervolting utility for 9750h. I wish to undervolt it by 0.125mV

* [i7-9750H](https://gist.github.com/cyring/8c781eb81a8d61e736920da70ad7fcf8) added. Thank you
* Beside BIOS, I don't see other _privileged_ mean to decrease the Vcore

* Your Processor is `HWP` capable.  
Go to `Menu > Window > Power & Thermal > Energy Policy - HWP EPP` to set it to a saving energy profile. This will lower Vcore, but unfortunately the Frequency also

Arthav24, 2020-03-27:
Happy to see i7-9750H in CPU support and Thanks for the mention :)
Yeah that I know can set it more like 175 or 200 but I was thinking to decrease Vcore just like Throttlestop does in windows. I am not so sure but it is done by changing MSR values. After plundervolt fix pushed by intel and dell its impossible to undervolt by softwares like XTU and Throttlestop in windows , now its on Dell they implement basic Vcore offfset settings in BIOS for supported CPSs.

cyring, 2020-03-27:
> ... Yeah that I know can set it more like 175 or 200 but I was thinking to decrease Vcore just like Throttlestop does in windows. I am not so sure but it is done by changing MSR values. ...

Can you export data from Throttlestop (or source code) to know which MSRs is/are involved ?

Arthav24, 2020-03-27:
let me see if I can or if something available online about it.

Arthav24, 2020-03-27:
0x150 is the msr , the values being written to. 
I found this 
https://github.com/georgewhewell/undervolt
https://www.intel.com/content/www/us/en/architecture-and-technology/64-ia-32-architectures-software-developer-vol-3b-part-2-manual.html
https://github.com/mihic/linux-intel-undervolt

cyring, 2020-03-27:
Yes I had searched for TVB, and I left some bits layout in the issue #121
 > MSR 0x150 ... set bit 63 to 1 and [39:32] to 18h
> Bit 3 shows you if TVB is enabled or disabled (0=disabled).
Using command line `rdmsr -ax`, plz post the output values of all Cores
* Also, using your Windows tool, change the settings to alter the Vcore as will; reading this MSR, before and after the change.

Arthav24, 2020-03-28:
surprisingly,
arthav@arthav-G7:~$ rdmsr -ax 0x150
arthav@arthav-G7:~$ sudo rdmsr -ax 0x150

No output in linux.
will post from windows soon

cyring, 2020-03-28:
> surprisingly,
> arthav@arthav-G7:~$ rdmsr -ax 0x150 arthav@arthav-G7:~$ sudo rdmsr -ax 0x150
> 
> No output in linux.
> will post from windows soon

1. was kernel module `msr.ko` correctly loaded ?
2. might be this register is not supported b/c TVB is not available in 9750H ?

Arthav24, 2020-03-31:
1) root@arthav-G7:/home/arthav/CoreFreq# modinfo msr.ko
modinfo: ERROR: Module msr.ko not found.

after I loaded the msr 
**root@arthav-G7:/home/arthav/CoreFreq# modinfo msr**
filename:       /lib/modules/5.3.0-42-generic/kernel/arch/x86/kernel/msr.ko
license:        GPL
description:    x86 generic MSR driver
author:         H. Peter Anvin <hpa@zytor.com>
srcversion:     E88DB62FB5C5B69B9D18A1F
depends:        
retpoline:      Y
intree:         Y
name:           msr
vermagic:       5.3.0-42-generic SMP mod_unload 
signat:         PKCS#7
signer:         
sig_key:        
sig_hashalgo:   md4
root@arthav-G7:/home/arthav/CoreFreq# rdmsr -ax 0x150
18593
18593
18593
18593
18593
18593
18593
18593
18593
18593
18593
18593
root@arthav-G7:/home/arthav/CoreFreq#

cyring, 2020-03-31:
converted to binary:
`11000010110010011`
* bit 3 equals zero, thus TVB is disabled
* let's set this bit to one, and modify register on all cores
`11000010110011011 = 1859B`
```
wrmsr -ax 0x150 0x1859b
rdmsr -ax 0x150
```
* the readings should output the same value as written to confirm TVB has been enabled
* Do some CoreFreq measurements before and after enabling TVB such as max frequency single CPU; Temperature ; Vcore;... You can stress from the Menu - Tools
* Plz post Frequency and Chpset view screenshots, before and after TVB

Arthav24, 2020-03-31:
I was onto same what you asked, but when I tried to insmod corefreqk.ko this came
root@arthav-G7:/home/arthav/CoreFreq# insmod corefreqk.ko
insmod: ERROR: could not insert module corefreqk.ko: Invalid module format

so I ran make clean and make all and then signed again now its working fine.
then 
root@arthav-G7:/home/arthav/CoreFreq# rdmsr -ax 0x150
185c4
185c4
185c4
185c4
185c4
185c4
185c4
185c4
185c4
185c4
185c4
185c4

185c4 is 11000010111000100 means the 3rd bit is 1 so TVB must be enabled.
I dont know what happened between restarts that it changed the value of 0x150 from 18593 to 185c4.
here are some screen shots 

[image]
[image]

Arthav24, 2020-03-31:
I was onto same what you asked, but when I tried to insmod corefreqk.ko this came
root@arthav-G7:/home/arthav/CoreFreq# insmod corefreqk.ko
insmod: ERROR: could not insert module corefreqk.ko: Invalid module format

so I ran make clean and make all and then signed again now its working fine.
then 
root@arthav-G7:/home/arthav/CoreFreq# rdmsr -ax 0x150
185c4
185c4
185c4
185c4
185c4
185c4
185c4
185c4
185c4
185c4
185c4
185c4

185c4 is 11000010111000100 means the 3rd bit is 0 so TVB must be disabled.
I dont know what happened between restarts that it changed the value of 0x150 from 18593 to 185c4.
so to enable it I tried to flip 3 bit to 1 so it will be 185cc or 11000010111001100 
but arthav@arthav-G7:~$ sudo wrmsr -a 0x150 0x80000000000185CC
wrmsr: pwrite: Operation not permitted

here are some screen shots before enabling TVB 
[image]
[image]

cyring, 2020-03-31:
I have searched _again_ in the last Intel SDM for MSR `0x150` specifications; and it is still undocumented.
So we have to rely on reversed engineering source codes...
So things found on the Internet say that:
* bit 3 is for the status : don't change it
* bit 63 allows to toggle the feature (which should reflects in bit 3)

* one good thing is reading / writing this MSR do not crash your Processor 
* I suggest you observe the bits values before /after tweaking TVB from any tools, or BIOS (if available)

cyring, 2020-03-31:
ark.intel.com/content/www/fr/fr/ark/products/191045/intel-core-i7-9750h-processor-12m-cache-up-to-4-50-ghz.html

I need to fix :
* Core i7-9750H of family `06_9E` belongs to Coffee Lake
* TVB is marked as None in Intel's datasheet. This might explain you can't write the MSR, but why can you still read something ?

Arthav24, 2020-04-01:
Yes the intel official page for i7-9750H marks support for TVB as No.
That explains the in ability to write the particular bit although it was wrong still.
the 0x150 msr might contain some other supported feature bits thats why its reading something may be

cyring, 2020-04-01:
Unfortunately we need another CPU to develop this feature.

Meanwhile I have pushed in the `develop` branch, some Kaby Lake  and Coffee Lake architecture name detection fixes. Could you give it a try and show me a screenshot please ?

Arthav24, 2020-04-03:
[image]
[image]

cyring, 2020-04-06:
[Updated](https://gist.github.com/cyring/8c781eb81a8d61e736920da70ad7fcf8), thank you.

Arthav24, 2020-04-25:
Apr 25 16:53:08 arthav-G7 undervolt.py[6256]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/4/msr
Apr 25 16:53:08 arthav-G7 undervolt.py[6256]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/5/msr
Apr 25 16:53:08 arthav-G7 undervolt.py[6256]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/6/msr
Apr 25 16:53:08 arthav-G7 undervolt.py[6256]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/7/msr
Apr 25 16:53:08 arthav-G7 undervolt.py[6256]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/8/msr
Apr 25 16:53:08 arthav-G7 undervolt.py[6256]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/9/msr
Apr 25 16:53:08 arthav-G7 undervolt.py[6256]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/10/msr
Apr 25 16:53:08 arthav-G7 undervolt.py[6256]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/11/msr
Apr 25 16:53:08 arthav-G7 undervolt.py[6256]: INFO:root:Read 0xf0000000 from /dev/cpu/0/msr

this I pulled from undervolt service output. May be you can integrate undervolting corefreq. That would be awesome Thanks

cyring, 2020-04-25:
```
Apr 25 16:53:08 arthav-G7 undervolt.py[6256]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/4/msr
Apr 25 16:53:08 arthav-G7 undervolt.py[6256]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/5/msr
Apr 25 16:53:08 arthav-G7 undervolt.py[6256]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/6/msr
Apr 25 16:53:08 arthav-G7 undervolt.py[6256]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/7/msr
Apr 25 16:53:08 arthav-G7 undervolt.py[6256]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/8/msr
Apr 25 16:53:08 arthav-G7 undervolt.py[6256]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/9/msr
Apr 25 16:53:08 arthav-G7 undervolt.py[6256]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/10/msr
Apr 25 16:53:08 arthav-G7 undervolt.py[6256]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/11/msr
Apr 25 16:53:08 arthav-G7 undervolt.py[6256]: INFO:root:Read 0xf0000000 from /dev/cpu/0/msr
```  

> this I pulled from undervolt service output. May be you can integrate undervolting corefreq. That would be awesome Thanks

cyring, 2020-04-25:
Did you got those msr from Core i7-9750H ?
(b/c last time we talked about, it was not TVB  capable)

Arthav24, 2020-04-25:
1) Yes I pulled this from i7-9750H equipped Dell G7 7590
2) Tbt it also triggered curiosity in me too that why only writing to processors 4 to 11 ! 
3) Yes I noticed immediate change in Vcore in corefreq upon writing msr through undervolt.

cyring, 2020-04-25:
* According to the few source codes about subject, AC or DC could make a difference when tweaking msr `0x150`. It may explain why our previous tests failed b/c of current mode.
* TVB looks declined in planes: that could be the reason of the CPU numbers confusion
* Programming the TJ offset can also be combined with TVB. That's definitely a feature I want to add into _CoreFreq_
* I still don't know what's the scope of the msr `0x150`, but if it follows a mailbox protocol, any and only one Core has to program it.
* Most TVB apps, I have quickly studied, don't feel comfortable with programming the voltage offset:
1. compute rounded negative value
2. reciprocal computal check
3. writing msr
4. reading, comparing written value

...

Arthav24, 2020-04-26:
Sorry, It from my end I couldn't get the whole output as it was truncating itself. 
Apr 26 11:57:11 arthav-G7 systemd[1]: Starting undervolt...
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Setting core offset to -125mV
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000011f0000000 to /dev/cpu/0/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000011f0000000 to /dev/cpu/1/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000011f0000000 to /dev/cpu/2/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000011f0000000 to /dev/cpu/3/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000011f0000000 to /dev/cpu/4/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000011f0000000 to /dev/cpu/5/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000011f0000000 to /dev/cpu/6/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000011f0000000 to /dev/cpu/7/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000011f0000000 to /dev/cpu/8/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000011f0000000 to /dev/cpu/9/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000011f0000000 to /dev/cpu/10/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000011f0000000 to /dev/cpu/11/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000001000000000 to /dev/cpu/0/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000001000000000 to /dev/cpu/1/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000001000000000 to /dev/cpu/2/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000001000000000 to /dev/cpu/3/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000001000000000 to /dev/cpu/4/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000001000000000 to /dev/cpu/5/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000001000000000 to /dev/cpu/6/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000001000000000 to /dev/cpu/7/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000001000000000 to /dev/cpu/8/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000001000000000 to /dev/cpu/9/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000001000000000 to /dev/cpu/10/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000001000000000 to /dev/cpu/11/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Read 0xf0000000 from /dev/cpu/0/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Setting cache offset to -125mV
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000211f0000000 to /dev/cpu/0/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000211f0000000 to /dev/cpu/1/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000211f0000000 to /dev/cpu/2/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000211f0000000 to /dev/cpu/3/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000211f0000000 to /dev/cpu/4/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000211f0000000 to /dev/cpu/5/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000211f0000000 to /dev/cpu/6/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000211f0000000 to /dev/cpu/7/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000211f0000000 to /dev/cpu/8/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000211f0000000 to /dev/cpu/9/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000211f0000000 to /dev/cpu/10/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000211f0000000 to /dev/cpu/11/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/0/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/1/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/2/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/3/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/4/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/5/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/6/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/7/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/8/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/9/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/10/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/11/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Read 0xf0000000 from /dev/cpu/0/msr
Apr 26 11:57:11 arthav-G7 systemd[1]: Started undervolt.

cyring, 2020-04-26:
# MSR 0x150

SET| CLR   | SEL   | RSV   | RSV | WRT|  OFS  | CLR
------|---------|---------|---------|----------|------|----------|--------
`63`|`62:44`|`43:40`|`39:36`|`35:33`|`32` |`31:21`|`20:0`

* `SEL` Plane Selector among those values

PLANE|SEL
---------|-------
CORE|`0`
GPU   |`1`
CACHE Level ? |`2`
UNCORE IMC ?|`3`
OTHER|`4`

* `OFS` Voltage Offset in decimal range [ `-1023` , `+1023` ]

SIGN|VALUE
-------|----------
 `31` | `30:21`

# Sources
* [undervolt.py](https://github.com/georgewhewell/undervolt/blob/master/undervolt.py)
* [Manjaro wiki](https://wiki.manjaro.org/index.php?title=Undervolt_intel_CPU)
* [ressman.org](https://ressman.org/posts/2018-06-26-voltage-temperature-tdp-thinkpad/)

cyring, 2020-12-07:
Moved to the Roadmap.

closed 2020-12-07 by cyring