cyring/CoreFreq #172: Regression:  wrong architecture displayed for Ryzen 5 3600

obaudys, 2020-04-07:
Up until commit 298af3fe5a8664d8b8094385277b533a44281a7a  my CPU was detected correctly as `Zen2/Matisse`.  eg:
`CoreFreq(4:10): Processor [ 8F_71] Architecture [Zen2/Matisse] SMT [12/12]`

From that commit onwards,   it is detected as `Zen/Summit Ridge` and in the latest commits as `Zen/Whitehaven`, eg:
`CoreFreq(3:9): Processor [ 8F_71] Architecture [Zen/Summit Ridge] SMT [12/12]`

The turbo frequency is also incorrect, I assume that is determined by the architecture.

BTW, this project is awesome!

cyring, 2020-04-07:
Thank you for letting me know.

* Can you please try this hot-fix:
```
git clone --branch hotfix_LookupProcessor --single-branch https://github.com/cyring/CoreFreq.git
```

* About Turbo:
1.  Intel provides registers to simply discover the Turbo multipliers.
2.  No such MSR documented within Zen. (_Except triggering a high load on the Boosted P-State_)
  The purpose of `LookupProcessor()` is to provide such values during the driver startup.

obaudys, 2020-04-07:
Unfortunately, I still get the following with the hotfix:
`CoreFreq(3:9): Processor [ 8F_71] Architecture [Zen/Whitehaven] SMT [12/12]`

cyring, 2020-04-07:
Can you show me the processor brand string resulting from `corefreq-cli -s`

obaudys, 2020-04-07:
Here are the strings you wanted, both of these are from hotfix_LookupProcessor:
```
Processor                                   [AMD Ryzen 5 3600 6-Core Processor ]
|- Architecture                                                 [Zen/Whitehaven]
|- Vendor ID                                                      [AuthenticAMD]
|- Microcode                                                        [ 141561875]
|- Signature                                                           [  8F_71]
|- Stepping                                                            [      0]
|- Online CPU                                                          [ 12/ 12]
```
```
./corefreq-cli -v
1.75.2
```

cyring, 2020-04-07:
Hello,
Can you try the new fix ?

cyring, 2020-04-07:
One more fix pushed. Sorry for the mess.

obaudys, 2020-04-07:
The new fix works for me! Thanks for your work!
```
CoreFreq(5:11): Processor [ 8F_71] Architecture [Zen2/Matisse] SMT [12/12]
```
All the speeds are correct too:
```
Processor                                   [AMD Ryzen 5 3600 6-Core Processor ]
|- Architecture                                                   [Zen2/Matisse]
|- Vendor ID                                                      [AuthenticAMD]
|- Microcode                                                        [ 141561875]
|- Signature                                                           [  8F_71]
|- Stepping                                                            [      0]
|- Online CPU                                                          [ 12/ 12]
|- Base Clock                                                          [100.001]
|- Frequency            (MHz)                      Ratio                        
                 Min    400.00                    <   4 >                       
                 Max   3600.03                    <  36 >                       
|- Factory                                                             [100.000]
                       3600                       [  36 ]                       
|- Performance                                                                  
   |- OSPM                                                                      
                 TGT   2200.02                    <  22 >                       
|- Turbo Boost                                                         [ UNLOCK]
                 XFR   4300.04                    [  43 ]                       
                 CPB   4200.04                    [  42 ]                       
                  1C   2800.02                    <  28 >                       
                  2C   2200.02                    <  22 >                       
```

cyring, 2020-04-07:
> The new fix works for me! Thanks for your work!
>
> ```
> CoreFreq(5:11): Processor [ 8F_71] Architecture [Zen2/Matisse] SMT [12/12]
> ```
>
> All the speeds are correct too:
>
> ``` [...]

Glad to see a `Base Clock` estimation close to the factory one at 100 MHz.
Usually it falls around [`99.8 MHz`](https://github.com/cyring/CoreFreq/wiki/CPU-support#amd)
What is your motherboard ? Can you post a screenshot of the _CoreFreq_ UI ?

obaudys, 2020-04-08:
My motherboard is an MSI Gaming Pro Carbon AC (B450 chipset)

[image]

cyring, 2020-04-08:
When using a plain Xterm, the UI will show better the dark gray color which is used for disabled features and other decorations. You may have a color profile option within your terminal, such as Linux theme.

obaudys, 2020-04-08:
Are you asking for a screenshot with dark gray background?  Well, here's one:
[image]

cyring, 2020-04-08:
Nom, I mean you are missing separators, unit symbols and labels, such as, in the footer, between SMT and BOOST

I'm also noticing a bug  that Min values Watt and Joules, per Core, are not shown or computed

obaudys, 2020-04-08:
Ah, I see it now, it looks even better!   To get it working with gnome-terminal, I enabled the "Show bold text in bright colors" preference.  I'm not sure how to get it working through MacOS terminal via ssh.

I had assumed the power values and certain C-states etc were not implemented or exported by the Ryzen CPUs.  If you need any help tracking down bugs, please let me know.

cyring, 2020-04-08:
> Ah, I see it now, it looks even better! To get it working with gnome-terminal, I enabled the "Show bold text in bright colors" preference. I'm not sure how to get it working through MacOS terminal via ssh.

I did make tests with OS/X terminals and was ok through ssh but have to set a standard color palette. 

> 
> I had assumed the power values and certain C-states etc were not implemented or exported by the Ryzen CPUs. If you need any help tracking down bugs, please let me know.

Indeed no AMD is far publishing the counters registers to measure the C-States. I'm understanding that the ACPI space can provide them. I believe most other tools are going through this route. 
But, I'm not sure if the ACPI provides the same accuracy that the full 64 bits MSR ? Can the ACPI be polled at 100 ms interval ?  Does it provide _all_ levels C3, C6, Cn per Core, Cluster CCX, CCD, Package ? 
I feel confident that if the TSC and Instructions counters are published, it is just a matter of time for the manufacturer to unveil the remaining. 
> please AMD, release the full BKDG

obaudys, 2020-04-09:
I'm satisfied that the issue is resolved, since the architecture and frequencies are correctly detected;  you can close this if you like.

cyring, 2020-04-10:
Using the `develop` branch, could you test again for non regression.  

I have indeed refactor the Zen tables to split the AMD families per CPUID signature ; which means that any future Zen processor should fall through into its inherent class.

Please post the full output of `corefreq-cli -s`
Thanks for your help.

obaudys, 2020-04-10:
The develop branch does not show the correct architecture or frequencies;  here's the output of `corefreq-cli -s`:
```
Processor                                   [AMD Ryzen 5 3600 6-Core Processor ]
|- Architecture                                                        [AMD Zen]
|- Vendor ID                                                      [AuthenticAMD]
|- Microcode                                                        [ 141561875]
|- Signature                                                           [  8F_71]
|- Stepping                                                            [      0]
|- Online CPU                                                          [ 12/ 12]
|- Base Clock                                                          [100.000]
|- Frequency            (MHz)                      Ratio                        
                 Min    400.00                    [   4 ]                       
                 Max   3599.99                    [  36 ]                       
|- Factory                                                             [100.000]
                       3600                       [  36 ]                       
|- Performance                                                                  
   |- OSPM                                                                      
                 TGT   2199.99                    <  22 >                       
|- Turbo Boost                                                         [   LOCK]
                 XFR   3599.99                    [  36 ]                       
                 CPB   3599.99                    [  36 ]                       
                  1C   2799.99                    [  28 ]                       
                  2C   2199.99                    [  22 ]                       
|- Uncore                                                              [   LOCK]
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AVX-512      [N]  BMI1/BMI2 [Y/Y]      CLFLUSH [Y]        CMOV [Y] 
|- CMPXCHG8B    [Y]   CMPXCHG16B [Y]         F16C [Y]         FPU [Y] 
|- FXSR         [Y]    LAHF/SAHF [Y]    MMX/Ext [Y/Y]  MONITOR/X[Y/Y] 
|- MOVBE        [Y]          MPX [N]    PCLMULQDQ [Y]      POPCNT [Y] 
|- RDRAND       [Y]       RDSEED [Y]       RDTSCP [Y]         SEP [Y] 
|- SGX          [N]          SSE [Y]         SSE2 [Y]        SSE3 [Y] 
|- SSSE3        [Y]  SSE4.1/4A [Y/Y]       SSE4.2 [Y]     SYSCALL [Y] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- 100 MHz multiplier Control                            100MHzSteps   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- Core Multi-Processing                                  CMP Legacy   [Capable]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Capable]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Missing]
|- CPL Qualified Debug Store                                  DS-CPL   [Missing]
|- 64-Bit Debug Store                                         DTES64   [Missing]
|- Fast-String Operation                                Fast-Strings   [Missing]
|- Fused Multiply Add                                     FMA | FMA4   [Capable]
|- Hardware Lock Elision                                         HLE   [Missing]
|- Long Mode 64 bits                                       IA64 | LM   [Capable]
|- LightWeight Profiling                                         LWP   [Missing]
|- Machine-Check Architecture                                    MCA   [Capable]
|- Model Specific Registers                                      MSR   [Capable]
|- Memory Type Range Registers                                  MTRR   [Capable]
|- No-Execute Page Protection                                     NX   [Capable]
|- OS-Enabled Ext. State Management                          OSXSAVE   [Capable]
|- Physical Address Extension                                    PAE   [Capable]
|- Page Attribute Table                                          PAT   [Capable]
|- Pending Break Enable                                          PBE   [Missing]
|- Process Context Identifiers                                  PCID   [Missing]
|- Perfmon and Debug Capability                                 PDCM   [Missing]
|- Page Global Enable                                            PGE   [Capable]
|- Page Size Extension                                           PSE   [Capable]
|- 36-bit Page Size Extension                                  PSE36   [Capable]
|- Processor Serial Number                                       PSN   [Missing]
|- Restricted Transactional Memory                               RTM   [Missing]
|- Safer Mode Extensions                                         SMX   [Missing]
|- Self-Snoop                                                     SS   [Missing]
[... cut ...]
|- Thermal Design Power                                          TDP   [Missing]
   |- Minimum Power                                              Min   [Missing]
   |- Maximum Power                                              Max   [Missing]
|- Units                                                                        
   |- Power                                               watt   [  0.125000000]
   |- Energy                                             joule   [  0.000015259]
   |- Window                                            second   [  0.000976562]
```

cyring, 2020-04-10:
Can you pull last source code in `develop` and test again.

Be sure to fully rebuild with a `make clean all`

Thank you

obaudys, 2020-04-10:
Done.  It seems to be good.  Here's a dump of `corefreq-cli -s`:
```
Processor                                   [AMD Ryzen 5 3600 6-Core Processor ]
|- Architecture                                                   [Zen2/Matisse]
|- Vendor ID                                                      [AuthenticAMD]
|- Microcode                                                        [ 141561875]
|- Signature                                                           [  8F_71]
|- Stepping                                                            [      0]
|- Online CPU                                                          [ 12/ 12]
|- Base Clock                                                          [100.000]
|- Frequency            (MHz)                      Ratio                        
                 Min    400.00                    <   4 >                       
                 Max   3600.00                    <  36 >                       
|- Factory                                                             [100.000]
                       3600                       [  36 ]                       
|- Performance                                                                  
   |- OSPM                                                                      
                 TGT   2200.00                    <  22 >                       
|- Turbo Boost                                                         [ UNLOCK]
                 XFR   4300.00                    [  43 ]                       
                 CPB   4200.00                    [  42 ]                       
                  1C   2800.00                    <  28 >                       
                  2C   2200.00                    <  22 >                       
|- Uncore                                                              [   LOCK]
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AVX-512      [N]  BMI1/BMI2 [Y/Y]      CLFLUSH [Y]        CMOV [Y] 
|- CMPXCHG8B    [Y]   CMPXCHG16B [Y]         F16C [Y]         FPU [Y] 
|- FXSR         [Y]    LAHF/SAHF [Y]    MMX/Ext [Y/Y]  MONITOR/X[Y/Y] 
|- MOVBE        [Y]          MPX [N]    PCLMULQDQ [Y]      POPCNT [Y] 
|- RDRAND       [Y]       RDSEED [Y]       RDTSCP [Y]         SEP [Y] 
|- SGX          [N]          SSE [Y]         SSE2 [Y]        SSE3 [Y] 
|- SSSE3        [Y]  SSE4.1/4A [Y/Y]       SSE4.2 [Y]     SYSCALL [Y] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- 100 MHz multiplier Control                            100MHzSteps   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- Core Multi-Processing                                  CMP Legacy   [Capable]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Capable]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Missing]
|- CPL Qualified Debug Store                                  DS-CPL   [Missing]
|- 64-Bit Debug Store                                         DTES64   [Missing]
|- Fast-String Operation                                Fast-Strings   [Missing]
|- Fused Multiply Add                                     FMA | FMA4   [Capable]
|- Hardware Lock Elision                                         HLE   [Missing]
|- Long Mode 64 bits                                       IA64 | LM   [Capable]
|- LightWeight Profiling                                         LWP   [Missing]
|- Machine-Check Architecture                                    MCA   [Capable]
|- Model Specific Registers                                      MSR   [Capable]
|- Memory Type Range Registers                                  MTRR   [Capable]
|- No-Execute Page Protection                                     NX   [Capable]
|- OS-Enabled Ext. State Management                          OSXSAVE   [Capable]
|- Physical Address Extension                                    PAE   [Capable]
|- Page Attribute Table                                          PAT   [Capable]
|- Pending Break Enable                                          PBE   [Missing]
|- Process Context Identifiers                                  PCID   [Missing]
|- Perfmon and Debug Capability                                 PDCM   [Missing]
|- Page Global Enable                                            PGE   [Capable]
|- Page Size Extension                                           PSE   [Capable]
|- 36-bit Page Size Extension                                  PSE36   [Capable]
|- Processor Serial Number                                       PSN   [Missing]
|- Restricted Transactional Memory                               RTM   [Missing]
|- Safer Mode Extensions                                         SMX   [Missing]
|- Self-Snoop                                                     SS   [Missing]
|- Supervisor-Mode Execution Prevention                         SMEP   [Capable]
[... cut ...]
|- Thermal Design Power                                          TDP   [Missing]
   |- Minimum Power                                              Min   [Missing]
   |- Maximum Power                                              Max   [Missing]
|- Units                                                                        
   |- Power                                               watt   [  0.125000000]
   |- Energy                                             joule   [  0.000015259]
   |- Window                                            second   [  0.000976562]
```

cyring, 2020-04-10:
# Brand string clean-up
Another bug which happens only AMD:
```
Processor                                   [AMD Ryzen 5 3600 6-Core Processor ]
```
* The last blank character of the processor brand string is not cleaned up.

# HWP
```
|- Hardware-Controlled Performance States                        HWP       <OFF>
```
* This is the Speed-Shift Intel technology. I have not found an AMD equivalent, thus I may hide in next UI release.

# Minimum Power and Energy values stuck to zero
* I'm currently working on this issue

cyring, 2020-04-10:
Hello,

Can you pull `develop` and test the Minimum Energy and Power fix.
To un-correlate the Joule and Watt values, please change the `Interval` to `500` ms (for example)  before your screenshot.
[image]

obaudys, 2020-04-10:
I have set my settings to match those above in the following screenshot... ie:  
```Interval 500ms
Thermal scope SMP
Voltage scope Pkg
Power scope None
```
[image]
In addition, the processor brand name still contains the trailing space:
```Processor                                   [AMD Ryzen 5 3600 6-Core Processor ]```

cyring, 2020-04-11:
* Can you leave all scopes as default, *untouched* ( b/c your AMD has power counters per Core, whereas my Intel has them only per Package )

* About the brand string, can you post the output of:
```sh
echo "[$(cat /proc/cpuinfo | grep -i 'model name' | tail -n 1 | cut -f 2 -d ':')]"
```

obaudys, 2020-04-11:
Here's with the scopes untouched:
[image]
I'd say the package power is correct;  in Windows Ryzen Master shows 80W when running with all cores at 100% (I'm running folding@home right now)

Here's the output you requested:
```
$ echo "[$(cat /proc/cpuinfo | grep -i 'model name' | tail -n 1 | cut -f 2 -d ':')]"
[ AMD Ryzen 5 3600 6-Core Processor]
```

cyring, 2020-04-11:
Thanks for your returns but I don't understand why the ACCU, Power and Energy are now all showing zero. The fix I made was only the Minimum columns.
So do you confirm that from scratch 
* PC rebooted; 
* download `develop`
* then build; 
* loading _CoreFreq_ 
* and go straight to the Power view; 

the values are zero ?

obaudys, 2020-04-11:
Yes I can confirm all of the following:
- PC rebooted
- Clean clone of `develop`
- built
- module loaded, daemon started
- straight into power view

All values are still zero except package power

cyring, 2020-04-11:
Switching to the `master` repo, do you get values ?
(be sure to unload current driver before switching)

obaudys, 2020-04-11:
Yes, switching to master I do get values:
[image]

cyring, 2020-04-11:
Can you download and test last `develop`
It will be version `1.76.1`

obaudys, 2020-04-11:
1.76.1 still shows all 0 values

cyring, 2020-04-11:
> 1.76.1 still shows all 0 values

Looking at the above `develop` screenshots, I can see that some values are garbage compared to `master`

For example: `L3=64` vs `L3=16384` ; also SMT is disabled in `develop`

Usually when values are inconsistent, it's a sign of a major change in the API carried by the memory shared between the driver and the daemon. As a footprint, the version is computed and stored in the shared memory and checked by Processes to be sure they are working on the same API

Don't you have any Oops in the kernel log ? see `dmesg`

cyring, 2020-04-11:
Do you have installed _CoreFreq_ from the AUR or enter `make install` ?

obaudys, 2020-04-11:
There are no oopses in the kernel log.   
```
Apr 11 20:08:14 orion kernel: [15033.524536] CoreFreq: Unload
Apr 11 20:09:52 orion kernel: [15131.722332] CoreFreq(0:-1): Processor [ 8F_71] Architecture [Zen2/Matisse] CPU [12/12]
```
I'm running CoreFreq from my local git repo, after pulling in changes and switching branches is required:
```
rmmod corefreqk; make clean all; sudo insmod ./corefreqk.ko; sudo ./corefreqd & sleep 1; ./corefreq-cli
 ```

cyring, 2020-04-11:
> There are no oopses in the kernel log.
> 
> ```
> Apr 11 20:08:14 orion kernel: [15033.524536] CoreFreq: Unload
> Apr 11 20:09:52 orion kernel: [15131.722332] CoreFreq(0:-1): Processor [ 8F_71] Architecture [Zen2/Matisse] CPU [12/12]
> ```
>

The log is strange:
`-1` confirmes no thread CPU detected thus SMT is disabled ; but we see the Zen2/Matisse CPUID family ID `8F_71`

> I'm running CoreFreq from my local git repo, after pulling in changes and switching branches is required:
> 
> ```
> rmmod corefreqk; make clean all; sudo insmod ./corefreqk.ko; sudo ./corefreqd & sleep 1; ./corefreq-cli
> ```

Is it virtualized ?

obaudys, 2020-04-11:
No, not virtualised, this is a bare metal PC.   Running Ubuntu 19.10 with a 5.4.0 kernel.

cyring, 2020-04-11:
> No, not virtualised, this is a bare metal PC. Running Ubuntu 19.10 with a 5.4.0 kernel.

OK, thanks for your answer. 
Doing a code review for regression but still don't know why SMT has for instance disappear.

cyring, 2020-04-11:
I think it comes from this commit f7c5dbecf14588fa3e68a8ec1db1ce411508b2f0

Can you remove it or restore the zeroings with `{{0}}` then test again.

obaudys, 2020-04-11:
I have pulled the latest develop, and built, but the issues persist:  No SMT,  wrong cache sizes, and per-core power shows  0.

cyring, 2020-04-12:
Can you please post the dump with `corefreq-cli -u`

CPU #0 will be enough

obaudys, 2020-04-12:
```
CPU #0   function         EAX          EBX          ECX          EDX            
|- 00000000:00000000    00000010     68747541     444d4163     69746e65         
   |- Largest Standard Function=00000010                                        
|- 80000000:00000000    80000020     68747541     444d4163     69746e65         
   |- Largest Extended Function=80000020                                        
|- 00000001:00000000    00870f10     000c0800     7ed8320b     178bfbff         
|- 00000002:00000000    00000000     00000000     00000000     00000000         
|- 00000003:00000000    00000000     00000000     00000000     00000000         
|- 00000004:00000000    00000000     00000000     00000000     00000000         
|- 00000004:00000001    00000000     00000000     00000000     00000000         
|- 00000004:00000002    00000000     00000000     00000000     00000000         
|- 00000004:00000003    00000000     00000000     00000000     00000000         
|- 00000005:00000000    00000040     00000040     00000003     00000011         
|- 00000006:00000000    00000004     00000000     00000001     00000000         
|- 00000007:00000000    00000000     219c91a9     00400004     00000000         
|- 00000007:00000001    00000000     00000000     00000000     00000000         
|- 00000009:00000000    00000000     00000000     00000000     00000000         
|- 0000000a:00000000    00000000     00000000     00000000     00000000         
|- 0000000b:00000000    00000001     00000002     00000100     00000000         
|- 0000000d:00000000    00000207     00000340     00000380     00000000         
|- 0000000d:00000001    0000000f     00000340     00000000     00000000         
|- 0000000d:00000002    00000100     00000240     00000000     00000000         
|- 0000000d:00000003    00000000     00000000     00000000     00000000         
|- 0000000d:00000004    00000000     00000000     00000000     00000000         
|- 0000000d:0000003e    00000000     00000000     00000000     00000000         
|- 0000000f:00000000    00000000     000000ff     00000000     00000002         
|- 0000000f:00000001    00000000     00000040     000000ff     00000007         
|- 00000010:00000000    00000000     00000002     00000000     00000000         
|- 00000010:00000001    0000000f     00000000     00000004     0000000f         
|- 00000010:00000002    00000000     00000000     00000000     00000000         
|- 00000010:00000003    00000000     00000000     00000000     00000000         
|- 00000012:00000000    00000000     00000000     00000000     00000000         
|- 00000012:00000001    00000000     00000000     00000000     00000000         
|- 00000012:00000002    00000000     00000000     00000000     00000000         
|- 00000014:00000000    00000000     00000000     00000000     00000000         
|- 00000014:00000001    00000000     00000000     00000000     00000000         
|- 00000015:00000000    00000000     00000000     00000000     00000000         
|- 00000016:00000000    00000000     00000000     00000000     00000000         
|- 00000017:00000000    00000000     00000000     00000000     00000000         
|- 00000017:00000001    00000000     00000000     00000000     00000000         
|- 00000017:00000002    00000000     00000000     00000000     00000000         
|- 00000017:00000003    00000000     00000000     00000000     00000000         
|- 00000018:00000000    00000000     00000000     00000000     00000000         
|- 00000018:00000001    00000000     00000000     00000000     00000000         
|- 0000001a:00000000    00000000     00000000     00000000     00000000         
|- 0000001b:00000000    00000000     00000000     00000000     00000000         
|- 0000001f:00000000    00000000     00000000     00000000     00000000         
|- 80000001:00000000    00870f10     20000000     75c237ff     2fd3fbff         
|- 80000002:00000000    20444d41     657a7952     2035206e     30303633         
|- 80000003:00000000    432d3620     2065726f     636f7250     6f737365         
|- 80000004:00000000    20202072     20202020     20202020     00202020         
|- 80000005:00000000    ff40ff40     ff40ff40     20080140     20080140         
|- 80000006:00000000    48006400     68006400     02006140     01009140         
|- 80000007:00000000    00000000     0000001b     00000000     00006799         
|- 80000008:00000000    00003030     010cb657     0000700b     00010000         
|- 8000000a:00000000    00000001     00008000     00000000     0013bcff         
|- 80000019:00000000    f040f040     00000000     00000000     00000000         
|- 8000001a:00000000    00000006     00000000     00000000     00000000         
|- 8000001b:00000000    000003ff     00000000     00000000     00000000         
|- 8000001c:00000000    00000000     00000000     00000000     00000000         
|- 8000001d:00000000    00004121     01c0003f     0000003f     00000000         
|- 8000001d:00000001    00004122     01c0003f     0000003f     00000000         
|- 8000001d:00000002    00004143     01c0003f     000003ff     00000002         
|- 8000001d:00000003    00014163     03c0003f     00003fff     00000001         
|- 8000001e:00000000    00000000     00000100     00000000     00000000         
|- 40000000:00000000    00000000     00000000     00000000     00000000         
|- 40000001:00000000    00000000     00000000     00000000     00000000         
|- 40000002:00000000    00000000     00000000     00000000     00000000         
|- 40000003:00000000    00000000     00000000     00000000     00000000         
|- 40000004:00000000    00000000     00000000     00000000     00000000         
|- 40000005:00000000    00000000     00000000     00000000     00000000         
|- 40000006:00000000    00000000     00000000     00000000     00000000         

```

cyring, 2020-04-12:
Yes those come from these commits 5322e6809102143db22e460b29c42a7645dd5097
ec3bb72be314034fa321f2ca769dff5dd1efb653

cyring, 2020-04-12:
Hello,
Can you please try the latest `develop` ?

obaudys, 2020-04-12:
Things look good now!
[image]

obaudys, 2020-04-12:
I spoke too soon, I see a glitch in the power summary:
[image]

cyring, 2020-04-12:
> When i resize the window:

This a bug in consequence of the previous glitch

cyring, 2020-04-12:
> I spoke too soon, I see a glitch in the power summary:

Looks like a bug due to negative values converted or printed as unsigned

cyring, 2020-04-12:
Do you get the same bug with `corefreq-cli -W` ?

obaudys, 2020-04-12:
Yes, I do:
```
CPU Freq(MHz)    Accumulator      Min  Energy(J) Max    Min  Power(W)  Max
000 3924.37  000000000000267281    0.02   4.08 281474976645124.06    0.02   4.08 281474976645124.06
```

cyring, 2020-04-12:
At this line: 
https://github.com/cyring/CoreFreq/blob/2b5775c82888fc9aecf95cab2b3ddbd560de1964/corefreqk.c#L9032
Can you replace with the following statement
```c
		/* Read the Physical Core RAPL counter. */
	    if (Core->T.ThreadID == 0)
	    {
		RDCOUNTER(Core->Counter[1].Power.ACCU,MSR_AMD_PP0_ENERGY_STATUS);
		Core->Counter[1].Power.ACCU &= 0xffffffff;

		if (Core->Counter[1].Power.ACCU > Core->Counter[0].Power.ACCU)
		{
			Core->Delta.Power.ACCU  = Core->Counter[1].Power.ACCU
						- Core->Counter[0].Power.ACCU;
		} else {
			Core->Delta.Power.ACCU  = 0;
		}

		Core->Counter[0].Power.ACCU = Core->Counter[1].Power.ACCU;
	    }
```

then, rebuild, reload and test, please.

obaudys, 2020-04-12:
Everything looks good now:
[image]

cyring, 2020-04-12:
So it's a counter overflow, and its fix will be part of next commit.
However I'm noticing in screenshot that Minimum is stuck again to zero: do you confirm ?

cyring, 2020-04-12:
> So it's a counter overflow, and its fix will be part of next commit.
> However I'm noticing in screenshot that Minimum is stuck again to zero: do you confirm ?

Edit: this answer is because I'm setting the counter delta to zero when the overflow happens.
So the fix should be to compute the absolute difference of counters...

cyring, 2020-04-12:
For your test, the counter overflow fix is available in `develop`

obaudys, 2020-04-12:
The counter is no longer stuck at 0. I've noticed that when I start the `corefreqd` process on an unloaded system the minimum power is always 0.01 or 0.00:
[image]
Starting `corefreqd` on a loaded system showed per core minimum at 4.xxW.  After the load all mimimums dropped to 0.01.  If this is correct then I guess the cores really do shut off completely!

cyring, 2020-04-13:
`0.01 or 0.00`

Nothing in your hardware. Looks like more a code side effect using floating numbers.

obaudys, 2020-04-13:
I thought I might take the time to record which values are zero for my Ryzen in the various views.  This is for the develop branch:

- In the Dashboard:
  - UNCORE and Bus is 0
- In the Frequency view:
  - C3, C6, and C7 states are 0.00 (probably not enabled in my hardware or Linux)
- In the Idle C-states view:
  - like above C3, C6, and C7 are 0.00
- In the Package cycles view:
  - PC02...PC10 are 0 for cycles and 0.00% for state
  - TSC is the only non-zero value
- In the System Interrupts view:
  - all SMI values 0 for all CPUs,
  - empty columns for `NMI[ LOCAL   UNKNOWN  PCI_SERR#  IO_CHECK]` fields
- In the Sensors View:
  - RAM and Uncore are both 0.0000
- In the Slice Counters view:
  - Cycles, Instructions, TSC, and PMC0 columns are all zeros for each thread.

cyring, 2020-04-13:
I really need to provide a manual, or hide unimplemented features, although I have the hope AMD will provide the missing BKDG datasheet 

> I thought I might take the time to record which values are zero for my Ryzen in the various views. This is for the develop branch:
> 
> * In the Dashboard:
>   
>   * UNCORE and Bus is 0

Unimplemented. Don't know, don't have specifications to measure its speed. I understand the Intel Uncore equivalent is one of the AMD Infinity Fabric or last Cache level. Do the Windows tools provide you something ? 

> * In the Frequency view:
>
>   * C3, C6, and C7 states are 0.00 (probably not enabled in my hardware or Linux)
> * In the Idle C-states view:
>
>   * like above C3, C6, and C7 are 0.00
> * In the Package [...]

I have no registers 
 in mind for these 

> * In the System Interrupts view:
>   
>   * all SMI values 0 for all CPUs,
>   * empty columns for `NMI[ LOCAL   UNKNOWN  PCI_SERR#  IO_CHECK]` fields

* Go to Settings and NMI registration, then inject a NMI (ACPI button found in Servers, or using a host command when Processor is virtualized) 
* Check BIOS for SMI state ? 

> * In the Sensors View:
>   
>   * RAM and Uncore are both 0.0000

No Intel equivalent of those found in AMD. Should it be hidden ? 

> * In the Slice Counters view:
>   
>   * Cycles, Instructions, TSC, and PMC0 columns are all zeros for each thread.

Go to menu Tools, selecting one of the stress functions, will increase these numbers

cyring, 2020-04-13:
[https://www.amd.com/system/files/TechDocs/56255_OSRR.pdf](https://www.amd.com/system/files/TechDocs/56255_OSRR.pdf) in chapter _2.1.11 Performance Monitor Counters_
* Those can be programmed, I need the hardware to experiment them

cyring, 2020-04-13:
I do believe Minimum issue comes from my thresholds, can you try the following change. 

https://github.com/cyring/CoreFreq/blob/a97c49fda19cce0cf0d840ab944455fa6c5299fa/corefreqd.c#L716

```c
void Core_ComputePowerLimits(CPU_STRUCT *Cpu, double Energy, double Power)
{	/* Per Core, computes the Min CPU Energy consumed.		*/
    if (((Cpu->Sensors.Energy.Limit[SENSOR_LOWEST] <= 0.5) && (Energy > 0.0))
    || ((Energy > 0.0) && (Energy < Cpu->Sensors.Energy.Limit[SENSOR_LOWEST])))
    {
	Cpu->Sensors.Energy.Limit[SENSOR_LOWEST] = Energy;
    }
	/* Per Core, computes the Max CPU Energy consumed.		*/
    if (Energy > Cpu->Sensors.Energy.Limit[SENSOR_HIGHEST]) {
	Cpu->Sensors.Energy.Limit[SENSOR_HIGHEST] = Energy;
    }
	/* Per Core, computes the Min CPU Power consumed.		*/
    if (((Cpu->Sensors.Power.Limit[SENSOR_LOWEST] <= 0.5) && (Power > 0.0))
    || ((Power > 0.0) && (Power < Cpu->Sensors.Power.Limit[SENSOR_LOWEST])))
    {
	Cpu->Sensors.Power.Limit[SENSOR_LOWEST] = Power;
    }
	/* Per Core, computes the Max CPU Power consumed.		*/
    if (Power > Cpu->Sensors.Power.Limit[SENSOR_HIGHEST]) {
	Cpu->Sensors.Power.Limit[SENSOR_HIGHEST] = Power;
    }
}
```

obaudys, 2020-04-13:
After running the change:
[image]
After a while, the values decay to 0.01 anyway.

[92 more replies]

closed 2020-08-03 by cyring