cyring/CoreFreq #173: can C1E be disabled on AMD cpu ?

gfanini, 2020-04-13:
can C1E be disabled on AMD cpu ? there is no such option in the BIOS

cyring, 2020-04-13:
> can C1E be disabled on AMD cpu ? there is no such option in the BIOS

The only bit I found comes from old architecture families: 
https://github.com/cyring/CoreFreq/blob/2f4cac48406c1e3e0bc3c7970c405c9bb94026f0/amdmsr.h#L426

In this function I'm reading it's state:
https://github.com/cyring/CoreFreq/blob/f6ba67a06676e3f83dfc4eeefb8b09683c39fc87/corefreqk.c#L4507

Presuming you are talking about recent Zen architecture, I will suggest you experiment it by inserting this function call:
```
	Query_AMD_Family_0Fh_C1E(Core);
```

at this line of the Zen initialization part `PerCore_AMD_Family_17h_Query`  :  
https://github.com/cyring/CoreFreq/blob/f6ba67a06676e3f83dfc4eeefb8b09683c39fc87/corefreqk.c#L5817

Then rebuild and test the driver. 
If no crash happens and the C1E is correctly queried then we will pursue with second part: writing its new state.

gfanini, 2020-04-13:
Are you aware of any programs (with source) that can detect how the o.s. (windows/linux) is setting the cpu frequency via ACPI, this description for linux http://acpi.sourceforge.net/documentation/processor.html

is there any way to detect setting of P C states by ACPI on windows, on linux I could look into acpi-cpufreq but I don't want to get into kernel module for my needs (to avoid o.s. down clocking), from user space side only.

cyring, 2020-04-13:
Multiple questions are somewhat confusing : is the original issue was about C1E ?
Do you will to show the CoreFreq output?

gfanini, 2020-04-13:
sorry it is in general how to detect windows cpu throttling via ACPI I now realise
this is corefreq ss for A10-9620P  family 15h revision 65h :

[image]

cyring, 2020-04-13:
So for this family, reading the C1E bit is implemented but not writing it to zero, aka disabled, if that works.

Thus, you will need to go to Linux, download the `develop` branch of _CoreFreq_; build and test it; with the iterative modifications I will send or ask to you.
Is this OK for you ?

gfanini, 2020-04-13:
I glanced through the code it's really well done, it compiles and runs ok in ubuntu linux, how to debug a kernel module in linux ? I need to set up a virtual machine and debug it from there that I cannot step into the kernel not ?

cyring, 2020-04-13:
> I glanced through the code it's really well done, it compiles and runs ok in ubuntu linux, how to debug a kernel module in linux ? 

* Thanks a lot
My way is to simply  trace/print things
* I also sandbox my algorithms in user-space and when ready put them back in driver
* But most of the time, things are made by _head_

> I need to set up a virtual machine and debug it from there that I cannot step into the kernel not ?

* We won't very far with a virtualized processor. Most registers won't be implemented. 

Please post me back a Markdown formatted output of `corefreq-cli -s` from the current development branch. 

I will then study the missing parts, such as temperature not shown in screenshot, in addition to the C1E order, in next commit for your tests

gfanini, 2020-04-14:
I think corefreq is way too advanced for my needs, I don't want to waste your time, in case what I "think" I am trying to do is, possibly like linux acpi-cpufreq :

https://elixir.bootlin.com/linux/latest/source/drivers/cpufreq/acpi-cpufreq.c

get_cur_freq_on_cpu(unsigned int cpu)

if (freq != cached_freq) {
		/*
		 * The dreaded BIOS frequency change behind our back.
		 * Force set the frequency on next target call.
		 */
		data->resume = 1;
	}

cyring, 2020-04-14:
Hope [mhz](https://github.com/cyring/Tips/blob/master/C/mhz.c) will be helpful

_all or per cpu_
```sh
$ sudo ./mhz 

"Intel(R) Xeon(R) CPU           W3690  @ 3.47GHz"       [06_2C stepping 2]

[  0] 3458      [  1] 3458      [  2] 1596      [  3] 1596
[  4] 1596      [  5] 1596      [  6] 1596      [  7] 1596
[  8] 1596      [  9] 1596      [ 10] 3458      [ 11] 3458
[ 12]    0      [ 13]    0      [ 14]    0      [ 15]    0

sudo ./mhz 4
3458
```

gfanini, 2020-04-14:
I try to write to AMD support a way of going about it might be to allow a MINIMUM (nominal not overclocked) FREQUENCY setting, i.e. never mind all the careful throttling mechanisms, if "plugged in" never say allow below 2.5 GHz 

"Maximum processor frequency" 
powercfg -attributes 54533251-82be-4824-96c1-47b60b740d00 75b0ae3f-bce0-45a7-8c89-c9611c25e100 -ATTRIB_HIDE

i.e. introduce a "minimum frequency" which is synthetic say, on linux for sure there are both min and max settable so I don't think it's harmful not:

/sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_max_freq

like this one :

/sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_min_freq

cyring, 2020-04-14:
I'm not sure to understand your needs but if you want to tweak your processor registers from Windows, you will have to program a driver ...
Here's one I did in the past
[WinMSR](https://github.com/cyring/WinMSR)

I can still provide some code to disable C1E, are you still OK to test it ?

cyring, 2020-04-15:
* To *fight* against the minimum frequency, you will have to master control the CPU idle loop of the Kernel.
I don't know if it is feasible under Windows, but that's a feature _CoreFreq_ provides for Intel families. Let me know if you want to contribute with the AMD case. 

* The key idea is to register as a CPU-Idle  driver, passing the residency of C-States through a table

* **Black-list** any previous drivers before (or build a custom kernel) . B/c only one can succeed in the registration. 

* When its driver is properly registered,  _CoreFreq_ , and according to the user selection, will limit the minimum C-state which is passed in register to the instruction `mwait`

* The Kernel window of corefreq-cli will show what is the current idle driver, and a pop-up will allow you to select a C-state level. You will observe the effects immediately in Frequency view where the `%` of values are moving from current to new state. 
You can even select `C0` which is in fact the Linux poll loop. 

* In the Q&A of my README.md are explained the installation instructions.

gfanini, 2020-04-16:
It would be interesting to read / be aware of what windows is doing to down clock, initially, not actual "fighting" but say "reading" 

User space applications do not need to worry about explicitly controlling the P state, unless they specifically wanted to. Within the Linux kernel, the CPU frequency scaling driver, CONFIG_CPU_FREQ, has traditionally been responsible for controlling the system’s P state transitions. This driver supports multiple governors for controlling the desired management policy. At the time of this writing, there are five standard governors performance, etc.

cyring, 2020-04-16:
Below the _CoreFreq_ parameters:
```
parm:           Register_CPU_Idle:Register the Kernel cpuidle driver (short)
parm:           Register_Governor:Register the Kernel governor (short)
parm:           Register_CPU_Freq:Register the Kernel cpufreq driver (short)
```
Set `=1` to activate to these options when inserting `corefreqk.ko`

Using Xeon Westmere , Sandy/Ivy bridge and Skylake, those options are part of my day-to-day operational mode, and they give me a fine control over the Processor + Kernel

That's a remaining implementation, also toggling C1E, to do for AMD. Are you OK to contribute ?

cyring, 2020-04-19:
As mentioned above, in the `develop` branch, you can trigger the C1E state in _CoreFreq_ Experimental mode from the UI `Settings` menu.
If not, just feel free to close your issue
Regards

gfanini, 2020-05-12:
I have no idea how to go about C1E, I don't think that is what I am after, what I'm thinking is to set the different north bridge multipliers, on A10 family 15h, to a single one, might it be possibile with this turion power control which supports 15h :

https://github.com/turionpowercontrol/tpc/issues/42

cyring, 2020-05-12:
So you will have to read-modify-write the VID onto the Northbridge PCI register, such as:
https://github.com/turionpowercontrol/tpc/blob/b43a14f3c4d2c3d14d1aab9ebb3233056284498c/Griffin.cpp#L783

Grab the BKDG specifications of your AMD family to find the register offset address. See my Wiki for references. 
`lspci` will confirm you the Northbridge PCI to  address
With `setpci` alter the VID

gfanini, 2020-05-12:
hmm that is void Griffin::setNBVid(DWORD nbvid) 
it seems to be AMD cpu "griffin" mobile of about 2008, if you notice you can't set NB fid multiplier in that one, what I wonder does this code for "interlagos" (?) family 15h work, and does it require any precaution before changing NB multiplier, from the KBDG it might be you need to allow the cpu to "self refresh" ram prior to attempting it ? or actual values in which order hmm 

https://github.com/turionpowercontrol/tpc/blob/master/Interlagos.cpp
void Interlagos::setNBFid(DWORD fid)

it might be of interest to you that also turion power control they seem to have trouble with C1E 
// There are a number of requirements for C1E on 15h
// These can be found on pg 85 of the BKDG for 15h
printf("Not configured for 15h\n");

cyring, 2020-05-12:
[Source](https://developer.amd.com/resources/developer-guides-manuals/): BIOS and Kernel Developer’s Guide (BKDG) for AMD Family 15h Models 60h-6Fh Processors
> 2.5.3 NB Power Management
2.5.3.1 NB P-states
>
>The processor supports several P-states, labled NBP0 (highest frequency) through NBPn (where n is the number of the lowest frequency P-state). Four of the NB P-states are specified in D18F5x16[C:0]; the processor may be configured to support additional P-states that are not visible in registers as well.
>
>Out of cold reset, the NB P-state is specified by D18F5x174[StartupNbPstate] and D0F0xBC_xD823_0F00[COF_VID_PROG]. The current NB P-state is specified by D18F5x174[CurNbFid, CurNbDid, CurNbVid].
>
>Although several NB P-states are defined, only two NB P-states are used at any given time, specified by D18F5x170[NbPstateHi, NbPstateLo].

* Northbridge VID `NbVid` of `COFVID Control` register in `31:24` bits

[image]

* Northbridge `NbPstateDis` access mode at bit position `0`
[image]

cyring, 2020-05-26:
Hello,

 Could you try the `develop` branch with your AMD A10 and print the Topology output:
`corefreq-cli -s -m`

Thank you

cyring, 2020-11-20:
When Zen C1E specs will be available, I will work on it again
Regards
CyrIng

closed 2020-11-20 by cyring