cyring/CoreFreq #174: `make install` fails

hakonhall, 2020-04-19:
After a successful `make`, the install failed:

```
CoreFreq$ sudo make install
make -C /lib/modules/5.3.0-46-generic/build M=/home/hakon/tmp/CoreFreq modules_install
make[1]: Entering directory '/usr/src/linux-headers-5.3.0-46-generic'
  INSTALL /home/hakon/tmp/CoreFreq/corefreqk.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:72
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:79
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  5.3.0-46-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-5.3.0-46-generic'
install -Dm 0755 corefreq-cli corefreqd -t /usr/bin
install -Dm 0644 corefreqd.service \
	/usr/lib/systemd/system/corefreqd.service
```

This is on Ubuntu 18.04 with `dkms` installed. Have I missed anything?

hakonhall, 2020-04-19:
Never mind, the SSL error is not fatal and I have been able to get it working!

Suggestions:

 * Add `sudo make install` to the Install section in README.md.
 * Tell to ignore the above SSL error.
 * Tell to run `depmod` if user get `modules_install: missing 'System.map' file. Skipping depmod.`
 * Fix service path to daemon: By default kcorefreqd gets installed in /usr/bin which can be overridden by setting PREFIX when installing. However corefreqd.service has hard-coded /bin/corefreqd.  At the very least, change .service file to /usr/bin/corefreqd to match default and update the install section to fix .service path.
 * Last step in install section should be to run `sudo systemctl daemon-reload`.
 * The next step would be to continue after "When installed"

cyring, 2020-04-20:
Thanks, I will take into account your remarks into the `develop` branch for your review.
Meanwhile we have to improve:
* sign-in the driver
* privileged installation into `/usr/bin` Not sure that `sudo make` is a safe thing

hakonhall, 2020-04-20:
Agree it should be non-sudo `make` but `sudo make install`.

cyring, 2020-04-21:
Hello,

For your review, additional documentation has been comited into the [README.md](https://github.com/cyring/CoreFreq/blob/develop/README.md)

cyring, 2020-11-20:
Module sign-in has to be handled by user.
Regards
CyrIng

closed 2020-11-20 by cyring