cyring/CoreFreq #175: Compiling issue

eperezv, 2020-04-20:
Hello,
I have recently upgraded my system to ubuntu 20.04 and I am not able to compile the code. My computer has an AMD Threadripper 3960X. I am using the last github version. Any clue about the error?
Thanks.

```
eduardo@eduardo-PC:~/Downloads/CoreFreq$ make
cc  -Wall -pthread -c corefreqd.c \
        -D FEAT_DBG=0 -D UBENCH=0 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 \
        -o corefreqd.o
cc  -Wall -c corefreqm.c \
        -D FEAT_DBG=0 -D UBENCH=0 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 \
        -o corefreqm.o
cc  -Wall corefreqd.c corefreqm.c \
        -D FEAT_DBG=0 -D UBENCH=0 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 \
        -o corefreqd -lpthread -lm -lrt
cc  -Wall -c corefreq-cli.c \
        -D FEAT_DBG=0 -D UBENCH=0 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 \
        -o corefreq-cli.o
cc  -Wall -c corefreq-ui.c \
        -D FEAT_DBG=0 -D UBENCH=0 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 \
        -o corefreq-ui.o
cc  -Wall -c corefreq-cli-rsc.c \
        -D FEAT_DBG=0 -D UBENCH=0 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 \
        -o corefreq-cli-rsc.o
cc  -Wall -c corefreq-cli-json.c \
        -D FEAT_DBG=0 -D UBENCH=0 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 \
        -o corefreq-cli-json.o
cc  -Wall -c corefreq-cli-extra.c \
        -D FEAT_DBG=0 -D UBENCH=0 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 \
        -o corefreq-cli-extra.o
cc  -Wall \
        corefreq-cli.c corefreq-ui.c corefreq-cli-rsc.c \
        corefreq-cli-json.c corefreq-cli-extra.c \
        -D FEAT_DBG=0 -D UBENCH=0 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 \
        -o corefreq-cli -lm -lrt
make -j1 -C /lib/modules/5.4.0-24-generic/build M=/home/eduardo/Downloads/CoreFreq modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-24-generic'
  CC [M]  /home/eduardo/Downloads/CoreFreq/corefreqk.o
/home/eduardo/Downloads/CoreFreq/corefreqk.c:238:26: error: initialization of ‘int (*)(struct cpufreq_policy_data *)’ from incompatible pointer type ‘int (*)(struct cpufreq_policy *)’ [-Werror=incompatible-pointer-types]
  238 |  /*MANDATORY*/ .verify = CoreFreqK_Policy_Verify,
      |                          ^~~~~~~~~~~~~~~~~~~~~~~
/home/eduardo/Downloads/CoreFreq/corefreqk.c:238:26: note: (near initialization for ‘CoreFreqK.FreqDriver.verify’)
/home/eduardo/Downloads/CoreFreq/corefreqk.c: In function ‘CoreFreqK_Policy_Verify’:
/home/eduardo/Downloads/CoreFreq/corefreqk.c:9376:36: error: passing argument 1 of ‘cpufreq_verify_within_cpu_limits’ from incompatible pointer type [-Werror=incompatible-pointer-types]
 9376 |   cpufreq_verify_within_cpu_limits(policy);
      |                                    ^~~~~~
      |                                    |
      |                                    struct cpufreq_policy *
In file included from /home/eduardo/Downloads/CoreFreq/corefreqk.c:22:
./include/linux/cpufreq.h:448:62: note: expected ‘struct cpufreq_policy_data *’ but argument is of type ‘struct cpufreq_policy *’
  448 | cpufreq_verify_within_cpu_limits(struct cpufreq_policy_data *policy)
      |                                  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:275: /home/eduardo/Downloads/CoreFreq/corefreqk.o] Error 1
make[1]: *** [Makefile:1719: /home/eduardo/Downloads/CoreFreq] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-24-generic'
make: *** [Makefile:38: all] Error 2
```

cyring, 2020-04-20:
Yes, it has been such a mess in Kernel when _one guy_ decided to change a function prototype.
I have tracked major kernel versions, before and after the change, in these code lines of the driver:
https://github.com/cyring/CoreFreq/blob/f6ba67a06676e3f83dfc4eeefb8b09683c39fc87/corefreqk.c#L9367
But I will need your help to adapt your kernel version, apparently `5.4.0`, to these build conditions.
Something like:
```
#if ((LINUX_VERSION_CODE >= KERNEL_VERSION(5, 4, 0))	\
  && (LINUX_VERSION_CODE <= KERNEL_VERSION(5, 4, 25)))	\
  || (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 5, 3))
```

eperezv, 2020-04-21:
Ok. I can help :)
Let me know how to do it. I am not sure if you are suggesting to replace the upper line (5, 4 ,19) by (5, 4, 0). I did it but nothing changed.

cyring, 2020-04-21:
> Ok. I can help :)

Thanks a lot

> Let me know how to do it. I am not sure if you are suggesting to replace the upper line (5, 4 ,19) by (5, 4, 0). I did it but nothing changed.
> 
> You are right. The kernel version is 5.4.0
> 
> ```
> Linux eduardo-PC 5.4.0-25-generic #29-Ubuntu SMP Fri Apr 17 15:06:57 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
> ```

It looks like Ubuntu is back-porting changes into its LTS
The structure expected by whatever version is your Kernel is `cpufreq_policy_data`
I don't find such structure into the [5.4.0](https://elixir.bootlin.com/linux/v5.4/source/include/linux/cpufreq.h) mainstream
But appears in Linux LTS [5.4.19](https://elixir.bootlin.com/linux/v5.4.19/source/include/linux/cpufreq.h#L157) then none in 5.5.0 and found again **since** [5.5.3](https://elixir.bootlin.com/linux/v5.5.3/source/include/linux/cpufreq.h#L157)

cyring, 2020-04-21:
I'm virtualizing Ubuntu 19.10 Desktop ISO (b/c I'm an Archer) and _CoreFreq_ builds OK
[image]

What is your Ubuntu ?

eperezv, 2020-04-21:
> It looks like Ubuntu is back-porting changes into its LTS
> The structure expected by whatever version is your Kernel is `cpufreq_policy_data`
> I don't find such structure into the [...]
What a mess. 

I originally installed ubuntu 18.04 but upgraded to the 20.04 beta when it was released. I have been updating it since there. Linux 5.4.0 was the default kernel.

cyring, 2020-04-21:
> I originally installed ubuntu 18.04 but upgraded to the 20.04 beta when it was released. I have been updating it since there. Linux 5.4.0 was the default kernel.

Please mention it in your issue kernel -;)
I'm so far trying to find/download a quick live CD of Ubuntu 20 to test and change code

cyring, 2020-04-21:
> > > It looks like Ubuntu is back-porting changes into its LTS
> > > The structure expected by whatever version is your Kernel is `cpufreq_policy_data`
> > > I don't find such structure into the [5.4.0](https://elixir.bootlin.com/linux/v5.4/source/include/linux/cpufreq.h) mainstream
> > > But appears in Linux LTS [5.4.19](https://elixir.bootlin.com/linux/v5.4.19/source/include/linux/cpufreq.h#L157) then none in 5.5.0 and found again **since** [5.5.3](https://elixir.bootlin.com/linux/v5.5.3/source/include/linux/cpufreq.h#L157)
> > > What a mess.
> > 
> > 
> > I originally installed ubuntu 18.04 but upgraded to the 20.04 beta when it was released. I have been updating it since there. Linux 5.4.0 was the default kernel.
> 
> Please mention it in your issue **title** -;)
> I'm so far trying to find/download a quick live CD of Ubuntu 20 to test and change code

cyring, 2020-04-21:
OK, Ubuntu 20 virtualized
Quick workarround in these 2 files:
https://github.com/cyring/CoreFreq/blob/f6ba67a06676e3f83dfc4eeefb8b09683c39fc87/corefreqk.h#L3485
https://github.com/cyring/CoreFreq/blob/f6ba67a06676e3f83dfc4eeefb8b09683c39fc87/corefreqk.c#L9367
replace conditions in both files with this single line
```c

 #if LINUX_VERSION_CODE >= KERNEL_VERSION(5, 4, 0)
```
* such as
[image]
[image]

* then rebuild with `make clean all`
[image]

cyring, 2020-04-21:
And what a surprise: Ubuntu `5.4.0.-21` is indeed `5.4.27`
[image]
[image]

eperezv, 2020-04-21:
It works! thanks a lot

cyring, 2020-04-21:
> It works! thanks a lot

Please, show me _CoreFreq_ screenshots of your dream 3960X Threadripper (This is the one I would like to have ...)

Especially please post the Topology `corefreq-cli -s -m`  

What is your Cooling solution ?

eperezv, 2020-04-21:
But now, I am not sure if it is working properly. CPU is full load right now, and the frequences are weird. Don't you think so?

[image]
[image]

eperezv, 2020-04-21:
> > It works! thanks a lot
> 
> Please, show me _CoreFreq_ screenshots of your dream 3960X Threadripper (This is the one I would like to have ...)
> 
> Especially please post the Topology `corefreq-cli -s -m`
> 
> What is your Cooling solution ?

I am still testing it ;) It looks promising in some ffmpeg and DNA-assembly tools, but it is still soon. It was assembled with the Noctua NH U14S TR4-SP3. It was indicated as compatible (and I was afraid of liquid cooling), so I tried. But I am not very happy. With ffmpeg full load, temperatures reach 95 ºC quite easily, so I am afraid of burning it. I am thinking about improving the cooling, but no decision so far. 

```
eduardo@eduardo-PC:~/Downloads/CoreFreq$ ./corefreq-cli -s -m
Processor                      [AMD Ryzen Threadripper 3960X 24-Core Processor ]
|- Architecture                                               [Zen2/Castle Peak]
|- Vendor ID                                                      [AuthenticAMD]
|- Microcode                                                        [ 137367589]
|- Signature                                                           [  8F_31]
|- Stepping                                                            [      0]
|- Online CPU                                                          [ 48/ 48]
|- Base Clock                                                          [ 99.807]
|- Frequency            (MHz)                      Ratio                        
                 Min    399.23                    <   4 >                       
                 Max   3792.68                    <  38 >                       
|- Factory                                                             [100.000]
                       3800                       [  38 ]                       
|- Performance                                                                  
   |- OSPM                                                                      
                 TGT   2195.76                    <  22 >                       
|- Turbo Boost                                                         [ UNLOCK]
                 XFR   4591.14                    [  46 ]                       
                 CPB   4491.33                    [  45 ]                       
                  1C   2794.61                    <  28 >                       
                  2C   2195.76                    <  22 >                       
|- Uncore                                                              [   LOCK]
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AVX-512      [N]  BMI1/BMI2 [Y/Y]      CLFLUSH [Y]        CMOV [Y] 
|- CMPXCHG8B    [Y]   CMPXCHG16B [Y]         F16C [Y]         FPU [Y] 
|- FXSR         [Y]    LAHF/SAHF [Y]    MMX/Ext [Y/Y]  MONITOR/X[Y/Y] 
|- MOVBE        [Y]          MPX [N]    PCLMULQDQ [Y]      POPCNT [Y] 
|- RDRAND       [Y]       RDSEED [Y]       RDTSCP [Y]         SEP [Y] 
|- SGX          [N]          SSE [Y]         SSE2 [Y]        SSE3 [Y] 
|- SSSE3        [Y]  SSE4.1/4A [Y/Y]       SSE4.2 [Y]     SYSCALL [Y] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- 100 MHz multiplier Control                            100MHzSteps   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- Core Multi-Processing                                  CMP Legacy   [Capable]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Capable]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Missing]
|- CPL Qualified Debug Store                                  DS-CPL   [Missing]
|- 64-Bit Debug Store                                         DTES64   [Missing]
|- Fast-String Operation                                Fast-Strings   [Missing]
|- Fused Multiply Add                                     FMA | FMA4   [Capable]
|- Hardware Lock Elision                                         HLE   [Missing]
|- Long Mode 64 bits                                       IA64 | LM   [Capable]
|- LightWeight Profiling                                         LWP   [Missing]
|- Machine-Check Architecture                                    MCA   [Capable]
|- Model Specific Registers                                      MSR   [Capable]
|- Memory Type Range Registers                                  MTRR   [Capable]
|- No-Execute Page Protection                                     NX   [Capable]
|- OS-Enabled Ext. State Management                          OSXSAVE   [Capable]
|- Physical Address Extension                                    PAE   [Capable]
|- Page Attribute Table                                          PAT   [Capable]
|- Pending Break Enable                                          PBE   [Missing]
[... 105 lines cut ...]
024:  0    45  1  22     1      32  8        32  8       512  8 i   65536  9w 
[... cut ...]
041:  0    60  1  30     0      32  8        32  8       512  8 i   65536  9w 
042:  0    49  0  24     1      32  8        32  8       512  8 i   65536  9w 
043:  0    51  0  25     1      32  8        32  8       512  8 i   65536  9w 
044:  0    53  0  26     1      32  8        32  8       512  8 i   65536  9w 
045:  0    57  1  28     1      32  8        32  8       512  8 i   65536  9w 
046:  0    59  1  29     1      32  8        32  8       512  8 i   65536  9w 
047:  0    61  1  30     1      32  8        32  8       512  8 i   65536  9w 
```

cyring, 2020-04-21:
Very weird, indeed. Target frequency `TGT` appears locked to ratio `22`
First time I see TRX40 

Can you instead use my integrated stress functions :
Press `[O]` for the Tools select:
1. `Atomic Burn` for all Cores
[image]

2. `Turbo Select CPU...`
[image]

**Press `[F10]` at any time to stop** the stress loop

Btw: _I will commit pretty soon commit the building fix in the `develop` branch_

eperezv, 2020-04-21:
Atomic burning
[image]
[image]

Turbo select CPU 
[image]
[image]

I wonder why CPU temperatures remain pretty well (below 70C) after these stress tests but are so crazy with ffmpeg...

cyring, 2020-04-21:
* Build fix released. You can now switch to the `develop` branch
1. Unload driver, remove the current _CoreFreq_ directory
2. Clone
```
git clone -b develop https://github.com/cyring/CoreFreq.git
```
3. Build and run

* The last dashboard screenshot is showing Turbo is reaching `4320` Mhz 
  was `Turbo Select CPU...` enough to reach this frequency or did you have to add some load produced by ffmpeg ?

eperezv, 2020-04-21:
Ok. Thanks for fixing the issue. 
Yes, Turbo Select CPU was enough to reach the frequency. Actually, I am monitoring it and it reaches up to 4378. Far from the 4500 that AMD sells.
Update: I have tried in another CPU and it is up to 4480

cyring, 2020-04-21:
Have you measured the time it takes to build a full Kernel ?

For example, my Xeon W3690 claims 30 minutes to build the Kernel core and modules, all `CONFIG_` options set, including also the post operations which lead to an initram fs

I'm seeing on the Web that it last less than 5 minutes with a Threadripper ! 

And there is a worst case: building an Android Kernel: 3 hours with 4C/8T Skylake

As an embedded engineer, those are tasks I'm doing oftenly

eperezv, 2020-04-21:
With Ubuntu 20.04 config, Kernel 5.4.34, with -j 24 was around 7.30, with -j 48 was around 6.15 

So yes, it is true. My threadripper does not have the best cooling and also, pbo is not activated (which could provide much better results).

cyring, 2020-04-21:
Bug: I'm just noticing that you have so much RAM that total is not displayed in the Dashboard card.

eperezv, 2020-04-22:
> I want a Threadripper, in a Small Form Factor, but cooling is also a matter of worrying...
> 
> In the UI, you can offline CPUs and thus maximize the best Core(s)
> 
> Target [TGT] , Turbo, Maximum frequency are still a work in progress but it's working better and better. So far, tested with a 3600X in issue #172
> Feel to try.

Thanks for the advice. I didn't know it was possible to offline some CPUs. I will try

cyring, 2020-04-22:
> Thanks for the advice. I didn't know it was possible to offline some CPUs. I will try

This a Kernel Offline, not a Hardware Offline. It means the Kernel will stop scheduling tasks to the selected CPU and put it in idle mode. However the Core behind the CPU is not gate off, but its usage should less
[image]

eperezv, 2020-04-23:
Ok. I've been using it. I see the CPU is deactivated in corefreq app but, does it affect to other (already running o new) tasks? I have seen in htop that it does not have an effect.

cyring, 2020-04-23:
> Ok. I've been using it. I see the CPU is deactivated in corefreq app but, does it affect to other (already running o new) tasks? I have seen in htop that it does not have an effect.

The other applications tasks have to be migrated by the Kernel scheduler to the remaining online CPUs ; even If CPU bound, and depending of the application CPU affinity bitmap :
* at least one CPU has to remain online to run the scheduler
* messages of 'affinity lost' kind will be logged in Kernel journal.
* offline processors will enter the idle loop
* top, htop should show the CPU number changing during migration: do you confirm ?

On its side, the CoreFreq framework is monitoring the CPU state changes and will end or begin its various services, CPU bound timers and threads, consequently.

eperezv, 2020-04-23:
Ok. thanks for letting me know.
One question that I have with this processor is about the multithreading. In the programs I am normally running, I manually select the number of threats I want. Since the processor is 24 (true) cores and 48 threads, I always have the doubt about selecting 24 or 48, thinking of whether it will perform faster or not. I guess it depends on the specific application but as a general rule, it is not clear to me.

cyring, 2020-04-23:
> Ok. thanks for letting me know.
> One question that I have with this processor is about the multithreading. In the programs I am normally running, I manually select the number of threats I want. Since [...]

My way would be to write small micro benchmark programs:
* are threads, bound respectively to physical core and to its sibling logical core, going faster because they are sharing the same caches
?
* which caches matter the most ?
* for which data size and alignment ?  
* cache partitionnement or not ? 
* how and which  instructions behave better in SMT ?

AMD is publishing a document about getting the most from the Zen architecture. See links in my wiki

eperezv, 2020-04-24:
I have tried your latest git version (fresh pull) and I cannot make it run. I have removed the module with rmmod and loaded the new one. The problem I am facing is this:
```
eduardo@eduardo-PC:~/Downloads/CoreFreq$ ./corefreq-cli 
Daemon connection error code 1
Version 1.76.1: 'Operation not permitted' @ line 11989
```
The daemon seems to be running fine

cyring, 2020-04-24:
> I have tried your latest git version (fresh pull) and I cannot make it run. I have removed the module with rmmod and loaded the new one. The problem I am facing is this:
>
> ``` [...]

I'm printing this error when API versions are wrongly mixed

Please make sure to build & run from only one project directory. 
* Unload, remove any previous projects. 
* Check if old driver is not stuck. 
* `rmmod -f` necessary when cpufreq or cpuidle registered

ShaneTursam, 2020-05-02:
/home/Shane/Documents/CoreFreq/corefreqk.c:10543:9: error: implicit declaration of function ‘cpu_down’ [-Werror=implicit-function-declaration]
10543 |    rc = cpu_down(cpu);
      |         ^~~~~~~~
/home/Shane/Documents/CoreFreq/corefreqk.c:10560:9: error: implicit declaration of function ‘cpu_up’ [-Werror=implicit-function-declaration]
10560 |    rc = cpu_up(cpu);
      |         ^~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:267: /home/Shane/Documents/CoreFreq/corefreqk.o] Error 1
make[1]: *** [Makefile:1732: /home/Shane/Documents/CoreFreq] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.7.0-0.rc3.20200429git1d2cc5ac6f66.1.fc32.x86_64'
make: *** [Makefile:39: all] Error 2

I'm assuming my problem is from using such a new kernel. Anything I can do to fix this?

cyring, 2020-05-02:
It is fixed in 5he

bubbleguuum, 2020-07-14:
I have the same failure to compile latest git version as of today for the newly released openSUSE Leap 15.2, using kernel 5.3.18 with lot of stuff backported (it compiles fines on Leap 15.1 and Tumbleweed):

```
[    4s] cc  -Wall \
[    4s] 	corefreq-cli.c corefreq-ui.c corefreq-cli-rsc.c \
[    4s] 	corefreq-cli-json.c corefreq-cli-extra.c \
[    4s] 	-D FEAT_DBG=0 -D UBENCH=0 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 \
[    4s] 	-o corefreq-cli -lm -lrt
[    4s] /usr/bin/make -j1 -C /lib/modules/5.3.18-lp152.19-default/build M=/home/abuild/rpmbuild/BUILD/CoreFreq modules
[    4s] make[1]: Entering directory '/usr/src/linux-5.3.18-lp152.19-obj/x86_64/default'
[    4s] make[1]: warning: -jN forced in submake: disabling jobserver mode.
[    4s] make[1]: Leaving directory '/usr/src/linux-5.3.18-lp152.19-obj/x86_64/default'
[    4s] make[1]: Entering directory '/usr/src/linux-5.3.18-lp152.19-obj/x86_64/default'
[    5s]   CC [M]  /home/abuild/rpmbuild/BUILD/CoreFreq/corefreqk.o
[    5s] /home/abuild/rpmbuild/BUILD/CoreFreq/corefreqk.c:258:26: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
[    5s]   /*MANDATORY*/ .verify = CoreFreqK_Policy_Verify,
[    5s]                           ^~~~~~~~~~~~~~~~~~~~~~~
[    5s] /home/abuild/rpmbuild/BUILD/CoreFreq/corefreqk.c:258:26: note: (near initialization for 'CoreFreqK.FreqDriver.verify')
[    5s] /home/abuild/rpmbuild/BUILD/CoreFreq/corefreqk.c: In function 'CoreFreqK_Policy_Verify':
[    5s] /home/abuild/rpmbuild/BUILD/CoreFreq/corefreqk.c:10941:36: error: passing argument 1 of 'cpufreq_verify_within_cpu_limits' from incompatible pointer type [-Werror=incompatible-pointer-types]
[    5s]    cpufreq_verify_within_cpu_limits(policy);
[    5s]                                     ^~~~~~
[    5s] In file included from /home/abuild/rpmbuild/BUILD/CoreFreq/corefreqk.c:25:0:
[    5s] /usr/src/linux-5.3.18-lp152.19/include/linux/cpufreq.h:448:1: note: expected 'struct cpufreq_policy_data *' but argument is of type 'struct cpufreq_policy *'
[    5s]  cpufreq_verify_within_cpu_limits(struct cpufreq_policy_data *policy)
[    5s]  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[    7s] cc1: some warnings being treated as errors
[    7s] make[3]: *** [/usr/src/linux-5.3.18-lp152.19/scripts/Makefile.build:282: /home/abuild/rpmbuild/BUILD/CoreFreq/corefreqk.o] Error 1
[    7s] make[2]: *** [/usr/src/linux-5.3.18-lp152.19/Makefile:1644: _module_/home/abuild/rpmbuild/BUILD/CoreFreq] Error 2
[    7s] make[1]: *** [../../../linux-5.3.18-lp152.19/Makefile:179: sub-make] Error 2
[    7s] make[1]: Leaving directory '/usr/src/linux-5.3.18-lp152.19-obj/x86_64/default'
[    7s] make: *** [Makefile:43: all] Error 2
[    7s] error: Bad exit status from /var/tmp/rpm-tmp.uP42uY (%build)
```

cyring, 2020-07-14:
https://github.com/cyring/CoreFreq/blob/befddf58b4f25e8b8955274dc89815e863fdbf0c/corefreqk.c#L10932

Above is the statement where you will adapt the version to fit your Suse distribution.

The thing is that looking at the 5.3.18 mainstream kernel, the Suse expected structure `cpufreq_policy_data` is not defined, bit still reference `cpufreq_policy_data`
https://elixir.bootlin.com/linux/v5.3.18/source/include/linux/cpufreq.h#L80

I don't see a way to manage compiling directives per distribution (and their back-port stuffs); I have to stick to the Linux kernel code versions to address a maximum cases of builds. Why Suse don't keep the kernel code coherency ?

Please let me know if it helps

bubbleguuum, 2020-07-14:
I will look if I can find such directive (also doubt it exists). Otherwise, as I'm buiding via a RPM, I will simply add a custom patch for the Leap 15.2 case.

cyring, 2020-07-14:
> I will look if I can find such directive (also doubt it exists). Otherwise, as I'm building via a RPM, I will simply add a custom patch for the Leap 15.2 case.

Off course, what you could also do is to build your kernel without `CONFIG_CPU_FREQ` , _CoreFreq_ driver will thus manage to build without.

bubbleguuum, 2020-07-14:
I just added a conditional patch in the RPM for Leap 15.2.
For reference, it is maintained there:
https://build.opensuse.org/package/show/hardware/CoreFreq

Interestingly, there are some Suse specific defines in the kernel (suse_version.h) but I did not understand how (if?) they could be used for module building.

cyring, 2020-07-14:
> I just added a conditional patch in the RPM for Leap 15.2.
> For reference, it is maintained there:
> https://build.opensuse.org/package/show/hardware/CoreFreq
>
> Interestingly, there are some Suse [...]

* So the patch is dedicated to this Suse release, isn't it ?

* This spec also shows a requirement:
```
# replace invocation in corefreqd.service of /bin/corefreqd  by /usr/bin/corefreqd
sed -i -e 's/\/bin\/corefreqd/\/usr\/bin\/corefreqd/g' %{buildroot}%{_unitdir}/corefreqd.service
```
Is there something I can improve for ?

bubbleguuum, 2020-07-14:
Maybe corefreqd.service shoud point to /usr/bin/corefreqd since it is the default PREFIX used in Makefile and where it will be installed ?

cyring, 2020-07-14:
> Maybe corefreqd.service shoud point to /usr/bin/corefreqd since it is the default PREFIX used in Makefile and where it will be installed ?
> 

It's an error, please open an issue, I will fix it.

Thank you

cyring, 2020-11-20:
Change done

closed 2020-11-20 by cyring