cyring/CoreFreq #176: UI - Dashboard - Hide temperature of the out of scope CPUs

cyring, 2020-04-21:
From [Threadripper tests](https://github.com/cyring/CoreFreq/issues/175#issuecomment-617093969), temperatures are left at `0C` or `32F` in the Dashboard for the thermal out of scope Cores.
Like the Frequency view, only the CPU involved into the sensor polling has to display the temperature, the others remain hidden.

eperezv, 2020-04-23:
I am not sure what the source of the problem is. As far as I know, to get proper ryzen temperatures on Linux is a nightmare. That was one of the reasons to update to ubuntu 20.04 beta, to be able to install kernel 5.4 that properly tracks ryzen 3000 temperatures https://www.phoronix.com/scan.php?page=news_item&px=Linux-5.4-Hwmon-Zen-2-Thermal At least from sensors, I only get 1 temperature for the whole CPU.

I am not sure if there is a way of getting more detailed temperatures in Linux 5.4 (or even with a later kernel version) for this processor.

cyring, 2020-04-23:
> I am not sure what the source of the problem is. As far as I know, to get proper ryzen temperatures on Linux is a nightmare. That was one of the reasons to update to ubuntu 20.04 beta, to be able to install kernel 5.4 that properly tracks ryzen 3000 temperatures https://www.phoronix.com/scan.php?page=news_item&px=Linux-5.4-Hwmon-Zen-2-Thermal At least from "sensors" app, I only get 1 temperature for the whole CPU (I see data from Tdie and Tctl, but both provide the same number).
> 
> I am not sure if there is a way of getting more detailed temperatures in Linux 5.4 (or even with a later kernel version) for this processor.

That's something I would like to develop, and I added it in the Roadmap #169 

> AMD EPYC & Threadripper : to provide the temperature sensor registers per CCD

I saw t
 a Kernel patch with those registers mentioned into it

I need a beta tester ?

eperezv, 2020-04-23:
Ok. I see.
I just found https://github.com/ocerman/zenpower and it seems to be working fine in my computer. With it, I can find temperatures on Tccd1, 3, 5 and 7. Hope it helps.

commit 2a111250 (2020-04-23): Issue Fix #176 #177 #178
closed 2020-04-24 by cyring