cyring/CoreFreq #181: cpuidle and cpufreq governor not working on AMD 3970X

justanerd, 2020-04-24:
The insmod should be enough if acpi_cpufreq is unloaded or am I missing something?

insmod corefreqk.ko Register_CPU_Freq=1 Register_CPU_Idle=1 Experimental=1 Register_Governor=1

dmesg:
```
[13864.517170] calling  CoreFreqK_init+0x0/0x1000 [corefreqk] @ 927568
[13864.780352] CoreFreq(0:32): Processor [ 8F_31] Architecture [Zen2/Castle Peak] SMT [64/64]
[13864.780950] initcall CoreFreqK_init+0x0/0x1000 [corefreqk] returned 0 after 257588 usecs
```
cmdline relavant part: idle=halt

cyring, 2020-04-24:
Hello,

* `Register_CPU_Idle` is not implemented for AMD but `Register_CPU_Freq` & `Register_Governor` are.

* Do you confirm from `/proc/modules` that any previous registered `cpufreq` is unloaded ?
  The current has to be force removed (_b/c Kernel does not like to live without a cpufreq driver once registered ! some Mutex reasons according to its source code_)

* My SysLinux/ArchLinux command line, for an Intel Westmere W3690` is the folowing:
```
BOOT_IMAGE=../vmlinuz-linux root=/dev/disk/by-label/ROOT rw rootdelay=0 quiet loglevel=3 rd.udev.log-priority=3 break=n agp=off ipv6.disable=1 selinux=0 plymouth.enable=0 consoleblank=0 vt.color=0x03 nmi_watchdog=0 modprobe.blacklist=pcspkr,iTCO_wdt,acpi_cpufreq,pcc_cpufreq,intel_cstate,intel_uncore,intel_powerclamp,i7core_edac,i5500_temp,coretemp,asus_atk0110,nouveau idle=halt intel_pstate=disable cpufreq.off=0 pcie_aspm=off intremap=off cpu0_hotplug audit=0 intel_idle.max_cstate=0 sysrq_always_enabled=1 acpi_enforce_resources=lax iommu=pt initrd=../intel-ucode.img,../initramfs-linux.img
```

* Btw: if you boot with Kernel parameter`cpufreq.off=1`, then _CoreFreq_ should take the leadership on the P-States (no driver options required).

justanerd, 2020-04-24:
I tried cpufreq.off=1 corefreq-cli shows all drivers off
Any way to verify if the P-State control is working?

cyring, 2020-04-24:
> I tried cpufreq.off=1 corefreq-cli shows all drivers off
> Any way to verify if the P-State control is working?

Yes, if you can increase the `Max` or any Boosted `1C` , `2C` ; it means that my driver is in control of the P-States. The Target `TGT` is also related to P-States, as a limiter.
[image]

_Please, gently raise the frequently ratio with one bin, as a starter_

justanerd, 2020-05-15:
I tried to change all the PSTATE settings I can choose the ratios but they will not get applied.
[image]

cyring, 2020-05-15:
Was this on the `master` branch ? May-be try the current `develop`

cyring, 2020-05-15:
I forgot to say : Core Performance Boost has to be disabled prior setting new ratios
I saw BIOS screenshots calling it Manual OC

justanerd, 2020-05-16:
That worked. Sadly I can't increase past 3.7Ghz.

cyring, 2020-05-16:
> That worked. Sadly I can't increase past 3.7Ghz.

Did you get a stable frequency  ? 
For the whole Processor or for distinct Core(s) ? 
What's the max frequency ratio you can set in BIOS or other tools ?

justanerd, 2020-05-16:
> > That worked. Sadly I can't increase past 3.7Ghz.
> 
> Did you get a stable frequency ?
Yes
> For the whole Processor or for distinct Core(s) ?
whole
> What's the max frequency ratio you can set in BIOS or other tools ?
With https://github.com/r4m0n/ZenStates-Linux I can enable the OC mode for all core to whatever I want. But the P-State stuff is also not working with higher frequencies.

cyring, 2020-05-16:
3970X is advertized with a frequency boost up to 4.5 GHz. How high your BIOS let you set in manual OC ? 
Do you them read the same frequency ratio between BIOS and Kernel  ?

justanerd, 2020-05-16:
I haven't tried the Custom P-State option yet but with a static divider I
can set what ever I want.
I mostly run the system with PBO because all core overclocks are not good
for single core performance.

cyring, 2020-05-26:
Hello,

Could you try the `develop` branch and post back the Topology in text mode: 
`corefreq-cli -s -m`

Thank you

justanerd, 2020-05-26:
sure
[output.txt](https://github.com/cyring/CoreFreq/files/4683514/output.txt)

cyring, 2020-05-26:
I'm not sure about the caches size: with 64 x SMT Cores, I'm decoding for a double value, compared to AMD datasheet.
Can you tell which sum is written for L1 L2 L3 in the UI header ?

justanerd, 2020-05-26:
[image]

cyring, 2020-05-26:
Thanks.
AMD says:
```
Total cache L1 = 2MB
Total cache L2 = 16MB
Total cache L3 = 128MB
```
Considering 32 physical Cores, L1 and L2 unit size in KB is ok, but L3 is wrong.
Still searching but I'm using the same algorithm for Ryzen and Threadripper

cyring, 2020-05-26:
Hello,
I'm rolling back to the AMD L3 cache size formula where register is given in 512 KB units.
Can you pull and try the latest `develop` branch.
Thank you.

cyring, 2020-06-21:
Hello,
I'm trying to read the thermal sensor per node which to my understanding counts more than one with EPYC & Threadripper
Could you please try the `develop` version. Once UI is running go to `Settings > Thermal scope` and change it to `Thread`, such as below.
[image]

justanerd, 2020-06-21:
[image]

cyring, 2020-06-21:
Thanks for trying
Regards

cyring, 2020-07-06:
Hello,
Among the last changes in issue #195 , an I/O ASM access to the FCH have been added: can you please check if the last _CoreFreq_ version is starting fine ?
Thank you

justanerd, 2020-07-07:
Yes, runs fine.

cyring, 2020-07-07:
> Yes, runs fine.

Thank you very much

cyring, 2020-08-18:
Hello,

For your testings, the cpufreq is available in the `develop` branch

You will have to blacklist any current cpufreq driver, like `acpi-cpufreq`, to leave the room to the _CoreFreq_ driver

CoreFreq's cpufreq can be registered from its kernel module using these two options:
```
insmod corefreqk.ko Register_CPU_Freq=1 Register_Governor=1
```
_or_ from the Client: you first register the governor then CPU-Freq

Remark: CPU-Idle is not implemented yet.

Once CoreFreq's cpufreq well registered, you can play with:
```
freq=3500000; for cpu in /sys/devices/system/cpu/cpufreq/policy*/scaling_setspeed ; do echo $freq > $cpu; done

### where freq is one of your available frequencies (ratio shown in cyan color)

echo 0 > /sys/devices/system/cpu/cpufreq/boost 
echo 1 > /sys/devices/system/cpu/cpufreq/boost

### and many other file attributes in the tree /sys/devices/system/cpu/cpufreq
```

Any change made on command line should be reflected in the UI

Cyril

cyring, 2020-09-03:
Hello,

In the `develop` branch is available a fix to monitor the temperature per CCD.
Could you please try with the Threadripper ?

* You go into `Settings` and change the `Thermal scope` to `Core` or `Thread`; next you stress per CPU with the `Tools > Turbo Select CPU...`
If things works OK, CPUs belonging to the same CCD should raise their temperature together.

justanerd, 2020-09-04:
[image]
It shows temps but they are completely in sync.

cyring, 2020-09-04:
> It shows temps but they are completely in sync.

Thanks for testing.
Can you post the full output of the topology using `corefreq-cli -m`
I have to the CCD resulting of Threadripper 

Be aware you have to stress differently CPUs to observe temps

You can now switch to the master branch

justanerd, 2020-09-04:
https://pastebin.ubuntu.com/p/3Z6yNrSd5R/

cyring, 2020-09-04:
* Very valuable output which shows that my CCD topology has an issue with more than 2 dies
https://github.com/cyring/CoreFreq/blob/d9e0202ccc792ff0047597401df7451b3a7baa44/corefreqk.c#L1718
* Referring to the EPYC PPR specifications, the architecture can go up to 8 CCDs with single and dual sockets
[image]
[image]

cyring, 2020-09-04:
Hello,

* In the `develop` branch is available a fix to enumerate the CCD and CCX identifiers.
  Can you please provide the topology output ?

* To my understanding CCX refers to the L3 cache shared by a group of 4 cores; whereas CCD is the complex die gathering 8 Cores

  Fyi here is my Ryzen 3950X map:
```
CPU Pkg  Apic  Core/Thread  Caches      (w)rite-Back (i)nclusive              
 #   ID   ID CCD CCX ID/ID L1-Inst Way  L1-Data Way      L2  Way      L3  Way 
000:BSP    0   0  0   0  0      32  8        32  8       512  8 i   65536 16w 
001:  0    2   0  0   1  0      32  8        32  8       512  8 i   65536 16w 
002:  0    4   0  0   2  0      32  8        32  8       512  8 i   65536 16w 
003:  0    6   0  0   3  0      32  8        32  8       512  8 i   65536 16w 
004:  0    8   0  1   4  0      32  8        32  8       512  8 i   65536 16w 
005:  0   10   0  1   5  0      32  8        32  8       512  8 i   65536 16w 
006:  0   12   0  1   6  0      32  8        32  8       512  8 i   65536 16w 
007:  0   14   0  1   7  0      32  8        32  8       512  8 i   65536 16w 
008:  0   16   1  0   8  0      32  8        32  8       512  8 i   65536 16w 
009:  0   18   1  0   9  0      32  8        32  8       512  8 i   65536 16w 
010:  0   20   1  0  10  0      32  8        32  8       512  8 i   65536 16w 
011:  0   22   1  0  11  0      32  8        32  8       512  8 i   65536 16w 
012:  0   24   1  1  12  0      32  8        32  8       512  8 i   65536 16w 
013:  0   26   1  1  13  0      32  8        32  8       512  8 i   65536 16w 
014:  0   28   1  1  14  0      32  8        32  8       512  8 i   65536 16w 
015:  0   30   1  1  15  0      32  8        32  8       512  8 i   65536 16w 
016:  0    1   0  0   0  1      32  8        32  8       512  8 i   65536 16w 
017:  0    3   0  0   1  1      32  8        32  8       512  8 i   65536 16w 
018:  0    5   0  0   2  1      32  8        32  8       512  8 i   65536 16w 
019:  0    7   0  0   3  1      32  8        32  8       512  8 i   65536 16w 
020:  0    9   0  1   4  1      32  8        32  8       512  8 i   65536 16w 
021:  0   11   0  1   5  1      32  8        32  8       512  8 i   65536 16w 
022:  0   13   0  1   6  1      32  8        32  8       512  8 i   65536 16w 
023:  0   15   0  1   7  1      32  8        32  8       512  8 i   65536 16w 
024:  0   17   1  0   8  1      32  8        32  8       512  8 i   65536 16w 
025:  0   19   1  0   9  1      32  8        32  8       512  8 i   65536 16w 
026:  0   21   1  0  10  1      32  8        32  8       512  8 i   65536 16w 
027:  0   23   1  0  11  1      32  8        32  8       512  8 i   65536 16w 
028:  0   25   1  1  12  1      32  8        32  8       512  8 i   65536 16w 
029:  0   27   1  1  13  1      32  8        32  8       512  8 i   65536 16w 
030:  0   29   1  1  14  1      32  8        32  8       512  8 i   65536 16w 
031:  0   31   1  1  15  1      32  8        32  8       512  8 i   65536 16w 
```
* If you find the Threadripper map OK, can you also try the per CCD temperature ?

Thank you

justanerd, 2020-09-04:
looks better:
[image]

justanerd, 2020-09-04:
zenpower output for reference:
[image]

cyring, 2020-09-04:
Thanks for your return.

* Do you mind to post the topology output ?

* To my concern, I don't understand why some groups of temperature remain at zero:
1. wrong address or no SMU register is backing the group ? 
2. is the Threadripper CCD numbering different from Epyc and/or Ryzen ?

justanerd, 2020-09-04:
https://pastebin.ubuntu.com/p/2Sct94kNyr/

cyring, 2020-09-04:
**Warning**: if you are using another monitoring tool while _CoreFreq_ is running, then my driver has to be build in a special way to share a kernel mutex among all tools going through the SMU (presuming they are doing the same)

The build for a _kernel protected SMU access_ is:
```
make HWM_CHIPSET=COMPATIBLE clean all
```
Remark `HWM_CHIPSET=COMPATIBLE` has not been intensively tested. So If you rather want to stay with the _CoreFreq_ driver standard way:
```
make clean all
```
then you have to make to sure no other software and their respective driver are not accessing the SMU: zenpower being one of them.

cyring, 2020-09-04:
New change again in the `develop` branch to count 4 Cores per CCD
_Previous results don't look like any screenshots of Threadrippers I can see on the Internet_
Sorry for this new change and thank you for providing the topology output.

justanerd, 2020-09-07:
[image]
First and third temps are always the same.
topology:
https://pastebin.ubuntu.com/p/S3sCNfghK9/

cyring, 2020-09-07:
> First and third temps are always the same.

Thank you for your returns

Based on the PPR specs above, it appears that one every two CCD is wired; may be because of the Threadripper design.

Based on the current `develop` branch, could you replace this source code:
https://github.com/cyring/CoreFreq/blob/dba5cc60d7f1fda33ef027e65d3072b6e6aa1698/corefreqk.c#L11023
with this:
```c
		Core_AMD_SMN_Read(	TccdSensor,
					SMU_AMD_THM_TCTL_CCD_REGISTER_F17H
					+ ((2 * Core->T.Cluster.CCD) << 2),
					SMU_AMD_INDEX_REGISTER_F17H,
					SMU_AMD_DATA_REGISTER_F17H );
```
Please build, reload driver all and test temperatures.  

Fyi, this change is interlacing the CCD when querying the SMU for sensors; a series of computed IDs, as follows `0 2 4 6`

justanerd, 2020-09-07:
looks better now:
[image]

justanerd, 2020-09-07:
first and last CCX still have the same temperature

cyring, 2020-09-07:
> first and last CCX still have the same temperature

* Probably applying load per CPU may help to understand the temperature relatively to the CCD and CCX clusters.

* Sharing my readings of the specs:

> up to eight core/cache complex dies (CCD) and a single I/O die (IOD).

> SP3 consists of from two to eight CCDs plus one IOD

> The two CCXs of a CCD share a single GMI2 Fabric port to the IOD.

> A single CCX consists of ... Four cores ... single-thread mode (1T) or two-thread SMT mode (2T) 

I'm trying to find in specs for a discriminant register to refine the topology among the 8 x CCD registers.  

We could also _count_ the CCD ID based on the architecture code-name but we have to consider the 3990X case and other EPYC 64 Cores where all the 8 CCD should be wired; and where the above code trick won't make it.

cyring, 2020-09-07:
Can you please post the CPUID dump
```
corefreq-cli -u
```
I'm especially interested by the latest physical Core of your Threadripper which would be CPU `#31`

justanerd, 2020-09-07:
https://pastebin.ubuntu.com/p/rJxxgzrwQS/

cyring, 2020-09-07:
Can you please output this ?
```
lspci -n
```

justanerd, 2020-09-07:
00:00.0 0600: 1022:1480
00:00.2 0806: 1022:1481
00:01.0 0600: 1022:1482
00:02.0 0600: 1022:1482
00:03.0 0600: 1022:1482
00:04.0 0600: 1022:1482
00:05.0 0600: 1022:1482
00:07.0 0600: 1022:1482
00:07.1 0604: 1022:1484
00:08.0 0600: 1022:1482
00:08.1 0604: 1022:1484
00:14.0 0c05: 1022:790b (rev 61)
00:14.3 0601: 1022:790e (rev 51)
00:18.0 0600: 1022:1490
00:18.1 0600: 1022:1491
00:18.2 0600: 1022:1492
00:18.3 0600: 1022:1493
00:18.4 0600: 1022:1494
00:18.5 0600: 1022:1495
00:18.6 0600: 1022:1496
00:18.7 0600: 1022:1497
01:00.0 1300: 1022:148a
02:00.0 1300: 1022:1485
02:00.3 0c03: 1022:148c
20:00.0 0600: 1022:1480
20:00.2 0806: 1022:1481
20:01.0 0600: 1022:1482
20:02.0 0600: 1022:1482
20:03.0 0600: 1022:1482
20:03.1 0604: 1022:1483
20:04.0 0600: 1022:1482
20:05.0 0600: 1022:1482
20:07.0 0600: 1022:1482
20:07.1 0604: 1022:1484
20:08.0 0600: 1022:1482
20:08.1 0604: 1022:1484
21:00.0 0300: 10de:1b06 (rev a1)
21:00.1 0403: 10de:10ef (rev a1)
22:00.0 1300: 1022:148a
23:00.0 1300: 1022:1485
23:00.1 1080: 1022:1486
23:00.3 0c03: 1022:148c
40:00.0 0600: 1022:1480
40:00.2 0806: 1022:1481
40:01.0 0600: 1022:1482
40:01.1 0604: 1022:1483
40:02.0 0600: 1022:1482
40:03.0 0600: 1022:1482
40:03.1 0604: 1022:1483
40:03.2 0604: 1022:1483
40:03.3 0604: 1022:1483
40:03.4 0604: 1022:1483
40:04.0 0600: 1022:1482
40:05.0 0600: 1022:1482
40:07.0 0600: 1022:1482
40:07.1 0604: 1022:1484
40:08.0 0600: 1022:1482
40:08.1 0604: 1022:1484
41:00.0 0604: 1022:57ad
42:01.0 0604: 1022:57a3
42:02.0 0604: 1022:57a3
42:03.0 0604: 1022:57a3
42:04.0 0604: 1022:57a3
42:05.0 0604: 1022:57a3
42:08.0 0604: 1022:57a4
42:09.0 0604: 1022:57a4
42:0a.0 0604: 1022:57a4
43:00.0 0108: 144d:a808
44:00.0 0200: 8086:1563 (rev 01)
44:00.1 0200: 8086:1563 (rev 01)
46:00.0 0c03: 1b21:2142
47:00.0 0106: 1b21:0612 (rev 02)
48:00.0 0280: 8086:2723 (rev 1a)
49:00.0 1300: 1022:1485
49:00.1 0c03: 1022:149c
49:00.3 0c03: 1022:149c
4a:00.0 0106: 1022:7901 (rev 51)
4b:00.0 0106: 1022:7901 (rev 51)
4c:00.0 0108: 144d:a808
4d:00.0 0108: 1987:5012 (rev 01)
4e:00.0 0108: 144d:a808
4f:00.0 0108: 144d:a808
50:00.0 1300: 1022:148a
51:00.0 1300: 1022:1485
60:00.0 0600: 1022:1480
60:00.2 0806: 1022:1481
60:01.0 0600: 1022:1482
60:02.0 0600: 1022:1482
60:03.0 0600: 1022:1482
60:04.0 0600: 1022:1482
60:05.0 0600: 1022:1482
60:07.0 0600: 1022:1482
60:07.1 0604: 1022:1484
60:08.0 0600: 1022:1482
60:08.1 0604: 1022:1484
61:00.0 1300: 1022:148a
62:00.0 1300: 1022:1485

cyring, 2020-09-07:
Thanks.

* I was expecting to decode this statement from the Specs:
> The processor includes configuration space registers located in both BCS and ECS. Processor configuration space is accessed through bus 0, devices 18h to 1Fh, where device 18h corresponds to node 0 and device 1Fh corresponds to node 7

   and find one device per enabled CCD; but in fact, comparing to my 3950X ...  
```
00:00.0 0600: 1022:1480
00:00.2 0806: 1022:1481
00:01.0 0600: 1022:1482
00:01.1 0604: 1022:1483
00:01.2 0604: 1022:1483
00:02.0 0600: 1022:1482
00:03.0 0600: 1022:1482
00:03.1 0604: 1022:1483
00:04.0 0600: 1022:1482
00:05.0 0600: 1022:1482
00:07.0 0600: 1022:1482
00:07.1 0604: 1022:1484
00:08.0 0600: 1022:1482
00:08.1 0604: 1022:1484
00:08.2 0604: 1022:1484
00:08.3 0604: 1022:1484
00:14.0 0c05: 1022:790b (rev 61)
00:14.3 0601: 1022:790e (rev 51)
00:18.0 0600: 1022:1440
00:18.1 0600: 1022:1441
00:18.2 0600: 1022:1442
00:18.3 0600: 1022:1443
00:18.4 0600: 1022:1444
00:18.5 0600: 1022:1445
00:18.6 0600: 1022:1446
00:18.7 0600: 1022:1447
01:00.0 0108: 1987:5016 (rev 01)
02:00.0 0604: 1022:57ad
03:03.0 0604: 1022:57a3
03:05.0 0604: 1022:57a3
03:06.0 0604: 1022:57a3
03:08.0 0604: 1022:57a4
03:09.0 0604: 1022:57a4
03:0a.0 0604: 1022:57a4
04:00.0 0200: 10ec:8125
05:00.0 0200: 8086:1539 (rev 03)
06:00.0 0280: 8086:2723 (rev 1a)
07:00.0 1300: 1022:1485
07:00.1 0c03: 1022:149c
07:00.3 0c03: 1022:149c
08:00.0 0106: 1022:7901 (rev 51)
09:00.0 0106: 1022:7901 (rev 51)
0a:00.0 0300: 10de:1086 (rev a1)
0a:00.1 0403: 10de:0e09 (rev a1)
0b:00.0 1300: 1022:148a
0c:00.0 1300: 1022:1485
0c:00.1 1080: 1022:1486
0c:00.3 0c03: 1022:149c
0c:00.4 0403: 1022:1487
0d:00.0 0106: 1022:7901 (rev 51)
0e:00.0 0106: 1022:7901 (rev 51)
```
   ... I'm just finding the same device `00:18.nn` which means node `0` and nothing like `00:19.nn` up to `00:1F.nn`

   I don't expect more solution from the PCI enumeration.

----

* About the CPUID, I'm trying to compute a discriminant from the leave `80000008:00000000`
  - 3970X : ECX = `0000703f` where `3f` means 63 + 1 = 64 SMT Cores
  - 3950X : ECX = `0000701f` where  `1f` means 31 + 1 = 32 SMT Cores

   To solve a _CCD factor_:

Example| Avail. Cores |   CPUID   | CCD_factor
------------|------------------|--------------|--------
3990X    |       64          |     128 ?    | 1
3970X    |       32          |       64       | 2
3960X    |       24          |       64 ?    | 2
3950X    |       16          |       32       | 1
3900X    |       12          |       32 ?    | 1

   and then to apply the CCD_factor into the SMU query

```C
		Core_AMD_SMN_Read(	TccdSensor,
					SMU_AMD_THM_TCTL_CCD_REGISTER_F17H
					+ (( CCD_factor * Core->T.Cluster.CCD) << 2),
					SMU_AMD_INDEX_REGISTER_F17H,
					SMU_AMD_DATA_REGISTER_F17H );
```

cyring, 2020-09-07:
I would also like to count the SMU registers by counting the number of UMC

1. can you dump the following address
```
modprobe msr
rdmsr -aX 0x0000017b
```
2. the output of
```
corefreq-cli -M
```

justanerd, 2020-09-07:
rdmsr:
FFFFFFFFFFFFFFEF
FFFFFFFFFFFFFFEF
FFFFFFFFFFFFFFEF
FFFFFFFFFFFFFFEF
FFFFFFFFFFFFFFEF
FFFFFFFFFFFFFFEF
FFFFFFFFFFFFFFEF
FFFFFFFFFFFFFFEF
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F

corefreq-cli -M
                              Zen UMC  [1493]                              
Controller #0                                                Quad Channel  
 Bus Rate     0 MT/s      Bus Speed    0 MHz           DRAM Speed    0 MHz 
                                                                           
 Cha   CL  RCDR RCDW  RP  RAS   RC  RRDS RRDL FAW  WTRS WTRL  WR  clRR clWW
  #0   18   18   18   18   39   57    4    6   26    3    9   18    3    3 
  #1   18   18   18   18   39   57    4    6   26    3    9   18    3    3 
  #2   16   15   14   14   32   46    4    6   20    4   12   12    4    4 
  #3   16   15   14   14   32   46    4    6   20    4   12   12    4    4 
      CWL  RTP RdWr WrRd scWW sdWW ddWW scRR sdRR ddRR drRR drWW drWR drRRD
  #0   12    9    5    1    1    3    3    1    3    3    0    0    0    0 
  #1   12    9    5    1    1    3    3    1    3    3    0    0    0    0 
  #2   16    8    8    4    1    7    7    1    5    5    0    0    0    0 
  #3   16    8    8    4    1    7    7    1    5    5    0    0    0    0 
      REFI RFC1 RFC2 RFC4 RCPB RPPB sFAW dFAW Ban  Page  CKE  CMD  GDM  ECC
  #0  9360  312  192  132   0    0    0    0  R0W0   0    6   1T   OFF   0 
  #1  9360  312  192  132   0    0    0    0  R0W0   0    6   1T   OFF   0 
  #2 14553  298  192  132   0    0    0    0  R1W1   0    1   1T    ON   0 
  #3 14553  298  192  132   0    0    0    0  R1W1   0    1   1T    ON   0 
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1                                                                  
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1                                                                  
 DIMM Geometry for channel #2                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1     2   16     65536      1024          16384                    
 DIMM Geometry for channel #3                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1     2   16     65536      1024          16384

justanerd, 2020-09-07:
There should be 2 more memory modules I have actually 4 x 16GB

cyring, 2020-09-07:
> There should be 2 more memory modules I have actually 4 x 16GB

Very interesting

* I have the following output from the MSR:
```
FFFFFFFFFFFFFFEF
FFFFFFFFFFFFFFEF
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
6F
```
3950X counts twice the pattern `FFFFFFFFFFFFFFEF` whereas 3970X , 8 times

* About the UMC, it's still a work in progress.  
Thanks for your output, I will check why it does not count 4 DIMMs ?
(_Fyi, to format code in Markdown, place 3 anti-quotes before after the code_)
```
                              Zen UMC  [1493]                              
Controller #0                                                Quad Channel  
 Bus Rate     0 MT/s      Bus Speed    0 MHz           DRAM Speed    0 MHz 
                                                                           
 Cha   CL  RCDR RCDW  RP  RAS   RC  RRDS RRDL FAW  WTRS WTRL  WR  clRR clWW
  #0   18   18   18   18   39   57    4    6   26    3    9   18    3    3 
  #1   18   18   18   18   39   57    4    6   26    3    9   18    3    3 
  #2   16   15   14   14   32   46    4    6   20    4   12   12    4    4 
  #3   16   15   14   14   32   46    4    6   20    4   12   12    4    4 
      CWL  RTP RdWr WrRd scWW sdWW ddWW scRR sdRR ddRR drRR drWW drWR drRRD
  #0   12    9    5    1    1    3    3    1    3    3    0    0    0    0 
  #1   12    9    5    1    1    3    3    1    3    3    0    0    0    0 
  #2   16    8    8    4    1    7    7    1    5    5    0    0    0    0 
  #3   16    8    8    4    1    7    7    1    5    5    0    0    0    0 
      REFI RFC1 RFC2 RFC4 RCPB RPPB sFAW dFAW Ban  Page  CKE  CMD  GDM  ECC
  #0  9360  312  192  132   0    0    0    0  R0W0   0    6   1T   OFF   0 
  #1  9360  312  192  132   0    0    0    0  R0W0   0    6   1T   OFF   0 
  #2 14553  298  192  132   0    0    0    0  R1W1   0    1   1T    ON   0 
  #3 14553  298  192  132   0    0    0    0  R1W1   0    1   1T    ON   0 
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1                                                                  
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1                                                                  
 DIMM Geometry for channel #2                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1     2   16     65536      1024          16384                    
 DIMM Geometry for channel #3                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1     2   16     65536      1024          16384                    
```

cyring, 2020-09-07:
For the memory module count, I'm refactoring the UMC code. Can you download, compile and run [zencli](https://github.com/cyring/Tips/blob/master/C/zencli.c)
```
cc zencli.c -o zencli
# as root:
./zencli umc 0x0
```

cyring, 2020-09-08:
**Edit**: can you also try the `develop` branch which has been improved to count the modules.
The Daemon will output a chips topology I'm interested in.

justanerd, 2020-09-08:
zencli umc 0x0
```
Welcome to the Data Fabric: UMC has 0 x Channel(s)

```
corefreq-cli -M
```
                              Zen UMC  [1493]                              
Controller #0                                                Quad Channel  
 Bus Rate     0 MT/s      Bus Speed    0 MHz           DRAM Speed    0 MHz 
                                                                           
 Cha   CL  RCDR RCDW  RP  RAS   RC  RRDS RRDL FAW  WTRS WTRL  WR  clRR clWW
  #0   18   18   18   18   39   57    4    6   26    3    9   18    3    3 
  #1   18   18   18   18   39   57    4    6   26    3    9   18    3    3 
  #2   16   15   14   14   32   46    4    6   20    4   12   12    4    4 
  #3   16   15   14   14   32   46    4    6   20    4   12   12    4    4 
      CWL  RTP RdWr WrRd scWW sdWW ddWW scRR sdRR ddRR drRR drWW drWR drRRD
  #0   12    9    5    1    1    3    3    1    3    3    0    0    0    0 
  #1   12    9    5    1    1    3    3    1    3    3    0    0    0    0 
  #2   16    8    8    4    1    7    7    1    5    5    0    0    0    0 
  #3   16    8    8    4    1    7    7    1    5    5    0    0    0    0 
      REFI RFC1 RFC2 RFC4 RCPB RPPB sFAW dFAW Ban  Page  CKE  CMD  GDM  ECC
  #0  9360  312  192  132   0    0    0    0  R0W0   0    6   1T   OFF   0 
  #1  9360  312  192  132   0    0    0    0  R0W0   0    6   1T   OFF   0 
  #2 14553  298  192  132   0    0    0    0  R1W1   0    1   1T    ON   0 
  #3 14553  298  192  132   0    0    0    0  R1W1   0    1   1T    ON   0 
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1     2   16     65536      1024          16384                    
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1     2   16     65536      1024          16384                    
 DIMM Geometry for channel #2                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1     2   16     65536      1024          16384                    
 DIMM Geometry for channel #3                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1     2   16     65536      1024          16384                    
```

I'm not sure why the timings differ between channel 0,1 and 2,3.

cyring, 2020-09-08:
> zencli umc 0x0
>
> ```
> Welcome to the Data Fabric: UMC has 0 x Channel(s)
> ```
>
> corefreq-cli -M
>
> ```
>                               Zen UMC  [1493]
> Controller [...]

Thanks. 

As a reference, do you have BIOS screenshots of Timings, or copy them here.

Be aware those registers are undocumented. Sources are Linux kernel for the ECC which includes Chip bits decoding. 
I wonder how they got those without AMD !
Timings and Frequency are SMU reversed engineering results spread over the web ! 

I'm also interested in the `corefreqd` output you got in terminal.

justanerd, 2020-09-08:
corefreqd output:
```
mc[0] cha[0/4]        chip[0] sec[0]  size[0]
mc[0]   cha[0/4]        chip[0] sec[1]  size[0]
mc[0]   cha[0/4]        chip[1] sec[0]  size[0]
mc[0]   cha[0/4]        chip[1] sec[1]  size[0]
mc[0]   cha[0/4]        chip[2] sec[0]  size[8388608]
mc[0]   cha[0/4]        chip[2] sec[1]  size[0]
mc[0]   cha[0/4]        chip[3] sec[0]  size[8388608]
mc[0]   cha[0/4]        chip[3] sec[1]  size[0]
mc[0]   cha[1/4]        chip[0] sec[0]  size[0]
mc[0]   cha[1/4]        chip[0] sec[1]  size[0]
mc[0]   cha[1/4]        chip[1] sec[0]  size[0]
mc[0]   cha[1/4]        chip[1] sec[1]  size[0]
mc[0]   cha[1/4]        chip[2] sec[0]  size[8388608]
mc[0]   cha[1/4]        chip[2] sec[1]  size[0]
mc[0]   cha[1/4]        chip[3] sec[0]  size[8388608]
mc[0]   cha[1/4]        chip[3] sec[1]  size[0]
mc[0]   cha[2/4]        chip[0] sec[0]  size[0]
mc[0]   cha[2/4]        chip[0] sec[1]  size[0]
mc[0]   cha[2/4]        chip[1] sec[0]  size[0]
mc[0]   cha[2/4]        chip[1] sec[1]  size[0]
mc[0]   cha[2/4]        chip[2] sec[0]  size[8388608]
mc[0]   cha[2/4]        chip[2] sec[1]  size[0]
mc[0]   cha[2/4]        chip[3] sec[0]  size[8388608]
mc[0]   cha[2/4]        chip[3] sec[1]  size[0]
mc[0]   cha[3/4]        chip[0] sec[0]  size[0]
mc[0]   cha[3/4]        chip[0] sec[1]  size[0]
mc[0]   cha[3/4]        chip[1] sec[0]  size[0]
mc[0]   cha[3/4]        chip[1] sec[1]  size[0]
mc[0]   cha[3/4]        chip[2] sec[0]  size[8388608]
mc[0]   cha[3/4]        chip[2] sec[1]  size[0]
mc[0]   cha[3/4]        chip[3] sec[0]  size[8388608]
mc[0]   cha[3/4]        chip[3] sec[1]  size[0]

```

I will later give you the BIOS screenhots.

cyring, 2020-09-08:
> I'm not sure why the timings differ between channel 0,1 and 2,3.

New code available to improve the CCD and SMU _affinity_.
It is available in `develop` for your Threadripper tests. 
Can you post the outputs of the followings:
* the Topology
* the temperature, SMT scope 
* the Memory Controller

justanerd, 2020-09-09:
[image]

```
CPU Pkg  Apic  Core/Thread  Caches      (w)rite-Back (i)nclusive              
 #   ID   ID CCD CCX ID/ID L1-Inst Way  L1-Data Way      L2  Way      L3  Way 
000:BSP    0   0  0   0  0      32  8        32  8       512  8 i  131072 16w 
001:  0    2   0  0   1  0      32  8        32  8       512  8 i  131072 16w 
002:  0    4   0  0   2  0      32  8        32  8       512  8 i  131072 16w 
003:  0    6   0  0   3  0      32  8        32  8       512  8 i  131072 16w 
004:  0    8   0  1   4  0      32  8        32  8       512  8 i  131072 16w 
005:  0   10   0  1   5  0      32  8        32  8       512  8 i  131072 16w 
006:  0   12   0  1   6  0      32  8        32  8       512  8 i  131072 16w 
007:  0   14   0  1   7  0      32  8        32  8       512  8 i  131072 16w 
008:  0   16   2  2   8  0      32  8        32  8       512  8 i  131072 16w 
009:  0   18   2  2   9  0      32  8        32  8       512  8 i  131072 16w 
010:  0   20   2  2  10  0      32  8        32  8       512  8 i  131072 16w 
011:  0   22   2  2  11  0      32  8        32  8       512  8 i  131072 16w 
012:  0   24   2  3  12  0      32  8        32  8       512  8 i  131072 16w 
013:  0   26   2  3  13  0      32  8        32  8       512  8 i  131072 16w 
014:  0   28   2  3  14  0      32  8        32  8       512  8 i  131072 16w 
015:  0   30   2  3  15  0      32  8        32  8       512  8 i  131072 16w 
016:  0   32   4  4  16  0      32  8        32  8       512  8 i  131072 16w 
017:  0   34   4  4  17  0      32  8        32  8       512  8 i  131072 16w 
018:  0   36   4  4  18  0      32  8        32  8       512  8 i  131072 16w 
019:  0   38   4  4  19  0      32  8        32  8       512  8 i  131072 16w 
020:  0   40   4  5  20  0      32  8        32  8       512  8 i  131072 16w 
021:  0   42   4  5  21  0      32  8        32  8       512  8 i  131072 16w 
022:  0   44   4  5  22  0      32  8        32  8       512  8 i  131072 16w 
023:  0   46   4  5  23  0      32  8        32  8       512  8 i  131072 16w 
024:  0   48   6  6  24  0      32  8        32  8       512  8 i  131072 16w 
025:  0   50   6  6  25  0      32  8        32  8       512  8 i  131072 16w 
026:  0   52   6  6  26  0      32  8        32  8       512  8 i  131072 16w 
027:  0   54   6  6  27  0      32  8        32  8       512  8 i  131072 16w 
028:  0   56   6  7  28  0      32  8        32  8       512  8 i  131072 16w 
029:  0   58   6  7  29  0      32  8        32  8       512  8 i  131072 16w 
030:  0   60   6  7  30  0      32  8        32  8       512  8 i  131072 16w 
031:  0   62   6  7  31  0      32  8        32  8       512  8 i  131072 16w 
032:  0    1   0  0   0  1      32  8        32  8       512  8 i  131072 16w 
033:  0    3   0  0   1  1      32  8        32  8       512  8 i  131072 16w 
034:  0    5   0  0   2  1      32  8        32  8       512  8 i  131072 16w 
035:  0    7   0  0   3  1      32  8        32  8       512  8 i  131072 16w 
036:  0    9   0  1   4  1      32  8        32  8       512  8 i  131072 16w 
037:  0   11   0  1   5  1      32  8        32  8       512  8 i  131072 16w 
038:  0   13   0  1   6  1      32  8        32  8       512  8 i  131072 16w 
039:  0   15   0  1   7  1      32  8        32  8       512  8 i  131072 16w 
040:  0   17   2  2   8  1      32  8        32  8       512  8 i  131072 16w 
041:  0   19   2  2   9  1      32  8        32  8       512  8 i  131072 16w 
042:  0   21   2  2  10  1      32  8        32  8       512  8 i  131072 16w 
043:  0   23   2  2  11  1      32  8        32  8       512  8 i  131072 16w 
044:  0   25   2  3  12  1      32  8        32  8       512  8 i  131072 16w 
045:  0   27   2  3  13  1      32  8        32  8       512  8 i  131072 16w 
046:  0   29   2  3  14  1      32  8        32  8       512  8 i  131072 16w 
047:  0   31   2  3  15  1      32  8        32  8       512  8 i  131072 16w 
048:  0   33   4  4  16  1      32  8        32  8       512  8 i  131072 16w 
049:  0   35   4  4  17  1      32  8        32  8       512  8 i  131072 16w 
050:  0   37   4  4  18  1      32  8        32  8       512  8 i  131072 16w 
051:  0   39   4  4  19  1      32  8        32  8       512  8 i  131072 16w 
052:  0   41   4  5  20  1      32  8        32  8       512  8 i  131072 16w 
053:  0   43   4  5  21  1      32  8        32  8       512  8 i  131072 16w 
054:  0   45   4  5  22  1      32  8        32  8       512  8 i  131072 16w 
055:  0   47   4  5  23  1      32  8        32  8       512  8 i  131072 16w 
056:  0   49   6  6  24  1      32  8        32  8       512  8 i  131072 16w 
057:  0   51   6  6  25  1      32  8        32  8       512  8 i  131072 16w 
058:  0   53   6  6  26  1      32  8        32  8       512  8 i  131072 16w 
059:  0   55   6  6  27  1      32  8        32  8       512  8 i  131072 16w 
060:  0   57   6  7  28  1      32  8        32  8       512  8 i  131072 16w 
061:  0   59   6  7  29  1      32  8        32  8       512  8 i  131072 16w 
062:  0   61   6  7  30  1      32  8        32  8       512  8 i  131072 16w 
063:  0   63   6  7  31  1      32  8        32  8       512  8 i  131072 16w 
```
[... 1 lines cut ...]
```
                              Zen UMC  [1493]                              
Controller #0                                                Dual Channel  
 Bus Rate  1866 MT/s      Bus Speed 1883 MHz           DRAM Speed 3767 MHz 
[... cut ...]
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1     2   16     65536      1024          16384                    
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1     2   16     65536      1024          16384                    

```

cyring, 2020-09-09:
**Thank you** for providing me all these test results.

* Things look much better. We now have decoded:
1. The particular CCD topology of the Threadripper 3970X and consequently the SMU addressing to the temperature sensors and the memory controller.
1. UMC, DRAM frequencies and timings are showing up.  
  All of them need to be compared to your BIOS settings.

* The counting of the 4 x 16GB DIMM is still inaccurate. That is a recurrent issue I'm encountering among Zen processors.
Due to a lack of specifications to blindly code a solution, I will still claim your contribute to code and test differentiate scenarios. Is this OK for you ?

Regards

justanerd, 2020-09-09:
Sure I'm glad to help.

Any plans to go down this rabbit hole?
https://pbs.twimg.com/media/EhKs3kIXcAMrzjc?format=png&name=small
Would be nice if he releases the source code or at least the PMU documantation or registers for adjusting CCX voltage etc.

cyring, 2020-09-09:
> Sure I'm glad to help.
> 
> Any plans to go down this rabbit hole?
> https://pbs.twimg.com/media/EhKs3kIXcAMrzjc?format=png&name=small
> Would be nice if he releases the source code or at least the PMU documantation or registers for adjusting CCX voltage etc.

It brought me to this [tweet](https://twitter.com/1usmus/status/1302293695518175233)
Does not look like a FOSS software, although most of those registers addresses are floating on the Internet; guessed by reverse engineering.
Without the BKDG specifications and the voltage usage, we risk to fry the hardware !
At least, I can start with read-only operations

[47 more replies]

closed 2020-12-07 by cyring