cyring/CoreFreq #184: Support for i5-4200U

Arthav24, 2020-05-15:
Hey, Thanks for the awesome tool.
I have added the tag at wiki page but could add gist page please do the honors.
here are some screenshots 
[image]
[image]
[image]

arthav@arthav-hdd:~/CoreFreq$ sudo ./corefreq-cli -s
[sudo] password for arthav: 
Processor                             [Intel(R) Core(TM) i5-4200U CPU @ 1.60GHz]
|- Architecture                                          [Haswell/Ultra Low TDP]
|- Vendor ID                                                      [GenuineIntel]
|- Microcode                                                        [        37]
|- Signature                                                           [  06_45]
|- Stepping                                                            [      1]
|- Online CPU                                                          [  4/  4]
|- Base Clock                                                          [ 99.868]
|- Frequency            (MHz)                      Ratio                        
                 Min    798.95                    [   8 ]                       
                 Max   2296.97                    [  23 ]                       
|- Factory                                                             [100.000]
                       1600                       [  16 ]                       
|- Performance                                                                  
   |- OSPM                                                                      
                 TGT   1498.02                    <  15 >                       
|- Turbo Boost                                                         [ UNLOCK]
                  1C   2596.57                    <  26 >                       
                  2C   2296.97                    <  23 >                       
                  3C   2296.97                    <  23 >                       
                  4C   2296.97                    <  23 >                       
|- Uncore                                                              [   LOCK]
                 Min    798.95                    [   8 ]                       
                 Max   2296.97                    [  23 ]                       
|- TDP                                                           Level [  0:3  ]
   |- Programmable                                                     [ UNLOCK]
   |- Configuration                                                    [ UNLOCK]
   |- Turbo Activation                                                 [ UNLOCK]
             Nominal   1597.89                    [  16 ]                       
              Level1    798.95                    [   8 ]                       
              Level2   2296.97                    [  23 ]                       
               Turbo   1498.02                    [  15 ]                       
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [N]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AVX512-F     [N]    AVX512-DQ [N]  AVX512-IFMA [N]   AVX512-PF [N] 
|- AVX512-ER    [N]    AVX512-CD [N]    AVX512-BW [N]   AVX512-VL [N] 
|- AVX512-VBMI  [N] AVX512-VBMI2 [N]  AVX512-VNMI [N]  AVX512-ALG [N] 
|- AVX512-VPOP  [N] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [N] 
|- AVX512-BF16  [N]  BMI1/BMI2 [Y/Y]      CLFLUSH [Y]        CMOV [Y] 
|- CMPXCHG8B    [Y]   CMPXCHG16B [Y]         F16C [Y]         FPU [Y] 
|- FXSR         [Y]    LAHF/SAHF [Y]    MMX/Ext [Y/N]  MONITOR/X[Y/N] 
|- MOVBE        [Y]    SERIALIZE [N]    PCLMULQDQ [Y]      POPCNT [Y] 
|- RDRAND       [Y]       RDSEED [N]       RDTSCP [Y]         SEP [Y] 
|- SGX          [N]          SSE [Y]         SSE2 [Y]        SSE3 [Y] 
|- SSSE3        [Y]  SSE4.1/4A [Y/N]       SSE4.2 [Y]     SYSCALL [Y] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- Core Multi-Processing                                  CMP Legacy   [Missing]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Capable]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Capable]
|- CPL Qualified Debug Store                                  DS-CPL   [Capable]
|- 64-Bit Debug Store                                         DTES64   [Capable]
|- Fast-String Operation                                Fast-Strings   [Capable]
|- Fused Multiply Add                                     FMA | FMA4   [Capable]
|- Hardware Lock Elision                                         HLE   [Missing]
[... 86 lines cut ...]
|- Last Level Cache Misses                                             [Capable]
|- Branch Instructions Retired                                         [Capable]
|- Branch Mispredicts Retired                                          [Capable]
                                                                                
Power & Thermal                                                                 
|- Clock Modulation                                             ODCM   <Disable>
   |- DutyCycle                                                        [ 50.00%]
|- Power Management                                         PWR MGMT   [   LOCK]
   |- Energy Policy                                        Bias Hint   <      6>
   |- Energy Policy                                          HWP EPP   [      0]
|- Junction Temperature                                        TjMax   [  0:100]
|- Digital Thermal Sensor                                        DTS   [Capable]
|- Power Limit Notification                                      PLN   [Capable]
|- Package Thermal Management                                    PTM   [Capable]
|- Thermal Monitor 1                                             TM1   [ Enable]
|- Thermal Monitor 2                                             TM2   [Capable]
|- Thermal Design Power                                          TDP   [     15]
   |- Minimum Power                                              Min   [Missing]
   |- Maximum Power                                              Max   [Missing]
|- Units                                                                        
   |- Power                                               watt   [  0.125000000]
   |- Energy                                             joule   [  0.000061035]
   |- Window                                            second   [  0.000976562]

Thanks

Arthav24, 2020-05-15:
arthav@arthav-hdd:~/CoreFreq$ sudo ./corefreq-cli -s
[sudo] password for arthav: 
Processor                             [Intel(R) Core(TM) i5-4200U CPU @ 1.60GHz]
|- Architecture                                          [Haswell/Ultra Low TDP]
|- Vendor ID                                                      [GenuineIntel]
|- Microcode                                                        [        37]
|- Signature                                                           [  06_45]
|- Stepping                                                            [      1]
|- Online CPU                                                          [  4/  4]
|- Base Clock                                                          [ 99.868]
|- Frequency            (MHz)                      Ratio                        
                 Min    798.95                    [   8 ]                       
                 Max   2296.97                    [  23 ]                       
|- Factory                                                             [100.000]
                       1600                       [  16 ]                       
|- Performance                                                                  
   |- OSPM                                                                      
                 TGT   1498.02                    <  15 >                       
|- Turbo Boost                                                         [ UNLOCK]
                  1C   2596.57                    <  26 >                       
                  2C   2296.97                    <  23 >                       
                  3C   2296.97                    <  23 >                       
                  4C   2296.97                    <  23 >                       
|- Uncore                                                              [   LOCK]
                 Min    798.95                    [   8 ]                       
                 Max   2296.97                    [  23 ]                       
|- TDP                                                           Level [  0:3  ]
   |- Programmable                                                     [ UNLOCK]
   |- Configuration                                                    [ UNLOCK]
   |- Turbo Activation                                                 [ UNLOCK]
             Nominal   1597.89                    [  16 ]                       
              Level1    798.95                    [   8 ]                       
              Level2   2296.97                    [  23 ]                       
               Turbo   1498.02                    [  15 ]                       
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [N]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AVX512-F     [N]    AVX512-DQ [N]  AVX512-IFMA [N]   AVX512-PF [N] 
|- AVX512-ER    [N]    AVX512-CD [N]    AVX512-BW [N]   AVX512-VL [N] 
|- AVX512-VBMI  [N] AVX512-VBMI2 [N]  AVX512-VNMI [N]  AVX512-ALG [N] 
|- AVX512-VPOP  [N] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [N] 
|- AVX512-BF16  [N]  BMI1/BMI2 [Y/Y]      CLFLUSH [Y]        CMOV [Y] 
|- CMPXCHG8B    [Y]   CMPXCHG16B [Y]         F16C [Y]         FPU [Y] 
|- FXSR         [Y]    LAHF/SAHF [Y]    MMX/Ext [Y/N]  MONITOR/X[Y/N] 
|- MOVBE        [Y]    SERIALIZE [N]    PCLMULQDQ [Y]      POPCNT [Y] 
|- RDRAND       [Y]       RDSEED [N]       RDTSCP [Y]         SEP [Y] 
|- SGX          [N]          SSE [Y]         SSE2 [Y]        SSE3 [Y] 
|- SSSE3        [Y]  SSE4.1/4A [Y/N]       SSE4.2 [Y]     SYSCALL [Y] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- Core Multi-Processing                                  CMP Legacy   [Missing]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Capable]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Capable]
|- CPL Qualified Debug Store                                  DS-CPL   [Capable]
|- 64-Bit Debug Store                                         DTES64   [Capable]
|- Fast-String Operation                                Fast-Strings   [Capable]
|- Fused Multiply Add                                     FMA | FMA4   [Capable]
|- Hardware Lock Elision                                         HLE   [Missing]
|- Long Mode 64 bits                                       IA64 | LM   [Capable]
|- LightWeight Profiling                                         LWP   [Missing]
|- Machine-Check Architecture                                    MCA   [Capable]
|- Memory Protection Extensions                                  MPX   [Missing]
|- Model Specific Registers                                      MSR   [Capable]
|- Memory Type Range Registers                                  MTRR   [Capable]
|- OS-Enabled Ext. State Management                          OSXSAVE   [Capable]
[... 77 lines cut ...]
[... cut ...]
|- Thermal Design Power                                          TDP   [     15]
   |- Minimum Power                                              Min   [Missing]
   |- Maximum Power                                              Max   [Missing]
|- Units                                                                        
   |- Power                                               watt   [  0.125000000]
   |- Energy                                             joule   [  0.000061035]
   |- Window                                            second   [  0.000976562]

cyring, 2020-05-15:
Thanks I will add a page.

Looks like there is remaining work w/ this Haswell ULT : 
* DRAM other timings
* Uncore counter
* Some Power RAPL
* Others I don't see yet

cyring, 2020-05-17:
[Here](https://gist.github.com/cyring/00191c52bc5958f52a53eb18435e4cdd) the page. Thank you.

I would suggest you remove & purge your screenshots above as they show your credentials

Arthav24, 2020-05-18:
ohh yess Sorry I should take care of it ! my bad.
I will remove them.
Do let me know how could I help you more with Corefreq :)

cyring, 2020-05-18:
Hello

# Memory Controller
* Can you post the output of
```
corefreq-cli -M
```
* I will need also need some memory speed and timings reference values to compare with : such as any screenshot from BIOS and/or `memtest86` or at least photos of DIMM timings stickers (plz mask the serial number)

 # Uncore
* Can you post screenshot of the Package view. 2 situations : idle, full Cores stressed (you can use menu Tools-Conics-2 planes)

Thank you

Arthav24, 2020-05-21:
```
sudo ./corefreq-cli -M
                         Lynx Point-M  [ A04]                         
Controller #0                                           Dual Channel  
 Bus Rate  5000 MT/s      Bus Speed 4988 MT/s     DRAM Speed 1600 MHz 
                                                                      
 Cha    CL  RCD   RP  RAS  RRD  RFC   WR RTPr WTPr  FAW  B2B  CWL Rate
  #0    11    0    0    0    0  208    0    0    0    0    0    8   1N
  #1    11    0    0    0    0  208    0    0    0    0    0    8   1N
      ddWR drWR srWR ddRW drRW srRW ddRR drRR srRR ddWW drWW srWW  ECC
  #0     4    3   20   11   11   11    6    6    4    7    7    4   0 
  #1     4    3   20   11   11   11    6    6    4    7    7    4   0 
                                                                      
 DIMM Geometry for channel #0                                         
      Slot Bank Rank     Rows   Columns    Memory Size (MB)           
       #0     4    2     16384      1024           4096               
 DIMM Geometry for channel #1                                         
      Slot Bank Rank     Rows   Columns    Memory Size (MB)           
       #0     4    2     16384      1024           4096 
```
I have this corsair DDR3L RAM installed (https://www.amazon.in/gp/product/B00G40TB6E/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1) in 2 x 4GB
 4GB Module (1x4GB) DDR3L 1600MHz SODIMM
1600MHz, 11-11-11-28 latency 

[image]
[image]

cyring, 2020-05-21:
Thanks, I'll check how to improve `RCD-RP-RAS`. I remember those were missing in datasheet.

Uncore is still tagged as experimental (b/c I had no hardware to test with)
https://github.com/cyring/CoreFreq/blob/d4483233be53413c99e1e68411e8e38e4de4bdaf/corefreqk.c#L8039
To test the Uncore counter, you have to start the driver with the Experimental mode enabled :
```
insmod corefreqk.ko Experimental=1
```
Be prepared for a potential processor crash. Save, close your files before shooting the driver. 

`HOT` is light on. It means processor has pending events, usually thermal throttling bit has been raised.
Press `[H]` and clear those events.
If processor load is idle, thermal event should stay cleared ; and CPU temperature not displayed in red

cyring, 2020-05-23:
* Can you replace function `Query_HSW_IMC` with the following code in source file `corefreqk.c` :
https://github.com/cyring/CoreFreq/blob/d4483233be53413c99e1e68411e8e38e4de4bdaf/corefreqk.c#L2826

```c
void Query_HSW_IMC(void __iomem *mchmap)
{	/*Source: Desktop 4th & 5th Generation Intel® Core™ Processor Family.*/
	unsigned short cha, dimmCount[2];

	Proc->Uncore.CtrlCount = 1;

	Proc->Uncore.MC[0].SNB.MAD0.value = readl(mchmap + 0x5004);
	Proc->Uncore.MC[0].SNB.MAD1.value = readl(mchmap + 0x5008);

	Proc->Uncore.MC[0].ChannelCount = 0;

	dimmCount[0] = (Proc->Uncore.MC[0].SNB.MAD0.Dimm_A_Size > 0)
		     + (Proc->Uncore.MC[0].SNB.MAD0.Dimm_B_Size > 0);
	dimmCount[1] = (Proc->Uncore.MC[0].SNB.MAD1.Dimm_A_Size > 0)
		     + (Proc->Uncore.MC[0].SNB.MAD1.Dimm_B_Size > 0);

	for (cha = 0; cha < 2; cha++)
		Proc->Uncore.MC[0].ChannelCount += (dimmCount[cha] > 0);

	for (cha = 0; cha < Proc->Uncore.MC[0].ChannelCount; cha++)
	{
		unsigned int R4C00 = readl(mchmap + 0x4c00);
		unsigned int R4C10 = readl(mchmap + 0x4c10);
		printk("CoreFreq: R4C00[%0x] R4C10[%0x]\n", R4C00, R4C10);

/*TODO( Unsolved: What is the channel #1 'X' factor of Haswell registers ? )
		Proc->Uncore.MC[0].Channel[cha].HSW.Timing.value =
					readl(mchmap + 0x4c04 + X * cha);
		Proc->Uncore.MC[0].Channel[cha].HSW.Rank_A.value =
					readl(mchmap + 0x4c08 + X * cha);
		Proc->Uncore.MC[0].Channel[cha].HSW.Rank_B.value =
					readl(mchmap + 0x4c0c + X * cha);
		Proc->Uncore.MC[0].Channel[cha].HSW.Rank.value =
					readl(mchmap + 0x4c14 + X * cha);
		Proc->Uncore.MC[0].Channel[cha].HSW.Refresh.value =
					readl(mchmap + 0x4e98 + X * cha);
*/
		Proc->Uncore.MC[0].Channel[cha].HSW.Timing.value =
					readl(mchmap + 0x4c04);

		Proc->Uncore.MC[0].Channel[cha].HSW.Rank_A.value =
					readl(mchmap + 0x4c08);

		Proc->Uncore.MC[0].Channel[cha].HSW.Rank_B.value =
					readl(mchmap + 0x4c0c);

		Proc->Uncore.MC[0].Channel[cha].HSW.Rank.value =
					readl(mchmap + 0x4c14);

		Proc->Uncore.MC[0].Channel[cha].HSW.Refresh.value =
					readl(mchmap + 0x4e98);
	}
	/* Is Dual DIMM Per Channel Disable ? */
	Proc->Uncore.MC[0].SlotCount = (Proc->Uncore.Bus.SNB_Cap.DDPCD == 1) ?
					1 : Proc->Uncore.MC[0].ChannelCount;

	Query_Turbo_TDP_Config(mchmap);
}
```

* Then rebuild, stop all, reload all and post the traces in kernel log
```
rmmod corefreqk.ko
make clean all
insmod corefreqk.ko
dmesg | grep CoreFreq
```

Thanks

cyring, 2020-05-25:
Hello,
Any return for above Uncore test and IMC debug
Regards

Arthav24, 2020-05-25:
> Thanks, I'll check how to improve `RCD-RP-RAS`. I remember those were missing in datasheet.
>
> Uncore is still tagged as experimental (b/c I had no hardware to test with) [...]
[image]

[image]
I cleared all events using H and also insmod corefreqk.ko Experimental=1

Arthav24, 2020-05-25:
> * Can you replace function `Query_HSW_IMC` with the following code in source file `corefreqk.c` : [...]

output of dmesg | grep CoreFreq

```
[  191.079817] CoreFreq(0:1): Processor [ 06_45] Architecture [Haswell/Ultra Low TDP] SMT [4/4]
[  621.307534] CoreFreq: Unload
[  789.209100] CoreFreq(0:1): Processor [ 06_45] Architecture [Haswell/Ultra Low TDP] SMT [4/4]
[  789.209207] CoreFreq: R4C00[1586716b] R4C10[1007]
[  789.209210] CoreFreq: R4C00[1586716b] R4C10[1007]
```

cyring, 2020-05-25:
> I cleared all events using H and also insmod corefreqk.ko Experimental=1

Thus thermal issues are still there. Although you press to clear each lighted events, the CPU temperatures remain RED ; it means the thermal register sticky bits are recurring.
[image]

cyring, 2020-05-25:
> > I cleared all events using H and also insmod corefreqk.ko Experimental=1
> 
> Thus thermal issues are still there. Although you press to clear each lighted events, the CPU temperatures remain RED ; it means the thermal register sticky bits are recurring.
> [image]

* Uncore is not counting so I don't know its register.

cyring, 2020-05-25:
* Register at offset 0x4C00
```
0x1586716b = 0b10101100001100111000101101011
```
[14-10] | [9-5] | [4-0]
-------- | ------- | --------
11100 | 01011 | 01011
28 | 11 | 11
RAS | RP? | RCD ?

cyring, 2020-05-25:
Hello,

Can you now switch to the `develop` branch of _CoreFreq_ ; 
unload, rebuild all ; 
and test for the timings tRP , tRRD , tRCD and tRAS

Plz show me a screenshot of the DIMM timings output

Arthav24, 2020-05-26:
yess I will do that give me some time actually that laptop isn't mine. will post results in some time.
and for uncore i think I was using master branch ! should I be using the develop branch for Uncore test too ?

Arthav24, 2020-05-26:
```
 dmesg | grep CoreFreq
[  680.644970] CoreFreq(2:3): Processor [ 06_45] Architecture [Haswell/Ultra Low TDP] SMT [4/4]

```
[image]

Arthav24, 2020-05-26:
I noticed that Hot was getting triggered due to power limitation and that too because on battery power so.
uncore testing (replace the query_hsw_imc in develop branch)
```
dmesg | grep CoreFreq
[  680.644970] CoreFreq(2:3): Processor [ 06_45] Architecture [Haswell/Ultra Low TDP] SMT [4/4]
[  976.849350] CoreFreq: Unload
[ 1057.716502] CoreFreq(2:3): Processor [ 06_45] Architecture [Haswell/Ultra Low TDP] SMT [4/4]
[ 1057.716618] CoreFreq: R4C00[1586716b] R4C10[1007]
[ 1057.716621] CoreFreq: R4C00[1586716b] R4C10[1007]
```
loaded driver with Experimental=1
[image]

[image]

cyring, 2020-05-26:
IMC looks better, thanks
I just missed the offset of registers for timings WR, FAW and B2B

You can stay on the development branch, and don't need to edit code any more. Changes have been back ported. 
If you have other tools, Bios to read the above timings ?
(not the SPD but the current configured timings)

cyring, 2020-05-26:
Another request: can you check if Power management features work ?
1. Enable the Experimental mode
2. Go to the window `Power & Thermal`
3. Check if you can enable the `Clock Modulation` ?
4. If enabled then change to another percent and stress Cores to observe a clock limitation
5. Disable `Clock Modulation` , while stressing, and check if Clock is restored to nominal ?
[image]

cyring, 2020-05-26:
* Update from the last development branch: In experimental mode, please try to read the Uncore counter.
Thanks

cyring, 2020-05-26:
* Update: If enabling the Uncore counter crashes Processor; please try this code:
https://github.com/cyring/CoreFreq/blob/73a5056db660fd865af273af62b35c7bee2230b1/corefreqk.c#L7294
```c
#define PKG_Counters_Haswell_ULT(Core, T)				\
({									\
    if (Proc->Registration.Experimental) {				\
	RDTSCP_COUNTERx7(Proc->Counter[T].PTSC,				\
			MSR_PKG_C2_RESIDENCY, Proc->Counter[T].PC02,	\
			MSR_PKG_C3_RESIDENCY, Proc->Counter[T].PC03,	\
			MSR_PKG_C6_RESIDENCY, Proc->Counter[T].PC06,	\
			MSR_PKG_C7_RESIDENCY, Proc->Counter[T].PC07,	\
			MSR_PKG_C8_RESIDENCY, Proc->Counter[T].PC08,	\
			MSR_PKG_C9_RESIDENCY, Proc->Counter[T].PC09,	\
			MSR_PKG_C10_RESIDENCY,Proc->Counter[T].PC10);	\
									\
	RDCOUNTER	(Proc->Counter[T].Uncore.FC0,			\
			MSR_SNB_UNCORE_PERF_FIXED_CTR0 );		\
    } else {								\
	RDTSCP_COUNTERx7(Proc->Counter[T].PTSC,				\
			MSR_PKG_C2_RESIDENCY, Proc->Counter[T].PC02,	\
			MSR_PKG_C3_RESIDENCY, Proc->Counter[T].PC03,	\
			MSR_PKG_C6_RESIDENCY, Proc->Counter[T].PC06,	\
			MSR_PKG_C7_RESIDENCY, Proc->Counter[T].PC07,	\
			MSR_PKG_C8_RESIDENCY, Proc->Counter[T].PC08,	\
			MSR_PKG_C9_RESIDENCY, Proc->Counter[T].PC09,	\
			MSR_PKG_C10_RESIDENCY,Proc->Counter[T].PC10);	\
    }									\
})
```
then rebuild, reload all

Arthav24, 2020-05-27:
> Another request: can you check if Power management features work ?
> 
>     1. Enable the Experimental mode
> 
>     2. Go to the window `Power & Thermal`
> 
>     3. Check if you can enable the `Clock Modulation` ?
> 
>     4. If enabled then change the `Duty Cycle` to another percent and stress Cores to observe a clock limitation
> 
>     5. Disable `Clock Modulation` , while stressing, and check if Clock is restored to nominal ?
>        [image]

yes the Clock Modulation works. while running conic 2 plane compute I observed the frequencies. changing the percentage changed the frequency and disable it, frequency is back to normal. :+1:
[image]
[image]

Arthav24, 2020-05-27:
>     * Update: If enabling the Uncore counter crashes Processor or counter value remains at zero ; please try this code:
>       https://github.com/cyring/CoreFreq/blob/73a5056db660fd865af273af62b35c7bee2230b1/corefreqk.c#L7294
> 
> 
> ```c
> #define PKG_Counters_Haswell_ULT(Core, T)				\
> ({									\
>     if (Proc->Registration.Experimental) {				\
> 	RDTSCP_COUNTERx7(Proc->Counter[T].PTSC,				\
> 			MSR_PKG_C2_RESIDENCY, Proc->Counter[T].PC02,	\
> 			MSR_PKG_C3_RESIDENCY, Proc->Counter[T].PC03,	\
> 			MSR_PKG_C6_RESIDENCY, Proc->Counter[T].PC06,	\
> 			MSR_PKG_C7_RESIDENCY, Proc->Counter[T].PC07,	\
> 			MSR_PKG_C8_RESIDENCY, Proc->Counter[T].PC08,	\
> 			MSR_PKG_C9_RESIDENCY, Proc->Counter[T].PC09,	\
> 			MSR_PKG_C10_RESIDENCY,Proc->Counter[T].PC10);	\
> 									\
> 	RDCOUNTER	(Proc->Counter[T].Uncore.FC0,			\
> 			MSR_SNB_UNCORE_PERF_FIXED_CTR0 );		\
>     } else {								\
> 	RDTSCP_COUNTERx7(Proc->Counter[T].PTSC,				\
> 			MSR_PKG_C2_RESIDENCY, Proc->Counter[T].PC02,	\
> 			MSR_PKG_C3_RESIDENCY, Proc->Counter[T].PC03,	\
> 			MSR_PKG_C6_RESIDENCY, Proc->Counter[T].PC06,	\
> 			MSR_PKG_C7_RESIDENCY, Proc->Counter[T].PC07,	\
> 			MSR_PKG_C8_RESIDENCY, Proc->Counter[T].PC08,	\
> 			MSR_PKG_C9_RESIDENCY, Proc->Counter[T].PC09,	\
> 			MSR_PKG_C10_RESIDENCY,Proc->Counter[T].PC10);	\
>     }									\
> })
> ```
> 
> then rebuild, reload all

pulled the develop branch but the uncore counter was stuck at 0. 
then changed the code snippet and results are below
[image]
[image]

cyring, 2020-05-27:
Thanks :100: 
Code pushed into the development branch.

cyring, 2020-11-20:
Change done.

closed 2020-11-20 by cyring