cyring/CoreFreq #186: Daemon Segmentation Fault on X570 + Ryzen 3

olejon, 2020-06-08:
I updated CoreFreq a couple of days ago using my automatic script and it worked just fine.

Now I did it again, and loading the module works but the daemon does not load and the UI shows nothing.

---

```
# corefreqd -d

CoreFreq Daemon 1.78.1  Copyright (C) 2015-2020 CYRIL INGENIERIE

Processor [AMD Ryzen 5 3600X 6-Core Processor]

Architecture [Zen2/Matisse] 12/12 CPU Online.

SleepInterval(1000), SysGate(2000), 2326 tasks

    Thread [7fa23dc86700] Init CHILD 000
    Thread [7fa23d485700] Init CHILD 001
    Thread [7fa23cc84700] Init CHILD 002
    Thread [7fa237fff700] Init CHILD 003
    Thread [7fa22ffff700] Init CHILD 004
    Thread [7fa235ffb700] Init CHILD 008
    Thread [7fa2367fc700] Init CHILD 007
    Thread [7fa236ffd700] Init CHILD 006
    Thread [7fa234ff9700] Init CHILD 010
    Thread [7fa22f7fe700] Init CHILD 011
    Thread [7fa2377fe700] Init CHILD 005
    Thread [7fa2357fa700] Init CHILD 009

Segmentation fault
```

---

Setup is same as before when you asked about my entire setup. I see you did some changes recently.

olejon, 2020-06-08:
**FYI:** It works fine on my Intel NUC (Haswell i3 Ultra Low TDP)

cyring, 2020-06-09:
Hi
Can you check Kernel log if `corefreqk.ko` has crashed ?

olejon, 2020-06-09:
Well, **it stays loaded**, and the `corefreqd-cmgr` process must be killed to unload it.

The kernel message is *as it used* to be **IIRC**. As it's not signed and loaded with `insmod`, although Secure Boot is disabled, but boots using EFI, not legacy.

```
corefreqk: loading out-of-tree module taints kernel.
corefreqk: module verification failed: signature and/or required key missing - tainting kernel
CoreFreq(1:7): Processor [ 8F_71] Architecture [Zen2/Matisse] SMT [12/12]
```

However the NUC, also using EFI boot partition, only shows:

```
CoreFreq(0:2): Processor [ 06_45] Architecture [Haswell/Ultra Low TDP] SMT [4/4]
```

But the X570 has a more modern BIOS for (U)EFI. The NUC is running the cleanest netinstall of Ubuntu 18.04. There are som differences there.

cyring, 2020-06-09:
I'm searching in source code for a change concerning an AMD instruction or RO page addressing but I don't find any so far.
It will be helpful if you start the Daemon with gdb and tell me in which function it segfaults
1. Build _CoreFreq_ with debug flag
```
make clean all CC='cc -g'
```
2. Load the driver and start Daemon in gdb
```
gdb corefreqd
run
```
3. Please post the output of GDB when it crashes and its call stack
```
backtrace
```

Thanks

cyring, 2020-06-09:
OK, I guess I found the issue.
Forget my gdb request above and just pull and try the last commit
Thanks

olejon, 2020-06-09:
Indeed. It works :+1: 

I still compile like this:

`make MSR_CORE_PERF_UCC=MSR_IA32_APERF MSR_CORE_PERF_URC=MSR_IA32_MPERF`

Cheers!

cyring, 2020-06-09:
Great. Thanks for your test.

olejon, 2020-06-09:
Also tested on Ubuntu 20.04 LTS with Live USB. Works fine there too :+1:

cyring, 2020-06-10:
> Also tested on Ubuntu 20.04 LTS with Live USB. Works fine there too

Thanks a lot

olejon, 2020-06-11:
> Also, by the end of the week, my new 3950X should be ready for the AMD developments

Cool! Finally ordered mine as well, with this cooler. What a beast! More than 1.3 kg... Only worry it'll make a nice part of the inside invisible through the tinted chassis window.

https://noctua.at/en/nh-d15-se-am4

cyring, 2020-06-11:
* Freshly setup
[image]

[image]

[image]

[image]

* I'll be out of coding for a while. There's some fans to setup, many tunings, close the case, and my GPU is still my old GTX570. etc
But so far those G.Skill DIMM die Samsung are amazing : 3600 MHz CL 16 with no effort
Fractal is making great products : Define 7 TG grey case + Celsius Dynamic S36. Temps looks goods.
Best for the end: Ryzen 3950X detected by _CoreFreq_

olejon, 2020-06-11:
Nice! :+1: 

If your RAM chips are 3600 MHz, IIRC from this video tuning Infinity Fabric to *not* be 50/50 shows benefit. My 3200 MHz ones don't. Also other tweaks, although little overclocking can be done on Ryzen, because it shouldn't be necessary... Intel sells on assuming people will do it. https://youtu.be/7w1EGPZUESU

cyring, 2020-06-11:
Thanks for the tips. Just gave a quick look but the video games intro makes me think about getting a new GPU which is Linux friendly. AMD RX 5600 is in my short list. But I'm not a gamer at all, and Pro GPU, such AMD WS series for PCI 4.0 is too expensive just to run a Linux desktop !
Really, manufacturers are totally ignoring that professionals need some thing else than RGB & gaming components

olejon, 2020-06-11:
Yes. Well they use like Cinebench etc a lot as it is a good benchmark tool. Games shows an overlay with tons of info including RAM performance/usage, GPU memory performance/usage if using MSI Afterburner (free and not limited to MSI cards nor AMD cards, works on Intel and NVIDIS GPUs too).

That's why I stick to my Radeon RX 570 (MSI), which was on discount and everything. It's still sold here, but the MSI one seems hard to find now.

Chose it after considering an NVIDIA 1660 as they've always been touted as very good vs price, and especially because almost always big discount on many sites. But didn't really want NVIDIA. Saw the Norwegian review and was compared to the RX 570. For me most important was of course out of the box Linux support (in kernel), then noise and performance at idle were basically the same (that review used Windows of course).

It gives 300 FPS using a FPS-testing map in CS:GO so... Why switch? Way above my monitors' refresh rates.

GPUs nowadays are the biggest single investment and they know it ( unless buying a 3950X :P ). They Ramp up prices because little competition. AMD tries to stay just below NVIDIA, that's about it. But at least their new 5XXX series is supported by the open radeon driver.

cyring, 2020-06-11:
* Does Linux reclocking work with the RX 570 ?
* The RX 5000 series brings PCIe 4. Not sure it makes a difference with previous generation. I'll avoid RX 5500 because of its 8x speed, although it's the cheapest.
* You may have read that kernel changelogs leak the ID of next Radeon GPU. Could be worth to wait for it. But I don't expect a cheap price, day one.

* NVIDIA is making the fastest GPU. I have been satisfied with its closed source until version 390xx got unsupported by Arch Linux. The only AUR package for 390xx is most of time in late with latest Linux kernel. 
Nvidia should really contribute to the nouveau driver.

olejon, 2020-06-12:
* Haven't tried, haven't even searched so don't really know how to (you do?). Pretty sure I tried the max OC which MSI Afterburner supports with a click, on Windows, and it's safe, never void warranty, and it worked just fine

BIOS says it runs `x16 Native` according to, *notice the OC in the name*, [Specifications of Radeon RX 570 ARMOR 8G OC](https://www.msi.com/Graphics-card/Radeon-RX-570-ARMOR-8G-OC/Specification)

[image]

* Yeah, maybe. Haven't followed tech for a while. I'm happy as long as the desktop is super smooth, HW video acceleration is smooth, Wayland works, Android Emulator is smooth, games are smooth although I don't game but have "played around a bit", online and with bots when last lived in Spain

* NVIDIA tends to be winning. But for me going from Intel iGPU to a dedicated AMD GPU was nice, TBH. Having 3 DisplayPorts and 1 HDMI is nice. Though NVIDIAs Control Center allows for some crazy setup `xrandr` simply can't do, although those changes can be written to `xorg.conf`. I use a pretty crazy multi-monitor setup on my monster laptop at my parents house (laptop 18 inches, VGA primary 17 inches + HDMI TV)

* Yeah, when I changed to Ryzen and realized it had no iGPU I looked first at NVIDIA. Then entered the world of AMD having a great driver built-in, and they actively contribute. I've seen improvements after kernel updates and graphics stack updates. I looked at the `nouveau` supported cards. None are really supported. They all lacked something I want. Was like "yay it supports X" then footnote "doesn't support Y, or is buggy". IIRC saw the Arch Wiki for helping me choose but basically said none have what one should expect today. And the performance sucks even if something is supported

* You won't be very impressed by the results of say Unigine Superposition benchmark for a RX 570. It performes a bit better on Windows where it uses DirectX, so maybe it isn't that optimized for OpenGL. At least I've found it just as good for games FPS, multi-monitor, video acceleration etc as Windows

* At least `lm-sensors` got an upgrade for Radeon 5XXX series, because the `edge` reading is for that series (and newer I guess), because they have a ton of sensors and the one on the edge decides. Also k10temp got Tdie and Tctl.

```
amdgpu-pci-0900
Adapter: PCI adapter
vddgfx:       +0.90 V  
fan1:        1004 RPM  (min =    0 RPM, max = 4500 RPM)
edge:         +44.0°C  (crit = +94.0°C, hyst = -273.1°C)
power1:       31.13 W  (cap = 120.00 W)

k10temp-pci-00c3
Adapter: PCI adapter
Tdie:         +60.4°C  (high = +70.0°C)
Tctl:         +60.4°C
```

* You get this good result on your NVMe? I get the same on both (CPU bound and MB bound)

[image]

---

Went through the process of booting Windows for the first time in months (Insider Program, Fast Track). Only 1 reboot (long) reboot. Wow. Ran Geekbench in CLI while system idle, so fair. Linux is still winning. Also in Blender Benchmark but didn't have time to run the new version on Windows, but surely is.

### Linux

[image]

### Windows

[image]

cyring, 2020-06-12:
* Having sensors, especially their source code, is driving me to Radeon.
[SAPPHIRE PULSE RX 5600 XT 6G](https://www.sapphiretech.com/en/consumer/pulse-radeon-rx-5600-xt-6g-gddr6) appears among the best.
* I've not decided yet how to partition the Sabrent nvme disk. Will I need Windows ? especially to ding into the Ryzen tools. Don't have an idea how to _clone_ correctly from previous SSD, the W10 part and it's boot sector.
I'm using Windows less than 10 times in a year. This OS became time consuming. Booting it once a while from a USB thumb key will be enough for me.
If such boot is feasible ?
* I'll post back some Sabrent benchmark using the `hdparm` speed test...

olejon, 2020-06-12:
Well I have Windows on a separate disk that was in my monster laptop (Windows accepted the total HW change). The only "annoying" thing is that I can't change it to a UEFI boot partition scheme, hence being able to set the BIOS to only allow detecting and booting UEFI bootable devices. Must now also have "Legacy OPROM".

Though, *since Windows 10 is activated through your Microsoft Account now* (make sure it is, if not convert it to it), I should be able to totally reinstall Windows 10 by downloading the Insider Program bootable ISO and install that and reactivate Windows without problems.

The same goes for you. You can always opt-out of receiving Insider Program updates, or choose the Slow Track, *although the Fast Track has only been a benefit for me. No bugs, no instability*. Just updates more frequently, something you want on newer HW.

Just register here with your Microsoft Account and you'll get a bootable ISO:

https://insider.windows.com/en-us/

Would definitely install Windows first (*without Secure Boot enabled at install*), then Linux. A user friendly distro should tap into UEFI just fine (or follow guides).

Since you use Windows so little, can't you just backup whatever is in you `C:\Users` folder (from say Linux, no cloning needee) and put it back again? Reinstalling programs shouldn't be a big deal?

IDK `hdparm` can't measure my NVMEs at all. GNOME Disks can. But not write speed when mounted, and not from Live USB either as the USB doesn't write fast enough. And Windows tools (for God's sake use ATTO Benchmark, only reliable) *can only check if there's an NTFS partition on them*.

Can always make some free space at the end, but well, Windows doesn't live on it so why... I trust GNOME Disks. It shows specification speeds, a little higher often (after BIOS + firmware updates, latter only possible from Windows...).

As for GPU choice can't help you more than buy one with a return policy that allows testing it as long as original packaging is not broken, test it and return if not satisfied. That's what I would've done.

olejon, 2020-06-12:
**EDIT:** Secure boot disabled at install of Windows.

olejon, 2020-06-12:
BTW the EFI partition should be the first partition on Linux and set to be mounted as `/boot/efi`. I just set to to 1 GB as per some guides, but Linux uses only 6 MB on it. Better to just let Windows set it up. Uses FAT32 with these boot flags:

```
# parted /dev/nvme1n1 print
Model: Force MP600 (nvme)
Disk /dev/nvme1n1: 1000GB
Sector size (logical/physical): 512B/512B
Partition Table: gpt
Disk Flags: 

Number  Start   End     Size    File system  Name  Flags
 1      1049kB  1075MB  1074MB  fat32              boot, esp
 2      1075MB  1000GB  999GB   ext4
```

It's a hazzle to set up afterwards, which I did. Had to use some fancy `fdisk` stuff to fix the ext4 partition. Windows doesn't let you add free space at beginning, but GParted does allow shrinking the ext4 from there and it worked just fine. Just on 1 of my PCs did I have to do some magic from Live USB to make it boot again. Think it got messed up when setting the specific EFI GPT FAT32 **partition table** to to `loop` which is what Ubuntu does at least.

```
# parted /dev/nvme1n1p1 print                                  Model: NVMe Device (nvme)
Disk /dev/nvme1n1p1: 1074MB
Sector size (logical/physical): 512B/512B
Partition Table: loop
Disk Flags: 

Number  Start  End     Size    File system  Flags
 1      0,00B  1074MB  1074MB  fat32
```

olejon, 2020-06-12:
Some edits...

cyring, 2020-06-13:
Thanks for all your tips
I have found this [tutorial](https://lecrabeinfo.net/creer-une-cle-usb-dinstallation-de-windows-10.html), in French unfortunately, to use the Windows Media Creation Tool to an 8GB USB bootable flash storage.

I will give a chance to a multiboot EFI/GPT partition.
I'm back in a while. 
Cheers

olejon, 2020-06-13:
It's easy. You go here and get the Insider Preview.

* If you're not already an insider, become one, it's free: https://insider.windows.com/en-us/

* https://www.microsoft.com/en-us/software-download/windowsinsiderpreviewadvanced?wa=wsignin1.0

* Sign in clicking the upper right User Icon

* Scroll down to **Select edition**

* Choose **First in List**

* Confirm

* Choose language

* You get links for 64-bit and 32-bit

* FAQ with links to Flashing: https://www.microsoft.com/en-us/software-download/faq

olejon, 2020-06-13:
Read above first.

**Edit:** Maybe installing Ubuntu 20.04 after Windows, which should set up everything correctly, then look at those partitions and mount points, and use those partitions for Arch Linux :-) which I suppose you still use. Just don't delete Windows EFI stuff on the EFI partitions. Though I assume Arch Wiki has a good page on this.

olejon, 2020-06-13:
And again, above-above and above. On Ubuntu it isntalls these packages, **key is the GRUB EFI packages** not the `grub-pc-x` packages.

Just this package pulls in the necessary, IIRC, as the dependencies pulls in all, like `efibootmgr` (indirectly) and `mokutil, sbsigntool ` directly and they GRUB2 EFI packages.

```
# apt show shim-signed

Package: shim-signed
Version: 1.37~18.04.3+15+1533136590.3beb971-0ubuntu1
Built-Using: shim (= 15+1533136590.3beb971-0ubuntu1)
Priority: optional
Section: utils
Source: shim-signed (1.37~18.04.3)
Origin: Ubuntu
Maintainer: Steve Langasek <steve.langasek@ubuntu.com>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 1 388 kB
Depends: debconf (>= 0.5) | debconf-2.0, shim (= 15+1533136590.3beb971-0ubuntu1), grub-efi-amd64-signed, grub2-common (>= 2.02-2ubuntu8.1), mokutil, sbsigntool
Recommends: secureboot-db
Breaks: grub-efi-amd64-signed (<< 1.93.7)
Task: ubuntu-core
Supported: 5y
Download-Size: 344 kB
APT-Manual-Installed: yes
APT-Sources: http://archive.ubuntu.com/ubuntu bionic-updates/main amd64 PackagesDescription: Secure Boot chain-loading bootloader (Microsoft-signed binary)
 This package provides a minimalist boot loader which allows verifying
 signatures of other UEFI binaries against either the Secure Boot DB/DBX or
 against a built-in signature database.  Its purpose is to allow a small,
 infrequently-changing binary to be signed by the UEFI CA, while allowing
 an OS distributor to revision their main bootloader independently of the CA.
 .
 This package contains the version of the bootloader binary signed by the
 Microsoft UEFI CA.
```

**NOTE:** Made for Secure Boot ON yes, *but it DOESN'T have to be on*! So why not? With everything in-kernel it can be on just fine, but say NVIDIA's driver must be signed.

Latest Ubuntu 20.04 let's you install it on install! Before Ubuntu had a solution, asks for password, just choose 12345678, reboots into UEFI and sign with that, reboot and done. Should also work with the CoreFreq module signing that way if in correct directory (dkms...)

cyring, 2020-06-14:
Thanks for all these tips.
I have a Windows 7 professional license upgraded to Windows 10 Pro. Installation is done onto a GPT 1GB EFI plus 192GB Windows data.
Now, using my _CoreFreq_ live image which embeddes the Arch Linux setup, I will follow directions of [systemd-boot](https://wiki.archlinux.org/index.php/Systemd-boot) wiki

olejon, 2020-06-14:
After playing a little with Ubuntu 20.04 using the Live USB, seems like this Ars Technica article is right on: [Ubuntu 20.04: Welcome to the future](https://arstechnica.com/information-technology/2020/05/ubuntu-20-04-welcome-to-the-future-linux-lts-disciples/)

Just switching to the dark theme... Much nicer than macOS, at least when macOS doesn't run on a very PPI monitor. Seemed to work fine in programs like LibreOffice, VLC, AudaCity etc, you know programs using a good mix of toolkits. On the fly.

All the theming changes, all the collaboration + contributions with GNOME and Debian devs. So it's no more Ubuntu-stuff on top of things. More Ubuntu things adopted by GNOME. Already seen in Fedora but... Like the Control Center and Nautilus. Remember when it looked like macOS. Still trying *too touch friendly*. It still is but just less need for `gnome-tweaks` (none for regular users, just like a macOS user would need TinkerTool).

Maybe it'll be my first time using the default Ubuntu desktop. In fact it probably will. Wallpaper and terminal theming change. About it. With all the theming. Heck even as planned but also as requested by HP, Dell and Lenovo Ubuntu now shows its splash screen loader during the POST logo, just like Windows does.

That is, if installing on a laptop sold with Ubuntu as an option. Probably can be activated for any PC, but my workstation boots so fast I can't see anything before suddenly it's time for login :-)

All of this theming is FOSS of course. Only reason to mess around with it becomes to "stand out and not have a little different desktop than others". For GNOME Shell there's only 2 extensions as I could see. Dock and legacy tray icons (IIRC deactivated by default). Hiding dock and you got Vanilla GNOME basically... Which is the metapackage I use.

They finally having an agreement with NVIDIA to install that driver at installation of the OS is nice for people with an NVIDIA GPU.

Hotswap of the kernel seems even easier. Now Ubuntu requires a Ubuntu SSO account for this. But you know it's possible, and has been, for a long time, without. It just makes it so dead easy and stable that first time users are encouraged to use it. People used to Windows must be like "WTH do I need to reboot this system, really?" :wink: 

And Software Center (GNOME Software but any compatible can use the underlying as you probably know). Well I prefer to install everything through APT. Maybe a stupid obsession as snaps are way more secure (Flatpak also supported, which comes with Fedora and snapd one click install). Even more fine grained permissions model than say Android.

The little I've seen makes it dead easy to get your program to the Software Center. If rejected, help to fix it. Usually devs adding access to files/folders the program doesn't need. Like the entire `~/.config` when only needing `~./config/<program>`. Even a lawyer chimes in if in doubt about naming/copyright.

Like the entire Microsoft Visual Studio for Linux is one click away. So is Spotify, Steam, Skype...

Just need more OEMs selling laptops with Ubuntu. This might be a first step.

BTW yeah the "What's New" celebrates the spin-offs too like Kubuntu, Ubuntu MATE etc.

Not that Arch Linux (and many others) isn't great, just won't see *that* on HP/Dell/Lenovos :P Like PopOS is already sold preinstalled, just not the big ones, yet.

---

Interesting.

According to this it works "A-OK" (I take it as it works A = Perfect) on Ubuntu 20.04 too (guide made for 18.04). Stack Overflow question says so.

https://blobfolio.com/2018/06/replace-grub2-with-systemd-boot-on-ubuntu-18-04/

Maybe I'll try it... For what? Faster boot? I never boot and if so never see GRUB anyways. As usual probably wait until the first point release, or even the second which comes with the HWE stack by default (that is when 20.10 is released + 3 months).

Login says:

```
Your Hardware Enablement Stack (HWE) is supported until April 2023.
```

Ubuntu 20.04 increases this to 2025 of course.

---

BTW for Windows, IDK about your motherboard's drivers listed on the website, but DON'T install the AMD chipset one that includes the graphics driver, unless that's the only chipset package there is. ASUS has one without. Just messes up things when you install the latest from AMD.com, where you should go for the "Optional" (latest) Adrenalin driver package, if you have an AMD GPU (if not probably best to let Windows 10 install them, that is if it actually has them all in Windows Update, but on Fast Track I'm pretty sure).

And always choose Advanced/Extra options during install and choose a "Hard install" (something like that). Completely uninstalls the previous. Not doing that I've seen an error various times when finished. Although works afterwards, really hard to see if it got installed if not comparing Windows Device Manager version before and after. The Radeon Software is pretty great. It can check for updates for you, just make sure to set it to check for both Standard + Optional.

olejon, 2020-06-14:
Would be really cool to see your 3950X results from Blender Benchmark (CPU). For comparison. Stresses my system more than Conic Compute.

Linux wins with a whopping 1 minute and 3 seconds on classroom. Seems to be ~9.3% win on average.

Just a launcher you can place anywhere. Data it downloads is simply placed in `~/.cache/` as it should... At first launch it downloads the necessary parts of Blender (use latest suggested version), and well, I just use the bmw + classroom renders.

### Linux

[image]

### Windows

[image]

cyring, 2020-06-14:
I will send you blender results when setting Arch is up.
So far, I have reinstall Win10Pro to "force" a GPT partitions table. I guess you warned me about this.

So far, I have installed systemd-boot with one Arch entry. Choice of UEFI entries is off course made straight from the fancy ASUS firmware. I plan to add one entry per kernel I'm working with.

Next, I have to set up all the development lab... 

Meanwhile I have played with Ryzen Master which is a powerful software but its CPU overhead is about 200 MHz, whatever the Windows performance policy is (balanced, high, saving, policies made by Microsoft or AMD)
Booting the _CoreFreq_ live shows a CPU overhead less than 10 MHz, but also less watts, voltage, lower Core temperature.
Interesting results.

P-States are working, no crash or locks, no TSC issues. Only some compute issues due to the fact that I don't have the registers to recall the original P-States values, once changed. 

For the next feature, I wish to measure the c-states, if general counters can be programmed for this; and specifications available.

cyring, 2020-06-14:
Btw, this issue being solved, feel free to close it, but keep it private for our tchat.
Regards 
Cyril

olejon, 2020-06-14:
Yup. Same here. UEFI shows simply `Ubuntu (Corsair + Model)`.

Since my Windows lives on a separate SATA SSD, I don't bother to convert it from Legacy. Must just set UEFI to, for internal disks, detect "UEFI + Legacy OPROM", and press F8 when booting and choose that disk.

Little annoyed at first when `update-grub`, which is run when new a new kernel version is installed detected Windows as well... My temporary solution of unplugging the SATA cable wasn't sustainable :P

Fixed setting in `/etc/default/grub`:

`GRUB_DISABLE_OS_PROBER=true`

So it doesn't check for other OS'

closed 2020-06-14 by olejon

cyring, 2020-06-14:
* Hardware done
[image]

Intel, left - AMD, right

cyring, 2020-06-14:
_CoreFreq_ build ready

As Ryzen Master says I have stressed the best Cores which are CPU number 0 and its SMT 16

[image]
[image]
[image]
[image]
[image]

Indeed best Cores can reach 4.7 GHz, not the other CCX

cyring, 2020-06-14:
I saw your Blender score, pushed the button; and said to myself: let's have a coffee
What a shock when  I came back !
[image]

Recommend you to try again under Linux...

olejon, 2020-06-14:
LOL. You thought it would take long? If your link sucks then the download would probably take longer :fr: 

* I suspected it would take approx. 3 times less, although not 3 times the cores. Almost right.
* What *Frequency & Temperature* do you get with Conic Compute, first choice in list?

---

* Assume you have latest BIOS
* Have you enabled PBO? It can be done in UEFI (but you must know what settings, then it'll show in Ryzen Master as `Auto Overclocking` profile == Enabled) or just use Ryzen Master, setting whatever profile to `Auto Overclocking` then `Apply & Test`, which will set it to 100 MHz (max is 200 MHz but makes no difference, no tests I've seen see benefit over 100 MHz)

---

* For me it enables in UEFI under "AMD Overclocking" these:
  * Memory overclocking: Enabled (Rest "Auto")
  * Precision Boost Override: Enabled (1 drop-down set to 100 MHz, rest "Auto")
  * That's it, does *not* change the PBO settings which are on the main AI Tweaker section of UEFI where you activate auto-memory overclock etc

olejon, 2020-06-14:
Edits...

cyring, 2020-06-14:
> LOL. But I did post Linux & Windows. You thought it would take long? If your link sucks then the download would probably take longer 
> 
> * I suspected it would take approx. 3 times less, although not 3 times the cores. Almost right.

Got familiar to wait with my Xeon
Now between the FTTH and Ryzen, I can't barely have a nap

> * What _Frequency & Temperature_ do you get with Conic Compute, first choice in list?

[image]

Apparently the Celsius 36 is doing a great cooling

> 
> * Assume you have latest BIOS
> * Have you enabled PBO? It can be done in UEFI (but you must know what settings, then it'll show in Ryzen Master as `OC Mode = Auto Overclocking`) or just use Ryzen Master, setting whatever profile to `Auto Overclocking` then `Apply & Test`, which will set it to 100 MHz (max is 200 MHz but makes no difference, no tests I've seen see benefit over 100 MHz)
> 
> * For me it enables in UEFI under "AMD Overclocking" these:
>   
>   * Memory overclocking: Enabled (Rest "Auto")
>   * Precision Boost Override: Enabled (1 drop-down set to 100 MHz, rest "Auto")
>   * That's it, does _not_ change the PBO settings which are on the main AI Tweaker section of UEFI (seems like a setting that does _nothing_)

Beside DRAM set to 3600 MHz and CL 16, all BIOS options are in AUTO mode.
RM confirms the OC mode. But I still have a lot to compare. Between the BIOS bits and the _CoreFreq_ readings... ie. setting BIOS options to [Enable] | [Disable] rather than [Auto] then query values in driver...

Here is one of the strange things I have noticed so far:
* CC6 [ON]
[image]
* CC6 [OFF]
[image]

Almost 1 MHz lost when disabling CC6

cyring, 2020-06-14:
CC6 _behavior_ confirmed by the power consumed; voltage to observe.
[image]
[image]

olejon, 2020-06-15:
**Question:** According to Ryzen Master's documentation:

> Gray and Gold stars are fastest per CCX and Gray Dots are second fastest per CCX

Here are screenshots from Ryzen Master and CoreFreq Topology. You can see CCX0 has 1 Gold Star and 1 Gray Dot, while CCX1 has 1 Gray Dot and 1 Gray Star.

**Which in the list should perform best in a CoreFreq:** `Turbo Select CPU`? I'm pretty sure `CPU 000` is the one with the Gold Star as it performs best. Then I get a little confused...

---

### Ryzen Master

[image]

### CoreFreq

[image]

cyring, 2020-06-15:
* Be aware that _CoreFreq_ numbering is zero based.
Yes, you will run the `Turbo Select CPU` function to stress the CPU number zero in _CoreFreq_ which is the best Core according to RM

* One thing I don't get is how RM find this best Core ?
At first, I would imagine RM stresses and measures each Core during its startup; but I don't notice such action on Processor while launching this software.
My guess is that some registers are storing those factory informations. 
Found nothing in specs.

olejon, 2020-06-17:
* [AnandTech: AMD Clarifies "Best Cores" vs "Preferred Cores" Discrepancies For Ryzen CPUs](https://www.anandtech.com/show/15137/amd-clarifies-best-cores-vs-preferred-cores)

* Think I got it. What RM shows is right according to MHz achieved in CoreFreq. I know it's 0-based of course. Just don't get the `CCX/ID` columns

* So say stressing 000 gives same result as stressing 006 (same core)

* RM can show when a core is sleeping. IDK if it actually is with its Windows drivers or it's just so low on MHz it says "Sleeping"

Oversized package for the 3950X, basically as those with a stock fan, but *MAN that Noctua package size!* :fearful: 

[image]

olejon, 2020-06-17:
Added one point

cyring, 2020-06-17:
* Congrats. Time for hardware build, now.

* About SMT Cores: AMD should have the micro-architecture answer. But you will have to run different load & idle cases and observe resulting Core states and sensors. 
With Nehalem, stressing one physical Core has "effects" on its SMT sibling: like voltage, temperature.

* Zen sleep states are my next duty...

olejon, 2020-06-18:
> Zen sleep states are my next duty

Nice :+1: 

Yeah I guess, no rush. I hate changing the CPU (and from now on MBs without an I/O block, really really hate the I/O bendy shields I sometimes forget to put in place until to late...).

If it wasn't for coolers it would be 10 seconds but... When my GF can help. My chassis' backplate there is much easier with 4 hands, especially *4 eyes* to see the screws come through correctly. Want it perfect the first time, after applying thermal paste and all.

I remember I did consider buying your MB, for the Wi-Fi and additional 2.5G LAN, but wasn't worth double the price. How's the Realtek working? I bought early on and remember reading some on reddit saying it worked but not exactly perfect (yet). Could crash or whatever.

Also the whole ROG, searched it to "Republic of Gamers", and "HERO" RGB put me off :laughing: instead of my subtle one, the white I/O block on mine makes it kind of look integrated into the chassis too.

Later we can compare benchs, and see who got luckiest with the silicon :wink:

cyring, 2020-06-18:
I have my components choices with the help of those awesome lists:

https://docs.google.com/spreadsheets/d/1wmsTYK9Z3-jUX5LGRoFnsZYZiW1pfiDZnKCjaXyzd1o/edit?usp=drivesdk

https://docs.google.com/spreadsheets/d/1c3RTF_ZPjep-Zfimgoca2Ef1gSjZM0rSHVWLkknbfUI/edit?usp=drivesdk

https://docs.google.com/spreadsheets/d/1dsu9K1Nt_7apHBdiy0MWVPcYjf6nOlr9CtkkfN78tSo/edit?usp=drivesdk

https://docs.google.com/spreadsheets/d/18aMZfSQFkVUQWjVt6RSFDh866D1wZlWAxSF7mAHWkbA/edit?usp=drivesdk

https://docs.google.com/spreadsheets/d/1Smj5dh97n32wJqm5dkdDcQt8ID7vH52-lKzaaXUUQx8/edit?usp=drivesdk

At first, I was looking for Pro components, like an EPYC, Threadripper, second; but Home-server components , for a Workstation form factor, was not so available. Cooling is also a matter of worrying. Thus gaming components, but with the best vrm motherboard, capable for the highest cores count among mainstream processor.

So far I'm not considering the ROG devices: network, wifi, BT, sound are just doing their job. Ryzen is my focus. ASUS BIOS has features I can compare _CoreFreq_ with.
This PC has to last a long time. Previous ROG Gene II, upgraded with W3690, is still alive 10 years after. I thrust this brand. Only 3rd MB was RMA and fixed by Mfr. 

Diving into Zen datasheets again, I don't find specifications of idle states I can program the PMC counters with.

olejon, 2020-06-18:
Oh, that's indeed some nice Docs.

Yeah the for the `vrm` cooling on your MB is probably better. Took a look and they seem pretty good protected.

---

## Any big difference? Seems to be the top ones under heatsink and the side ones on both next to the side heatsink (and 2 next to RAM?)

Images comparison.

### ASUS PRIME X570-PRO

[image]

---

### ASUS ROG Crosshair VIII Hero

[image]

---

According to [This doc - found on Reddit but one of yours](https://docs.google.com/spreadsheets/d/1d9_E3h8bLp-TXr-0zTJFqqVxdCR9daIVNyMatydkpFA/htmlview#gid=639584818)

* My `Prime X570-Pro` ticks all the Green Boxes. But not sure how to interpret under `Vcore VRM`:
  * `Prime X570-Pro` = Phases 4 - 3x SiC639 50A Power Stage
  * `ROG Crosshair VIII Hero [Wi-Fi]` = Phases 7 - 2x IR3555 60A Power Stage

---

Just saw in Reddit (which has the Docs link), one reply to using the 3950X on a cheap MB **that is no longer sold - here at least**.

> If he doesn't manually OC (stupid thing anyway with these CPU's) the worst that will happen under heavy load is the CPU won't clock as high due to insufficient VRM power (cooling). At least that's the way I understand these processors work.

olejon, 2020-06-19:
* Nice when everything works the firs time and BIOS says "New CPU Installed, press F1 for Setup"

* As always with new heatsinks, difficult to fasten the first time, had to take it off and lay the chassis down, hence fearing the Noctua thermal paste would need cleaning, but I had put it on perfect so it covers the surfaces perfectly, not too much, not too little

* Often the packages are way bigger than the content, like the 3950X is the size of a big stamp in a standard box for coolers, but the Noctua heatsink - *wow* - I can't see my nice subtle I/O block's RGB anymore, looks like there's an engine in there now... but RAM and PCH fan lights, yes (just color pulse)...

* And I haven't put on the second fan, with just the one in the center, the system struggles to reach even 70 Celcius (have to use Blender Benchmark and can probably avoid it by ramping up the CPU fan to 100% at a lower temperature), also the ambient temperature, now it's 30 Celcius outside and inside, so... Just the one fan works excellent, and *so* much quieter than the stock for the 3600X, and the idle sound, as it should be for that size, could ramp it up to 40% duty cycle at idle, no difference, and not using the noise-lowering cables that come with the package...

* I think the second fan will crash with my RAM chips, any RAM chips on an X570 ATX MB, unless you mount it vertically (really just 2 mount options because apparantly there are 2 types of backplates) which is possible, but then air will just be dragged in from nothing (top) and down to the GPU...

[image]

---

This is the best comparison IMO, against an Intel i9 Xeon with 18C/36T (more):

[image]

cyring, 2020-06-20:
Definitely it looks like a car engine. Can't way to see your Conic temperature.
Remember, I got 70C with the Celsius S36, not at its highest fans rotation.

My HTPC is using a low profile Noctua nh-c12p to cool the i7-920 processor. Yesterday, removing dust, I changed TIM with MX-4 which resulted to temperature as low as with Noctua TIM.
Off course I'm using a TIM solution cleaner for both, processor and heat-sink, before applying a "big" drop of TIM, but it becomes difficult to find a cooling benefit among the TIM top 10

olejon, 2020-06-20:
* Lol, and definitely is a **good one** too
* *Remember I just have the middle fan installed*, but nice to have a spare one then. Doubt it would've made much difference considering temperatures below and previous experience with this chassis' airflow
* IDK which Chassis + X570 MB is made so that one can use both without clashing with the RAM chips.
* SSH screenshots from my phone:

### Idle Temperature - This is just 4 degrees above Room Temperature!

[image]

### Conic Temperature - Approx 2 minutes - It never goes above 70

[image]

---

* **Note:** When running Blender Benchmark all cores stay at almost 4000-4100 MHz, seems like it gets more out of the CPU than Conic *despite stressing more, see next*
* Blender Benchmark manages to get it to some 71-72 degrees, *and cores run at higher MHz than seen in the screenshots above, so as before Blender Benchmark is the best stresstest for temperature and MHz*
* This Noctua 140mm fan is so quiet and nice I can set it to a higher duty cycle when idle and also to gear up 5 degrees before the chassis fans do at 70 degrees, so it's very silent running Conic
* Ryzen Master shows % of 142 Watts, but with the 3950X CoreFreq actually shows a significantly *higher Package Watts than the TDP* of 105 Watts, *when stressed* (logical?)
* This wasn't the case with my 3600X with a TDP of 95 Watts, where Ryzen Master showed % of 128 Watts, and CoreFreq Package Watts stayed at around 87-90 Watts *when stressed* (seemed logical)

---

### More benchmarks and comparisons here:

https://www.olejon.net/files/X570-3950X/

cyring, 2020-06-20:
* Around 70C is a good result, especially if fan stays quiet
* Can you reach 4.7 GHz single Core ?
* You Blender result is 10 seconds more than mine in `classroom` (worst w/ Windows): do you have CPB disabled in BIOS ? or DRAM speed and latency matter ...

* Nice Linux Software of the day
[image]
[OpenRGB](https://gitlab.com/CalcProgrammer1/OpenRGB) works fine Aura

olejon, 2020-06-20:
* Yes I get a little above 4.7 GHz on "gold star cores"
* Can't remember having touched CPB, will check tomorrow
* Yeah, your faster RAM came to mind, it definitely has an impact and with 3600 MHz one can achieve even better performance not using the default Infinity 50/50 (1800/1800 MHz), *but also the newer kernel you run came to mind*... Can check with a live USB easily later if there's gains there, if so Ubuntu 20.04 upgrade probably sooner, BUT not that it should noticable outside benchmarks :wink: Got a graphics stack update recently and made the radeon driver even slightly better, small things also like I don't have a splash screen of course since system boots so fast, and a tiny flicker when GDM is loaded just disappeared with that update, no kernel, just X.org, drivers and Wayland...
* Nice. As I bought early, didn't bother to search, but knew it either existed already but probably buggy, but knew it would come. *Like...:*
* Was pretty surprised when I found [this for Razer peripherals](https://openrazer.github.io/), just driver and daemon and let others make nice GUIs or just CLI using it. Totally works but my keyboard only has backlit keys with one color (that's why I bought it, good keys, backlit and no RGB fancy), so can be used to change the level *which of course can be done by Fn + Keys anyways* - also the just annoying "Fade in/out" effect can be applied, and speed. Can look cool when idle but absolutely not when using it. But bet it's nice for other keyboards with more features. Razer's own Synapse software is *worse* IMO, especially on macOS when I used my MacBook as a workstation temporarily living in Spain

cyring, 2020-06-20:
> * Ryzen Master shows % of 142 Watts, but with the 3950X CoreFreq actually shows a significantly _higher Package Watts than the TDP_ of 105 Watts, _when stressed_ (logical?)
> * This wasn't the case [...]

* Please check this [AnandTech 3950X review](https://www.anandtech.com/show/15043/the-amd-ryzen-9-3950x-review-16-cores-on-7nm-with-pcie-40/2)
What it says:
> At its peak, the 3950X draws 137 W for the cores when 10 cores are loaded. The chip as a whole hits ~144-145W at that level, well above the 105 W TDP rating on the box ...

* Now _CoreFreq_ power estimation; stressing Processor with `Conic > Hyperboloid of two sheets`
Look at the whole Package measurement. 
Same conclusion ?
[image]

olejon, 2020-06-20:
* Checked the CBP (Core Boost Performance I assume) and it was `"Auto" == Enabled`. Setting that to `Disabled` obviously made it slower
* Set 2 more under `AMD Overclocking > DDR`. I already have disabled `Gear Down == Disabled` (remember that from YouTube video being something that could give make benchs/games perform *a little better, BUT not CoreFreq stressing`
* Did get about 135 Watts so also set some DDR `Power Down/Saving == Disabled` but think it's the `SoC Uncore` which makes the big difference here, *just look* :open_mouth: : 

[image]

---

* If you think this is "cheating", well, not with this Noctua fan. Reason I bought it instead of the slimmer also just for AM4 is because it has the logo for the first image above for the 3950X, so my fan is listed as `best turbo/overclocking headroom`
* Yes it gets hotter as you can see, and my 3600X didn't like this with the stock fan, obviously. In fact the PC suddenly rebooted a couple of times when stressed, but not always... :thinking: 
* As you probably know AMD considers 75°C as "high" as per `sensors` but that means little, Intel used 85°C. To reach the "hysterical" point AMD has 95°C before throttling or shutting down, and Intel 100°C (Intel always gets hotter)

[image]

olejon, 2020-06-20:
Some edits...

olejon, 2020-06-20:
Some more edits...

cyring, 2020-06-21:
More than 195W is scarry and with 82°C, I wonder when thermal throatling is going to engage ?
I'm gonna review the driver to see if I had set a thermal sticky bit into code, which translates into `HOT` being highlighted.

olejon, 2020-06-21:
* First of all it's an official *Enabled/Disabled* setting in UEFI and the Help Text simply says something like "*Can boost performance at the cost of power saving*"
* But when Idle, Temperature and Watts = normal
* Also my 3600X *with the stock cooler* often got above 75°C, which AMD considers "High" for all these CPUs, and I've never had problems with the temperature being about what Intel (85°C) or AMD consider "High"
* A MacBook Air, many other laptops and iMacs too with Intel reach close to 100°C ("Hysterical") quickly with just some moderate multitasking, as easily seen as they start to throttle. Then CoreFreq correctly for my MacBook Air with Haswell reports 3 `HOT`, but there's no way to avoid that and they're built that way, and BIOS set to throttle instead of shut down for CPU protection
* When my workstation ran Skylake i5 I set all cores to Turbo and overclocked all so the CPU reached 95°C under stress without problems, just used a sensor tool to simply *reboot the PC* if I wasn't using it, and for some reason say some process went crazy and it reached 96°C (never happened but tested and worked fine)
* I have only read that the "High" value is a temperature you generally want to stay below but it definitely can go higher in periods, just prolonged close to "Hysterical" can damage it over time, but reaching "Hysterical" causes the PC to shut off (or throttle if that's set in BIOS)
* If any Intel laptop goes 85°C+ often it can't be a problem for an AMD desktop CPU to go 82°C for a few minutes? Only benchmarks causes this
* And regarding benchmarks, Blender Benchmark did *NOT* cause an even higher temperature *BUT* 203 Watts. Temperature actually floated at a slightly lower one than Conic IIRC
* Now, Watts is another thing. Blender Benchmark showing 203 Watts, wow. *My PSU is just 500 Watts so kind of surprised*, so:
* **Must check Ryzen Master if CoreFreq's reading is right**
* Also must check if Ryzen Master gives any warning when running Blender Benchmark on Windows
* BTW with this I got exactly 1 second less on *both your results* from Blender Benchmark :smiley: 
* Will read a bit more about this option before stressing the system again, *but it's nowhere close to this at regular use*, it is now at 14 Watts and 35°C as ambient is 29°C

cyring, 2020-06-21:
Thanks for your temperature calibration effort. Can't wait to see your results.

Meanwhile I've checked code and driver has no ThermalTrip query in families 17h & 18h.
Thus no `HOT` will appear with Zen processor.
Perhaps kernel is logging it. 
I will search for such thermal status register which may be queried onto SMU

olejon, 2020-06-21:
Many bullet points. Please read them all.

* You made me curious although I haven't slept yet :stuck_out_tongue: 
* First, correction, according to `k10temp` "HIGH" is 70°C, **BUT forget that because:**
* *Have to look at Ryzen Master*, which only lists 1 temperature and that's called **`Limit 95°C`**, as said before. Goes for all the Gen 3 CPUs
* I ran a full Blender Benchmark `classroom` and the *system was totally stable*
* Ryzen Master *did not even reach Orange color for temperature or any other*
* I am almost 100% sure I saw the Orange color on my previous 3600X when stressing that
* In fact I *manually* overclocked my 3600X to Ratio 41 (4100 MHz), just 1 above standard 40, and when stressed it didn't like that, temperature rose and the system simply shut off, *and I could without problems boot into UEFI and revert that setting to default*. Same with my Skylake i5
* Temperature towards the end was at 85°C, *but still Yellow color and NO warning*
* As said, on Linux that temperature is *lower* during the same Blender Benchmark run
* Ryzen Master now shows `395W`, and 50% of that is `197.5W` as in Screenshot 2
* **Ryzen Master Screenshots when Idle (almost, I did some things) and During Blender Benchmark (also see points below these):**

### Almost Idle

[image]

### During Blender Benchmark

[image]

---

* Absolutely no warnings from the kernel when pushed up above 80°C. I checked with `dmesg`, `journalctl`, `sensors` (which shows an alarm/alert for certain HW) and used `grep` on ALL files in `/var/log` and there was nothing matching any relevant keyword
* The last message was simply that CoreFreq was loaded, others were standard stuff from server services and desktop environment
* Blender Benchmark `classroom` finished faster on Windows as well, but result isn't valid as I had Ryzen Master open + I took a screenshot during the run
* Blender Benchmark is very sensitive to any other stuff going on (system should be totally idle), and it stresses both Windows and Linux so much for me that the desktop lags a bit and is a bit slower to open say the file manager
* The below says *"... expense of idle power savings"*, which may be true for the RAM but I don't see the CPU being more active when idle (and there's no integrated graphics as you know so that doesn't count)
* **Screenshot of the Name and Description of the Setting in UEFI. I see nothing that screams Danger! Danger! Do you?**

### Both Red Rectangles say the same

[image]

cyring, 2020-06-21:
Very interesting investigation you did, although I will go very slowly on those settings t'il we have a clear understanding on what is going on, and their side effects. I'm still not sure if hardware is well protected against over-voltage, over-consumption and so on.
Don't fry your brand new Ryzen !

* This morning I went through the last PPR, this document revision applies to model 20h which concerns APU to my understanding.
What we found is a list of registers not unveiled in previous families. Most of them can be queried through wire protocols: SM-Bus, i2C, GPIO, etc
I bet AMD has some MSR to ease the readings but those are not public yet.

We find wanted registers a bit every-where, sometimes through :
- ProcHot, ThermalTrip : on SMIstatus,  SmiControl5 & Power Management (PM) Registers
- C1e : `PMx07E [CStateEn]`
- IdleCount (aka "idle count from the latest monitored period") at `MISCx060 [IdleCntrl] (FCH::MISC::IdleCntrl)`. Is this what RM called Sleep from ?

* Have a rest, you deserve it !

[110 more replies]