cyring/CoreFreq #190: UI improvement / clarity

olejon, 2020-06-25:
**Most important and easy fix. Under `View > Power` or `W` shortcut at the bottom:**

* `Cores` value says `X.YYYY(W)` making it clear it measures Watts
* BUT `Package` only says `X.YYYY` *without the* the `(W)`
* And remember `Package` comes before `Cores` in the UI, and most people are interested in `Package` as it's the value read by say Ryzen Master and similar tools

---

Also, to get at least some information in `Window > Memory Controller` or `M` shortcut *on X570 + Ryzen 3*, this command does at least shows correct data for my RAM chips:

* **NOTE:** *Questions comes below all*
* `# dmidecode -t memory`
* It shows data for the 2 slots I have RAM chips in, and just `Unknown` for slots without, like comparing RAM slot with/without a RAM chip:
* **NO RAM chip:** `Form Factor: Unknown`
* **RAM chip:** `Form Factor: DIMM`

* So for a slot with a RAM chip these values read correctly, *EXCEPT voltage, BUT that goes for various Windows tools too except Ryzen Master, which correctly shows `1.35V`*

```
Handle 0x003F, DMI type 17, 84 bytes
Memory Device
Array Handle: 0x0035
Error Information Handle: 0x003E
Total Width: 64 bits
Data Width: 64 bits
Size: 16384 MB
Form Factor: DIMM
Set: None
Locator: DIMM_A2
Bank Locator: BANK 1
Type: DDR4
Type Detail: Synchronous Unbuffered (Unregistered)
Speed: 3200 MT/s
Manufacturer: Corsair
Serial Number: 00000000
Asset Tag: Not Specified
Part Number: CMR32GX4M2C3200C16  
Rank: 2
Configured Clock Speed: 3200 MT/s
Minimum Voltage: 1.2 V
Maximum Voltage: 1.2 V
Configured Voltage: 1.2 V
```

**QUESTION:** Possible to at least use code from `dmidecode` - *I assume the code is easy to find and can be incorporated into CoreFrew without licensing problems* - to at least show, as a temporary workaround until you maybe find a way to read these straight from the arch:

* RAM size per slot
* RAM speed per slot *in MHz instead of MT/s*
* (Manufacturer)
* (Part Number)
* Because remember, the more values *people typically look for* CoreFreq can show, the better
* YES I know `dmidecode` is not perfect. You can add e.g. a `EXPERIMENTAL AND NOT TO BE TRUSTED` label, simply

```
# dmidecode 3.1
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.
# SMBIOS implementations newer than version 3.1.1 are not
# fully supported by this version of dmidecode.
```

olejon, 2020-06-25:
Some edits :arrow_up:

cyring, 2020-06-25:
Thanks a lot @olejon for this review

True, UI needs improvements. I will make changes in the meantime I'm progressing on Zen ...

> BUT Package only says X.YYYY without the (W)

Yeah, it's not obvious that the unit `(W)`, applying to `RAM: Uncore: Package: Cores:`, is part of the last trailing characters.
In fact, if one press `$`, it will change for Joule `(J)`

> QUESTION: Possible to at least use code from `dmidecode`

DRAM values from DMI (in fact SMBios) are SPD or XMP values (to my knowledge)
What you read from SPD is not necessary what you get after RAM _bootstrap_

_CoreFreq_ brings something else: the **configured** DRAM values.
Those left in Memory Controller by your wills when setting DRAM timings within BIOS.

This is feasible with Intel and previous AMD because registers are specified.
I have hope to find those registers for Zen...

olejon, 2020-06-25:
* Thanks. Think everyone wants the measurement value. And yes, only `Cores` change to `(J)`...
* For Memory Controller I'm sure you WON'T find timings in the dmidecode code. Just showing RAM chip size and RAM chip speed in MHz *would be better than nothing*
* Although you're deep into registers, most users simply want to see frequencies, temperature, power consumption and system info like RAM info
* Like try free [HWMonitor](https://www.cpuid.com/softwares/hwmonitor.html), which most people use, can even read RAM temperature, NVMe temperatures even without Corsair software installed (GNOME Disks can't yet but can for all SATA disks), for Intel it's complete but not accurate for CPU for Ryzen 3, so Ryzen Master for that (seems like everyone uses RM anyways), voltages, accurate CPU core speeds at least (those shown when `!` in CoreFreq) and maybe power consumption is correct too *much much more*, just temperature don't correspond, *unless stressed*, then suddenly one CPU temperature value is correct

olejon, 2020-06-25:
:arrow_up: first.

> CoreFreq brings something else: the configured DRAM values

* At least `dmidecode` shows 3200 MHz, which other tools don't, not even many Windows tools, so it reads something right...

cyring, 2020-06-25:
* So trying `dmidecode -t memory`
Significant items of last DIMM show:
```
Memory Device
        ...
        Total Width: 64 bits
        Data Width: 64 bits
        Size: 16384 MB
        Form Factor: DIMM
        Set: None
        Locator: DIMM_B2
        Bank Locator: BANK 3
        Type: DDR4
        Type Detail: Synchronous Unbuffered (Unregistered)
        Speed: 3600 MT/s
        Manufacturer: G-Skill
        Serial Number: 00000000
        Asset Tag: Not Specified
        Part Number: F4-3600C16-16GTZN
        Rank: 2
        Configured Memory Speed: 3600 MT/s
        Minimum Voltage: 1.2 V
        Maximum Voltage: 1.2 V
        Configured Voltage: 1.2 V
        Memory Technology: DRAM
        Memory Operating Mode Capability: Volatile memory
        Firmware Version: Unknown
        ...
        Volatile Size: 16 GB
        Cache Size: None
        Logical Size: None
```
Most values are good, others not: my DRAM is for instance set in BIOS with a voltage of `1.35`
dmidecode through i2c scans identifiers of DIMM which are then looked up into a database.

In the past, I have already sandbox this algorithm and It requires to embed and maintain a devices database. Not the direction of _CoreFreq_
The other values are strings obtain in SMBios memory.
See my previous project [CoreMod](https://github.com/cyring/CoreMod) and its SMBios memory map
[image]

The _danger_ of this DMI/SMBios way is that values can be badly or not enough populated by the BIOS motherboard. (remind you the ACPI issues). Also DMI, depending of its version, will make you program several parsing conditions, and some can be non compliant. I hate that.

The benefits of the Processor's Memory Controller is that there's only _one_ and fixed way to get data, usually through a PCI base address and its registers.
Look at the corefredq.c, each controller just need less than 300 lines of aggregation code.
And most IMC decoding benefits to various architecture families

Please AMD, release the Zen BKDG !

olejon, 2020-06-25:
* Yeah I already wrote the Voltage is wrong
* CoreFreq *did not show correct speed on my Intel Skylake i5 with XMP enabled* (3200 MHz + 1.35V and 16-18-36). It simply showed default DDR frequency, as if no XMP enabled, IIRC, at least no Linux tool showed 3200 MHz, but then I didn't know about the `dmidecode` command, which probably would've showed the correct one
* Gotta check my NUC later, pretty sure it's incorrect there too

cyring, 2020-06-25:
DRAM calculator annouces in its changelog of 1.7.0
> Added functionality to read current memory timings for Zen 2 (AM4).

Last time I asked them for info, they said they were just using tables ?

cyring, 2020-06-25:
> * Yeah I already wrote the Voltage is wrong
> * CoreFreq _did not show correct speed on my Intel Skylake i5 with XMP enabled_ (3200 MHz + 1.35V and 16-18-36). It simply showed default DDR frequency, [...]

Definitely we have to debug those issues. 
Usually when OC, bits are out of specs. It means we have to dump your Skylake IMC registers to debug which value is not handled by the bit mapping. Then we can add new conditions for frequency association.

Voltage is a different issue. If you are talking about a wrong Vcore, it means that my formula (VID to voltage) is wrong. But we change it, this will impact all the other processors of same family whom Vcore is good.
So, what is the architectural exception your processor belongs to ?

olejon, 2020-06-25:
Unfortinately that PC is packed away in pieces in a box. But at least I gave you the data for this:

https://gist.github.com/cyring/63aa0c84aec874a0e2f1c4782d338c4a

Which has this info. This was with RAM in *3200 MHz, 1.35V and 16-18-36 (XMP auto)*.

```

                         Sunrise Point  [191F]                        
Controller #0                                          Single Channel 
 Bus Rate  8000 MT/s      Bus Speed 8009 MT/s     DRAM Speed 2133 MHz 
                                                                      
 Cha    CL  RCD   RP  RAS  RRD  RFC   WR RTPr WTPr  FAW  B2B  CWL Rate
  #0     5    8    8   28    0  180    0    6   24    0    0    6   1N
      ddWR drWR srWR ddRW drRW srRW ddRR drRR srRR ddWW drWW srWW  ECC
  #0     0    0    0    0    0    0    0    0    0    0    0    0   0 
                                                                      
 DIMM Geometry for channel #0                                         
      Slot Bank Rank     Rows   Columns    Memory Size (MB)           
       #0
```

So it was a `Intel Core i5-6600K`, and CoreFreq just defaulted to the 2133 MHz speed here: https://ark.intel.com/content/www/us/en/ark/products/88191/intel-core-i5-6600k-processor-6m-cache-up-to-3-90-ghz.html

Maybe that is fixed now 🤷‍♂️ You did have a Skylake too, no? My Intel NUC Haswell i3 Ultra Low TDP shows 1600 MHz, which is kind of correct, but the SODIMM chip can go 2666 MHz but is used with another that can not. My MacBook Air has same CPU, but is a much faster i7, so hyperthreading, and goes hot with pretty normal use and hence throttles, also on macOS using the official Intel Power Gadget tool.

Remember Memory Controller Window was not even available before I gave you some data from these PCs and you managed to fix it?

Look at temperatures at stress (98/99˚C). Bad screenshots regarding colors using Fedora's GNOME Terminal but can be seen:

https://gist.github.com/cyring/bbb8c8eec31eb0a0b8f3b2d7e8fc82bf

cyring, 2020-06-26:
I had access to this [i7-6700](https://gist.github.com/cyring/7509077959ccc24930526b949380d796) Skylake for a while but can't do non regression tests anymore .

olejon, 2020-06-26:
Well, if you can at least add the measurements values, that would be great. Especially `(W)` the values that use that, and toggle to `(J)` respectively.

closed 2020-06-26 by olejon