cyring/CoreFreq #191: CoreFreq for Windows Subsystem for Linux kernel

cyring, 2020-06-26:
# Windows
## Prerequisites
### Programs
* Select WSL for Linux and Hyper-V
[image]
### Store
[image]
### Configure
```
notepad %USERPROFILE%\.wslconfig

[wsl2]
# kernel=<path>              # An absolute Windows path to a custom Linux kernel.
# memory=<size>              # How much memory to assign to the WSL2 VM.
# processors=<number>        # How many processors to assign to the WSL2 VM.
# swap=<size>                # How much swap space to add to the WSL2 VM. 0 for no swap file.
# swapFile=<path>            # An absolute Windows path to the swap vhd.
# localhostForwarding=<bool> # Boolean specifying if ports bound to wildcard or localhost in the WSL2 VM should be connectable from the host via localhost:port (default true).
# <path> entries must be absolute Windows paths with escaped backslashes, for example C:\\Users\\Ben\\kernel
# <size> entries must be size followed by unit, for example 8GB or 512MB
swap=0
```
# WSL
```
sudo apt-get install make gcc git
git clone https://github.com/microsoft/WSL2-Linux-Kernel.git
cd  WSL2-Linux-Kernel
cat README-Microsoft.WSL2
sudo apt install build-essential flex bison libssl-dev libelf-dev

make KCONFIG_CONFIG=Microsoft/config-wsl

```

cyring, 2020-06-26:
# Work-in-progress

* TSC is virtualized
* APERF/MPERF counters crash the kernel.
* Hypervisor ID string to be integrated

[image]

cyring, 2020-06-28:
# [HV_X64_MSR_VP_RUNTIME](https://elixir.bootlin.com/linux/latest/source/arch/x86/include/asm/hyperv-tlfs.h#L212)
* Can give CPU usage inside VM
```c
 8057                         RDTSC64(Core->Counter[1].TSC);
 8058                         /* HV_X64_MSR_VP_RUNTIME:vcpu runtime in 100ns units */
 8059                         RDCOUNTER(Core->Counter[1].C0.URC, 0x40000010);
 8060
 8061                         Core->Counter[1].C0.URC = Core->Counter[1].C0.URC
 8062                                 * PUBLIC(RO(Proc))->Features.Factory.Ratio
 8063                                 * 10;
 8064
 8065                         Core->Counter[1].C0.UCC = Core->Counter[1].C0.URC;
```

cyring, 2020-06-28:
# VirtualMachine CPU Loop
## KVM
[image]

cyring, 2020-07-01:
# KVM
## 255 x EPYC VCPU
```
qemu-system-x86_64 -accel accel=kvm -m 16384 -smp cpus=255 -cpu EPYC -cdrom archlinux-corefreq.iso -display gtk
```
[image]

olejon, 2020-07-01:
After `make deb-pkg -j 16`, you don't have a guide for which to choose:

```
TSX enable mode
> 1. off (X86_INTEL_TSX_MODE_OFF) (NEW)
  2. on (X86_INTEL_TSX_MODE_ON) (NEW)
  3. auto (X86_INTEL_TSX_MODE_AUTO) (NEW)
choice[1-3?]:
```

?

olejon, 2020-07-01:
* I do get it to run on Windows (WSL-2), but a lot of work for *what?*
* There are a gazillion tools on Windows to monitor and tweak stuff CoreFreq can not do
* Mounting of `C:\` fails...
* Starting Ubuntu takes way longer...
* Correct way to list installed packages is: `dpkg -l | grep linux`, *NOT* using `apt` with `grep`
* No one uses `apt-get install` as in first step anymore, but `apt install`, though it works for backwards compat.
* Worth mentioning that the `.wslconfig` does not exist by default, meaning must be created as asked for
* One can just run `sudo ./corefreqd &` and without `bash -c` and press Enter to continue...

### Running `dmesg` as it tells to...

[image]

### CoreFreq

olejon, 2020-07-01:
Edit. Forgot screenshot of CoreFreq running ⬆️

cyring, 2020-07-02:
> After `make deb-pkg -j 16`, you don't have a guide for which to choose:
> 
> ```
> TSX enable mode
> > 1. off (X86_INTEL_TSX_MODE_OFF) (NEW)
>   2. on (X86_INTEL_TSX_MODE_ON) (NEW)
>   3. auto (X86_INTEL_TSX_MODE_AUTO) (NEW)
> choice[1-3?]:
> ```
> 
> ?

* Probably the Microsoft/config-wsl used has not defined an answer to this choice

1. Using an AMD, keep it OFF. Not hardware relevant.

2. With Intel, it depends of your BIOS TSX state and if you decided to make it use in the host.

* various readings
[Wikipedia](https://en.wikipedia.org/wiki/Transactional_Synchronization_Extensions)
[kernel.org](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=c2955f270a84762343000f103e0640d29c7a96f3)

cyring, 2020-07-02:
> * I do get it to run on Windows (WSL-2), but a lot of work for _what?_
> * There are a gazillion tools on Windows to monitor and tweak stuff CoreFreq can not do

What can I say ?
_CoreFreq_ for WSL is a POC

> * Mounting of `C:\` fails...
> * Starting Ubuntu takes way longer...
> * Correct way to list installed packages is: `dpkg -l | grep linux`, _NOT_ using `apt` with `grep`
> * No one uses `apt-get install` as in first step anymore, but `apt install`, though it works for backwards compat.

I'm an Archer.

> * Worth mentioning that the `.wslconfig` does not exist by default, meaning must be created as asked for

[WSL2](https://github.com/microsoft/WSL2-Linux-Kernel) in [Microsoft](https://github.com/microsoft/WSL2-Linux-Kernel/tree/master/Microsoft) directory

> * One can just run `sudo ./corefreqd &` and without `bash -c` and press Enter to continue...

Indeed many ways to workaround the predefined Ubuntu/WSL `su`

> ### Running `dmesg` as it tells to...
Are WSL2 prerequisites missing in the Windows host ?

> ### CoreFreq
Thanks a lot for the results. I'm noticing in it the expected data.

olejon, 2020-07-02:
Yeah. Using `sudo bash -c` is how I would've too. Always use that. Or `su -l -c "bash -c 'something if spaces and arguments to say $1'" <username>` for stuff already running as root to run as regular user, say a ncurses UI in a `screen` (like easy boot command where systemd is overkill). And yeah `sudo -s` and `sudo -i` is what you want for regular old school `su`.

**FYI AND TO OTHERS, this breaks Ryzen Master:** What breaks it is the Windows Extra Feature `Windows Hypervisor Platform`. Took a little while to find the reason.

**Error:**

> Ryzen Master can only run with Virtualization Based Security (VBS) disabled in the Windows operating system. Please disable VBS and re-start Ryzen Master. Virtualization-Based Security (VBS)!

[image]

First I thought IOMMU, but nope, not SVM either. Uninstalled Ubuntu, then Microsoft WSL Update from `Control Center > Programs and Features`, then `Windows Subsystem for Linux`, then the above, `Windows Hypervisor Platform`.

*Rebooted every time to find the reason*. Also had to set `wsl --set-default-version 1` before reinstalling Ubuntu. Otherwise it would not work, although WSL-2 was *uninstalled*. Stupid Windows. The WSL-2 update was uninstalled as said.

**BUT:** I did not have to install `Hyper-V` Windows feature for this to work.

[Powershell Commands in Microsoft's documentation do not do that](https://docs.microsoft.com/en-us/windows/wsl/install-win10), **points 2-6 in order as per Microsoft:**

1. `Install the Windows Subsystem for Linux` :heavy_check_mark: 
2. `Update to WSL 2` :heavy_check_mark: 
3. **`Enable the 'Virtual Machine Platform' optional component`** :heavy_check_mark: 
4. Set WSL to version 2 by default :heavy_check_mark: 
5. Install Ubuntu, or another distro from Microsoft Store :heavy_check_mark: 

Powershell *confirmed Ubuntu ran in WSL version 2*

olejon, 2020-07-02:
Edits :arrow_up:

olejon, 2020-07-02:
Edit again :arrow_up:

cyring, 2020-07-02:
I better understand now, thank you.
Indeed Ryzen Master is not installed, and Windows is straight out of the box. Including the certified apps from the Store, such as the Ubuntu ready to play button, I try to stick to the Microsoft ecosystem.
But I've also struggled with WSL locked to version one until I was notified by Windows-Update to upgrade _kernel_
Then the magic happened: WSL set for version 2

Don't ask me why open-source Kernels are my favorite !
You just have to read the source code to understand what's going on...

olejon, 2020-07-02:
* No, I don't ask you about that. As I read from Microsoft WSL-2 is fully Open Source so kind if needed I guess, or better at least
* There was a major *Windows Build Update* when I checked yesterday, yes, but I installed it *first* (I have Windows Update on Pause always so I check manually then Pause again)
* Also clean Windows, apart from the mentioned software in the other issue for my HW *which is not running, not even Services for them*
* In fact I disable several Windows Services so it's *totally idle*
* Your guide should update Ubuntu as *first step before installing any package or CoreFreq*, with `sudo apt update && sudo apt dist-upgrade`

**Note:** There is a weird error (Windows only thing) if you do this immediately, like it says the update files "are not available yet", and says they're not ready before X minutes (never seen on Ubuntu outside Windows), so close Ubuntu and wait, try again after X minutes, or reboot and maybe it works then. At least I have update my Ubuntu in Windows *Several Times*. It *must* be done automatically, as Windows does *not* update Ubuntu from the repos, logically.

olejon, 2020-07-02:
You asked to try on any Hypervisor. Here is Headless Ubuntu 18.04 VM in *CPU + CPU Cache passthrough mode* on Headless Ubuntu 18.04 host running KVM + QEMU, running same kernel version, but VM uses the Ubuntu `virtual` kernel. Pretty much as expected. Stresstests just as fast as on host, with host showing same stress level (running CoreFreq on both). Both APERF/MPERF.

[image]

cyring, 2020-07-02:
> You asked to try on any Hypervisor...

Thank you very much

> Both APERF/MPERF.

Nop: when Virtualization is detected by _CoreFreq_ , an unique counter is used for both UCC & URC.
```
HV_X64_MSR_VP_RUNTIME
```
This counter is a MSR read at address `0x40000010` and its implementation is free to the Hypervisor.

Source code of KVM shows that it uses some accounting based on task runtime per CPU.

Thus, although the driver inside the VM reads this counter like a hardware register such as APERF/MPERF, though the asm instruction `rdmsr`; this counter is in fact a software counter.

olejon, 2020-07-02:
> Nop: when Virtualization is detected by CoreFreq , an unique counter is used for both UCC & URC

I meant I *compiled* as on the *host*, with APERF/MPERF. Should I simply run `make` without those in the *VM*? Or `make something-else`?

cyring, 2020-07-02:
> > Nop: when Virtualization is detected by CoreFreq , an unique counter is used for both UCC & URC
> 
> I meant I _compiled_ as on the _host_, with APERF/MPERF. Should I simply run `make` without those in the _VM_? Or `make something-else`?

Symply run `make`

olejon, 2020-07-03:
Alright. Done. Don't see any difference apart from APERF/MPERF saying OFF.

Have downloaded and installed Manjaro XFCE on a VM on the 3950X. Will check tomorrow how that one works.

**Q:** On AMD host, also just `make` in the VM?

cyring, 2020-07-03:
> **Q:** On AMD host, also just `make` in the VM?

Yes, the same, just `make`

* FYI, here are the instructions to read counter for the Virtual Machine case
https://github.com/cyring/CoreFreq/blob/6074dd6496d87d1ad1750d25bb4fa52ae920d4e1/corefreqk.c#L7419
As stated previously, the address `0x40000010` is employed. No Makefile options can change it.

cyring, 2020-07-07:
# Fetch all branches
```sh
for branch in `git branch -r|grep -v ' -> '|cut -d"/" -f2`; do git checkout $branch; git fetch; done;
```

cyring, 2020-08-05:
To sum up the issue: `0x40000010` is the new counter introduced by Hypervisors. _CoreFreq_ makes use of it in a dedicated loop of the driver. The resulted indicator is not as accurate as running bare-metal,  but it gives a seamless preview of the VCPU usage. 
As a reminder, _CoreFreq_ is made for bare-metal CPU under the Linux kernel.  
Besides features regression, or occasional findings, I don't plan to pursue the work on this enhancement. Closing the issue.

 CyrIng

closed 2020-08-05 by cyring