cyring/CoreFreq #192: Is HOT > Power Limitation a bug on Haswell/Ultra Low TDP] when (re)booting

olejon, 2020-06-26:
* My Intel NUC D34010WYK always (re)boots with indicator `HOT > Power Limitation`
* This makes little sense when it idles at 45°C and a looong `Conic Compute` struggles to get it to 55°C
* A (re)boot is much *less stressful* than that. The system runs minimal Ubuntu (netinstall)
* What exactly does that `Power Limitation` mean? It *definitely* has enough power (power brick is pretty big, CPU is slow)
* To add to that, my MacBook Air 13" Summer 2013 max-specs with an `i7-4650U Haswell/Ultra Low TDP]` *ALSO (re)boots with this indicator*
* **Note:** On the MacBook it is logical that other `HOT` indicators get triggered. IIRC it's `PROCHOT# Agent`, `Thermal Treshold` *and/or* `Critical Temperature`. At least 3, so maybe 2 of those and `Power Limitation` again
* If necessary for data or stresstesting, I can boot my MacBook into Fedora again. It's been a while so must update the system and compile CoreFreq again
* But [as screenshots show here](https://gist.github.com/cyring/bbb8c8eec31eb0a0b8f3b2d7e8fc82bf), the MacBook reaches 99°C just by stressing 1 Core, then throttles, *so it's possible it does that when booting too*
* The NUC however maxes out at its spec during `Conic Compute` - `1696 MHz` - and *dead stable there* as temperature is low
* So `HOT` is NEVER triggered EVER by stressing the NUC to the MAX with CoreFreq and VMs and whatnot, *just (re)boot causes it*
* So is it a Haswell bug, *possibly only for Ultra Low TDP*, or a CoreFreq bug? If it is a Haswell bug, is it possible to detect if was caused by boot and can hence by *discarded* as "false", as it truly is false
* Not that it seems to do anything, but since I (re)boot it so rarely, I tend to go into CoreFreq and turn that indicator OFF anyway, as it annoys me :stuck_out_tongue: 

### Screenshot - Intel NUC D34010WYK

[image]

cyring, 2020-06-26:
This is something I learned from Intel datasheets: the Power Limitation bit is raised when current is restored.
Thus, when Processor is powered on, or, resumed from suspend
Nothing to worry, but it's good to know it exists. One could imagine use-cases to script with this bit ...

olejon, 2020-06-27:
Aha, good to know. But IIRC wasn't like this on my Skylake. Is it still used on Intel CPUs? Only laptop CPUs maybe, or?

cyring, 2020-06-27:
> Aha, good to know. But IIRC wasn't like this on my Skylake. Is it still used on Intel CPUs? Only laptop CPUs maybe, or?

I noticed the same happens on Desktop Skylake.

If irs annoying I could add an event mask bitmap argument to the Client.
What do you think about this ?

olejon, 2020-06-27:
You saw it on Desktop Skylake or not? Didn't quite understand you there. I can not remember I did.

But no thanks. Explanation was :heavy_check_mark:

closed 2020-06-27 by olejon

cyring, 2020-06-27:
> You saw it on Desktop Skylake or not? Didn't quite understand you there. I can not remember I did.
> 
> But no thanks. Explanation was

Yes, HOT lights ON when booting Desktop SkyLake

Regards