cyring/CoreFreq #193: [AMD][Family 17h][Zen] IOMMU Enablement State

cyring, 2020-07-01:
# _CoreFreq_ version [`1.79.5`]
## Git branch availability [`develop`]
* MATISSE and APU of models 20h can now be queried for their IOMMU state
[image]
* The same option found in BIOS at:
[image]

* **Contribution**: 
If you want to get yours, please post your `lspci -nn` output regarding the IOMMU device

olejon, 2020-07-01:
Yes. Don't know where you found that option, since you searched in UEFI. But on my Motherboard it's called `SR-IOV` and it's the only Setting under some Section in Advanced. A drop-down.

`00:00.2 IOMMU [0806]: Advanced Micro Devices, Inc. [AMD] Device [1022:1481]`

[image]

cyring, 2020-07-01:
Thanks for your test but if the device shows up in `lspci` then it has to be enabled or auto in BIOS.
Meanwhile I will check my BIOS if `SR-IOV` and  `IOMMU` are or not separated options ?

olejon, 2020-07-01:
It's enabled. Only other option is disabled (no Auto). I've been through ALL Settings 100 times and NOWHERE have I've seen IOMMU...  As I understand it it's the same as `VT-d` on Intel (a website said so).

I can try searching...

I read this: https://access.redhat.com/documentation/en-us/red_hat_virtualization/4.0/html/hardware_considerations_for_implementing_sr-iov/index

olejon, 2020-07-01:
**Found it!**

* It's under
  * `Advanved > AMD CBS`
* Not that it changed *anything*. Seems like `Auto` was enough to enable it, with `SR-IOV = Enabled`
* Pretty darn sure `Technologies` before said `I/O MMU = [OFF]`

```
# lspci -nn
00:00.2 IOMMU [0806]: Advanced Micro Devices, Inc. [AMD] Device [1022:1481]
```

[image]

cyring, 2020-07-01:
Thanks for your efforts. 

So do you mean you get a match between the BIOS state and the _CoreFreq_ state ?

IOMMU disabled in both ?
IOMMU enabled in both ?

olejon, 2020-07-01:
* First has `SR-IOV = Enabled` only and was then `IOMMU = Auto` before I found it (seems enough to enable SR-IOV only)
* Second has `Both = Enabled` (not that it does anything it seems)

Default is `SR-IOV = Disabled` and `IOMMU = Auto`. Which seems to disable AMD-V.

cyring, 2020-07-02:
IOMMU state confirmed with Ryzen 2700X
https://github.com/cyring/CoreFreq/issues/54#issuecomment-652788534

olejon, 2020-07-02:
You got it right? Just `SR-IOV = Enabled` and let `IOMMU = Auto` and it enables it, according to `lspci -nn` and `CoreFreq > Technologies`. I guess you can reproduce that. You have ASUS too.

closed 2020-11-20 by cyring