cyring/CoreFreq #195: [AMD][Family 17h] Globale C-States

cyring, 2020-07-03:
# BIOS C-States option
```
C-STATES [OFF]
	HWCR(0xc0010015) = 0x0000000149000011
	CC6 (0xc0010296) = 0x0000000000080808
	PC6 (0xc0010292) = 0x0000000104000012
	PST (0xc0010064) = 0x8000000048D2088C
	Ratio = 45.71
	Vcore = 1.0562   1.0562   1.4812
C-STATES [ON]
	HWCR(0xc0010015) = 0x0000000149000011
	CC6 (0xc0010296) = 0x0000000000484848
	PC6 (0xc0010292) = 0x0000000104000012
	PST (0xc0010064) = 0x8000000048D2088C
	Ratio = 47.00
	Vcore = 0.9062   0.9062   1.4875
```
* `MSR 0xc0010296` with a mask of `0x0000000000404040` seems to be the C-States enablement mode

cyring, 2020-07-03:
CC6 activation fixed in commit 47b46d6a5d08ab9b57f9f6522d360c9c7be55b4b

adatum, 2020-07-03:
CC6 disabled:
[image]

CC6 enabled:
[image]

cyring, 2020-07-03:
> CC6 disabled:

> CC6 enabled:

Thanks. I clearly see the differences, especially the temperature.

adatum, 2020-07-03:
> I clearly see the differences, especially the temperature.

I expect there is a difference in temperature and power, but I wouldn't read too much into what the screenshots show. 

The system was mostly idle but with browser tabs open, and the normal temperature fluctuations are greater than the average difference between the enabled/disabled states, especially with warm ambient temperatures ~30C.

I did a similarly idle test over time, and I wouldn't say that I can distinguish enabled vs disabled in the CPU temperatures. Maybe the VRM temperature shows a small and more stable difference. 

[image]

cyring, 2020-07-04:
> ... and I wouldn't say that I can distinguish enabled vs disabled ... 

Minimum Vcore can also be observed in single user console mode. 

A hardware C-state counter will be the best way to measure the idle residency; but no such register is found in Zen specs.
The ACPI space may offer a _similar_ counter, that's a work in progress.

cyring, 2020-07-05:
As part of commit 8e4add933bfa22d886a2c1dc651b91fb083dc46a , the BIOS Global C-State is reported through _CoreFreq_ `CCx`
[image]
[image]
* After combinary tests, the state is enabled when the Base Address (BAR) is different from zero
* C-State BAR register is specified as read-write. I'm doing a read-only access.
https://github.com/cyring/CoreFreq/blob/ec1c73178e6e7d4a514d62223cc53ffb2b0e3feb/amdmsr.h#L440

cyring, 2020-07-05:
# PMx07E [CStateEn] (FCH::PM::CStateEn)
```
C1eToC3En. Read-write. Reset: 1. 1=Put APU into C3 state in C1e state. 
C1eToC2En. Read-write. Reset: 0. 1=Put APU into C2 state in C1e state. 
```
A part of commit f1c0e8e3c48685736f73c345a636a50e7f97295f , the _CoreFreq_ **Experimental** mode is required to read the Power Management (PM) Registers
[image]
* _Remarks_: 
1. Not sure if the AMD is specifying UnDemotion or Auto Demotion
2. `C1U` (the Intel variable) is used as a placeholder for Zen `C2U`
3. Can we say than Zen `C1E` is de facto enabled at boot ?

adatum, 2020-07-05:
Global C-States disabled in BIOS:

[image]

Global C-States enabled (Auto) in BIOS:

[image]

cyring, 2020-07-05:
@adatum :+1:  Thank you for your tests.
According to your screenshots, you also have tested `C1U C3U` which means that my assembly instructions to read the FCH did not crash the 2700X.
Removing Experimental and showing `C2 UnDemotion` in commit d9b2663f7fbdcbfd4e6b52db28de96ce3b7bd7bb
[image]

adatum, 2020-07-05:
You're welcome. 

One side note: using the scroll wheel up or down in the UI causes an error message which I do not know how to close:

[image]

The upper left corner shows [c306:41] on scroll wheel up, and [c306:42] on scroll wheel down.

cyring, 2020-07-05:
> One side note: using the scroll wheel up or down in the UI causes an error message which I do not know how to close:

> The upper left corner shows [c306:41] on scroll wheel up, and [c306:42] on scroll wheel down.

For some reasons, a key pressed may have been injected into the UI which handled it as a command to the driver. Could have been a mouse move, mouse buttons ?
Meanwhile I will debug and harden this ...

* Writing to the FCH is now part of commit 3be3ff46d5a8178defc0bee71b26f377c6ce7d00
Bellow `C2U` and  `C3U` state have been toggled but I don't notice a major change in Voltage, Temps, Consumption ...
[image]

cyring, 2020-07-05:
I don't know if these PM registers managed by the FCH are SMT or per Core or CCX or whole Package ?

Current implementation in driver: any CPU is reading / writing through the FCH; and no access protection of mutex, semaphore kinds, is applying.

adatum, 2020-07-06:
> For some reasons, a key pressed may have been injected into the UI which handled it as a command to the driver. Could have been a mouse move, mouse buttons ?

Yes, mouse scroll wheel, up and down.

cyring, 2020-07-06:
> > For some reasons, a key pressed may have been injected into the UI which handled it as a command to the driver. Could have been a mouse move, mouse buttons ?
> 
> Yes, mouse scroll wheel, up and down.

Do you get the same in a plain shell terminal  or console ?
Can you read the ASCII dump of those mouse events  ?

cyring, 2020-07-06:
Depending on the number of characters pasted, you will have to escape many times the Driver message in red until the exchange ring gets empty (at most 16 commands)

adatum, 2020-07-06:
No I don't get issues in a plain shell terminal. 
I tried the getkey.c program, but scrolling does not register as an input and just scrolls the terminal window as normal. 
I tried `xev` which does register events from the scroll wheel but I don't know if any of the output are ASCII codes.

You're right, pressing [Escape] enough times finally gets rid of the red Driver message. 
I realized that mouse wheel scrolling does not always cause the error. At some times it is taken as user input, for example opening a menu or scrolling down some list sometimes. The behavior is not obvious to me.

cyring, 2020-07-06:
> No I don't get issues in a plain shell terminal.
> I tried the getkey.c program, but scrolling does not register as an input and just scrolls the terminal window as normal.
> I tried `xev` which does [...]

May be try catching mouse codes with:
```
./getkey -x
```

However, I'm far to be ready to support the Mouse in the _CoreFreq_ UI.  
This is reserved for [_CoreFreq.gui_](https://github.com/cyring/CoreFreq.gui)

adatum, 2020-07-06:
The output is not consistent and changes each time. 
The first time scroll down got: `Poll(1)[262e614d5b1b]` and scroll up got: `Poll(1)[2370604d5b1b]` but the part before 4d5b1b changes each time. 
Also, after quitting `getkey` with Ctrl-C, moving the scroll wheel enters gibberish into the terminal, and the terminal text color is changed. I have to `reset` to return to normal behavior.

I don't expect mouse support for CoreFreq. I was just surprised by the unexpected behavior from an accidental mouse scrolling action.

cyring, 2020-07-06:
> The output is not consistent and changes each time.
> The first time scroll down got: `Poll(1)[262e614d5b1b]` and scroll up got: `Poll(1)[2370604d5b1b]` but the part before 4d5b1b changes each time. [...]

Thanks. 
The codes you dump are fixed Id of mouse event completed with motion, coordinates, or relative values like up, down incréments. 

It's still a draft code which supports X mouse, thus terminal is badly restored. 

In the UI to-do list: I'll try to catch and discard the mouse events using masks of the fixed ASCII part.

cyring, 2020-07-06:
Could you please test the latest commit and toggle several times `C2U` and `C3U`

This commit contains asm codes to atomically access the FCH.
In concurrent access case, CPU will retry 10 times, spaced by 5 ms

It works with my 3950X but be prepared for any lock, thus save your files.

adatum, 2020-07-06:
All 4 combinations of `C2U` and `C3U` OFF/ON were tried with no ill effects noticed.

cyring, 2020-07-06:
> All 4 combinations of `C2U` and `C3U` OFF/ON were tried with no ill effects noticed.

Yes, Yes, Yes !
Thank you.

closed 2020-11-20 by cyring