cyring/CoreFreq #197: corefreqd.service should point to instead of /bin/corefreqd /usr/bin/corefreqd

bubbleguuum, 2020-07-14:
By default, corefreqd is installed in /usr/bin so it makes sense to have  corefreqd.service use /usr/bin/corefreqd instead /usr/bin/corefreqd. It could also use the value of PREFIX in Makefile to dynamically modify corefreqd.service  to use the proper path

cyring, 2020-09-06:
Hello,

_Sorry for being long to provide a fix._

To my surprise, `ExecStart` can work without an absolute path to `corefreqd`

For your tests, you can try the `develop` branch

ArchLinux Users can also test this branch by changing the `source` line in the `PKGBUILD`
```
source=(git+${url}.git#branch=develop)
```

cyring, 2020-09-11:
Hello,

I'm tagging [1.80-rc.1](https://github.com/cyring/CoreFreq/tree/1.80-rc.1) as ready for packaging

cyring, 2020-11-20:
Change had been provided.
Regards
CyrIng

closed 2020-11-20 by cyring