cyring/CoreFreq #201: Feature request: multiple column view for core/thread usage bars

hvico, 2020-08-12:
Hello,

I really love this tooI! 

I have a workstation that has 12 core/24 threads, and another that has 8/16. In order to see the usage bars of all cores/threads I need to resize the terminal in a way that fills all my screen. It would be nice to have a multi column view with shorter bars, maybe split in two that part of the screen and then monitor half of the threads/cores in each half. Then it would be possible to monitor more cores/threads in a smaller window.

Thanks!

cyring, 2020-08-12:
Hello,

I'm also missing space with my 16C/32T on a plain console but it's OK on 2K display
[image]

B/c it takes hours to code some changes, could you suggest draft drawings of the expected UI layout ?

hvico, 2020-08-12:
Hi! 

This was my idea of a multi-column view, just share the space with shorter bars:

[image]

Hope it helps.

Regards,

cyring, 2020-08-12:
> Hi!
> 
> This was my idea of a multi-column view, just share the space with shorter bars:
> 
> [image]
> 
> Hope it helps.
> 
> Regards,

Such as htop. That's indeed a good idea. Let me think about the impacts, like odd or even ordering...

cyring, 2020-08-12:
Noticing that you don't have the voltage Vcore. Probably, your hardware provides the Winbond Super-I/O chipset.
(my Westmere is connected to one W83627) 

 Try to build this way:

```
make HWM_CHIPSET=W83627 clean all
```

Then restart all: driver, daemon, cli.

hvico, 2020-08-12:
I'll try that, thank you. 

It's a Mac Pro 2009 (with 2010 firmware, Mac Pro 5.1). But it is running Ubuntu Linux 18.04. Yes, Corefreq runs perfectly well in this hardware.

cyring, 2020-08-12:
> It's a Mac Pro 2009 (with 2010 firmware, Mac Pro 5.1). But it is running Ubuntu Linux 18.04. Yes, Corefreq runs perfectly well in this hardware.

I don't believe my eyes !

Can you please the output of:
```
corefreq-cli -s -M
```

hvico, 2020-08-12:
I tried "make HWM_CHIPSET=W83627 clean all", no change.

corefreq-cli -s -M:

Processor                                 [Intel(R) Xeon(R) CPU X5690 @ 3.47GHz]
|- Architecture                                                    [Westmere/EP]
|- Vendor ID                                                      [GenuineIntel]
|- Microcode                                                        [        31]
|- Signature                                                           [  06_2C]
|- Stepping                                                            [      2]
|- Online CPU                                                          [ 24/ 24]
|- Base Clock                                                          [132.986]
|- Frequency            (MHz)                      Ratio                        
                 Min   1595.84                    <  12 >                       
                 Max   3457.65                    <  26 >                       
|- Factory                                                             [133.333]
                       3470                       [  26 ]                       
|- Performance                                                                  
   |- OSPM                                                                      
                 TGT   8378.15                    <  63 >                       
|- Turbo Boost                                                         [   LOCK]
                  1C   3723.62                    <  28 >                       
                  2C   3723.62                    <  28 >                       
                  3C   3590.63                    <  27 >                       
                  4C   3590.63                    <  27 >                       
                  5C   3590.63                    <  27 >                       
                  6C   3590.63                    <  27 >                       
|- Uncore                                                              [   LOCK]
|- TDP                                                           Level [  0:0  ]
   |- Programmable                                                     [   LOCK]
   |- Configuration                                                    [   LOCK]
   |- Turbo Activation                                                 [   LOCK]
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [N]          AES [Y]  AVX/AVX2 [N/N] 
|- AVX512-F     [N]    AVX512-DQ [N]  AVX512-IFMA [N]   AVX512-PF [N] 
|- AVX512-ER    [N]    AVX512-CD [N]    AVX512-BW [N]   AVX512-VL [N] 
|- AVX512-VBMI  [N] AVX512-VBMI2 [N]  AVX512-VNMI [N]  AVX512-ALG [N] 
|- AVX512-VPOP  [N] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [N] 
|- AVX512-BF16  [N]  BMI1/BMI2 [N/N]         CLWB [N] CLFLUSH/O [Y/N] 
|- CLAC-STAC    [N]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y] 
|- F16C         [N]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y] 
|- MMX/Ext    [Y/N] MON/MWAITX [Y/N]        MOVBE [N]   PCLMULQDQ [Y] 
|- POPCNT       [Y]       RDRAND [N]       RDSEED [N]      RDTSCP [Y] 
|- SEP          [Y]          SHA [N]          SSE [Y]        SSE2 [Y] 
|- SSE3         [Y]        SSSE3 [Y]  SSE4.1/4A [Y/N]      SSE4.2 [Y] 
|- SERIALIZE    [N]      SYSCALL [Y]          SGX [N]       RDPID [N] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- Core Multi-Processing                                  CMP Legacy   [Missing]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Direct Cache Access                                           DCA   [Capable]
|- Debugging Extension                                            DE   [Capable]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Capable]
|- CPL Qualified Debug Store                                  DS-CPL   [Capable]
|- 64-Bit Debug Store                                         DTES64   [Capable]
|- Fast-String Operation                                Fast-Strings   [Missing]
|- Fused Multiply Add                                     FMA | FMA4   [Missing]
|- Hardware Lock Elision                                         HLE   [Missing]
|- Long Mode 64 bits                                       IA64 | LM   [Capable]
|- LightWeight Profiling                                         LWP   [Missing]
|- Machine-Check Architecture                                    MCA   [Capable]
|- Memory Protection Extensions                                  MPX   [Missing]
|- Model Specific Registers                                      MSR   [Capable]
|- Memory Type Range Registers                                  MTRR   [Capable]
|- OS-Enabled Ext. State Management                          OSXSAVE   [Missing]
|- Physical Address Extension                                    PAE   [Capable]
[... 77 lines cut ...]
|- Reference Cycles                                                    [Missing]
|- Last Level Cache References                                         [Capable]
[... cut ...]
|- Thermal Design Power                                          TDP   [Missing]
   |- Minimum Power                                              Min   [Missing]
   |- Maximum Power                                              Max   [Missing]
|- Units                                                                        
   |- Power                                               watt   [      Missing]
   |- Energy                                             joule   [      Missing]
   |- Window                                            second   [      Missing]

cyring, 2020-08-12:
> I tried "make HWM_CHIPSET=W83627 clean all", no change.
> 
> corefreq-cli -s -M:
> 
> Processor [Intel(R) Xeon(R) CPU X5690 @ 3.47GHz]
> |- Architecture [Westmere/EP]
> |- Vendor ID [GenuineIntel]
> |- Microcode [ 31]
> |- Signature [ 06_2C]
> |- Stepping [ 2]
> |- Online CPU [ 24/ 24]
> |- Base Clock [132.986]
> |- Frequency (MHz) Ratio
> Min 1595.84 < 12 >
> Max 3457.65 < 26 >
> |- Factory [133.333]
> 3470 [ 26 ]
> |- Performance
> |- OSPM
> TGT 8378.15 < 63 >
> |- Turbo Boost [ LOCK]
> 1C 3723.62 < 28 >
> 2C 3723.62 < 28 >
> 3C 3590.63 < 27 >
> 4C 3590.63 < 27 >
> 5C 3590.63 < 27 >
> 6C 3590.63 < 27 >
> |- Uncore [ LOCK]
> |- TDP Level [ 0:0 ]
> |- Programmable [ LOCK]
> |- Configuration [ LOCK]
> |- Turbo Activation [ LOCK]
> 
> Instruction Set Extensions
> |- 3DNow!/Ext [N/N] ADX [N] AES [Y] AVX/AVX2 [N/N]
> |- AVX512-F [N] AVX512-DQ [N] AVX512-IFMA [N] AVX512-PF [N]
> |- AVX512-ER [N] AVX512-CD [N] AVX512-BW [N] AVX512-VL [N]
> |- AVX512-VBMI [N] AVX512-VBMI2 [N] AVX512-VNMI [N] AVX512-ALG [N]
> |- AVX512-VPOP [N] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [N]
> |- AVX512-BF16 [N] BMI1/BMI2 [N/N] CLWB [N] CLFLUSH/O [Y/N]
> |- CLAC-STAC [N] CMOV [Y] CMPXCHG8B [Y] CMPXCHG16B [Y]
> |- F16C [N] FPU [Y] FXSR [Y] LAHF-SAHF [Y]
> |- MMX/Ext [Y/N] MON/MWAITX [Y/N] MOVBE [N] PCLMULQDQ [Y]
> |- POPCNT [Y] RDRAND [N] RDSEED [N] RDTSCP [Y]
> |- SEP [Y] SHA [N] SSE [Y] SSE2 [Y]
> |- SSE3 [Y] SSSE3 [Y] SSE4.1/4A [Y/N] SSE4.2 [Y]
> |- SERIALIZE [N] SYSCALL [Y] SGX [N] RDPID [N]
> 
> Features
> |- 1 GB Pages Support 1GB-PAGES [Capable]
> |- Advanced Configuration & Power Interface ACPI [Capable]
> |- Advanced Programmable Interrupt Controller APIC [Capable]
> |- Core Multi-Processing CMP Legacy [Missing]
> |- L1 Data Cache Context ID CNXT-ID [Missing]
> |- Direct Cache Access DCA [Capable]
> |- Debugging Extension DE [Capable]
> |- Debug Store & Precise Event Based Sampling DS, PEBS [Capable]
> |- CPL Qualified Debug Store DS-CPL [Capable]
> |- 64-Bit Debug Store DTES64 [Capable]
> |- Fast-String Operation Fast-Strings [Missing]
> |- Fused Multiply Add FMA | FMA4 [Missing]
> |- Hardware Lock Elision HLE [Missing]
> |- Long Mode 64 bits IA64 | LM [Capable]
> |- LightWeight Profiling LWP [Missing]
> |- Machine-Check Architecture MCA [Capable]
> |- Memory Protection Extensions MPX [Missing]
> |- Model Specific Registers MSR [Capable]
> |- Memory Type Range Registers MTRR [Capable]
> |- OS-Enabled Ext. State Management OSXSAVE [Missing]
> |- Physical Address Extension PAE [Capable]
[... 78 lines cut ...]
> |- Last Level Cache References [Capable]
> |- Last Level Cache Misses [Capable]
> |- Branch Instructions Retired [Capable]
> |- Branch Mispredicts Retired [Capable]
> 
> Power & Thermal
> |- Clock Modulation ODCM
> |- DutyCycle [ 0.00%]
> |- Power Management PWR MGMT [ LOCK]
> |- Energy Policy Bias Hint [ 0]
> |- Energy Policy HWP EPP [ 0]
> |- Junction Temperature TjMax [ 0:101]
> |- Digital Thermal Sensor DTS [Capable]
> |- Power Limit Notification PLN [Missing]
> |- Package Thermal Management PTM [Missing]
> |- Thermal Monitor 1 TM1 [ Enable]
> |- Thermal Monitor 2 TM2 [Capable]
> |- Thermal Design Power TDP [Missing]
> |- Minimum Power Min [Missing]
> |- Maximum Power Max [Missing]
> |- Units
> |- Power watt [ Missing]
> |- Energy joule [ Missing]
> |- Window second [ Missing]
```

cyring, 2020-08-12:
> I tried `make HWM_CHIPSET=W83627 clean all`, no change.

Perhaps, MAC is using another chipset. You can confirm with command `sensors`

> corefreq-cli -s -M:
```
Processor                                 [Intel(R) Xeon(R) CPU X5690 @ 3.47GHz]
|- Architecture                                                    [Westmere/EP]
|- Vendor ID                                                      [GenuineIntel]
|- Microcode                                                        [        31]
|- Signature                                                           [  06_2C]
|- Stepping                                                            [      2]
|- Online CPU                                                          [ 24/ 24]
|- Base Clock                                                          [132.986]
|- Frequency            (MHz)                      Ratio                        
                 Min   1595.84                    <  12 >                       
                 Max   3457.65                    <  26 >                       
|- Factory                                                             [133.333]
                       3470                       [  26 ]                       
|- Performance                                                                  
   |- OSPM                                                                      
                 TGT   8378.15                    <  63 >                       
|- Turbo Boost                                                         [   LOCK]
                  1C   3723.62                    <  28 >                       
                  2C   3723.62                    <  28 >                       
                  3C   3590.63                    <  27 >                       
                  4C   3590.63                    <  27 >                       
                  5C   3590.63                    <  27 >                       
                  6C   3590.63                    <  27 >                       
|- Uncore                                                              [   LOCK]
|- TDP                                                           Level [  0:0  ]
   |- Programmable                                                     [   LOCK]
   |- Configuration                                                    [   LOCK]
   |- Turbo Activation                                                 [   LOCK]
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [N]          AES [Y]  AVX/AVX2 [N/N] 
|- AVX512-F     [N]    AVX512-DQ [N]  AVX512-IFMA [N]   AVX512-PF [N] 
|- AVX512-ER    [N]    AVX512-CD [N]    AVX512-BW [N]   AVX512-VL [N] 
|- AVX512-VBMI  [N] AVX512-VBMI2 [N]  AVX512-VNMI [N]  AVX512-ALG [N] 
|- AVX512-VPOP  [N] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [N] 
|- AVX512-BF16  [N]  BMI1/BMI2 [N/N]         CLWB [N] CLFLUSH/O [Y/N] 
|- CLAC-STAC    [N]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y] 
|- F16C         [N]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y] 
|- MMX/Ext    [Y/N] MON/MWAITX [Y/N]        MOVBE [N]   PCLMULQDQ [Y] 
|- POPCNT       [Y]       RDRAND [N]       RDSEED [N]      RDTSCP [Y] 
|- SEP          [Y]          SHA [N]          SSE [Y]        SSE2 [Y] 
|- SSE3         [Y]        SSSE3 [Y]  SSE4.1/4A [Y/N]      SSE4.2 [Y] 
|- SERIALIZE    [N]      SYSCALL [Y]          SGX [N]       RDPID [N] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- Core Multi-Processing                                  CMP Legacy   [Missing]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Direct Cache Access                                           DCA   [Capable]
|- Debugging Extension                                            DE   [Capable]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Capable]
|- CPL Qualified Debug Store                                  DS-CPL   [Capable]
|- 64-Bit Debug Store                                         DTES64   [Capable]
|- Fast-String Operation                                Fast-Strings   [Missing]
|- Fused Multiply Add                                     FMA | FMA4   [Missing]
|- Hardware Lock Elision                                         HLE   [Missing]
|- Long Mode 64 bits                                       IA64 | LM   [Capable]
|- LightWeight Profiling                                         LWP   [Missing]
|- Machine-Check Architecture                                    MCA   [Capable]
|- Memory Protection Extensions                                  MPX   [Missing]
|- Model Specific Registers                                      MSR   [Capable]
|- Memory Type Range Registers                                  MTRR   [Capable]
[... 92 lines cut ...]
|- Digital Thermal Sensor                                        DTS   [Capable]
|- Power Limit Notification                                      PLN   [Missing]
|- Package Thermal Management                                    PTM   [Missing]
[... cut ...]
It may be set by the current cpufreq driver whom name is provided with shortcut `[k]` 
You could also start the driver with the options:
```
insmod corefreqk.ko Register_CPU_Freq=1 Register_Governor=1
```
These will give the control to _CoreFreq_ to govern the target frequencies you will select in the UI at `OSPM` rows when `TGT` ratio is surrounded by `<` and `>`
To allow registration, you however have to blacklist the current cpufreq driver, see Readme.
* Memory controller has not be decoded. `lspci -nn` should list the device ids I can into code.

hvico, 2020-08-12:
This is my sensors output:

```hvico@MacProLinux:~$ sensors
applesmc-isa-0300
Adapter: ISA adapter
PCI :         599 RPM  (min =  800 RPM, max = 4500 RPM)
PS :          599 RPM  (min =  600 RPM, max = 2800 RPM)
EXHAUST :     599 RPM  (min =  600 RPM, max = 2800 RPM)
INTAKE :      600 RPM  (min =  600 RPM, max = 2800 RPM)
BOOSTA :      601 RPM  (min =  800 RPM, max = 5200 RPM)
BOOSTB :      694 RPM  (min =  800 RPM, max = 5200 RPM)
TA0P:         +29.8°C  
TCAC:         +52.5°C  
TCAD:         +49.0°C  
TCAG:         +80.0°C  
TCAH:         +47.8°C  
TCAS:         +48.0°C  
TCBC:         +68.2°C  
TCBD:         +38.0°C  
TCBG:         +80.0°C  
TCBH:         +35.8°C  
TCBS:         +35.8°C  
TH1F:         +30.8°C  
TH1P:         +30.8°C  
TH1V:         +34.5°C  
TH2F:         +34.2°C  
TH2P:         +34.2°C  
TH2V:         +38.2°C  
TH3F:         +36.5°C  
TH3P:         +36.5°C  
TH3V:         +43.0°C  
TH4F:         +36.0°C  
TH4P:         +36.0°C  
TH4V:         +44.5°C  
THPS:         +44.5°C  
THTG:         +58.0°C  
TM1P:         +46.8°C  
TM2P:         +48.2°C  
TM2V:        -127.0°C  
TM3P:         +51.2°C  
TM3V:        -127.0°C  
TM4P:         +51.5°C  
TM5P:         +52.0°C  
TM6P:         +53.8°C  
TM6V:         +52.8°C  
TM7P:         +44.0°C  
TM7V:         +46.0°C  
TM8P:         +32.8°C  
TM8V:         +50.8°C  
TM9V:         +52.5°C  
TMA1:         +55.8°C  
TMA2:         +59.8°C  
TMA3:         +59.5°C  
TMA4:         +55.8°C  
TMB1:        -127.0°C  
TMB2:         +37.8°C  
TMB3:        -127.0°C  
TMB4:         +32.8°C  
TMHS:         +59.8°C  
TMLS:         +32.8°C  
TMPS:         +59.8°C  
TMPV:         +50.8°C  
TMTG:         +75.0°C  
TN0D:         +90.2°C  
TN0H:         +75.2°C  
TNTG:         +65.0°C  
Te1F:        -127.0°C  
Te1P:         +39.5°C  
Te1S:        -127.0°C  
Te2F:        -127.0°C  
[... 266 lines cut ...]
00:1e.0 PCI bridge [0604]: Intel Corporation 82801 PCI Bridge [8086:244e] (rev 90)
00:1f.0 ISA bridge [0601]: Intel Corporation 82801JIB (ICH10) LPC Interface Controller [8086:3a18]
00:1f.2 SATA controller [0106]: Intel Corporation 82801JI (ICH10 Family) SATA AHCI Controller [8086:3a22]
00:1f.3 SMBus [0c05]: Intel Corporation 82801JI (ICH10 Family) SMBus Controller [8086:3a30]
01:00.0 PCI bridge [0604]: Microsemi / PMC / IDT PES12T3G2 PCI Express Gen2 Switch [111d:8061] (rev 01)
02:02.0 PCI bridge [0604]: Microsemi / PMC / IDT PES12T3G2 PCI Express Gen2 Switch [111d:8061] (rev 01)
02:04.0 PCI bridge [0604]: Microsemi / PMC / IDT PES12T3G2 PCI Express Gen2 Switch [111d:8061] (rev 01)
03:00.0 USB controller [0c03]: Fresco Logic FL1100 USB 3.0 Host Controller [1b73:1100] (rev 10)
04:00.0 SATA controller [0106]: Samsung Electronics Co Ltd Device [144d:a801] (rev 01)
05:00.0 VGA compatible controller [0300]: NVIDIA Corporation GK104 [GeForce GTX 680] [10de:1180] (rev a1)
05:00.1 Audio device [0403]: NVIDIA Corporation GK104 HDMI Audio Controller [10de:0e0a] (rev a1)
06:00.0 PCI bridge [0604]: ASMedia Technology Inc. Device [1b21:2824] (rev 01)
07:00.0 PCI bridge [0604]: ASMedia Technology Inc. Device [1b21:2824] (rev 01)
07:08.0 PCI bridge [0604]: ASMedia Technology Inc. Device [1b21:2824] (rev 01)
08:00.0 SATA controller [0106]: Samsung Electronics Co Ltd Device [144d:a801] (rev 01)
09:00.0 Non-Volatile memory controller [0108]: Samsung Electronics Co Ltd NVMe SSD Controller SM981/PM981 [144d:a808]
0c:00.0 Ethernet controller [0200]: Intel Corporation 82574L Gigabit Network Connection [8086:10f6]
0d:00.0 Ethernet controller [0200]: Intel Corporation 82574L Gigabit Network Connection [8086:10f6]
0e:00.0 PCI bridge [0604]: Texas Instruments XIO2213A/B/XIO2221 PCI Express to PCI Bridge [Cheetah Express] [104c:823e] (rev 01)
0f:00.0 FireWire (IEEE 1394) [0c00]: Texas Instruments XIO2213A/B/XIO2221 IEEE-1394b OHCI Controller [Cheetah Express] [104c:823f] (rev 01)
10:00.0 Network controller [0280]: Broadcom Inc. and subsidiaries BCM4322 802.11a/b/g/n Wireless LAN Controller [14e4:432b] (rev 01)```

The only "memory controller" there is related to the NVM SSD I have, it might be one of that long list of Intel Chipsets.

BTW, thanks for the antiquote tip.

cyring, 2020-08-13:
* According to your PCI list, you may already have the IOMMU state ? 
Identifiers here:
https://github.com/cyring/CoreFreq/blob/befddf58b4f25e8b8955274dc89815e863fdbf0c/corefreq-api.h#L921

The Memory Controller should fall in the `8086:34xx` series. I have to find and check against the `X5690` datasheet which I hope is similar to my `W3690`

* The Apple ISA chipset will be more tuff, b/c no Vcore already listed  by sensors . I've to check its kernel driver source code. Let me know if you find any registers datasheet.

* Setting a target ratio above than the available ratios brings no performance. Reason I'm providing a cpufreq compatible code in _CoreFreq_

cyring, 2020-08-13:
Checking the DID in Table 2.1 of  Xeon 5600 [datasheet volume 2](https://www.intel.com/content/dam/www/public/us/en/documents/datasheets/xeon-5600-vol-2-datasheet.pdf), the IMC should be listed on Bus `0xff`

My Xeon W3690 complete `lspci -n`returns:
```
00:00.0 0600: 8086:3405 (rev 12)
00:01.0 0604: 8086:3408 (rev 12)
00:03.0 0604: 8086:340a (rev 12)
00:07.0 0604: 8086:340e (rev 12)
00:14.0 0800: 8086:342e (rev 12)
00:14.1 0800: 8086:3422 (rev 12)
00:14.2 0800: 8086:3423 (rev 12)
00:14.3 0800: 8086:3438 (rev 12)
00:1a.0 0c03: 8086:3a37
00:1a.1 0c03: 8086:3a38
00:1a.2 0c03: 8086:3a39
00:1a.7 0c03: 8086:3a3c
00:1b.0 0403: 8086:3a3e
00:1c.0 0604: 8086:3a40
00:1c.2 0604: 8086:3a44
00:1c.4 0604: 8086:3a48
00:1d.0 0c03: 8086:3a34
00:1d.1 0c03: 8086:3a35
00:1d.2 0c03: 8086:3a36
00:1d.7 0c03: 8086:3a3a
00:1e.0 0604: 8086:244e (rev 90)
00:1f.0 0601: 8086:3a16
00:1f.2 0106: 8086:3a22
00:1f.3 0c05: 8086:3a30
02:00.0 0300: 1002:3e50
05:00.0 0200: 10ec:8168 (rev 02)
ff:00.0 0600: 8086:2c70 (rev 02)
ff:00.1 0600: 8086:2d81 (rev 02)
ff:02.0 0600: 8086:2d90 (rev 02)
ff:02.1 0600: 8086:2d91 (rev 02)
ff:02.2 0600: 8086:2d92 (rev 02)
ff:02.3 0600: 8086:2d93 (rev 02)
ff:03.0 0600: 8086:2d98 (rev 02)
ff:03.1 0600: 8086:2d99 (rev 02)
ff:03.4 0600: 8086:2d9c (rev 02)
ff:04.0 0600: 8086:2da0 (rev 02)
ff:04.1 0600: 8086:2da1 (rev 02)
ff:04.2 0600: 8086:2da2 (rev 02)
ff:04.3 0600: 8086:2da3 (rev 02)
ff:05.0 0600: 8086:2da8 (rev 02)
ff:05.1 0600: 8086:2da9 (rev 02)
ff:05.2 0600: 8086:2daa (rev 02)
ff:05.3 0600: 8086:2dab (rev 02)
ff:06.0 0600: 8086:2db0 (rev 02)
ff:06.1 0600: 8086:2db1 (rev 02)
ff:06.2 0600: 8086:2db2 (rev 02)
ff:06.3 0600: 8086:2db3 (rev 02)
```

To list the missing DID, try one of these solutions:
1. Query the PCI ID database for all ID's via DNS (see `lspci -h`)
```
lspci -nn -Q
```
2. Boot my ArchLinux [Live Image](https://github.com/cyring/CoreFreq/wiki/Live-CD) which has a kernel 5.x version.  
Exit from the _CoreFreq_ UI to go back to the Shell as root then run the `lspci`

3. Update the Ubuntu PCI package & probably its kernel version

4. If this MAC has a BIOS, check if some PCIs are marked as hidden

cyring, 2020-08-19:
I came across the [applesmc source code](https://elixir.bootlin.com/linux/latest/source/drivers/hwmon/applesmc.c), I'm afraid no Core voltage is available.

hvico, 2020-08-19:
OK, thanks.

I'll give this a try:

[https://github.com/dkorunic/iSMC](url)

cyring, 2020-08-19:
> OK, thanks.
> 
> I'll give this a try:
> 
> [https://github.com/dkorunic/iSMC](url)

Nice finding. 
Probably the key `VCOC` 
 https://github.com/dkorunic/iSMC/blob/985e01dbd2020bbd25b77b66ac3b0e66c90f9d8c/smc/sensors.go#L390

I'm also noticing keys for Powers. It would be great to add them into the driver.

cyring, 2020-09-14:
* For your tests, here is the alpha version of the multiple column upper view.
[CoreFreq_alpha_20200914.tar.gz](https://github.com/cyring/CoreFreq/files/5218369/CoreFreq_alpha_20200914.tar.gz)

* The number of columns is increased with key **`>`** and decreased with **`<`**
  Up to 8 columns
  As a power of 2 when above than 1

* Results are not great and various **things** are still **broken** 😞

* I'm thinking about rolling back and try something else.
  Using my windowing API : _any view as a window_

[image]
[image]
[image]

cyring, 2020-09-18:
* I'm aware it does not answer to the Multiple Column view but to gain some space the current `develop` branch now allows you to do this:

[image]
  Just build this way:
```sh
make NO_LOWER=1 NO_HEADER=1 NO_FOOTER=1
```
* You can also build other layouts.

[image]

* Or

[image]

* The Makefile recalls these new options
```sh
make help
```
```
...
|                                                               |
|  User Interface Layout:                                       |
|    NO_HEADER=<F>  NO_FOOTER=<F>  NO_UPPER=<F>  NO_LOWER=<F>   |
|      when <F> is 1 don't build and display this area part     |
|                                                               |
```

hvico, 2020-09-19:
Hello! Thanks for the update. Tried the alpha version but it seems the GUI seems to be buggy as you said.

This is how it looks in my machine, some cores do not show and instead I get some hyphens:

[image]

I'll try the development branch, that compact view it is somehow a workaround. Thank you!

cyring, 2020-09-20:
> I'll try the development branch, that compact view it is somehow a workaround. Thank you!

* Thanks for trying. It would be easier to program those columns with a GUI framework.
  I save the idea for future developments.

* PCI probing has been refactored in the `develop` branch.
  Please let me also know if your IMC, , Uncore frequency ratios, VT-d and other X58 's are showing up ?

cyring, 2020-11-20:
Some day _CoreFreq_ will have a GUI.
So far closing the issue
Regards
CyrIng

closed 2020-11-20 by cyring