cyring/CoreFreq #202: dkms breaks when /bin/sh is not bash

John-Gee, 2020-09-06:
With dash this is what I get:

==> dkms install --no-depmod -m corefreqk -v 1.80 -k 5.8.7-arch1-1
/var/lib/dkms/corefreqk/1.80/build/scripter.sh: 7: 9: not found
/var/lib/dkms/corefreqk/1.80/source/scripter.sh: 7: 10: not found

If those scripts depend strongly on bashisms, it probably would be best to then ask for /bin/bash instead of /bin/sh

cyring, 2020-09-06:
Hello,

Before I bring some changes, can you please edit `scripter.sh` and replace the first line with:
```
#!/bin/bash
```
Regards

John-Gee, 2020-09-06:
Yeah I already did that before writing that bug report. :) (Well that and replacing sh by bash in my system too, both options work).

Thank you!

cyring, 2020-09-06:
> Yeah I already did that before writing that bug report. :) (Well that and replacing sh by bash in my system too, both options work).
> 
> Thank you!

Thank you very much, I will send this fix in next commit.

closed 2020-09-06 by John-Gee