cyring/CoreFreq #204: compatibility with 5.9 kernel

thor2002ro, 2020-09-15:
https://github.com/cyring/CoreFreq/blob/13d69a8dbf757ea472b55c76a013ec10de45ec87/corefreqk.c#L11261

corefreqk.c:11353:19: error: assignment to ‘int (*)(struct cpuidle_device *, struct cpuidle_driver *, int)’ from incompatible pointer type ‘void (*)(struct cpuidle_device *, struct cpuidle_driver *, int)’ [-Werror=incompatible-pointer-types]
11353 |    ].enter_s2idle = CoreFreqK_S2IdleHandler;
      |                   ^

probably changing into this will do it....
`static int CoreFreqK_S2IdleHandler(struct cpuidle_device *pIdleDevice,
				struct cpuidle_driver *pIdleDriver, int index)
{
	unsigned long MWAIT=(CoreFreqK.IdleDriver.states[index].flags>>24)&0xff;
	mwait_idle_with_hints(MWAIT, 1UL);
	return index;
}`

cyring, 2020-09-15:
Thank you for letting me know. 
I'm not in 5.9 yet but I will because of this [patch](https://patchwork.kernel.org/patch/11686311/). 
Fix will be delivered within the development branch in a few hours.

cyring, 2020-09-16:
For your tests, the Linux v5.9 hot-fix is available in the branch `develop`

thor2002ro, 2020-09-16:
I don't know if its an issue with 5.9 kernel(using 5.9RC5) or my distro(unraid based on Slack)

module loads fine 
```
[  386.583297] corefreqk: module is from the staging directory, the quality is unknown, you have been warned.
[  386.650290] CoreFreq(5:13): Processor [ 8F_08] Architecture [Zen+ Pinnacle Ridge] SMT [16/16]
```

but daemon fails to run....
```
root@Tower:/dev# corefreqd -q
Driver connection error code 13
Version 1.80.9: 'Permission denied' @ line 5940
```

says 1.80.9 because its a version built with dev and the patch I suggested on top basically the same that you commited on 1.81

any ideas?

cyring, 2020-09-16:
* You can't mix versions. 
The version has to be the same among the 3 programs: Driver, Daemon and Client.
The version number is stored into the shared memories to make sure the structured data are consistent.
* I you prefer I can push to the GitHub the hot-fix branch which is `master` branch based. Using it you may have a chance to apply your patch.

thor2002ro, 2020-09-16:
success... that was it I made a mistake in the build script and didn't pull the develop branch

interesting that I/O MMU shows as "off" and I'm pretty sure its on I'm using it frequently using passthrough gpu to vm

[image]

```
root@Tower:~# corefreq-cli -s -M
Processor                                [AMD Ryzen 7 2700 Eight-Core Processor]
|- Architecture                                            [Zen+ Pinnacle Ridge]
|- Vendor ID                                                      [AuthenticAMD]
|- Microcode                                                        [0x0800820d]
|- Signature                                                           [  8F_08]
|- Stepping                                                            [      2]
|- Online CPU                                                          [ 16/ 16]
|- Base Clock                                                          [100.001]
|- Frequency            (MHz)                      Ratio                        
                 Min   1500.01                    <  15 >                       
                 Max   3200.02                    <  32 >                       
|- Factory                                                             [100.000]
                       3200                       [  32 ]                       
|- Performance                                                                  
   |- P-State                                                                   
                 TGT   3200.05                    <  32 >                       
|- Turbo Boost                                                         [ UNLOCK]
                 XFR   4200.02                    [  42 ]                       
                 CPB   4000.02                    [  40 ]                       
                  1C   2800.02                    <  28 >                       
                  2C   1500.01                    <  15 >                       
|- Uncore                                                              [   LOCK]
                 Min   1600.01                    [  16 ]                       
                 Max   1600.01                    [  16 ]                       
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AVX512-F     [N]    AVX512-DQ [N]  AVX512-IFMA [N]   AVX512-PF [N] 
|- AVX512-ER    [N]    AVX512-CD [N]    AVX512-BW [N]   AVX512-VL [N] 
|- AVX512-VBMI  [N] AVX512-VBMI2 [N]  AVX512-VNMI [N]  AVX512-ALG [N] 
|- AVX512-VPOP  [N] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [N] 
|- AVX512-BF16  [N]  BMI1/BMI2 [Y/Y]         CLWB [N] CLFLUSH/O [Y/Y] 
|- CLAC-STAC    [Y]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y] 
|- F16C         [Y]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y] 
|- MMX/Ext    [Y/Y] MON/MWAITX [Y/Y]        MOVBE [Y]   PCLMULQDQ [Y] 
|- POPCNT       [Y]       RDRAND [Y]       RDSEED [Y]      RDTSCP [Y] 
|- SEP          [Y]          SHA [Y]          SSE [Y]        SSE2 [Y] 
|- SSE3         [Y]        SSSE3 [Y]  SSE4.1/4A [Y/Y]      SSE4.2 [Y] 
|- SERIALIZE    [N]      SYSCALL [Y]        RDPID [N]        UMIP [N] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- 100 MHz multiplier Control                            100MHzSteps   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- Core Multi-Processing                                  CMP Legacy   [Capable]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Capable]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Missing]
|- CPL Qualified Debug Store                                  DS-CPL   [Missing]
|- 64-Bit Debug Store                                         DTES64   [Missing]
|- Fast-String Operation                                Fast-Strings   [Missing]
|- Fused Multiply Add                                     FMA | FMA4   [Capable]
|- Hardware Lock Elision                                         HLE   [Missing]
|- Instruction Based Sampling                                    IBS   [Missing]
|- Long Mode 64 bits                                       IA64 | LM   [Capable]
|- LightWeight Profiling                                         LWP   [Missing]
|- Machine-Check Architecture                                    MCA   [Capable]
|- Memory Protection Extensions                                  MPX   [Missing]
|- Model Specific Registers                                      MSR   [Capable]
|- Memory Type Range Registers                                  MTRR   [Capable]
[... 88 lines cut ...]
   |- Window                                            second   [  0.000976562]
                              Zen UMC  [1463]                              
Controller #0                                                Dual Channel  
 Bus Rate  1666 MT/s      Bus Speed 1666 MHz           DRAM Speed 3333 MHz 
[... cut ...]
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1    16    2     65536      1024          16384                    
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1    16    2     65536      1024          16384                    
root@Tower:~# 
```

cyring, 2020-09-16:
> success... that was it I made a mistake in the build script and didn't pull the develop branch

Thanks for your confirmation and your outputs.

> interesting that I/O MMU shows as "off" and I'm pretty sure its on I'm using it frequently using passthrough gpu to vm

Same issue here with Matisse. Thanks for the tip :100: 
Tested in the Host and the Guest (with QEMU/KVM enabled)

[image]
[image]

To debug:
* query happens at :
https://github.com/cyring/CoreFreq/blob/048d00e5d1c46014faa34b76db74506f4a942e4f/corefreqk.c#L4082
* probing list starts in :
https://github.com/cyring/CoreFreq/blob/048d00e5d1c46014faa34b76db74506f4a942e4f/corefreqk.h#L1768
* aggregation status at :
https://github.com/cyring/CoreFreq/blob/048d00e5d1c46014faa34b76db74506f4a942e4f/corefreqd.c#L3832
* printed at :
https://github.com/cyring/CoreFreq/blob/048d00e5d1c46014faa34b76db74506f4a942e4f/corefreq-cli.c#L2498

thor2002ro, 2020-09-16:
the pci device looks fine for Pinnacle Ridge probably the problem is with the aggregation status....

```
root@Tower:~# lspci -nn
00:00.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Root Complex [1022:1450]
00:00.2 IOMMU [0806]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) I/O Memory Management Unit [1022:1451]
00:01.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge [1022:1452]
00:01.3 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe GPP Bridge [1022:1453]
00:02.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge [1022:1452]
00:03.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge [1022:1452]
00:03.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe GPP Bridge [1022:1453]
00:03.2 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe GPP Bridge [1022:1453]
00:04.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge [1022:1452]
00:07.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge [1022:1452]
00:07.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Internal PCIe GPP Bridge 0 to Bus B [1022:1454]
00:08.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge [1022:1452]
00:08.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Internal PCIe GPP Bridge 0 to Bus B [1022:1454]
00:14.0 SMBus [0c05]: Advanced Micro Devices, Inc. [AMD] FCH SMBus Controller [1022:790b] (rev 59)
00:14.3 ISA bridge [0601]: Advanced Micro Devices, Inc. [AMD] FCH LPC Bridge [1022:790e] (rev 51)
00:18.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 0 [1022:1460]
00:18.1 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 1 [1022:1461]
00:18.2 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 2 [1022:1462]
00:18.3 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 3 [1022:1463]
00:18.4 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 4 [1022:1464]
00:18.5 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 5 [1022:1465]
00:18.6 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 6 [1022:1466]
00:18.7 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 7 [1022:1467]
01:00.0 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] Device [1022:43d0] (rev 01)
01:00.1 SATA controller [0106]: Advanced Micro Devices, Inc. [AMD] 400 Series Chipset SATA Controller [1022:43c8] (rev 01)
01:00.2 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 400 Series Chipset PCIe Bridge [1022:43c6] (rev 01)
02:00.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 400 Series Chipset PCIe Port [1022:43c7] (rev 01)
02:01.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 400 Series Chipset PCIe Port [1022:43c7] (rev 01)
02:02.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 400 Series Chipset PCIe Port [1022:43c7] (rev 01)
02:03.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 400 Series Chipset PCIe Port [1022:43c7] (rev 01)
02:04.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 400 Series Chipset PCIe Port [1022:43c7] (rev 01)
02:09.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 400 Series Chipset PCIe Port [1022:43c7] (rev 01)
03:00.0 SATA controller [0106]: Marvell Technology Group Ltd. Device [1b4b:9215] (rev 11)
05:00.0 Ethernet controller [0200]: Intel Corporation I211 Gigabit Network Connection [8086:1539] (rev 03)
06:00.0 Network controller [0280]: Intel Corporation Wireless-AC 9260 [8086:2526] (rev 29)
08:00.0 USB controller [0c03]: ASMedia Technology Inc. ASM1143 USB 3.1 Host Controller [1b21:1343]
09:00.0 VGA compatible controller [0300]: NVIDIA Corporation GM204 [GeForce GTX 970] [10de:13c2] (rev a1)
09:00.1 Audio device [0403]: NVIDIA Corporation GM204 High Definition Audio Controller [10de:0fbb] (rev a1)
0a:00.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Baffin [Radeon RX 550 640SP / RX 560/560X] [1002:67ff] (rev ff)
0a:00.1 Audio device [0403]: Advanced Micro Devices, Inc. [AMD/ATI] Baffin HDMI/DP Audio [Radeon RX 550 640SP / RX 560/560X] [1002:aae0]
0b:00.0 Non-Essential Instrumentation [1300]: Advanced Micro Devices, Inc. [AMD] Zeppelin/Raven/Raven2 PCIe Dummy Function [1022:145a]
0b:00.2 Encryption controller [1080]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Platform Security Processor [1022:1456]
0b:00.3 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] Zeppelin USB 3.0 Host controller [1022:145f]
0c:00.0 Non-Essential Instrumentation [1300]: Advanced Micro Devices, Inc. [AMD] Zeppelin/Renoir PCIe Dummy Function [1022:1455]
0c:00.2 SATA controller [0106]: Advanced Micro Devices, Inc. [AMD] FCH SATA Controller [AHCI mode] [1022:7901] (rev 51)
0c:00.3 Audio device [0403]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) HD Audio Controller [1022:1457]
root@Tower:~# 
```

thor2002ro, 2020-09-16:
i got a question does doing something like an execution script be any problem? assuming its ran as root of course... 
```
#!/bin/bash
modprobe corefreqk

corefreqd -q &
corefreq-cli
pkill -f -- "corefreqd -q"
sleep 1
modprobe -r corefreqk
```

cyring, 2020-09-16:
* We had successfully tested the IOMMU _BIOS state_ in the issue #193 
* Version 1.79.5 was the code related to this feature
https://github.com/cyring/CoreFreq/commit/d38ca25c61ecfb2f46d6ff86cd40db7cb72e5077
* But today we don't have the expected result ! 
I'm checking my current BIOS in which all settings are enabled:
1. `IOMMU` is enabled
1. `SR-IOV` is enabled
1. `AMD-V` is enabled

thor2002ro, 2020-09-16:
here is a 3500u laptop same  iommu problem....
```
corefreq-cli -s -M
Processor                        [AMD Ryzen 5 3500U with Radeon Vega Mobile Gfx]
|- Architecture                                                   [Zen+ Picasso]
|- Vendor ID                                                      [AuthenticAMD]
|- Microcode                                                        [0x08108102]
|- Signature                                                           [  8F_18]
|- Stepping                                                            [      1]
|- Online CPU                                                          [  8/  8]
|- Base Clock                                                          [ 99.813]
|- Frequency            (MHz)                      Ratio                        
                 Min   1397.38                    <  14 >                       
                 Max   2096.07                    <  21 >                       
|- Factory                                                             [100.000]
                       2100                       [  21 ]                       
|- Performance                                                                  
   |- P-State                                                                   
                 TGT   2096.02                    <  21 >                       
|- Turbo Boost                                                         [   LOCK]
                  1C   1696.82                    <  17 >                       
                  2C   1397.38                    <  14 >                       
|- Uncore                                                              [   LOCK]
                 Min   1197.76                    [  12 ]                       
                 Max   1197.76                    [  12 ]                       
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AVX512-F     [N]    AVX512-DQ [N]  AVX512-IFMA [N]   AVX512-PF [N] 
|- AVX512-ER    [N]    AVX512-CD [N]    AVX512-BW [N]   AVX512-VL [N] 
|- AVX512-VBMI  [N] AVX512-VBMI2 [N]  AVX512-VNMI [N]  AVX512-ALG [N] 
|- AVX512-VPOP  [N] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [N] 
|- AVX512-BF16  [N]  BMI1/BMI2 [Y/Y]         CLWB [N] CLFLUSH/O [Y/Y] 
|- CLAC-STAC    [Y]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y] 
|- F16C         [Y]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y] 
|- MMX/Ext    [Y/Y] MON/MWAITX [Y/Y]        MOVBE [Y]   PCLMULQDQ [Y] 
|- POPCNT       [Y]       RDRAND [Y]       RDSEED [Y]      RDTSCP [Y] 
|- SEP          [Y]          SHA [Y]          SSE [Y]        SSE2 [Y] 
|- SSE3         [Y]        SSSE3 [Y]  SSE4.1/4A [Y/Y]      SSE4.2 [Y] 
|- SERIALIZE    [N]      SYSCALL [Y]        RDPID [N]        UMIP [N] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- 100 MHz multiplier Control                            100MHzSteps   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- Core Multi-Processing                                  CMP Legacy   [Capable]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Capable]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Missing]
|- CPL Qualified Debug Store                                  DS-CPL   [Missing]
|- 64-Bit Debug Store                                         DTES64   [Missing]
|- Fast-String Operation                                Fast-Strings   [Missing]
|- Fused Multiply Add                                     FMA | FMA4   [Capable]
|- Hardware Lock Elision                                         HLE   [Missing]
|- Instruction Based Sampling                                    IBS   [Missing]
|- Long Mode 64 bits                                       IA64 | LM   [Capable]
|- LightWeight Profiling                                         LWP   [Missing]
|- Machine-Check Architecture                                    MCA   [Capable]
|- Memory Protection Extensions                                  MPX   [Missing]
|- Model Specific Registers                                      MSR   [Capable]
|- Memory Type Range Registers                                  MTRR   [Capable]
|- No-Execute Page Protection                                     NX   [Capable]
|- OS-Enabled Ext. State Management                          OSXSAVE   [Capable]
|- Physical Address Extension                                    PAE   [Capable]
|- Page Attribute Table                                          PAT   [Capable]
|- Pending Break Enable                                          PBE   [Missing]
|- Process Context Identifiers                                  PCID   [Missing]
|- Perfmon and Debug Capability                                 PDCM   [Missing]
[... 161 lines cut ...]
[... cut ...]
	00:18.5 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 Device 24: Function 5 [1022:15ed]
	00:18.6 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 Device 24: Function 6 [1022:15ee]
	00:18.7 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 Device 24: Function 7 [1022:15ef]
IOMMU Group 8:
	01:00.0 SD Host controller [0805]: O2 Micro, Inc. SD/MMC Card Reader Controller [1217:8621] (rev 01)
IOMMU Group 9:
	02:00.0 Network controller [0280]: Realtek Semiconductor Co., Ltd. RTL8822BE 802.11a/b/g/n/ac WiFi adapter [10ec:b822]

```

cyring, 2020-09-16:
> i got a question does doing something like an execution script be any problem? assuming its ran as root of course...
> 
> ```
> #!/bin/bash
> modprobe corefreqk
> 
> corefreqd -q &
> corefreq-cli
> pkill -f -- "corefreqd -q"
> sleep 1
> modprobe -r corefreqk
> ```

Script looks fine. 

By design, only ony driver, one daemon, and one client (optionally one GUI) are granted to make the CoreFreq _chain_

Although Programs are built with access protections, you are doing right to sleep in the script.

Be however aware that the Daemon is forked into two processes: The first to exchange / collect from the Driver, the second to run the stress algorithms. 
Both spawn one thread per CPU and additional ring and signaling threads. 
Thus you will have to kill the first Daemon process which in turn will terminate the second process as its child process. 
Usually the main process has a lower `pid` than the child process.
Another discriminant way is to kill the `pid` owning the Manager thread named `"corefreqd-pmgr"` (whereas its child is called `"corefreqd-cmgr"`)

Just send standard signals to the Daemon and Client, like `SIGTERM` and `SIGQUIT`
`SIGKILL` has to be avoided during a nominal execution of _CoreFreq_

Don't force remove the kernel module while the Daemon is still running.

cyring, 2020-09-16:
The _CoreFreq_ [live image](http://corefreq.blogspot.com/2018/07/how-to-build-corefreq-live-cd.html) embeds scripts you may find useful. 
Go to the section:
> add the CoreFreq helper scripts

thor2002ro, 2020-09-16:
cool thanks
I usually avoid using pids 

this should send SIGQUIT to the daemon after the cli is stooped... :) 
`pkill -QUIT -f -- "corefreqd -q"`

cyring, 2020-09-16:
> cool thanks
> I usually avoid using pids
> 
> this should send SIGQUIT to the daemon after the cli is stooped... :)
> `pkill -QUIT -f -- "corefreqd -q"`

I would recommend to add a condition before removing the driver at:
```
modprobe -r corefreqk
```
For example test if the Daemon exit code is zero

cyring, 2020-09-19:
Hello,

I have found the IOMMU state regression (which was introduced with the IMC implementation)

Here is my result:
[image]

For your tests, the fix is part of the branch `develop`

Regards

thor2002ro, 2020-09-23:
[image]
works

cyring, 2020-09-23:
* Perfect, thank you for your confirmation.

* Do you have disable any `SPREAD SPECTRUM` in your BIOS ?
  If not, please disable it and let me know if you get  a _CoreFreq_ Base Clock estimation closer to `100.0` MHz

thor2002ro, 2020-09-25:
nope no SPREAD SPECTRUM in bios....

cyring, 2020-09-25:
Here is the current unpublished CPU-Idle developments for Zen

[CoreFreq_develop.tar.gz](https://github.com/cyring/CoreFreq/files/5283267/CoreFreq_develop.tar.gz)

The backend kernel handler I'm registering will invoke `mwait` but my 3950X appears to freeze somewhere during its sleep states.

I was thinking it was a bug within the first generation of Zen , solved by using `halt` instead of `mwait`

Please let me know if you can give it a try:
1. first, you ask the registration in `Settings`
[image]
1. next you check in `Kernel` if `corefreqk-idle` is registered
[image]
1. let system goes to idle for a while ... 
1. and if Processor is freezing or not ?

* I'm also verifying if disabling any C6 make a difference
[image]

cyring, 2020-09-25:
**EDIT** b/c I'm registering all my subdrivers, my boot cmdline has the followings:
```
modprobe.blacklist=acpi_cpufreq idle=halt notsc nowatchdog
```

cyring, 2020-09-26:
It appears the freezing issue is due to a SMU concurrent access with the `edac_mce_amd` module
See operations in https://github.com/cyring/CoreFreq/issues/115#issuecomment-699469580

cyring, 2020-11-20:
Hello,
Can you tell if you have remaining issues ?
Regards,
CyrIng

closed 2020-11-21 by thor2002ro