cyring/CoreFreq #205: Unable to adjust Turbo Frequencies on E5-1650 V2 Processor

Gamebag, 2020-10-01:
When I try to change turbo frequencies to any value, I get this message. 

[image]
However, it applies the turbo frequency to the MSR read from i7z, but it is still ignored by the system if higher than 3.6 GHz on any core. 
[image]

If there is any more information you need I'd be glad to provide it.

cyring, 2020-10-01:
Hello,
Do you confirm you are following this path ? 
```
Processor > Turbo Boost > 1C
```
And Turbo ratios are unlocked for this Xeon ?

Gamebag, 2020-10-01:
That is a Yes to both questions. When I am in windows, I am able to overclock my Xeon processor with their XTU utility.

cyring, 2020-10-02:
> That is a Yes to both questions. When I am in windows, I am able to overclock my Xeon processor with their XTU utility.

OK, try this driver argument:
```
insmod corefreqk.ko Turbo_Ratio_Unlock=1
```

cyring, 2020-10-02:
* TODO for future version:
[STH](https://forums.servethehome.com/index.php?threads/xeons-with-an-open-multiplier.15983/post-153522) unlocked Xeon list for specific tables addition into `corefreqk.h`

Gamebag, 2020-10-02:
Turbo_Ratio_Unlock is equal to 1 but no change has happened

Processor                            [Intel(R) Xeon(R) CPU E5-1650 v2 @ 3.50GHz]
|- Architecture                                                   [IvyBridge/EP]
|- Vendor ID                                                      [GenuineIntel]
|- Microcode                                                        [0x0000042e]
|- Signature                                                           [  06_3E]
|- Stepping                                                            [      4]
|- Online CPU                                                          [ 12/ 12]
|- Base Clock                                                          [ 99.759]
|- Frequency            (MHz)                      Ratio                        
                 Min   1197.11                    <  12 >                       
                 Max   3491.58                    <  35 >                       
|- Factory                                                             [100.000]
                       3500                       [  35 ]                       
|- Performance                                                                  
   |- P-State                                                                   
                 TGT   3990.37                    <  40 >                       
|- Turbo Boost                                                         [ UNLOCK]
                  1C   3990.37                    <  40 >                       
                  2C   3990.37                    <  40 >                       
                  3C   3990.37                    <  40 >                       
                  4C   3990.37                    <  40 >                       
                  5C   3990.37                    <  40 >                       
                  6C   3990.37                    <  40 >                       
                  7C   3990.37                    <  40 >                       
                  8C   3990.37                    <  40 >                       
|- Uncore                                                              [   LOCK]
                 Min   1197.11                    [  12 ]                       
                 Max   3491.58                    [  35 ]                       
|- TDP                                                           Level [  0:0  ]
   |- Programmable                                                     [ UNLOCK]
   |- Configuration                                                    [   LOCK]
   |- Turbo Activation                                                 [   LOCK]
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [N]          AES [Y]  AVX/AVX2 [Y/N] 
|- AVX512-F     [N]    AVX512-DQ [N]  AVX512-IFMA [N]   AVX512-PF [N] 
|- AVX512-ER    [N]    AVX512-CD [N]    AVX512-BW [N]   AVX512-VL [N] 
|- AVX512-VBMI  [N] AVX512-VBMI2 [N]  AVX512-VNMI [N]  AVX512-ALG [N] 
|- AVX512-VPOP  [N] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [N] 
|- AVX512-BF16  [N]  BMI1/BMI2 [N/N]         CLWB [N] CLFLUSH/O [Y/N] 
|- CLAC-STAC    [N]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y] 
|- F16C         [Y]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y] 
|- MMX/Ext    [Y/N] MON/MWAITX [Y/N]        MOVBE [N]   PCLMULQDQ [Y] 
|- POPCNT       [Y]       RDRAND [Y]       RDSEED [N]      RDTSCP [Y] 
|- SEP          [Y]          SHA [N]          SSE [Y]        SSE2 [Y] 
|- SSE3         [Y]        SSSE3 [Y]  SSE4.1/4A [Y/N]      SSE4.2 [Y] 
|- SERIALIZE    [N]      SYSCALL [Y]          SGX [N]       RDPID [N] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- Core Multi-Processing                                  CMP Legacy   [Missing]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Direct Cache Access                                           DCA   [Capable]
|- Debugging Extension                                            DE   [Capable]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Capable]
|- CPL Qualified Debug Store                                  DS-CPL   [Capable]
|- 64-Bit Debug Store                                         DTES64   [Capable]
|- Fast-String Operation                                Fast-Strings   [Capable]
|- Fused Multiply Add                                     FMA | FMA4   [Missing]
|- Hardware Lock Elision                                         HLE   [Missing]
|- Instruction Based Sampling                                    IBS   [Missing]
|- Long Mode 64 bits                                       IA64 | LM   [Capable]
|- LightWeight Profiling                                         LWP   [Missing]
|- Machine-Check Architecture                                    MCA   [Capable]
|- Memory Protection Extensions                                  MPX   [Missing]
|- Model Specific Registers                                      MSR   [Capable]
[... 82 lines cut ...]
[... cut ...]
|- Units                                                                        
   |- Power                                               watt   [  0.125000000]
   |- Energy                                             joule   [  0.000015259]
   |- Window                                            second   [  0.000976562]

The multipliers are all at 40x from previous testing. I've also realized that the CPU doesn't turbo above 3.6 even in single threads when it should be at 3.9. I don't know if its because I changed the multiplier or not.

cyring, 2020-10-02:
* What my `Xeon W3690`  seems to share with your `Xeon E5-1650 v2` is that they have unpublished Turbo unlock capability.
  `W3690` is already registered as an exception into _CoreFreq_ tables. `E5-1650 v2` is something to add in source code; reason I have asked to add the argument `Turbo_Ratio_Unlock=1`
  Check with `dmesg` for no spelling error warned by the driver.
* Depending on the architecture, Boost is triggered, depending on the Target P-State `TGT`, and off course the Turbo Boost bit enabled
  With Nehalem/Westmere, Target ratio must be the Max ratio + one
  With Sandy Bridge and above, Target ratio is any value above the Max ratio (yours is `35`)
* Thus I'm finding more _confortable_ to have a full control over the P-States by disabling any other drivers
  Here is my boot command line:
```
nmi_watchdog=0 modprobe.blacklist=pcspkr,iTCO_wdt,acpi_cpufreq,pcc_cpufreq,intel_cstate,intel_uncore,intel_powerclamp,i7core_edac,i5500_temp,coretemp,asus_atk0110 idle=halt intel_pstate=disable cpufreq.off=0
```
* Then I load _CoreFreq_ driver with those arguments:
```
insmod corefreqk.ko RDPMC_Enable=1 Register_ClockSource=1 Register_Governor=1 Register_CPU_Freq=1 Register_CPU_Idle=1 Turbo_Ratio_Unlock=1
```
Note: `Turbo_Ratio_Unlock=1` is added for your case
* Next just start the Daemon and the Client with the following usage in the UI
1. Turbo is OFF
[image]
2. Enable Turbo 
[image]
3. Turbo is ON
  You are noticing that the Target ratio `TGT` has been increased
[image]
4. Change the single Core Turbo ratio in `1C`
  Select the first line `Processor` because ratio is common to all Cores
[image]
  Here going to `4 GHz`
[image]
  The new frequency is confirmed in both windows at `1C`
[image]
5. Stress a single Core
  Press `O` for Tools and go to `Turbo Select CPU...`
[image]
  Enter one of your Core. Here CPU number 0
[image]
  Press `ESC` to exit any window and observe the resulted frequency
[image]
  Observe that ruler has been updated with the new frequency ratio
  Press `F10` to stop the Stress

* Turbo Boost Prerequisites:
1. Turbo bit unlocked (or overrided is supported)
1. SpeedStep ON
1. C-States ON
1. IDA capable
1. Turbo ON
1. P-State ratio well defined
1. Clock Modulation ODCM is disabled

Gamebag, 2020-10-02:
I followed your steps but when I started the program I noticed turbo was still enabled. I then followed the rest of your steps ignoring that and I was still met with "turbo must be disabled". However, I am now getting the turbo boost to 3.9GHz I am supposed to on single core.

Gamebag, 2020-10-02:
I tried again and while I was able to get it to start without turbo and then being able to enable turbo, changing the frequencies still results in the turbo must be disabled message

Gamebag, 2020-10-02:
Turning up the TGT to 40 makes the system respect the default turbo settings

cyring, 2020-10-02:
* I believe I have a remaining issue synchronizing the Turbo state in the UI from changes happening in Driver.
At any time, you can press key `*` to request a full refresh

* I had programmed those _atomic_ algorithms in Tools to provide enough stress to trigger Turbo. It may explain why it now reaches 4 GHz
* For multi Cores or high Temperature, I recommend one of my Conics stress loops
* If not done yet, disable SPREAD SPECTRUM in your BIOS to get a better `100 MHz` Base Clock in _CoreFreq_
* Tomorrow, you should get a CoreFreq update which cares about your Xeon

Gamebag, 2020-10-02:
My bad I meant that it now respects the 39/37/37/36/36/36 default Xeon settings. I am not getting requested 40x turbo frequencies. SPREAD SPECTRUM is not a thing in my BIOS, as I am using an HP Z420 Workstation. I wonder whats causing it to ignore my settings in Linux but do just fine in Windows.

cyring, 2020-10-04:
Hello,

Can you switch to the [`develop` branch](https://github.com/cyring/CoreFreq/tree/develop)

You will get some specific codes to unlock your Processor; thus you don't need any driver argument.

As a reminder, don't forget to fully rebuild and reload all (Driver, Daemon, Cli) when new code is available.
```
make clean all
insmod corefreqk.ko
corefreqd
corefreq-cli
```

Gamebag, 2020-10-04:
I have switched to develop branch, but I am still getting the same "Turbo Boost must be disabled" message. However, uncore is now unlocked

cyring, 2020-10-04:
Can you please pull the latest `develop` and try again ?

Gamebag, 2020-10-04:
You've fixed the Turbo must be enabled message! However the CPU is still only boosting to 3.6GHz.

cyring, 2020-10-04:
> You've fixed the Turbo must be enabled message! However the CPU is still only boosting to 3.6GHz.

So Processor is able to reach its `3.9 GHz` nominal Turbo boost frequency but can not go above ?

Gamebag, 2020-10-04:
I don't know what changed but single core only boosts to 3.6 again unfortunately

cyring, 2020-10-04:
Start the single Core stress tools on CPU 0 and let it run. 
It the mean time, toogle the features below, one after one:

TGT = Highest ratio
1C =  Your requested ratio
2C, 3C, ... 7C, 8C = Default boot ratios
BIAS Hint = 0 (no power savings) 
ODCM = Disable 
IOMWAIT = Disable
***
2nd use case (still stressing) 
Turbo OFF
EIST OFF
EIST ON
Turbo ON

Gamebag, 2020-10-04:
IOMWAIT does not disable on my machine in this software. I followed your 2nd use case and the frequency went back up from base right to 3.6 again

cyring, 2020-10-05:
Let's try the Semaphore bit.
New code available in `develop` for your tests

Gamebag, 2020-10-05:
Still boosting to 3.6 max

cyring, 2020-10-05:
It's the only relevant [topic](https://community.intel.com/t5/Software-Tuning-Performance/Change-in-Turbo-policy-for-Xeon-E5-v3/td-p/1060747) I have found on the subject and the SDM specs say:
> Semaphore for Turbo Ratio Limit Configuration
If 1, the processor uses override configuration1 specified in MSR_TURBO_RATIO_LIMIT and
MSR_TURBO_RATIO_LIMIT1.
If 0, the processor uses factory-set configuration (Default).

Gamebag, 2020-10-05:
At this point this feels like a linux thing. You guys have tried everything and it just refuses to boost above 3.6

Gamebag, 2020-10-05:
I still have all the CPU drivers blacklisted for what its worth

cyring, 2020-10-05:
My bad. Please pull `develop` and try again.

Gamebag, 2020-10-05:
IT WORKS! Thank you so much! You're a genius!

cyring, 2020-10-05:
Thank you for providing me all these tests. Main things I have learned from EP processors:
1. purpose of the Semaphore bit
1. EP with less than 8 physical Cores requires conditional Turbo limits MSR writing 
1. Some EP Xeons are unofficially Turbo unlocked

Later I'll will code back to you with code refactoring and clean up. 

Meanwhile feel free to try any other _CoreFreq_ features, including screenshots.
 ## State of:
* Memory controller
* All expected Package C-States are showing up
* Clock Modulation with percent of your choice is limiting frequency
* Accuracy of Temps, Voltages, Energy counters
* Temperature of the whole Processor (in footer) vs value per Core
* Effectiveness of the CPU-Idle to ask Cores for deep sleeping
* Limiting the C-States level in the Kernel window
* Any value compared with BIOS or OS, such as Microcode version, etc
* BIAS hint changed with the Frequency/Sensors effects

## Enhancements:
* IOMWAIT, I probably have to white-list your architecture
* Find/Add the Voltage register of the whole Processor vs Voltage Core

Qualifying and tuning _CoreFreq_ for a x86 architecture may takes us several days. I hope you're OK to pursue this work.

Gamebag, 2020-10-05:
I can confirm some of them now. All package C-States are showing up, Clock modulation works, Temp monitoring works very well, CPU-Idle also works well on this, All the BIOS / OS values that I know the value of are right, Voltage monitoring is correct. I haven't tried limiting C-states yet.

I have starred the program :+1: It's very good!

cyring, 2020-10-05:
Thanks a lot for all your tests.
Do you mind to show me a screenshot of the Memory Controller ?
(_those registers are indeed tuff to find_)

Gamebag, 2020-10-05:
Sure! 
[image]

cyring, 2020-10-05:
Thanks

I have to work on:
* DRAM speed
* Missing timings on last 2 channels

So, you have been able to overclock Turbo to `4.2` GHz ;-)

Gamebag, 2020-10-05:
Yes! Turbo overclocking works perfectly now! Once voltage is implemented it'll be unstoppable

cyring, 2020-10-05:
> Yes! Turbo overclocking works perfectly now! Once voltage is implemented it'll be unstoppable

I've been looking in datasheet a way to alter the `VID` but unfortunately Voltages appears to be read-only. Same with the Nuveton SuperIO controller.

Gamebag, 2020-10-05:
In XTU there's a slider for additional voltage in Turbo mode. I wonder if this is undocumented as well?

cyring, 2020-10-06:
> In XTU there's a slider for additional voltage in Turbo mode. I wonder if this is undocumented as well?

Read `IA32_PERF_CTL` msr register at `0x199` each time the voltage is slide up/down.
Reproduce the same write _by hand_ to verify if change is acknowledged.

Most upper bits of `IA32_PERF_CTL` are left reserved in https://software.intel.com/content/www/us/en/develop/download/intel-64-and-ia-32-architectures-sdm-combined-volumes-1-2a-2b-2c-2d-3a-3b-3c-3d-and-4.html

Here's the state of my research so far:
https://github.com/cyring/CoreFreq/blob/5da83ae6c6691eea5ba22834e4fb0d7392cce18c/intelmsr.h#L422

cyring, 2020-10-10:
Hello,

For the missing DRAM speed, can you post the output of:
```
lspci -nn
```

Also, how many DIMMs are there ? How the slots are populated (including the empty ones) ?

Thank you.

Gamebag, 2020-10-11:
Speed is still missing. There are 4 DIMMS, in positions 1,3,5,7. Unpopulated are 2,4,6,8. my LSPCI is here: 
[lspci.txt](https://github.com/cyring/CoreFreq/files/5361264/lspci.txt)

cyring, 2021-02-17:
Hello 

Can you pull and test the latest `develop` branch which should fix the bus rate speed.

About DRAM speed, it may be an unspecified value and I will need your help to add a print into code at this line:
https://github.com/cyring/CoreFreq/blob/eef2d33ca0143effc7c3d348f33f97547618eae2/corefreqd.c#L3322
insert above that line:
```
printf("DMFC=%x\n", Proc->Uncore.Bus.SNB_EP_Cap1.DMFC);
```

Value will be printed after building, and starting Daemon.  

About frequency, if you can't get it from BIOS, try `dmidecode -t memory`

Thank you

(SmartPhone edition)

cyring, 2021-02-28:
Maybe one day, please help with above question.

closed 2021-02-28 by cyring

islampro400, 2023-04-30:
> 

activating turbo boost on z420 can you help me