cyring/CoreFreq #208: Bug in Intel Nuc Haswell i3-4010U Ultra Low TDP

olejon, 2020-10-30:
1. This was not like this before
2. On Ryzen 3/X570 this *does not happen*
3. **BUG:** The Temperature at the Always-Present Bottom "Tech" bar does not reflect the Core with the highest Temperature
4. On my Ryzen 3 *it does ALWAYS reflect the Core with the highest temperature*

**See Screenshot, the Core with the highest Temperature is at 43°C while the Arrow pointing to the Bottom Bar's Temperature shows 47°C, a whopping 4 degrees more. Not even the `Max` reading has been at 47°C**

[image]

?

cyring, 2020-10-30:
Yes, I bought some changes while coding for Zen :
In footer you get the Package temperature.

olejon, 2020-10-30:
I see. Ok.

closed 2020-10-30 by olejon

cyring, 2020-11-09:
Hello,
I'm not able to reproduce your issue. 
I don't have an HSW-ULT to test but the aggregation looks fine with other Intel architectures.

cyring, 2020-11-09:
Please verify the issue with the `corefreq-cli -c` output. Both cases: idle and high load.

_Remark: Ryzen sample bellow_
```
CPU Freq(MHz) Ratio  Turbo  C0(%)  C1(%)  C3(%)  C6(%)  C7(%)  Min TMP:TS  Max
000    0.90 ( 0.01)   0.03   0.04  99.96   0.00   0.00   0.00  28 / 36:682/ 63
...
031    6.45 ( 0.06)   0.18   0.29  99.71   0.00   0.00   0.00  29 / 37:688/ 61

    Averages:        Turbo  C0(%)  C1(%)  C3(%)  C6(%)  C7(%)    TjMax:    Pkg:
                      0.32   0.51  99.49   0.00   0.00   0.00      95 C    35 C
```
```
CPU Freq(MHz) Ratio  Turbo  C0(%)  C1(%)  C3(%)  C6(%)  C7(%)  Min TMP:TS  Max
000 4057.97 (40.58) 115.94 100.00   0.00   0.00   0.00   0.00  28 / 71:962/ 72
...
031 4057.47 (40.57) 115.93 100.00   0.00   0.00   0.00   0.00  29 / 70:952/ 70

    Averages:        Turbo  C0(%)  C1(%)  C3(%)  C6(%)  C7(%)    TjMax:    Pkg:
                    115.94 100.00   0.00   0.00   0.00   0.00      95 C    70 C
```