cyring/CoreFreq #209: [Semi-off topic question] MSR and Memory timings

leinardi, 2020-11-03:
Hi, I saw that CoreFreq is able to correctly read the memory timings of my AMD system (Ryzen 3800X).
Looking at the code I figure out that you are getting this data from the Model-specific register (MSR) and I saw that some kind of mapping is available here: https://github.com/cyring/CoreFreq/blob/master/amdmsr.h
I'm not really familiar with C syntax (I'm a Java/Python dev) so I'm having hard time to understand how this mapping works.

I was wondering if you could give me a little help to figure out how can I get the value for the tCL form the MSR.
I think this value is somehow related to the struct `AMD_0F_DRAM_TIMING_LOW` but I cannot understand what is the address of the registry is providing this data and how to interpret the pair of integer associated with each field of the struct (`tCL		:  3-0`): is it an index/length pair? as in index = 3 and length = 0?

I tried to google on how to access this kind of info but I was not able to find anything and, as far as I know, CoreFreq is the only Linux software that actually shows this kind of information.

cyring, 2020-11-04:
I'm not aware of any MSR which provides the UMC timings.
Those are in fact queried from the SMU. 
You will read the SMU using a index/data protocol through the PCI space.
* The raw way is:
https://github.com/cyring/CoreFreq/blob/933df839bf494061a9488bb600a2fff18bdf6731/corefreqk.h#L555
* The compatible way is through a kernel exported function `amd_smn_read` which might be more appropriated in Java code. 

Next, you need the UMC like in this function:
https://github.com/cyring/CoreFreq/blob/933df839bf494061a9488bb600a2fff18bdf6731/corefreqk.c#L4271

leinardi, 2020-11-04:
Oh wow, it seems to be way more complex than I anticipated. I would like to use Python as language and I tried to google on how to access `amd_smn_read()` or `amd_nb` from Python but I found absolutely nothing...

I have also an Intel laptop (i7-9750H): could you tell me if on Intel side it is also used the PCI space?

One more thing: most probably you are already fully aware but, just in case you are not, I would like to point out this very cool Windows software that shows a lot of information on memory timings: [ZenTimings](https://github.com/irusanov/ZenTimings).

It's opensource and I thought that could be useful to you if you are missing some of the values that this software shows:
https://github.com/irusanov/ZenTimings/blob/318eae2acc456f4f40222d16d3efe3cd238046f6/Windows%20Forms/MainForm.cs#L501

It could be useful also to compare the implementation of Zen3 support, once it is out.

closed 2020-11-04 by leinardi

cyring, 2020-11-04:
To my knowledge all software, including _CoreFreq_ are using the same route to the Zen UMC.
Be aware that the SMU won't deal with parallel and simultaneous queries. So you will have to serialize the access through it. 
Intel IMC: you will address the timings registers by their offset added to a base address which has to be remap in your process memory space.
Indeed I won't program such accesses with Python. Things are much easier in a driver that you Python code will forward its queries to.