cyring/CoreFreq #210: [Question] Would you be interested in a GTK client cooperation?

leinardi, 2020-11-05:
Hi, I'm the developer of [GreenWithEnvy](https://gitlab.com/leinardi/gwe) and [other small GTK apps](https://gitlab.com/leinardi) and I was wondering if you would be interested in a cooperation to build a GTK client for CoreFreq.

Initially I was looking at the code of CoreFreq to figure out how to add the memory timings to [GST](https://gitlab.com/leinardi/gst) but, after our chat on #209, I figured out that doing this in GTK + Python would be pretty messy. For this reason I would like to offer my GTK knowledge to build a new GTK client for CoreFreq in C.

BTW, I saw that you already started working on a [CoreFreq GUI](https://github.com/cyring/CoreFreq.gui) on a separate repository, but I was hoping to start a new GTK client.

If you are interested, my suggestion would be to talk about it on Skype or similar voice chat software.

cyring, 2020-11-06:
Hi Roberto,

Looking at your GUIs, I'm sure they will be a great addition to the _CoreFreq_ project.

I also believe we have to exchange about the C & ASM instructions and other bits structures the Client is making use of. Mainly for latency purpose. I can't tell how Python can manage that.

Depending of your time zone; mine is Paris/France; I'm available for a talk starting this weekend:
Please find my email by running `corefreq-cli -h`.

Regards,
Cyril

leinardi, 2020-11-06:
Email sent, thanks! :+1:

closed 2020-11-06 by leinardi