cyring/CoreFreq #211: Consider splitting the source code

inglor, 2020-11-11:
Hi @cyring 

Great project and interesting UI. Personally I found the UI a bit hard to navigate but I blame myself for not reading the manual :)

Looking on the source code as far as I understand there are 3 components on this project
- kernel module
- server
- cli or UI

It would make sense to split those into separate folders and adjust the Makefile to be able to compile individually each. Then one can install for example on kernel, server and cli (as {s}he wouldn't be interested in UI). It might also attract more devs instead of a spread out source code since it would be easier to find something and enhance it.

Keep up the good work and I hope this is extended to support Zen 3 :+1:

cyring, 2020-11-11:
Hello, 

Yes, definitely I'm thinking about splitting the source code tree in multiple programs, and especially to provide a breakdown per architecture and their models ; like the Linux kernel source tree is.
I'm aware my code is too big to _invite_ contributors to come in.
I'm looking forward a quiet period to do it and not to introduce regressions among the previous tested hardware I don't have any more under my hands.

Indeed the UI has been made with "data centers" in mind: the bare minimum with no X-Window, just a plain  enhanced VT. But stay on the line, I have made a Front-End friend ...

Zen3: Anyone who owns this processor is welcome to test my alpha version, but no one has shown up so far.

inglor, 2020-11-11:
That's good to hear and looking forward to the split up.

As for Zen 3 I think very limited people have got their hands on them. I was thinking of grabbing one but I think I will wait a see how it rolls out - I'm going to stick to my 3950X for now.

cyring, 2020-11-12:
The way recipes is provided, can't you make those individual actions ?
* Getting help for options which can be added to the make command line
```
make help
```
* Cleaning and building individually
1. Daemon
```
make clean corefreqd
```
2. Client
```
make clean corefreq-cli
```
* Installing individually 
1. Daemon
```
make install corefreqd
```
2. Client
```
make install corefreq-cli
```

inglor, 2020-11-12:
Thanks for this - it's useful. Currently the package in Archlinux copy everything to `/usr/src` but not everything is required to build just the module. I'm looking at a split PKGBUILD which will create 3 packages
- corefreq-daemon
- corefreq-module
- corefreq-cli

I think I'm miss configuring the `dkpg.conf`

Later we can extend it to UI also ;)

What do you think?

My AUR username is artafinde and I can be a co-maintainer in the Archlinux package so you can focus more on development if you want.

[corefreq-1.80_rc.1-1.src.tar.gz](https://github.com/cyring/CoreFreq/files/5531843/corefreq-1.80_rc.1-1.src.tar.gz)

cyring, 2021-05-28:
@inglor Thank you for providing the AUR [packages](https://aur.archlinux.org/packages/?O=0&K=CoreFreq)
Issue is closed

closed 2021-05-28 by cyring