cyring/CoreFreq #213: 3950X doesn't boost until 4700

inglor, 2020-11-18:
I have a pretty similar system with yours @cyring.
```
CPU: 3950X
Motherboard: Asus Motherboard Crosshair Hero VIII WiFi (BIOS 2502)
RAM: F4-3200C14Q-32GFX (4x DIMM)
CPU Cooling: NZXT Kraken Z63
OS: Archlinux
```
So my BIOS settings are just defaults with XMP profile for 3200Mhz CL 14 and with these settings the max I can get on the `corefreq` is ~4500Mhz.

From your video you posted in [here] I see you can easily boost to 4700MHz and wondering how you got this? Is there a kernel parameter I'm missing?

[here]: https://github.com/cyring/CoreFreq/issues/181#issuecomment-729740919

cyring, 2020-11-19:
My BIOS version is the `2206`
Like you, most settings are `AUTO` beside DIMM set to its advertised speed of `3600` MHz and the primary timings `16-16-16-16-36` copied from the `SPD`
`DOCP` or Manual, both giving `4700` on the best Core which is CPU `#0`

Having chosen `AUTO` means that I allow CPB, XFR, and other OC parameters which the ASUS will enable on its own.

I built the setup in April, Crosshair was delivered with a BIOS v12XX; Boost was naturally operational with Linux and Windows. With the last one, I have run the AMD Ryzen Master for checking and understanding of this new architecture. This software had pop up a prompt to warn about OC and warranty. Not sure if it had fused something. No specs on this.

Do I have an issue in _CoreFreq_ to monitor and trigger the `CPB` ? Probably, I still don't have a clear understanding of the published only registers.
Reading the code, again & again, I don't see where could be the gap.

The first code for Ryzen has been implemented blindly with the help of 2700, 19xx, and EPYC owners, showing me their screenshots. 
Getting a '3950X' allowed me to code the final touch for Zen 2

Is a Kernel version involved in the CPB. I don't think so: data appears to be feed by the ACPI tables. Thus BIOS ! Make sure this is enabled in yours; that any table are put on the Kernel disposal.

In short if Windows can't make your processor boost (optionally with the AMD drivers kit) then BIOS is the center of investigations.

If Windows can, but not Linux, thus try booting another distribution: this check if some kernel and modules compilation flags were missing/different.

Despite everything :
If _CoreFreq_ can not report boost, we could then script a simple loop to read every second the performance counters, using the `rdmsr` command. Next compute the difference to estimate the relative frequency. I will supply you the script for this.

cyring, 2020-11-19:
Bellow is the bash script.
* you will need the MSR-Tools package
* load the driver
```
modprobe msr
```
* any CPU burn software like the 7-Zip benchmark
---
* as root in a terminal, start monitoring your best Core. Change the CPU number in the script.
```bash
CPU=0; REG=0xe8; while true; do CTR0=$(rdmsr --processor ${CPU} --decimal ${REG}); sleep 1; CTR1=$(rdmsr --processor ${CPU} --decimal ${REG}); (( FREQ=${CTR1} - ${CTR0} )); echo ${FREQ}; done
```
* in another terminal, as a user, burn the same CPU
```
taskset -c 0 7z b
```
----
* frequency output example
```
4694190478
4701950441
4696374211
4703442090
4702920638
4695896154
4700392651
4694149912
4703419400
4694399019
4702284225
4696449037
4703608365
4701421357
4695394645
4702811024
```

inglor, 2020-11-19:
Hi @cyring and thanks for the details on how-to test this.

- I downgraded to the same BIOS `2206`
- FCLK set to `1600` MHz
- DRAM settings are the XMP profile set `Manual` to `3200` MHz with Timings as per screenshot below
[image]

I've tested these RAM settings extensively so I doubt this prevents the CPU from Boosting all the way to 4.7GHz.

My results are as follow:
```
# CPU=0; REG=0xe8; while true; do CTR0=$(rdmsr --processor ${CPU} --decimal ${REG}); sleep 1; CTR1=$(rdmsr --processor ${CPU} --decimal ${REG}); (( FREQ=${CTR1} - ${CTR0} )); echo ${FREQ}; done
5760659
22360406
7620159
16644243
37317380
10618680
426375926
11214555
37411865
211541182
1016076336
531570479
651363557
69548075
44703720
27162100
561803707
475676343
887549996
12843659
7128716
2415858
2835474
2221526454
4626183195       # <-- Here is where I started the 7z benchmark.
4623634549
4624347706
4626349744
4630066483
4624044382
4632650725
4633719214
4633854642
4617750205
4615342788
4616498355
4620820788
4624491697
4628065394
4630986091
4631744797
4633497083
4633614086
4623421549
4612259524
4615896560
4618717434
4624136202
4629802348
4632266049
4630887624
4634141835
4634618023
4625548870
4620663907
4637934062
4628896146
4637468234
4624454790
4633225568
4639584828
4633149024
```

So my CPU is advertized for `4700` Mhz and only gets to `4630`. Considering the `FMAX` of 3950X is `4725` Mhz (taken from dmi-decode) I'm missing ~ 100Mhz :(

I wonder how Windows can boost to `4700` Mhz..

cyring, 2020-11-19:
> So my CPU is advertized for `4700` Mhz and only gets to `4630`. Considering the `FMAX` of 3950X is `4725` Mhz (taken from dmi-decode) I'm missing ~ 100Mhz :(
> 
> I wonder how Windows can boost to `4700` Mhz..

* I don't have oftenly `4700` MHz and briefly above.
[image]

* Open the `Processor > P-State:TGT` window and stress your best Core

  On CPU `#0` the Linux acpi-cpufreq module is governing or listening to the boosted P-States of `47` which is then multiplied by the Base Clock. 
The BCLK should be ideally equaled to `100` MHz; but if your BIOS is protected against the EMI, the BCLK can be slightly lower.

Be sure that any SPREAD SPECTRUM options are disabled in your BIOS.

inglor, 2020-11-19:
>     * Open the `Processor > P-State:TGT` window and stress your best Core
>       On CPU `#0` the Linux acpi-cpufreq module is governing or listening to the boosted P-States of `47` which is then multiplied by the Base Clock.
>       The BCLK should be ideally equaled to `100` MHz; but if your BIOS is protecting against the EMI, the BCLK can be slightly lower.
> 
> 
> Be sure that any SPREAD SPECTRUM options are disabled in your BIOS.

I adjusted my kernel params to
```
$ cat /proc/cmdline
BOOT_IMAGE=/boot/vmlinuz-linux root=UUID=c7d325eb-b760-466f-85ef-854f6a1d77a5 rw nvidia-drm.modeset=1 acpi_enforce_resources=lax add_efi_memmap nmi_watchdog=0 selinux=0 modprobe.blacklist=nouveau,k10temp cpu0_hotplug audit=0 nowatchdog sysrq_always_enabled
```
I tried your suggestion to manually set the P-State multiplier to 47 and failed with an error message: 
[image]

[image]
My BCLK is ~`99.78` - it is jumping around so I assume it's the SPREAD SPECTRUM.

I will update once I disable the SPREAD SPECTRUM stuff in BIOS.

cyring, 2020-11-19:
> I adjusted my kernel params to
> 
> ```
> $ cat /proc/cmdline
> BOOT_IMAGE=/boot/vmlinuz-linux root=UUID=c7d325eb-b760-466f-85ef-854f6a1d77a5 rw nvidia-drm.modeset=1 acpi_enforce_resources=lax add_efi_memmap nmi_watchdog=0 selinux=0 modprobe.blacklist=nouveau,k10temp cpu0_hotplug audit=0 nowatchdog sysrq_always_enabled
> ```
> 
> > Open the `Processor > P-State:TGT` window and stress your best Core
> > On CPU `#0` the Linux acpi-cpufreq module is governing or listening to the boosted P-States of `47` which is then multiplied by the Base Clock.
> 
> I tried your suggestion to manually set the P-State multiplier to 47 and failed with an error message:
> [image]

Contrary to Intel Proc, I noticed with the AMD Zen you can only target the valid P-states up to `P0`. That's why I put them in blue in the list. 
The boosted P-States are barely undocumented. I don't have the registers and explanation how to target them, beside that if CPU is in `P0` then start the boosted states. 
However I'm offering a way to alter the `Max Ratio`. You have to activate the `Experimental Mode` in `Settings`. Prerequisites are that _CoreFreq_ is the current Linux clocksource and CPB disabled. 
I have been able to OC to `4700` MHz

> > The BCLK should be ideally equaled to `100` MHz; but if your BIOS is protecting against the EMI, the BCLK can be slightly lower.
> 
> My BCLK is ~`99.78` - it is jumping around so I assume it's the SPREAD SPECTRUM.
> [image]
> 
> > Be sure that any SPREAD SPECTRUM options are disabled in your BIOS.
> 
> @cyring
> Which of the two SPREAD SPECTRUM settings in BIOS ?
> 
> * VRM Spread Spectrum
> * SB Clock Spread Spectrum
> 
> [image]
> [image]

I got to the conclusion that disabling SB Clock Spread Spectrum is sufficient to get a nice 100 MHz BCLK.
In BIOS you can verify the same in the CPU frequency (third tab) when frequency reports 3500-3502 MHz

inglor, 2020-11-19:
> Contrary to Intel Proc, I noticed with the AMD Zen you can only target the valid P-states up to `P0`. That's why I put them in blue in the list.
> The boosted P-States are barely undocumented. I don't [...]

Yeah, I don't understand what you are suggestion and the CoreFreq UI is not friendly enough for me to understand it. The message I got is cryptic `Numerical argument out of domain` and your suggestion above regarding the experimental feature is not working because CoreFreq says `Turbo Boost must be disabled`. Is that what you mean with `CPB` disabled?

How do you make `CoreFreq` the current Linux clocksource?

How do you disable `CPB`?

Thank you

cyring, 2020-11-19:
OK tomorrow I should find some time to record an OC demo with _CoreFreq_

UI is made for Terminals.

cyring, 2020-11-20:
1. Kernel boot parameters
```
modprobe.blacklist=acpi_cpufreq idle=halt tsc=unstable
```
1. Build with TSC implementation
```
make DELAY_TSC=1 clean all
```
1. Start the driver in the experimental mode with all sub-drivers registration
```
insmod corefreqk.ko Register_ClockSource=1 Register_Governor=1 Register_CPU_Freq=1 Register_CPU_Idle=1 Experimental=1
```
1. Switch and check the current kernel clock source
```
echo "corefreq" > /sys/devices/system/clocksource/clocksource0/current_clocksource
cat /sys/devices/system/clocksource/clocksource0/current_clocksource
corefreq
```
1. Start Daemon; next, the Client
* Disable CPB
[image]
* Change the Max Ratio
[image]
[image]

cyring, 2020-11-20:
Hi,

I'm back in BIOS today

* Both VRM options are disabled
* Trying CPB2 in AMD Overclocking section.
  Rather than `AUTO`, I'm setting those parameters:
  - Limits = Motherboard
  - Override = +75 MHz
  - Scalar = x10

[image]

An override of 75 MHz is giving a better frequency bump than [100 to 200] MHz !

cyring, 2020-11-20:
* I don't have the same relative ratio among different boots
[image]
* Either a CoreFreq issue, either a Hardware issue, either Kernel ...

inglor, 2020-11-20:
I followed the steps to enable CPB2 in ASEGA as above and it helps.

I did not add any of the kernel params of _CoreFreq_ compiler params.

This helps quite a bit - first time I see 4.7GHz - ever.

_CoreFreq_ even reports 4725.

cyring, 2020-11-21:
> I followed the steps to enable CPB2 in ASEGA as above and it helps.
> 
> I did not add any of the kernel params of _CoreFreq_ compiler params.
> 
> This helps quite a bit - first time I see 4.7GHz - ever.
> 
> _CoreFreq_ even reports 4725.

Can we say it's a good news ! 

I can however remember that my 3950X was reaching easily 4.7 GHz, single best Core, with the out of the box BIOS  
Now, version 2206, requires me to tweak those PB options for the same.

Feel free to close the issue.

inglor, 2020-11-21:
Thanks for your assistance - I'm quite disappointed with Asus and their BIOS updates - I'll be looking for a different manufacturer for my next motherboard.

Thanks

closed 2020-11-21 by inglor

cyring, 2020-11-21:
They just need some help (D

Meanwhile I may have found the source of my maximum frequencies inconsistency: COLD BOOT

I'm setting all BIOS settings and then save/reboot.
Next, shutdown, and **Power Off** the PSU

Power On PSU ---> Boot --> Best frequency !

inglor, 2020-11-21:
> They just need some help (D
> 
> Meanwhile I may have found the source of my maximum frequencies inconsistency: COLD BOOT
> 
> I'm setting all BIOS settings and then save/reboot.
> Next, shutdown, and **Power Off** the PSU
> 
> Power On PSU ---> Boot --> Best frequency !

Interesting, that only means the BIOS is more messed up than I thought.. non deterministic results between reboots and / or cold/ warm reboots sounds really bad. 

Maybe reach out to others in overclock.net to check if others see similar behaviour?