cyring/CoreFreq #214: Cannot run perf when corefreq is loaded

marco44, 2020-12-01:
perf tries to read /proc/kallsyms. And this occurs when the corefreq module is loaded:
```
[root@marco ~]# cat /proc/kallsyms > /dev/null
[root@marco ~]# modprobe corefreqk 
[root@marco ~]# cat /proc/kallsyms > /dev/null
Killed
```
an extract from dmesg:
```
[60695.322004] BUG: unable to handle page fault for address: fffffffff5dbb850
[60695.322007] #PF: supervisor read access in kernel mode
[60695.322008] #PF: error_code(0x0000) - not-present page
[60695.322008] PGD 575613067 P4D 575613067 PUD 575615067 PMD 0 
[60695.322011] Oops: 0000 [#21] PREEMPT SMP NOPTI
```
Do I need to provide something else ?

cyring, 2020-12-01:
_CoreFreq_ was not designed with perf compatibility. It is explained in project Wiki

However with Intel processors I have been able to make them run simultaneously, if no PMC counters programming conflicts happen. 
With Ryzen, `perf top` just crashes so far with the latest Arch built kernel.

What's your processor ?

For security purpose `corefreqk.ko` also implements read-only and read-write pages, it might not please perf.

marco44, 2020-12-02:
That's a Ryzen, 3900X

But that's what I've been looking into, perf top crashes, and it seems it crashed because of the previous message, from a strace. perf tries to read kallsyms, and gets killed because of the page fault

cyring, 2020-12-03:
Arch Linux: after a full system rolling upgrade with a `5.9.11` kernel built and no command line boot options.
_CoreFreq_ is not running, not even started
```
# cat /proc/kallsyms > /dev/null

# dmesg -t | tail 
Dazed and confused, but trying to continue
Uhhuh. NMI received for unknown reason 2c on CPU 26.
Do you have a strange power saving mode enabled?
Dazed and confused, but trying to continue
Uhhuh. NMI received for unknown reason 3c on CPU 26.
Do you have a strange power saving mode enabled?
Dazed and confused, but trying to continue
Uhhuh. NMI received for unknown reason 3c on CPU 25.
Do you have a strange power saving mode enabled?
Dazed and confused, but trying to continue

# perf top
Samples: 4K of event 'cycles', 4000 Hz, Event count (approx.): 1273347411 lost: 
Overhead  Shared Object            Symbol                                       
   2.79%  liblzma.so.5.2.5         [.] lzma_crc64
   1.06%  [kernel]                 [k] __x86_indirect_thunk_rax
   0.91%  perf                     [.] __symbols__insert
   0.59%  [kernel]                 [k] clear_page_rep
   0.54%  [kernel]                 [k] psi_group_change
   0.49%  libc-2.32.so             [.] __memmove_avx_unaligned_erms
   0.45%  perf                     [.] dso__load_sym
   0.45%  [kernel]                 [k] copy_user_generic_string
   0.42%  [vdso]                   [.] 0x0000000000000675
   0.42%  [kernel]                 [k] check_preemption_disabled
...
```

EDIT: now rebooting for _CoreFreq_...

cyring, 2020-12-03:
This could happen in those two `strlcpy`
https://elixir.bootlin.com/linux/latest/source/kernel/module.c#L4237
Perhaps a `KSYM_NAME_LEN` of `128` bytes limitation
I'm trying to sandbox a simple module with the same number of corefreqk.c parameters but can't reproduce the `kallsyms` issue...

marco44, 2020-12-03:
Can I provide anything ?

cyring, 2020-12-03:
> Can I provide anything ?

It looks like _CoreFreq_ driver has some much symbols that `kallsyms` is running out of resources.
That's why I would say it's not a `perf` issue but a consequence because `perf` is loading _all_ symbols.

I have searched into the kernel documentation for such limitations. Nothing found. Although I'm aware that with time `corefreqk.c` became so huge in lines of code for a driver that it may induce too many symbols.

To track the issue I don't see another immediate solution than tracing the kernel on these parts:
```
			strlcpy(name, kallsyms_symbol_name(kallsyms, symnum), KSYM_NAME_LEN);
			strlcpy(module_name, mod->name, MODULE_NAME_LEN);
```
* `printk` the return code and the variables involved into `strlcpy` when error happens
* build and boot this kernel
* tracks if some _symbol_ variables are null or out of buffer; which could give a hint on limitation.

Could you help with this or another debug way ?

marco44, 2020-12-04:
Yeah sure. Can you give me a patch or just an example printk so I do exactly what you want ?

cyring, 2020-12-04:
[Arch Build System](https://wiki.archlinux.org/index.php/Arch_Build_System)
```
# pacman -Sy
$ asp update && git pull
$ makepkg -osr
```
```c
int module_get_kallsym(unsigned int symnum, unsigned long *value, char *type,
                        char *name, char *module_name, int *exported)
{
        struct module *mod;

        preempt_disable();
        list_for_each_entry_rcu(mod, &modules, list) {
                struct mod_kallsyms *kallsyms;

                if (mod->state == MODULE_STATE_UNFORMED)
                        continue;
                kallsyms = rcu_dereference_sched(mod->kallsyms);
                if (symnum < kallsyms->num_symtab) {
                        const Elf_Sym *sym = &kallsyms->symtab[symnum];
                        size_t len = 0;

                        *value = kallsyms_symbol_value(sym);
                        *type = kallsyms->typetab[symnum];
                        len = strlcpy(name, kallsyms_symbol_name(kallsyms, symnum), KSYM_NAME_LEN);
                        if (len == 0) {
                                pr_warn("strlcpy(name[%s])>%t\n",name,len);
                        }
                        len = strlcpy(module_name, mod->name, MODULE_NAME_LEN);
                        if (len == 0) {
                                pr_warn("strlcpy(module_name, mod->name)>%t\n",
                                        module_name, mod->name, len);
                        }
                        *exported = is_exported(name, *value, mod);
                        preempt_enable();
                        return 0;
                }
                symnum -= kallsyms->num_symtab;
        }
        preempt_enable();
        return -ERANGE;
}
```
```
$ makepkg -eisr
```

cyring, 2020-12-04:
```
# cat /proc/kallsyms
Killed
```
```
strlcpy(name[])>0
BUG: unable to handle page fault for address: 00000000b0160940
#PF: supervisor read access in kernel mode
#PF: error_code(0x0000) - not-present page
```

cyring, 2020-12-04:
OK, now changing to print **all** names going through this function ...
```c
int module_get_kallsym(unsigned int symnum, unsigned long *value, char *type,
                        char *name, char *module_name, int *exported)
{
        struct module *mod;

        preempt_disable();
        list_for_each_entry_rcu(mod, &modules, list) {
                struct mod_kallsyms *kallsyms;

                if (mod->state == MODULE_STATE_UNFORMED)
                        continue;
                kallsyms = rcu_dereference_sched(mod->kallsyms);
                if (symnum < kallsyms->num_symtab) {
                        const Elf_Sym *sym = &kallsyms->symtab[symnum];
                        size_t len = 0;

                        *value = kallsyms_symbol_value(sym);
                        *type = kallsyms->typetab[symnum];
                        len = strlcpy(name, kallsyms_symbol_name(kallsyms, symn>

                        printk("strlcpy(name[%s])>%ld\n",name,len);

                        len = strlcpy(module_name, mod->name, MODULE_NAME_LEN);
                        if (len == 0) {
                                printk("strlcpy(module_name[%s], mod->name[%s])>
                                        module_name, mod->name, len);
                        }
                        *exported = is_exported(name, *value, mod);
                        preempt_enable();
                        return 0;
                }
                symnum -= kallsyms->num_symtab;
        }
        preempt_enable();
        return -ERANGE;
}
```

cyring, 2020-12-04:
* All traces
```
strlcpy(name[])>0
strlcpy(name[_note_7])>7
strlcpy(name[KPublic])>7
strlcpy(name[CSWTCH.4036])>11
strlcpy(name[CSWTCH.4037])>11
strlcpy(name[CSWTCH.4042])>11
strlcpy(name[CSWTCH.4043])>11
strlcpy(name[CSWTCH.4048])>11
strlcpy(name[CSWTCH.4049])>11
strlcpy(name[CSWTCH.4054])>11
strlcpy(name[CSWTCH.4055])>11
strlcpy(name[Intel_Turbo_Cfg8C_PerCore])>25
strlcpy(name[Intel_Turbo_Cfg15C_PerCore])>26
strlcpy(name[Intel_Turbo_Cfg16C_PerCore])>26
strlcpy(name[Intel_Turbo_Cfg18C_PerCore])>26
strlcpy(name[Intel_Turbo_Cfg_SKL_X_PerCore])>29
strlcpy(name[SNB_EP_HB])>9
strlcpy(name[AMD_17h_ZenIF])>13
strlcpy(name[KPrivate])>8
strlcpy(name[Set_Core2_Target])>16
strlcpy(name[Set_Nehalem_Target])>18
strlcpy(name[Set_SandyBridge_Target])>22
strlcpy(name[Get_Core2_Target])>16
strlcpy(name[Get_Nehalem_Target])>18
strlcpy(name[Get_SandyBridge_Target])>22
strlcpy(name[Cmp_Core2_Target])>16
strlcpy(name[Cmp_Nehalem_Target])>18
strlcpy(name[Cmp_SandyBridge_Target])>22
strlcpy(name[Start_Uncore_Nehalem])>20
strlcpy(name[Stop_Uncore_Nehalem])>19
strlcpy(name[Start_Uncore_SandyBridge])>24
strlcpy(name[Stop_Uncore_SandyBridge])>23
strlcpy(name[Start_Uncore_SandyBridge_EP])>27
strlcpy(name[Stop_Uncore_SandyBridge_EP])>26
strlcpy(name[Start_Uncore_Haswell_ULT])>24
strlcpy(name[Stop_Uncore_Haswell_ULT])>23
strlcpy(name[Start_Uncore_Haswell_EP])>23
strlcpy(name[Stop_Uncore_Haswell_EP])>22
strlcpy(name[Start_Uncore_Skylake])>20
strlcpy(name[Stop_Uncore_Skylake])>19
strlcpy(name[Start_Uncore_Skylake_X])>22
strlcpy(name[CoreFreqK_Idle_State_Withdraw])>29
strlcpy(name[CoreFreqK])>9
strlcpy(name[CoreFreqK_Policy_Exit])>21
strlcpy(name[CoreFreqK_Policy_Init])>21
strlcpy(name[Register_Governor])>17
strlcpy(name[Policy_GetFreq])>14
strlcpy(name[CoreFreqK_DevNode])>17
strlcpy(name[CoreFreqK_Empty_Func_Level_Down])>31
strlcpy(name[CoreFreqK_ShutDown])>18
strlcpy(name[CoreFreqK_Alloc_Features_Level_Down])>35
strlcpy(name[CoreFreqK_Alloc_Device_Level_Down])>33
strlcpy(name[CoreFreqK_Make_Device_Level_Down])>32
strlcpy(name[CoreFreqK_Create_Device_Level_Down])>34
strlcpy(name[CoreFreqK_Alloc_Public_Level_Down])>33
strlcpy(name[CoreFreqK_Alloc_Public_Level_Down])>33
strlcpy(name[CoreFreqK_Alloc_Private_Level_Down])>34
strlcpy(name[CoreFreqK_Alloc_Processor_RO_Level_Down])>39
strlcpy(name[CoreFreqK_Alloc_Processor_RW_Level_Down])>39
strlcpy(name[CoreFreqK_Alloc_Public_Cache_Level_Down])>39
strlcpy(name[CoreFreqK_Alloc_Private_Cache_Level_Down])>40
strlcpy(name[CoreFreqK_Alloc_Per_CPU_Level_Down])>34
strlcpy(name[CoreFreqK_Ignition_Level_Down])>29
strlcpy(name[RunLevel])>8
strlcpy(name[ClockMod_HWP_PerCore])>20
strlcpy(name[CoreFreqK_NMI_Handler])>21
BUG: unable to handle page fault for address: 00000000a4da4940
#PF: supervisor read access in kernel mode
#PF: error_code(0x0000) - not-present page
```

cyring, 2020-12-04:
I don't find a solution.
I've created another kernel module with many functions, static or not, parameters, callbacks, and can't reproduce this bug in a sandbox. 
_CoreFreq_ as kvm or nvidia is a big driver but those last don't encounter such issue.
I still believe the error comes from my coding. I might do something something which is not compliant with the kernel rules but i don't have any clue. 
If kallsyms has bug it would appear in the past and we could find some facts about it. But nothing found about this kind of issue. 
Any help is welcomed.

cyring, 2020-12-06:
A fix is available in the `develop` branch.  
Can you please give a try ?

marco44, 2020-12-07:
Hi,

Problem fixed for me

Thanks a lot

cyring, 2020-12-07:
> Hi,
> 
> Problem fixed for me
> 
> Thanks a lot

You are welcome.

cyring, 2022-01-22:
Only with AMD/Zen, running `perf top` is getting killed!
With an Intel Tiger Lake/U `perf` starts nicely.

cyring, 2022-01-22:
> When `corefreqk.ko` is already running:
> 
> * Only with AMD/Zen, running `perf top` is getting killed!
> * With an Intel Tiger Lake/U `perf` starts nicely.

This is indeed linked with `cat /proc/kallsyms` but using the same disk, same build, plugged on Intel/TGL has no issue but plugged on AMD/Matisse has the issue.
I'm reviewing the whole driver code for any mismatch in function prototypes, especially _static_, but don't find any difference in coding ...

cyring, 2022-01-24:
`nm corefreqk.ko`
```
0000000000000000 t CoreFreqK_Read_CS_From_Invariant_TSC
```
Why ?

cyring, 2022-01-26:
What! `5.15.16-xanmod1-1`: no kallsyms issue

cyring, 2022-05-08:
Mainline Kernel does not want but latest [xanmod](https://aur.archlinux.org/packages/linux-xanmod) of 5.15.36 is doing fine with kallsyms plus _CoreFreq_

Unsolved. Issue is postponed.

closed 2022-05-08 by cyring

cyring, 2022-05-10:
> _Much Ado About Nothing_

# Solution

> **`no-inline`** flag compiler is the key answer

1. Build and install _CoreFreq_ as below
 ```sh
make CC='cc -fno-inline' clean all
```
```sh
ln -s /home/cyril/src/CoreFreq/corefreqk.ko /lib/modules/$(uname -r)/kernel/drivers/misc/

depmod -a
```
2. Start _CoreFreq_
3. Check for the kernel module symbols
```
grep corefreqk /proc/kallsyms
```
# Perf profiling use-case
```sh
insmod corefreqk.ko Register_ClockSource=1 Register_Governor=1 Register_CPU_Freq=1 Register_CPU_Idle=1 NMI_Disable=0

./corefreqd
```
```sh
corefreq-cli -t interrupts
```

[image]

[image]

cyring, 2022-05-26:
Root cause was address relocation.
See commit bdfe6bc9cf114497d4996d31c0af5aa5e00d9b56