cyring/CoreFreq #216: Wrong DTS shown on Ryzen 5800X

leinardi, 2021-01-16:
When running `corefreq-cli -c` I get some very strange DTS temperature with my Ryzen 5800X:

```
CPU Freq(MHz) Ratio  Turbo  C0(%)  C1(%)  C3(%)  C6(%)  C7(%)  Min TMP:TS  Max
000  273.76 ( 2.74)   7.20   8.10  91.90   0.00   0.00   0.00  30 / 30:636/ 73
001  173.15 ( 1.73)   4.56   4.48  95.52   0.00   0.00   0.00  30 / 30:636/ 73
002  161.34 ( 1.61)   4.25   5.31  94.69   0.00   0.00   0.00  30 / 30:636/ 73
003  124.02 ( 1.24)   3.26   3.28  96.72   0.00   0.00   0.00  30 / 30:636/ 73
004   50.89 ( 0.51)   1.34   1.36  98.64   0.00   0.00   0.00  30 / 30:636/ 73
005   69.69 ( 0.70)   1.83   2.24  97.76   0.00   0.00   0.00  30 / 30:636/ 73
006   95.22 ( 0.95)   2.51   3.05  96.95   0.00   0.00   0.00  30 / 30:636/ 73
007   40.60 ( 0.41)   1.07   1.33  98.67   0.00   0.00   0.00  30 / 30:636/ 73
008   29.00 ( 0.29)   0.76   0.95  99.05   0.00   0.00   0.00  30 / 30:636/ 73
009   41.00 ( 0.41)   1.08   1.05  98.95   0.00   0.00   0.00  30 / 30:636/ 73
010  142.42 ( 1.42)   3.75   4.74  95.26   0.00   0.00   0.00  30 / 30:636/ 73
011  149.43 ( 1.49)   3.93   3.93  96.07   0.00   0.00   0.00  30 / 30:636/ 73
012  118.45 ( 1.18)   3.12   3.35  96.65   0.00   0.00   0.00  30 / 30:636/ 73
013   41.19 ( 0.41)   1.08   1.28  98.72   0.00   0.00   0.00  30 / 30:636/ 73
014   28.69 ( 0.29)   0.76   0.91  99.09   0.00   0.00   0.00  30 / 30:636/ 73
015   70.94 ( 0.71)   1.87   2.37  97.63   0.00   0.00   0.00  30 / 30:636/ 73

    Averages:        Turbo  C0(%)  C1(%)  C3(%)  C6(%)  C7(%)    TjMax:    Pkg:
                      2.65   2.98  97.02   0.00   0.00   0.00      90 C    35 C
```

leinardi, 2021-01-16:
Never mind, looks like it's OK like this.

cyring, 2021-01-16:
> Never mind, looks like it's OK like this.

* Indeed it is _explained_ by this formula:
https://github.com/cyring/CoreFreq/blob/5890ec7d75c6d6dd12ebc0418ad889be03a61fc4/corefreq.h#L582

  So your CPU number `000` temperature  is solved as:
```
Temp = ( (1016 * 5 / 40) - 49 ) - 0 = 78°C
```
_Let's keep this thread opened for anybody wondering about the same question._

* In addition, I would appreciate other results from your Zen3 to check my source code ?

1. The output of `Power, Current & Thermal` window to verify `TjMax TDP Min Max PPT EDC TDC`
2. The output of the power consumed with all CPUs stressed by  `Conic Compute > Hyperboloid of two sheets` from the  `Tools` menu

  _You can put both answers in a single screenshot like mine bellow_
[image]

3. Do you mind also to show me the resulted CPU Topology with the `Voltage` view  

_Press shortcut keys [`V`] then [`m`] . You can move the window with [SHIFT] + [Arrows]_  
[image]

In advance, thank you.

leinardi, 2021-01-17:
Hey sure, no problem:

1.  [image]
2. [image]
3. [image]

Let me know if you need anything else.

cyring, 2021-03-27:
Hello,

In this [thread](https://www.overclock.net/threads/corefreq-call-for-ryzen-5900-testers.1776015/post-28763410), something odd encountered with a Vermeer Ryzen 5 5600X.
Total CPU count is one !

Whatever the SMT state is, don't you have such bug with the 5800X, do you ?

cyring, 2021-04-05:
Hi @leinardi 
Original `DTS` question seems covered. I believe we can close the issue.
Without a Vermeer processor, I can't tell if other _CoreFreq_ features are unavailable.
My assumption is that Vermeer may be the same registers as Matisse.

closed 2021-04-05 by cyring