cyring/CoreFreq #218: [Intel] DDR4 layout missing and many values missing

leinardi, 2021-01-17:
I'm trying CoreFreq on my Intel laptop and I noticed that the Memory Controller layout is not showing the same DDR4 layout that I see on my AMD desktop. On top of that, many fields are showing a 0 value:

[image]

I tested with both the current master and develop branch.

cyring, 2021-01-17:
Thanks for this report.

At the time, I was indeed missing the specifications of the registers adresses to query those DDR extended timings

Zen came also later with additionnal timings I found somehow the decoding of.

With your hardware contribution, we can re-open this Intel case.
Let's see if we can get more specs online ...

leinardi, 2021-01-17:
Sure! Let me know how can I help :+1:

cyring, 2021-01-17:
> Sure! Let me know how can I help 👍

Yes, it will be helpful to post BIOS screenshots of the Memory Controller data: main and extended timings, memclock speed or frequency, any IMC features, and so on ...

leinardi, 2021-01-17:
Unfortuately it's just a Dell laptop, so the bios doesn't show this kind of information. And I do not have a Windows installation to check those values there either...

cyring, 2021-01-18:
Is pushed in the `develop` branch the ACT register unveiled in the 10th generation specs.

You may now read these 3 additional timings:
* RRD
* WR
* FAW

_Without a hardware reference, it's hard to tell if bits decoding is OK. 
I suggest you to get the SPD from any DMI tools_

For your tests, thank you.

leinardi, 2021-01-19:
Hi @cyring, this is a screenshot I just took from the latest `develop` branch:

[image]

> I suggest you to get the SPD from any DMI tools

Sure! Could you please point me to some Linux tool that I can use to read the SPD values?

cyring, 2021-01-19:
> Hi @cyring, this is a screenshot I just took from the latest `develop` branch:
> 

Nice! We have some values showing up in `RRD` and `FAW` but none in `WR`
Encouraging, I will program other registers from that datasheet.

[image]

> > I suggest you to get the SPD from any DMI tools
> 
> Sure! Could you please point me to some Linux tool that I can use to read the SPD values?

Wrong direction, my DMI is giving me no timings.
```
sudo dmidecode -t memory
```
But i2c is doing the job
You may need modules: `i2c_dev` , `i2c_piix4` and bellow
```
sudo modprobe eeprom
sudo decode-dimms | less
```
----
```
SPD Revision                                     1.0
...
AA-RCD-RP-RAS (cycles)                           15-15-15-36
Supported CAS Latencies                          16T, 15T, 14T, 13T, 12T, 11T, 10T
AA-RCD-RP-RAS (cycles) as DDR4-1866              13-13-13-31
AA-RCD-RP-RAS (cycles) as DDR4-1600              11-11-11-27
...
Minimum Cycle Time (tCKmin)                      0.938 ns
Maximum Cycle Time (tCKmax)                      1.500 ns
Minimum CAS Latency Time (tAA)                   13.750 ns
Minimum RAS to CAS Delay (tRCD)                  13.750 ns
Minimum Row Precharge Delay (tRP)                13.750 ns
Minimum Active to Precharge Delay (tRAS)         33.000 ns
Minimum Active to Auto-Refresh Delay (tRC)       46.750 ns
Minimum Recovery Delay (tRFC1)                   350.000 ns
Minimum Recovery Delay (tRFC2)                   260.000 ns
Minimum Recovery Delay (tRFC4)                   160.000 ns
Minimum Four Activate Window Delay (tFAW)        21.000 ns
Minimum Row Active to Row Active Delay (tRRD_S)  3.700 ns
Minimum Row Active to Row Active Delay (tRRD_L)  5.300 ns
Minimum CAS to CAS Delay (tCCD_L)                5.625 ns
...
```
I let you do the maths -;)

leinardi, 2021-01-19:
So, this is the relevant output of `decode-dimms`:
```
Decoding EEPROM: /sys/bus/i2c/drivers/ee1004/10-0050
Guessing DIMM is in                              bank 1

---=== SPD EEPROM Information ===---
EEPROM CRC of bytes 0-125                        OK (0xD430)
# of bytes written to SDRAM EEPROM               384
Total number of bytes in EEPROM                  512
Fundamental Memory type                          DDR4 SDRAM
SPD Revision                                     1.1
Module Type                                      SO-DIMM
EEPROM CRC of bytes 128-253                      OK (0x2355)

---=== Memory Characteristics ===---
Maximum module speed                             2666 MHz (PC4-21300)
Size                                             16384 MB
Banks x Rows x Columns x Bits                    16 x 16 x 10 x 64
SDRAM Device Width                               8 bits
Ranks                                            2
Rank Mix                                         Symmetrical
AA-RCD-RP-RAS (cycles)                           19-19-19-43
Supported CAS Latencies                          20T, 19T, 18T, 17T, 16T, 15T, 14T, 13T, 12T, 11T, 10T

---=== Timings at Standard Speeds ===---
AA-RCD-RP-RAS (cycles) as DDR4-2666              19-19-19-43
AA-RCD-RP-RAS (cycles) as DDR4-2400              17-17-17-39
AA-RCD-RP-RAS (cycles) as DDR4-2133              15-15-15-35
AA-RCD-RP-RAS (cycles) as DDR4-1866              13-13-13-30
AA-RCD-RP-RAS (cycles) as DDR4-1600              11-11-11-26

---=== Timing Parameters ===---
Minimum Cycle Time (tCKmin)                      0.750 ns
Maximum Cycle Time (tCKmax)                      1.600 ns
Minimum CAS Latency Time (tAA)                   13.750 ns
Minimum RAS to CAS Delay (tRCD)                  13.750 ns
Minimum Row Precharge Delay (tRP)                13.750 ns
Minimum Active to Precharge Delay (tRAS)         32.000 ns
Minimum Active to Auto-Refresh Delay (tRC)       45.750 ns
Minimum Recovery Delay (tRFC1)                   350.000 ns
Minimum Recovery Delay (tRFC2)                   260.000 ns
Minimum Recovery Delay (tRFC4)                   160.000 ns
Minimum Four Activate Window Delay (tFAW)        21.000 ns
Minimum Row Active to Row Active Delay (tRRD_S)  3.000 ns
Minimum Row Active to Row Active Delay (tRRD_L)  4.900 ns
Minimum CAS to CAS Delay (tCCD_L)                5.000 ns
Minimum Write Recovery Time (tWR)                15.000 ns
Minimum Write to Read Time (tWTR_S)              2.500 ns
Minimum Write to Read Time (tWTR_L)              7.500 ns

---=== Other Information ===---
Package Type                                     Monolithic
Maximum Activate Count                           Unlimited
Post Package Repair                              One row per bank group
Soft PPR                                         Supported
Module Nominal Voltage                           1.2 V
Thermal Sensor                                   No

---=== Physical Characteristics ===---
Module Height                                    30 mm
Module Thickness                                 2 mm front, 2 mm back
Module Reference Card                            E revision 1

---=== Manufacturer Data ===---
Module Manufacturer                              SK Hynix (former Hyundai Electronics)
DRAM Manufacturer                                SK Hynix (former Hyundai Electronics)
Manufacturing Location Code                      0x02
Manufacturing Date                               2019-W31
Assembly Serial Number                           0x2F9E1768
Part Number                                      HMA82GS6JJR8N-VK    
```

I tried to do the math:
[image]

There seems to be some matches, like `RAS`, `RFC` and `FAW` and others are slightly off, like `RP`.
According to my calculation `WR` should be 20.

If you want to play with the spreadsheet, it's available here:
https://docs.google.com/spreadsheets/d/1Ej8oEUCPEtKFRchUWGko12H9AJ1PURY3pQdxpUtChrE/edit?usp=sharing

cyring, 2021-01-19:
That is just an awesome calculator!

cyring, 2021-01-19:
So we're chasing `tWR` which could be one of these fields starting at:

`SKL_IMC_CR_TC_PRE`

https://github.com/cyring/CoreFreq/blob/43658a362d59b2a21beb9a93d2a8daf4200581d6/intelmsr.h#L2699

Or in registers `SKL_IMC_CR_TC_ACT` ;  `SKL_IMC_CR_SC_CFG` ; `SKL_IMC_CR_TC_ODT` ; `SKL_IMC_REFRESH_TC`
**but** could also be stored among the undocumented bits `ReservedBits`. (this happens!)

In Daemon code, here's the line where the field is _transferred_

https://github.com/cyring/CoreFreq/blob/43658a362d59b2a21beb9a93d2a8daf4200581d6/corefreqd.c#L3585

Changing `ACT.tRCD_WR` then rebuild, reload CoreFreq to try the other field. Off course we won't bother with fields already in used in that `SKL_IMC()`

I'm aware it's a very fastidious way but I've no clue where is that `tWR`

leinardi, 2021-01-20:
Sorry but I'm not a C developer and I'm not sure how can I go from `SKL_IMC_CR_TC_PRE` to `Uncore.MC[mc].Channel[cha].SKL.ACT.tRCD_WR`. Could you please push on a branch all the changes so that I can simply checkout that branch, build and give you the output? Alternatively I'll be fine with a git patch.

cyring, 2021-01-20:
> Sorry but I'm not a C developer and I'm not sure how can I go from `SKL_IMC_CR_TC_PRE` to `Uncore.MC[mc].Channel[cha].SKL.ACT.tRCD_WR`. Could you please push on a branch all the changes so that I [...]

I'll provide something for debug purpose.

About `SKL-S/i7-6820HQ` ( _if tests are based on `develop` ?_ ), we clearly see that latest addition of registers does not provide:
* `RRD` (but in `CFL/i7-9750H`)
* `FAW` (but in `CFL/i7-9750H`)
* `WR` (still missing in both architectures)

## As a remainder, the ToDo list for SKL and Superiors:
1. advanced timings
* `B2B`
* `WR`
2. extended timings
* `tRDWR`
* `tWRRD`
* `tRCPB`
* `tRPPB`
* Different Bank Group
* Different Rank
* Different DIMM
* Different Logical Rank

leinardi, 2021-01-20:
Hey, sorry, I forgot to test on `develop`. I think we can see more or less the same timings also on this i7-6820HQ:

[image]

[image]

[image]

[image]

As you can see, `WR` is missing also for this laptop.

cyring, 2021-01-20:
* Nice progress, thank you.

# TODO
## Main timings
* `WR`
## Advanced timings
* `B2B`
## Extended timings
* `tRDWR`
* `tWRRD`
* `tRCPB`
* `tRPPB`
* Different Bank Group
* Different Rank
* Different DIMM
* Different Logical Rank

cyring, 2021-05-09:
Hello,

If you can try the latest `develop`  with Skylake (and others), I'm providing as much TDP as possible.

[image]

There is also the new driver parameter ` Custom_TDP_Limit` as an array of TDP tuples (PL1, PL2). 
`insmod corefreqk.ko Custom_TDP_Limit=50,100` which alters PL1 and PL2 of Package limits to 50 and 100 watt
Very experimental!

cyring, 2021-06-16:
> * Nice progress, thank you.
> 
> # TODO
> ## Main timings
> * `WR`
> 
> ## Advanced timings
> * `B2B`
> 
> ## Extended timings
> * `tRDWR`
> * `tWRRD`
> * `tRCPB`
> * `tRPPB`
> * Different Bank Group
> * Different Rank
> * Different DIMM
> * Different Logical Rank

Better progress shows that tWR should be the latest timing. I keep  this in Roadmap. B2B might be relevant with DDR 3 only.
 
https://github.com/cyring/CoreFreq/discussions/238#-875928

closed 2021-06-16 by cyring