cyring/CoreFreq #220: corefreqk.ko invalid module format

svmlegacy, 2021-02-07:
[image]

Seemed to have run fine in a previous kernel patch. No luck loading the module today.

cyring, 2021-02-07:
10pm in Paris until I got notified about your critical issue.
Let me switch on Computer and find a solution. Please stay online...

cyring, 2021-02-07:
OK, here running `5.10.13` with a Ryzen: no issue

`Invalid module format`: did you recently change for a signed kernel ?
Can you show me your kernel log about CoreFreq kernel module problem ?

svmlegacy, 2021-02-07:
I haven't explicitly changed to a signed kernel, as far as I know it's still unsigned.
Fedora 33 pulled in 5.10.12 last night for me.

dmesg is showing " corefreqk: version magic '5.10.11-200.fc33.x86_64 SMP mod_unload ' should be '5.10.12-200.fc33.x86_64 SMP mod_unload ' "

svmlegacy, 2021-02-07:
Remaking the package in a different folder has resolved the issue for me.

I did originally rerun the make command in the existing folder before reporting the issue, that did not resolve it.
Sorry for disturbing your night...

cyring, 2021-02-07:
> I haven't explicitly changed to a signed kernel, as far as I know it's still unsigned.
> Fedora 33 pulled in 5.10.12 last night for me.
> 
> dmesg is showing " corefreqk: version magic '5.10.11-200.fc33.x86_64 SMP mod_unload ' should be '5.10.12-200.fc33.x86_64 SMP mod_unload ' "

I believe either your kernel header files are not aligned with your last update and/or `corefreqk.ko` has to be fully rebuild.

Safe way is (if _CoreFreq_ is manually built):
```
make clean all
```

Or if _CoreFreq_ comes from a repo, try uninstall then install package again. This should trigger dkms to fully rebuild `corefreqk.ko`

closed 2021-02-07 by svmlegacy

cyring, 2021-02-07:
> Remaking the package in a different folder has resolved the issue for me.
> 
> I did originally rerun the make command in the existing folder before reporting the issue, that did not resolve it.
> Sorry for disturbing your night...

Don't worry. I appreciate to get any returns from _CoreFreq_ Users; especially b/c I can't imagine all use-cases.

cyring, 2021-02-08:
Thank you for providing the UMC.
`👍`

cyring, 2021-02-14:
Hello, thank you for your latest i3 addition into the wiki.

I'm also making some cleanups in the 1700X by deleting my comment. Feel free to remove yours at:
https://gist.github.com/svmlegacy/bc927a032b1555458a597596b8195b13#gistcomment-3623676

Regards
Cyril