cyring/CoreFreq #221: [Enhancement] Way to save/load config file

Jojonintendo, 2021-02-13:
I've been using CoreFreq on my system (Ryzen 1700 on MSI B450 Tomahawk MAX), and I really find it powerful and clean to use. The overclocking works fine in my case with experimental mode (a simple bump on P-State P0, which my MB doesn't allow modifying), however I couldn't find a way to export or save the current setup in order for it to be enabled and loaded on reboot. The settings that work for me have to be manually set on every boot.

I think it would be really neat, especially since the systemd service can be enabled, if it could also load a previously saved config. Or maybe a way to pass arguments through the command line, because if I'm not wrong, it currently only allows data to be read, while the CLI allows changes to be made.

Thank you for this amazing software, any guidance is welcome!

cyring, 2021-02-13:
Hello,

In the current release, those settings can be set with the driver during its start-up.

# Starting Experimental w/o TurboBoost
* Depending of the _CoreFreq_ installation, from source or from a package, 2 cases:
## Manual installation
```
insmod /your_path/CoreFreq/corefreqk.ko Experimental=1 TurboBoost_Enable=0
```

## Package installation
Arguments should be sources from `/etc/modprobe.d/modprobe.conf` or as parameters
```
modprobe corefreqk Experimental=1 TurboBoost_Enable=0
```

# P-State P0
* You still need the Cli to select a new frequency ratio, thus the following enhancement.

# My Todo list
* Make use of driver arguments `PState_FID` and `PState_VID`  to select Frequency and Voltage Core with AMD Families 17h ... 19h

_What do you think ?_

Jojonintendo, 2021-02-13:
Hi,
This is definitely working fine, I was looking in the wrong place and loading the module with the options works. I've looked into modinfo but couldn't find an option to pass directly a MAX frequency value or ratio, only for unlocking it. Would that be worth an enhancement too? Especially on CPUs that don't still accept PStates changes. Thanks again.

cyring, 2021-02-13:
> Hi,
> ... I've looked into modinfo but couldn't find an option to pass directly a MAX frequency value or ratio, only for unlocking it. Would that be worth an enhancement too? Especially on CPUs that don't still accept PStates changes. Thanks again.

This was indeed my proposal and I have started to work on it.  
The principle is that if it is available from Cli then it will be also possible from a driver _startup_ argument.  

However if no Max frequency function exists in _CoreFreq_ for some processors architecture, then such argument won't make any difference. At most, a warning message will be printed into the kernel log.

cyring, 2021-02-14:
The enhancement is available for testings.
Please pull the `develop` branch.

# Recommended operations
* Make sur to unload any previous instance of the driver.
* Uninstall any _CoreFreq_ package to avoid a mix of binaries
* Work from Source code
## Boot
* Kernel command line
```
idle=halt tsc=unstable
```
## Build
```
make DELAY_TSC=1 clean all
```
## Run
```
echo "corefreq" > /sys/devices/system/clocksource/clocksource0/current_clocksource

insmod corefreqk.ko Register_ClockSource=1 Experimental=1 TurboBoost_Enable=0 PState_FID=42
```
# Kernel Log
* Reading the following message in the kernel log ...
```
kernel: clocksource: Switched to clocksource corefreq
```
... means the Kernel clock is now sourced from _CoreFreq_ which uses the TSC.

cyring, 2021-02-19:
Hello,
Any comment from testing `PState_FID` ?
Does it fit your original request for enhancement ?
Thank you

cyring, 2021-02-28:
OK, feel free to come back with answers to my above request.

closed 2021-02-28 by cyring