cyring/CoreFreq #223: single print of a value

LouisCyfer, 2021-03-17:
Hi, I don't know if it's just me or my lack of experience in utilizing bash-scripts.

I really love the idea of this tool and need itto have any idea about my CPU temperature (lm-sensors doesn't work well for my AMD Ryzen 2700).

I figured the option -C prints out a sensor monitor continuously. is there a way to make that one single-print so it can be used like grepping a value from sensors?

cyring, 2021-03-17:
> Hi, I don't know if it's just me or my lack of experience in utilizing bash-scripts.
>
> I really love the idea of this tool and need itto have any idea about my CPU temperature (lm-sensors doesn't [...]

Hi  

It is made to be continuous but I'm adding your idea into the todo list.  

Meanwhile, you can use the JSON output, argument `-j`  
There are a lot of data but piping through a json parser, you could limit the output to the temperature values. Those are mainly per CPU for each core.

cyring, 2021-04-04:
Hello,

For your testings, the enhancement is available in the `develop` branch.
* Here is the new syntax:
```
corefreq-cli -h
CoreFreq.  Copyright (C) 2015-2021 CYRIL INGENIERIE

Usage:  corefreq-cli [-option <arguments>]

    Interface options
        -0,1,2  Memory unit in K,M,G Byte
        -F      Temperature in Fahrenheit
        -J  #   SMBIOS string index number
        -Y      Show Secret Data
        -t <v>  Show Top (default) with optional <v>iew:
                {       frequency, instructions, core, idle, package,
                        tasks, interrupts, sensors, voltage, power, slices  }
        -d      Show Dashboard

    Command options
        -C <#>  Monitor Sensors
        -V <#>  Monitor Voltage
        -W <#>  Monitor Power
        -g <#>  Monitor Package
        -c <#>  Monitor Counters
        -i <#>  Monitor Instructions
        -s      Print System Information
        -j      Print System Information (json-encoded)
        -M      Print Memory Controller
        -R      Print System Registers
        -m      Print Topology
        -u      Print CPUID
        -B      Print SMBIOS
        -k      Print Kernel
        -h      Print out this message
        -v      Print the version number

Exit status:
        0       SUCCESS         Successful execution
        3       CMD_SYNTAX      Command syntax error
        4       SHM_FILE        Shared memory file error
        5       SHM_MMAP        Shared memory mapping error
        6       PERM_ERR        Execution not permitted
        7       MEM_ERR         Memory operation error
        8       EXEC_ERR        General execution error
        9       SYS_CALL        System call error

Report bugs to labs[at]cyring.fr
```

* To make a single-print from the sensor monitor:
```
corefreq-cli -C 1
CPU Freq(MHz) VID  Vcore  TMP(C)    Accumulator       Energy(J)     Power(W)
000    4.84   103  0.9062   30  000000000000000349    0.005325317   0.005325317
001    4.35   103  0.9062   30  000000000000000201    0.003067017   0.003067017
002    4.06   101  0.9187   30  000000000000000185    0.002822876   0.002822876
003    4.39   101  0.9187   30  000000000000000178    0.002716064   0.002716064
...
031    3.80   101  0.9187   31  000000000000000000    0.000000000   0.000000000

              Package        Cores          Uncore         Memory
Energy(J):   16.452529907    0.051834106   16.002548218    0.000000000
Power(W) :   16.452529907    0.051834106   16.002548218    0.000000000
```

* Going further, you can also combine various iterations
```
corefreq-cli -C 3 -V 2 -W 5
```
>  which means 3 Sensors loops, 2 Voltages loops, 5 Power loops

Thanks for any comment.

LouisCyfer, 2021-04-04:
That is awesome, ty for the insight. This helps alot already!

olejon, 2021-04-04:
Have you checked `sensors` recently? The module `k10temp`, after some kernel updates, has added more and more, and now shows a lot of values for at least X570 and any Ryzen 3 CPU. A bit strange if it doesn't go for the 2XXX CPUs as well?

I get all this from my 3950X, ccds and everything:
```
k10temp-pci-00c3
Adapter: PCI adapter
Vcore:         1.48 V  
Vsoc:          1.08 V  
Tctl:         +35.4°C  
Tdie:         +35.4°C  
Tccd1:        +34.0°C  
Tccd2:        +37.5°C  
Icore:        14.00 A  
Isoc:          8.50 A
```

I don't discourage the use of CoreFreq of course. It's awesome 👍  But for simple grepping etc, just using the built-in kernel module is simpler.

NOTE: There's NO reason to run `# sensors-detect`. As long as `k10temp` loads at boot it just works, as with Coretemp for Intel.

cyring, 2021-04-05:
Closing the issue.
As a reminder for anyone reading this: don't run any `k10temp` or any other sensors driver in the **same time** than _CoreFreq_. This will overwhelm the SMU of Ryzen, EPYC, Threadripper processors which may freeze sooner or later.

closed 2021-04-05 by cyring