cyring/CoreFreq #224: calng errors

SitanshuNanavati, 2021-03-17:
With clang compiler,  following erros show up. 

~/CoreFreq $ make
make -j1 -C /kernel-5_10 M=/CoreFreq modules
make[1]: Entering directory '/kernel-5_10'
  CC [M]  /CoreFreq/corefreqk.o
clang-12: error: unknown argument: '-mno-fp-ret-in-387'
clang-12: error: unknown argument: '-mpreferred-stack-boundary=3'
clang-12: error: unknown argument: '-mskip-rax-setup'
clang-12: error: unknown argument: '-mindirect-branch=thunk-extern'
clang-12: error: unknown argument: '-mindirect-branch-register'
clang-12: error: unknown argument: '-fno-allow-store-data-races'
clang-12: error: optimization flag '-falign-jumps=1' is not supported [-Werror,-Wignored-optimization-argument]
clang-12: error: optimization flag '-falign-loops=1' is not supported [-Werror,-Wignored-optimization-argument]
$

clang version = 12.0.0

Any fix/workaround ?

Thanks.

cyring, 2021-03-17:
Hello,

clang is not able to build the Linux kernel driver.
Only gcc can do that.
No workaround, it's impossible, although Daemon and Cli can.

Why do you need that ?

Regards

cyring, 2021-04-03:
1. Compile the CoreFreq driver with `gcc`
```
make clean all CC='gcc'
```
2. Start immediately the driver
```
insmod corefreqk.ko
```
3.  Compile the Driver and the Client with `clang`
```
make clean corefreqd corefreq-cli CC='clang'
```
4. Start the Daemon
```
corefreqd
```
5. Start the Client
```
corefreq-cli
```

closed 2021-04-03 by cyring