cyring/CoreFreq #226: error on kernel 5.12?

DocMAX, 2021-03-24:
```
[root@htpc]: /home/docmax># cat /var/lib/dkms/corefreqk/1.84/build/make.log
DKMS make.log for corefreqk-1.84 for kernel 5.12.0-rc4-00007-g7acac4b3196c-dirty (x86_64)
Wed Mar 24 01:33:27 CET 2021
make: Entering directory '/usr/src/corefreqk-1.84'
cc  -Wall corefreqd.c corefreqm.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0 \
  -o corefreqd -lpthread -lm -lrt
cc  -Wall \
  corefreq-cli.c corefreq-ui.c corefreq-cli-rsc.c \
  corefreq-cli-json.c corefreq-cli-extra.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0  \
  -o corefreq-cli -lm -lrt
make -j1 -C /usr/lib/modules/5.12.0-rc4-00007-g7acac4b3196c-dirty/build M=/var/lib/dkms/corefreqk/1.84/build modules
make[1]: Entering directory '/usr/src/corefreqk-1.84'
make[1]: warning: -j1 forced in submake: resetting jobserver mode.
  CC [M]  /var/lib/dkms/corefreqk/1.84/build/corefreqk.o
/var/lib/dkms/corefreqk/1.84/build/corefreqk.c:52:26: error: expected declaration specifiers or '...' before string constant
   52 | MODULE_SUPPORTED_DEVICE ("Intel,AMD,HYGON");
      |                          ^~~~~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:271: /var/lib/dkms/corefreqk/1.84/build/corefreqk.o] Error 1
make[1]: *** [Makefile:1851: /var/lib/dkms/corefreqk/1.84/build] Error 2
make[1]: Leaving directory '/mnt/t420/root/opt/sources/linux/linux_htpc'
make: *** [Makefile:70: all] Error 2
make: Leaving directory '/usr/src/corefreqk-1.84'
```

cyring, 2021-03-24:
Thks.

That `MODULE_SUPPORTED_DEVICE` macro clearly disappears between kernel rc3 and rc4

https://elixir.bootlin.com/linux/v5.12-rc3/source/include/linux/module.h#L34

Sounds like I will have a fun time to fix various impacts introduced by `5.12`

cyring, 2021-03-24:
I would like to tell to our friends:
_Yo guys, when you change your mind, just make it a placebo_

Changes at https://github.com/gregkh/linux/commit/6417f03132a6952cd17ddd8eaddbac92b61b17e0

cyring, 2021-04-03:
In the `develop` branch you are now getting this fix f6a350bafae6cc67ac9a5ec636f126a01c7df0f7 .
Please let me know how it runs ?

aqxa1, 2021-04-05:
I tested the branch and it appears to fix the issue on 5.12.0-rc6, for me.

cyring, 2021-04-05:
Closing the issue; feel free to come back here if other 5.12 changes impact _CoreFreq_

closed 2021-04-05 by cyring