cyring/CoreFreq #227: `modprobe corefreqk` fails with system stall, Segmentation fault and trace in kernel log/ `dmesg`.

felixbecker2, 2021-04-05:
## Problem description

Trying to insert the module `corefreqk` fails with sime call traces in the kernel log, a few seconds system stall and a subsequent segmentation fault of `modprobe`:

`modprobe corefreqk`:  
\[few seconds system stall\]  
```
Segmentation fault
```

Not reproducible without reboots since subsequent invocations of `modprobe corefreqk` just seem to not terminate without system stall and without other noticeable effects.

Relevant parts in `dmesg`:  
```
[  996.724383] CoreFreq(1:-1): Processor [ 06_7A] Architecture [Atom/Gemini Lake] CPU [4/4]
[  996.724445] general protection fault: 0000 [#1] PREEMPT SMP PTI
[  996.724454] CPU: 1 PID: 8546 Comm: modprobe Tainted: G           OE     5.11.0-pf6-longcmdline-custom #1
[  996.724459] Hardware name: GPD MicroPC/Default string, BIOS 4.10 06/14/2019
[  996.724462] RIP: 0010:Start_Haswell_ULT+0x2bd/0x330 [corefreqk]
[  996.724498] Code: 48 09 d0 49 89 c0 4c 89 16 4c 89 5e 08 4c 89 66 10 4c 89 6e 20 4c 89 76 28 4c 89 7e 30 4c 89 4e 38 4c 89 46 40 b9 95 03 00 00 <0f> 32 48 8b 35 f2 c6 01 00 48 c1 e2 20 48 8b 0e 89 c0 48 09 c2 48
[  996.724502] RSP: 0018:ffffb9aa0bab7a58 EFLAGS: 00010046
[  996.724507] RAX: 0000000000000000 RBX: ffff9d4362dfd000 RCX: 0000000000000395
[  996.724510] RDX: 0000000000000000 RSI: ffff9d4302d38000 RDI: ffff9d430eceb780
[  996.724512] RBP: 0000000000000001 R08: 0000000000000000 R09: 0000000000000000
[  996.724514] R10: 000000f1541487d4 R11: 00000004c49f0766 R12: 0000000000000000
[  996.724517] R13: 0000006e02ffb1ef R14: 0000000000000000 R15: 0000000000000000
[  996.724519] FS:  00007fd40411e740(0000) GS:ffff9d4476c80000(0000) knlGS:0000000000000000
[  996.724522] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  996.724525] CR2: 00007fd403d28000 CR3: 000000018d502000 CR4: 0000000000350ee0
[  996.724528] Call Trace:
[  996.724535]  smp_call_function_single+0x189/0x1d0
[  996.724545]  ? printk+0x64/0x7b
[  996.724552]  Controller_Start+0xe3/0x110 [corefreqk]
[  996.724573]  CoreFreqK_Ignition_Level_Up.cold+0x9c/0x1d8 [corefreqk]
[  996.724594]  ? 0xffffffffc0a3b000
[  996.724598]  CoreFreqK_StartUp+0xcf/0x12d [corefreqk]
[  996.724617]  ? CoreFreqK_Alloc_Public_Level_Down+0x20/0x20 [corefreqk]
[  996.724637]  ? Query_Features+0x80/0x80 [corefreqk]
[  996.724656]  ? CoreFreqK_Make_Device_Level_Up+0x40/0x40 [corefreqk]
[  996.724674]  ? CoreFreqK_Create_Device_Level_Up+0x60/0x60 [corefreqk]
[  996.724692]  ? CoreFreqK_ProbePCI+0x170/0x170 [corefreqk]
[  996.724709]  ? CoreFreqK_Alloc_Features_Level_Up+0x80/0x80 [corefreqk]
[  996.724726]  ? CoreFreqK_Alloc_Public_Level_Up+0x70/0x70 [corefreqk]
[  996.724744]  ? CoreFreqK_Alloc_Private_Level_Up+0x40/0x40 [corefreqk]
[  996.724761]  ? CoreFreqK_Alloc_Processor_RO_Level_Up+0xc0/0xc0 [corefreqk]
[  996.724778]  ? Compute_Interval+0xa0/0xa0 [corefreqk]
[  996.724796]  ? CoreFreqK_Alloc_Private_Cache_Level_Up+0x40/0x40 [corefreqk]
[  996.724813]  ? CoreFreqK_Alloc_Per_CPU_Level_Up+0x190/0x190 [corefreqk]
[  996.724831]  ? CoreFreqK_Alloc_Processor_RW_Level_Up+0xc0/0xc0 [corefreqk]
[  996.724848]  ? SMBIOS_Collect+0x290/0x290 [corefreqk]
[  996.724867]  ? Controller_Stop+0x110/0x110 [corefreqk]
[  996.724886]  CoreFreqK_Init+0x7/0x1000 [corefreqk]
[  996.724904]  do_one_initcall+0x5b/0x190
[  996.724910]  do_init_module+0x51/0x230
[  996.724916]  load_module+0x27dc/0x2980
[  996.724920]  ? m_show+0x1d0/0x1d0
[  996.724926]  __do_sys_init_module+0x17d/0x1c0
[  996.724931]  do_syscall_64+0x33/0x80
[  996.724935]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[  996.724941] RIP: 0033:0x7fd40424ad9e
[  996.724945] Code: 48 8b 0d d5 10 0c 00 f7 d8 64 89 01 48 83 c8 ff c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 49 89 ca b8 af 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d a2 10 0c 00 f7 d8 64 89 01 48
[  996.724949] RSP: 002b:00007ffc718cbb28 EFLAGS: 00000246 ORIG_RAX: 00000000000000af
[  996.724953] RAX: ffffffffffffffda RBX: 0000560ab2bf2bd0 RCX: 00007fd40424ad9e
[  996.724955] RDX: 0000560ab1a21270 RSI: 0000000000086668 RDI: 00007fd403be1010
[  996.724957] RBP: 00007fd403be1010 R08: 00007fd40472e000 R09: 0000000000000000
[  996.724960] R10: ffffffffffffffb8 R11: 0000000000000246 R12: 0000560ab1a21270
[  996.724962] R13: 0000000000000000 R14: 0000560ab2bf2cc0 R15: 0000560ab2bf2bd0
[  996.724965] Modules linked in: corefreqk(OE+) sch_fq_codel snd_hda_codec_hdmi snd_hda_codec_realtek snd_hda_codec_generic vmmon(OE) vmw_vmci vboxnetflt(OE) vboxnetadp(OE) vboxdrv(OE) v4l2loopback videodev mc usbip_host usbip_core snd_seq_dummy snd_seq_oss snd_seq_midi_event snd_seq snd_seq_device cuse fuse netatop(OE) intel_atomisp2_pm haxm(OE) sg binfmt_misc acpi_call(OE) snd_sof_pci snd_sof_intel_hda_common snd_sof_intel_hda snd_sof_intel_byt snd_sof_intel_ipc snd_sof snd_sof_xtensa_dsp intel_telemetry_pltdrv snd_soc_skl intel_punit_ipc intel_telemetry_core intel_powerclamp snd_soc_hdac_hda kvm_intel snd_hda_ext_core kvm snd_soc_sst_ipc snd_soc_sst_dsp snd_soc_acpi_intel_match snd_soc_acpi irqbypass snd_hda_intel snd_intel_dspcfg soundwire_intel soundwire_generic_allocation soundwire_cadence snd_hda_codec snd_hda_core snd_hwdep soundwire_bus snd_soc_core iwlmvm r8169 btusb btrtl snd_compress btbcm ac97_bus realtek snd_pcm_dmaengine btintel mac80211 bluetooth snd_pcm
[  996.725034]  spi_pxa2xx_platform mdio_devres nxp_nci_i2c snd_timer iwlwifi nxp_nci ucsi_acpi of_mdio nci snd fixed_phy typec_ucsi intel_ish_ipc cfg80211 libphy intel_ishtp nfc rfkill_gpio soundcore typec i2c_hid rfkill tpm_crb tpm_tis tpm_tis_core intel_hid tpm mac_hid soc_button_array sparse_keymap uas usb_storage usbhid pcspkr sdhci_pci cqhci sdhci serio_raw mmc_core xhci_pci xhci_pci_renesas xhci_hcd
[  996.725079] ---[ end trace bfe68545bb08f02a ]---
[... 71 lines cut ...]
parm:           R2H_Disable:Disable Race to Halt (short)
parm:           Clear_Events:Clear Thermal and Power Events (uint)
parm:           ThermalScope:[0:None; 1:SMT; 2:Core; 3:Package] (int)
parm:           VoltageScope:[0:None; 1:SMT; 2:Core; 3:Package] (int)
parm:           PowerScope:[0:None; 1:SMT; 2:Core; 3:Package] (int)
parm:           Register_CPU_Idle:Register the Kernel cpuidle driver (short)
parm:           Register_CPU_Freq:Register the Kernel cpufreq driver (short)
parm:           Register_Governor:Register the Kernel governor (short)
parm:           Register_ClockSource:Register Clock Source driver (short)
parm:           Idle_Route:[0:Default; 1:I/O; 2:HALT; 3:MWAIT] (short)
parm:           Mech_IBRS:Mitigation Mechanism IBRS (short)
parm:           Mech_STIBP:Mitigation Mechanism STIBP (short)
parm:           Mech_SSBD:Mitigation Mechanism SSBD (short)
parm:           Mech_IBPB:Mitigation Mechanism IBPB (short)
parm:           Mech_L1D_FLUSH:Mitigation Mechanism Cache L1D Flush (short)
```

## More hardware details

`dmidecode`:

see [&rarr;&nbsp;here](http://ix.io/2V6q).

`hwinfo`:  
*(too big to post here or upload to ix.io, thus not provided.)*

felixbecker2, 2021-04-05:
Addendum: `corefreqk` is loaded afterwards according to `lsmod`, and cannot be unloaded ("`Module corefreqk is in use`").

felixbecker2, 2021-04-05:
Addendum 2 (the edit functionality here on github does not work for me, so I have to add posts):

`corefreqd` start up afterwards without complaining;  
`corefreqd -d`:

```
CoreFreq Daemon 1.84.1  Copyright (C) 2015-2021 CYRIL INGENIERIE

  Processor [Intel(R) Celeron(R) N4100 CPU @ 1.10GHz]
  Architecture [Atom/Gemini Lake] 4/4 CPU Online.
  SleepInterval(1000), SysGate(2000), 2326 tasks

    CPU #000 @ 1094.41 MHz
    CPU #001 @ 1094.41 MHz
    CPU #002 @ 0.00 MHz
    CPU #003 @ 0.00 MHz
        NTFY || ....
    Thread [7f3482084640] Init CHILD 003
    Thread [7f3482084640] Init CYCLE 003
    Thread [7f3483086640] Init CHILD 001
    Thread [7f3482885640] Init CHILD 002
    Thread [7f3483086640] Init CYCLE 001
    Thread [7f3483887640] Init CHILD 000
    Thread [7f3483887640] Init CYCLE 000
    Thread [7f3482885640] Init CYCLE 002
```

`corefreq-cli` does not show any values different from zero:

```
 _  _  _  _  Processor [Intel(R) Celeron(R) N4100 CPU @ 1.10GHz    ]   4/4   CPU
| || || || | Architecture [Atom/Gemini Lake       ] Caches L1 Inst=32 Data=24 KB
|_||_||_||_| Base Clock ~ 000,000,000 Hz                   L2=4096   L3=0     KB
--- ! Relative frequency ---8-------11------------------------------------23--24
000~                                                                            
001                                                                             
002                                                                             
003                                                                             
--- Freq(MHz) Ratio - Turbo -- C0 ---- C1 -- C2:C3 - C4:C6 --- C7 -- Min TMP Max
000    0.00 ( 0.00)   0.00%   0.00%   0.00%   0.00%   0.00%   0.00%  44 /  0/ 44
001    0.00 ( 0.00)   0.00%   0.00%   0.00%   0.00%   0.00%   0.00%  0  /  0/  0
002    0.00 ( 0.00)   0.00%   0.00%   0.00%   0.00%   0.00%   0.00%  0  /  0/  0
003    0.00 ( 0.00)   0.00%   0.00%   0.00%   0.00%   0.00%   0.00%  0  /  0/  0
------ % Averages [   0.00%   0.00%   0.00%   0.00%   0.00%   0.00% ]-----------
Tech [TSC-INV,HTT,EIST,IDA,TURBO,C1E,PM4,C3A,C1A,C3U,C1U,TM,HOT] V[    ] T[  0C]
Linux [5.11.0]     Default string     Tasks [     0] Mem [        0/0        KB]
```

A subsequent invocation of `corefreq-cli` shows only empty fields (no zeros anymore), until I kill and restart `corefreqd` too:

```
             Processor [Intel(R) Celeron(R) N4100 CPU @ 1.10GHz    ]   4/4   CPU
 o  o  o  o  Architecture [Atom/Gemini Lake       ] Caches L1 Inst=32 Data=24 KB
 o  o  o  o  Base Clock ~ 000,000,000 Hz                   L2=4096   L3=0     KB
--- ! Relative frequency ---8-------11--12--------------------------------23--24
000                                                                             
001                                                                             
002                                                                             
003                                                                             
--- Freq(MHz) Ratio - Turbo -- C0 ---- C1 -- C2:C3 - C4:C6 --- C7 -- Min TMP Max
000                                                                             
001                                                                             
002                                                                             
003                                                                             
------ % Averages [                                                 ]-----------
Tech [TSC-INV,HTT,EIST,IDA,TURBO,C1E,PM4,C3A,C1A,C3U,C1U,TM,HOT] V[ .  ] T[   C]
Linux [5.11.0]     Default string     Tasks [     0] Mem [        0/0        KB]
```

cyring, 2021-04-05:
* Sorry for this issue. _First time I have a report from Gemini Lake_
* Bug has to be found on the Gemini Lake call flow starting at:
https://github.com/cyring/CoreFreq/blob/39953f71d2ed5cde1ec80c72c642f7f2c74d60ab/corefreqk.h#L6430
* Probably one/some registers are not available or wrongly used.
-------
* To debug, could you start the driver with the most generic architecture identifier:
https://github.com/cyring/CoreFreq/blob/39953f71d2ed5cde1ec80c72c642f7f2c74d60ab/corefreqk.h#L5311

1. Start the driver, directly from the build directory, like:
```
insmod corefreqk.ko ArchID=11
```
2. If no issue dump into kernel log, proceed with Daemon and Client

felixbecker2, 2021-04-05:
Hey,

*On Mon, 05 Apr 2021 03:58:47 -0700, CYRIL INGENIERIE ***@***.***> wrote about "Re: [cyring/CoreFreq] `modprobe corefreqk` fails with system stall, Segmentation fault and trace in kernel log/ `dmesg`. (#227)":*

> * To debug, could you start the driver with the most generic
> architecture identifier:
> https://github.com/cyring/CoreFreq/blob/39953f71d2ed5cde1ec80c72c642f7f2c74d60ab/corefreqk.h#L5311
>
> 1. Start the [...]

Thanks for that instructions!

Though, I cannot do it right now, because I will not reboot my machine for some time. I need it for productivity, and reboot (instead of suspend to disk) would impact that. And without reboot I cannot reload the module.

So this has to wait some time.

Regards!

-- 
I'll catch you if you fall.
                      - floor

cyring, 2021-04-05:
Looking at the `RCX` register `= 0x395` is the Uncore fixed performance counter which exists since SandyBridge but **not** for Gemini Lake.

https://github.com/cyring/CoreFreq/blob/39953f71d2ed5cde1ec80c72c642f7f2c74d60ab/intelmsr.h#L113

Back tracing the call flow, MSR error may happen at this line:

https://github.com/cyring/CoreFreq/blob/39953f71d2ed5cde1ec80c72c642f7f2c74d60ab/corefreqk.c#L9357

That faulty MSR call has to be commented in `corefreqk.c`
```c
#define PKG_Counters_Haswell_ULT(Core, T)				\
({									\
    RDTSCP_COUNTERx7(PUBLIC(RO(Proc))->Counter[T].PTSC,			\
		MSR_PKG_C2_RESIDENCY, PUBLIC(RO(Proc))->Counter[T].PC02,\
		MSR_PKG_C3_RESIDENCY, PUBLIC(RO(Proc))->Counter[T].PC03,\
		MSR_PKG_C6_RESIDENCY, PUBLIC(RO(Proc))->Counter[T].PC06,\
		MSR_PKG_C7_RESIDENCY, PUBLIC(RO(Proc))->Counter[T].PC07,\
		MSR_PKG_C8_RESIDENCY, PUBLIC(RO(Proc))->Counter[T].PC08,\
		MSR_PKG_C9_RESIDENCY, PUBLIC(RO(Proc))->Counter[T].PC09,\
		MSR_PKG_C10_RESIDENCY,PUBLIC(RO(Proc))->Counter[T].PC10);\
									\
/*	RDCOUNTER	(PUBLIC(RO(Proc))->Counter[T].Uncore.FC0,	\
			MSR_SNB_UNCORE_PERF_FIXED_CTR0 );*/		\
})
```
* Once above change applied, fully rebuild/force-unload/reload the driver:
```
rmmod -f corefreqk
make clean all
insmod corefreqk.ko
```
* Oftenly, a wrong register call may leave Processor in a bad state. Force unloading the driver won't help and system has to be rebooted

_Sure, whenever you can and w/o Production running._

felixbecker2, 2021-04-05:
First, for safety reasons I _strongly_ recommend that the module fails to load on non-tested architectures unless some force-load module parameter is given.

On Mon, 05 Apr 2021 07:08:59 -0700, CYRIL INGENIERIE ***@***.***> wrote about "Re: [cyring/CoreFreq] `modprobe corefreqk` fails with system stall, Segmentation fault and trace in kernel log/ `dmesg`. (#227)":

> That faulty MSR call has to be commented in `corefreqk.c`
> ```c
> #define PKG_Counters_Haswell_ULT(Core, T)				\
> ({									\
>     RDTSCP_COUNTERx7(PUBLIC(RO(Proc))->Counter[T].PTSC,			\ [...]

Attached a .patch which applies that changes,

> * Once above change applied,

'make clean`  
`make`  
`insmod corefreqk.ko` (on a freshly rebooted machine):

Works.

In `dmesg`, there is just  
```
[  664.631402] CoreFreq(2:-1): Processor [ 06_7A] Architecture [Atom/Gemini Lake] CPU [4/4]
```

`./corefreqd -d`:  
```
CoreFreq Daemon 1.84.1  Copyright (C) 2015-2021 CYRIL INGENIERIE

  Processor [Intel(R) Celeron(R) N4100 CPU @ 1.10GHz]
  Architecture [Atom/Gemini Lake] 4/4 CPU Online.
  SleepInterval(1000), SysGate(2000), 2326 tasks

    CPU #000 @ 1094.40 MHz
    CPU #001 @ 1094.40 MHz
    CPU #002 @ 1094.40 MHz
    CPU #003 @ 1094.40 MHz
        NTFY || ....
    Thread [7fd59636d640] Init CHILD 003
    Thread [7fd59736f640] Init CYCLE 001
    Thread [7fd596b6e640] Init CHILD 002
    Thread [7fd596b6e640] Init CYCLE 002
    Thread [7fd59736f640] Init CHILD 001
    Thread [7fd597b70640] Init CYCLE 000
    Thread [7fd59636d640] Init CYCLE 003
    Thread [7fd597b70640] Init CHILD 000
```

`./corefreq-cli` seems to show useful stuff; I haven't looked into everything; voltage and power readings for example stay empty.

-- 
Machen ist wie wollen, nur krasser.

felixbecker2, 2021-04-05:
On Mon, 05 Apr 2021 03:58:47 -0700, CYRIL INGENIERIE ***@***.***> wrote about "Re: [cyring/CoreFreq] `modprobe corefreqk` fails with system stall, Segmentation fault and trace in kernel log/ `dmesg`. (#227)":

> * To debug, could you start the driver with the most generic
> architecture identifier:
> https://github.com/cyring/CoreFreq/blob/39953f71d2ed5cde1ec80c72c642f7f2c74d60ab/corefreqk.h#L5311
> 
> 1. Start the driver, directly from the build directory, like:
> ```
> insmod corefreqk.ko ArchID=11
> ```

`cd /usr/src/corefreqk-1.84+r1306.20210228.39953f7`  
`make`  
`insmod corefreqk.ko ArchID=11`:  
```
Segmentation fault
```

In kernel log, `dmesg`:  
```
[  773.752332] general protection fault: 0000 [#1] PREEMPT SMP PTI
[  773.752343] CPU: 2 PID: 5819 Comm: insmod Tainted: G           OE     5.11.0-pf6-longcmdline-custom #1
[  773.752348] Hardware name: GPD MicroPC/Default string, BIOS 4.10 06/14/2019
[  773.752350] RIP: 0010:BaseClock_Core+0x8/0x80 [corefreqk]
[  773.752372] Code: 00 00 48 89 f8 48 c7 07 64 00 00 00 48 c7 47 08 00 00 00 00 48 c7 47 10 00 e1 f5 05 c3 0f 1f 44 00 00 49 89 f8 b9 cd 00 00 00 <0f> 32 83 e0 07 3c 01 89 f6 74 3f 3c 03 75 20 48 69 f6 0a 1a 00 00
[  773.752375] RSP: 0018:ffff9fc1025a7a88 EFLAGS: 00010286
[  773.752380] RAX: ffffffffc0cfb130 RBX: ffff9fc1025a7bb8 RCX: 00000000000000cd
[  773.752383] RDX: 000000000000000b RSI: 000000000000000b RDI: ffff9fc1025a7aa0
[  773.752385] RBP: ffff9fc1025a7b40 R08: ffff9fc1025a7aa0 R09: 0000000000000000
[  773.752387] R10: 0000000000000000 R11: 0000000000000001 R12: 0000000000000004
[  773.752389] R13: 000000000000000b R14: 0000000000000000 R15: 0000000000000000
[  773.752391] FS:  00007fdc3e402740(0000) GS:ffff9332f6d00000(0000) knlGS:0000000000000000
[  773.752394] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  773.752397] CR2: 00007fdc3e5c4730 CR3: 000000016cdc8000 CR4: 0000000000350ee0
[  773.752400] Call Trace:
[  773.752405]  Controller_Init+0x143/0x490 [corefreqk]
[  773.752429]  CoreFreqK_Ignition_Level_Up+0x1e8/0x7d7 [corefreqk]
[  773.752449]  ? 0xffffffffc0b69000
[  773.752453]  CoreFreqK_StartUp+0xcf/0x12d [corefreqk]
[  773.752472]  ? CoreFreqK_Alloc_Public_Level_Down+0x20/0x20 [corefreqk]
[  773.752491]  ? Query_Features+0x80/0x80 [corefreqk]
[  773.752510]  ? CoreFreqK_Make_Device_Level_Up+0x40/0x40 [corefreqk]
[  773.752528]  ? CoreFreqK_Create_Device_Level_Up+0x60/0x60 [corefreqk]
[  773.752545]  ? CoreFreqK_ProbePCI+0x170/0x170 [corefreqk]
[  773.752562]  ? CoreFreqK_Alloc_Features_Level_Up+0x80/0x80 [corefreqk]
[  773.752579]  ? CoreFreqK_Alloc_Public_Level_Up+0x70/0x70 [corefreqk]
[  773.752596]  ? CoreFreqK_Alloc_Private_Level_Up+0x40/0x40 [corefreqk]
[  773.752613]  ? CoreFreqK_Alloc_Processor_RO_Level_Up+0xc0/0xc0 [corefreqk]
[  773.752629]  ? Compute_Interval+0xa0/0xa0 [corefreqk]
[  773.752647]  ? CoreFreqK_Alloc_Private_Cache_Level_Up+0x40/0x40 [corefreqk]
[  773.752664]  ? CoreFreqK_Alloc_Per_CPU_Level_Up+0x190/0x190 [corefreqk]
[  773.752681]  ? CoreFreqK_Alloc_Processor_RW_Level_Up+0xc0/0xc0 [corefreqk]
[  773.752698]  ? SMBIOS_Collect+0x290/0x290 [corefreqk]
[  773.752717]  ? Controller_Stop+0x110/0x110 [corefreqk]
[  773.752736]  CoreFreqK_Init+0x7/0x1000 [corefreqk]
[  773.752754]  do_one_initcall+0x5b/0x190
[  773.752763]  do_init_module+0x51/0x230
[  773.752769]  load_module+0x27dc/0x2980
[  773.752774]  ? m_show+0x1d0/0x1d0
[  773.752779]  __do_sys_finit_module+0xb7/0x120
[  773.752785]  do_syscall_64+0x33/0x80
[  773.752791]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[  773.752797] RIP: 0033:0x7fdc3e528a9d
[  773.752800] Code: bb 0c 00 0f 05 eb a9 66 0f 1f 44 00 00 f3 0f 1e fa 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d a3 73 0c 00 f7 d8 64 89 01 48
[  773.752803] RSP: 002b:00007ffd615a6df8 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[  773.752807] RAX: ffffffffffffffda RBX: 0000564254bf6830 RCX: 00007fdc3e528a9d
[  773.752810] RDX: 0000000000000000 RSI: 0000564254bf62a0 RDI: 0000000000000003
[  773.752812] RBP: 0000000000000000 R08: 0000000000000000 R09: 00007ffd615a6fb8
[  773.752814] R10: 0000000000000003 R11: 0000000000000246 R12: 0000564254bf62a0
[  773.752816] R13: 0000000000000000 R14: 0000564254bf67e0 R15: 0000564254bf62a0
[... 71 lines cut ...]
[  773.757080] RIP: 0033:0x7fdc3e528a9d
[  773.757083] Code: bb 0c 00 0f 05 eb a9 66 0f 1f 44 00 00 f3 0f 1e fa 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d a3 73 0c 00 f7 d8 64 89 01 48
[  773.757086] RSP: 002b:00007ffd615a6df8 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[  773.757089] RAX: ffffffffffffffda RBX: 0000564254bf6830 RCX: 00007fdc3e528a9d
[  773.757091] RDX: 0000000000000000 RSI: 0000564254bf62a0 RDI: 0000000000000003
[  773.757094] RBP: 0000000000000000 R08: 0000000000000000 R09: 00007ffd615a6fb8
[  773.757096] R10: 0000000000000003 R11: 0000000000000246 R12: 0000564254bf62a0
[  773.757098] R13: 0000000000000000 R14: 0000564254bf67e0 R15: 0000564254bf62a0
[  773.757101] ---[ end trace f59a783be914e221 ]---
[  775.054456] RIP: 0010:BaseClock_Core+0x8/0x80 [corefreqk]
[  775.054500] Code: 00 00 48 89 f8 48 c7 07 64 00 00 00 48 c7 47 08 00 00 00 00 48 c7 47 10 00 e1 f5 05 c3 0f 1f 44 00 00 49 89 f8 b9 cd 00 00 00 <0f> 32 83 e0 07 3c 01 89 f6 74 3f 3c 03 75 20 48 69 f6 0a 1a 00 00
[  775.054504] RSP: 0018:ffff9fc1025a7a88 EFLAGS: 00010286
[... cut ...]
[  775.054895] RBP: ffff9fc1025a7b40 R08: ffff9fc1025a7aa0 R09: 0000000000000000
[  775.054899] R10: 0000000000000000 R11: 0000000000000001 R12: 0000000000000004
[  775.054901] R13: 000000000000000b R14: 0000000000000000 R15: 0000000000000000
[  775.054903] FS:  00007fdc3e402740(0000) GS:ffff9332f6d80000(0000) knlGS:0000000000000000
[  775.054906] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  775.054908] CR2: 00007ffe6a188c80 CR3: 000000016cdc8000 CR4:
0000000000350ee0
```

-- 
Machen ist wie wollen, nur krasser.

cyring, 2021-04-05:
According to the Intel SDM:
[image]

So I'm now routing all Gemini Lake entries to Silvermont.

Can you please try the new version `1.84.4` of _CoreFreq_ in the `develop` [branch](https://github.com/cyring/CoreFreq/tree/develop) ?

felixbecker2, 2021-04-08:
On Mon, 05 Apr 2021 14:00:12 -0700, CYRIL INGENIERIE
***@***.***> wrote about "Re: [cyring/CoreFreq] `modprobe
corefreqk` fails with system stall, Segmentation fault and trace in
kernel log/ `dmesg`. (#227)":

> Can you please try the new version `1.84.4` of _CoreFreq_ in the
> `develop` [branch](https://github.com/cyring/CoreFreq/tree/develop) ?

Can you tell me how I get this branch? I don't know git more than `git
clone` and `git pull`.

-- 
ich warte gerne [...] Wenn ich warten muss, da hab' ich ja viel Zeit.

(Quelle: Peter Lustig, http://youtube.com/watch?v=5ippbVCKbjQ&t=113)

cyring, 2021-04-09:
* [Github](https://github.com/git-guides/git-clone): delete any previous downloaded directories of _CoreFreq_ , next clone again from the development branch
```
git clone [url] --branch [branch] --single-branch
```
where `[url]` is replaced by `https://github.com/cyring/CoreFreq` and `[branch]` is `develop`

------
_(Smartphone edit)_

felixbecker2, 2021-04-09:
On Mon, 05 Apr 2021 14:00:12 -0700, CYRIL INGENIERIE ***@***.***> wrote about "Re: [cyring/CoreFreq] `modprobe corefreqk` fails with system stall, Segmentation fault and trace in kernel log/ `dmesg`. (#227)":

> Can you please try the new version `1.84.4` of _CoreFreq_ in the `develop` [branch](https://github.com/cyring/CoreFreq/tree/develop) ?

`git clone https://github.com/cyring/CoreFreq --branch develop`
`cd CoreFreq`
`make`
`insmod ./corefreqk.ko`
```
Segmentation Fault
```

In `dmesg`:
```
[ 7785.073412] general protection fault: 0000 [#1] PREEMPT SMP PTI
[ 7785.073423] CPU: 0 PID: 5451 Comm: insmod Tainted: G           OE     5.11.0-pf6-longcmdline-custom #1
[ 7785.073428] Hardware name: GPD MicroPC/Default string, BIOS 4.10 06/14/2019
[ 7785.073430] RIP: 0010:BaseClock_Silvermont+0x8/0x60 [corefreqk]
[ 7785.073452] Code: af f0 49 89 10 49 89 48 10 4c 89 c0 49 89 70 08 c3 b9 b4 90 f7 04 b8 05 0d 00 00 ba 53 00 00 00 eb da 49 89 f8 b9 cd 00 00 00 <0f> 32 83 e0 07 3c 04 77 3b 0f b6 c0 48 8b 14 c5 80 3d c2 c0 48 8b
[ 7785.073456] RSP: 0018:ffff9f810af0fa88 EFLAGS: 00010282
[ 7785.073460] RAX: ffffffffc0bf32e0 RBX: ffff9f810af0fbb8 RCX: 00000000000000cd
[ 7785.073463] RDX: 0000000000000037 RSI: 000000000000000b RDI: ffff9f810af0faa0
[ 7785.073465] RBP: ffff9f810af0fb40 R08: ffff9f810af0faa0 R09: 0000000000000000
[ 7785.073467] R10: 0000000000000000 R11: 0000000000000001 R12: 0000000000000004
[ 7785.073469] R13: 000000000000000b R14: ffff8f4cf2a4a000 R15: 0000000000000000
[ 7785.073472] FS:  00007ffb88ba3740(0000) GS:ffff8f4cf6c00000(0000) knlGS:0000000000000000
[ 7785.073475] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 7785.073477] CR2: 00007ffb88d65730 CR3: 0000000269e50000 CR4: 0000000000350ef0
[ 7785.073480] Call Trace:
[ 7785.073485]  Controller_Init+0x143/0x490 [corefreqk]
[ 7785.073508]  CoreFreqK_Ignition_Level_Up+0x1e8/0x7d7 [corefreqk]
[ 7785.073529]  ? 0xffffffffc0c45000
[ 7785.073532]  CoreFreqK_StartUp+0xcf/0x12d [corefreqk]
[ 7785.073551]  ? CoreFreqK_Alloc_Public_Level_Down+0x20/0x20 [corefreqk]
[ 7785.073569]  ? Query_Features+0x80/0x80 [corefreqk]
[ 7785.073588]  ? CoreFreqK_Make_Device_Level_Up+0x40/0x40 [corefreqk]
[ 7785.073606]  ? CoreFreqK_Create_Device_Level_Up+0x60/0x60 [corefreqk]
[ 7785.073623]  ? CoreFreqK_ProbePCI+0x170/0x170 [corefreqk]
[ 7785.073640]  ? CoreFreqK_Alloc_Features_Level_Up+0x80/0x80 [corefreqk]
[ 7785.073656]  ? CoreFreqK_Alloc_Public_Level_Up+0x70/0x70 [corefreqk]
[ 7785.073673]  ? CoreFreqK_Alloc_Private_Level_Up+0x40/0x40 [corefreqk]
[ 7785.073690]  ? CoreFreqK_Alloc_Processor_RO_Level_Up+0xc0/0xc0 [corefreqk]
[ 7785.073707]  ? Compute_Interval+0xa0/0xa0 [corefreqk]
[ 7785.073725]  ? CoreFreqK_Alloc_Private_Cache_Level_Up+0x40/0x40 [corefreqk]
[ 7785.073742]  ? CoreFreqK_Alloc_Per_CPU_Level_Up+0x190/0x190 [corefreqk]
[ 7785.073759]  ? CoreFreqK_Alloc_Processor_RW_Level_Up+0xc0/0xc0 [corefreqk]
[ 7785.073775]  ? SMBIOS_Collect+0x290/0x290 [corefreqk]
[ 7785.073794]  ? Controller_Stop+0x110/0x110 [corefreqk]
[ 7785.073812]  CoreFreqK_Init+0x7/0x1000 [corefreqk]
[ 7785.073832]  do_one_initcall+0x5b/0x190
[ 7785.073840]  do_init_module+0x51/0x230
[ 7785.073847]  load_module+0x27dc/0x2980
[ 7785.073851]  ? m_show+0x1d0/0x1d0
[ 7785.073857]  __do_sys_finit_module+0xb7/0x120
[ 7785.073862]  do_syscall_64+0x33/0x80
[ 7785.073868]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[ 7785.073875] RIP: 0033:0x7ffb88cc9a9d
[ 7785.073879] Code: bb 0c 00 0f 05 eb a9 66 0f 1f 44 00 00 f3 0f 1e fa 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d a3 73 0c 00 f7 d8 64 89 01 48
[ 7785.073882] RSP: 002b:00007fffca7f4b78 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[ 7785.073886] RAX: ffffffffffffffda RBX: 0000564625dc77c0 RCX: 00007ffb88cc9a9d
[ 7785.073889] RDX: 0000000000000000 RSI: 0000564624eb7270 RDI: 0000000000000003
[ 7785.073891] RBP: 0000000000000000 R08: 0000000000000000 R09: 00007ffb88d95380
[ 7785.073893] R10: 0000000000000003 R11: 0000000000000246 R12: 0000564624eb7270
[ 7785.073895] R13: 0000000000000000 R14: 0000564625dca1f0 R15: 0000000000000000
[ 7785.073898] Modules linked in: corefreqk(OE+) r8153_ecm cdc_ether usbnet r8152 mii sch_fq_codel vmmon(OE) vmw_vmci vboxnetflt(OE) vboxnetadp(OE) vboxdrv(OE) v4l2loopback videodev mc usbip_host usbip_core snd_seq_dummy snd_seq_oss snd_seq_midi_event snd_seq snd_seq_device cuse fuse netatop(OE) intel_atomisp2_pm haxm(OE) sg binfmt_misc acpi_call(OE) snd_hda_codec_hdmi snd_sof_pci snd_sof_intel_hda_common snd_hda_codec_realtek snd_sof_intel_hda snd_sof_intel_byt snd_hda_codec_generic snd_sof_intel_ipc intel_telemetry_pltdrv snd_sof intel_punit_ipc intel_telemetry_core intel_powerclamp snd_sof_xtensa_dsp snd_soc_skl kvm_intel snd_soc_hdac_hda snd_hda_ext_core snd_soc_sst_ipc kvm snd_soc_sst_dsp snd_soc_acpi_intel_match snd_soc_acpi snd_hda_intel irqbypass snd_intel_dspcfg soundwire_intel soundwire_generic_allocation soundwire_cadence snd_hda_codec snd_hda_core snd_hwdep soundwire_bus snd_soc_core btusb btrtl btbcm snd_compress btintel ac97_bus snd_pcm_dmaengine iwlmvm r8169
[ 7785.073969]  bluetooth realtek snd_pcm mdio_devres mac80211 snd_timer iwlwifi spi_pxa2xx_platform of_mdio ucsi_acpi nxp_nci_i2c fixed_phy typec_ucsi snd nxp_nci cfg80211 libphy soundcore typec intel_ish_ipc nci intel_ishtp tpm_crb rfkill_gpio nfc rfkill i2c_hid tpm_tis mac_hid tpm_tis_core intel_hid soc_button_array tpm sparse_keymap uas usb_storage usbhid pcspkr sdhci_pci cqhci  [... cut ...]
[ 7786.752381] CR2: 00007fa7ee6cefd8 CR3: 0000000269e50000 CR4:
0000000000350ee0
```

`./corefreq-cli -v` and `./corefreqd -v`: `1.84.4`.

`rmmod -f corefreqk`:  
```
rmmod: ERROR: could not remove 'corefreqk': Device or resource busy
rmmod: ERROR: could not remove module corefreqk: Device or resource busy
```

(Forced module unloading I have compiled into the kernel:  
`zgrep MODULE_FORCE_UNLOAD /proc/config.gz`:  
```
CONFIG_MODULE_FORCE_UNLOAD=y
```
)

-- 
Planung ist der Ersatz des Zufalls durch den Irrtum.

felixbecker2, 2021-04-09:
.. for safety and security reasons, **Please make the module refusing to load on untested architectures, until some overriding module parameter is provided!**

-- 
ich warte gerne [...] Wenn ich warten muss, da hab' ich ja viel Zeit.

(Quelle: Peter Lustig, http://youtube.com/watch?v=5ippbVCKbjQ&t=113)

cyring, 2021-04-10:
`MSR_FSB_FREQ (0xcd)` is the error.
GDM Plus: only registers from tables 2.6, 2.12, 2.13 and probably other IA and compatible MSR.
Tremont may also crashed.
I need to access and develop directly on these architectures.

cyring, 2021-04-10:
Hello,
I'm refactoring Goldmont and subsequent architectures (Gemini Lake)
Can you please get the latest commit from `develop` and try again ?
```
git pull

make clean all
insmod corefreqk.ko
## if no error happened then proceed with daemon and client
./corefreqd -d
./corefreq-cli -s -m -R -c 1 -i 1 -g 1 -C 1
```

felixbecker2, 2021-04-11:
Sorry, I do have no capacities anymore to do frequent testing & communication.

If you have done major work, than I can test for you.

Regards!

On Sat, 10 Apr 2021 15:49:44 -0700, CYRIL INGENIERIE
***@***.***> wrote about "Re: [cyring/CoreFreq] `modprobe
corefreqk` fails with system stall, Segmentation fault and trace in
kernel log/ `dmesg`. (#227)":

> Hello,
> I'm refactoring Goldmont and subsequent architectures (Gemini Lake)
> Can you please get the latest commit from `develop` and try again ?
> ```
> git pull
>
> make clean all
> insmod corefreqk.ko
> ## if [...]

-- 
Wählt man eine der Antworten zu dieser Frage zufällig aus, wie hoch ist
die Wahrscheinlichkeit, dass sie korrekt ist?

a) 1/3
b) 1/3
c) sonstige, und zwar: ____

cyring, 2021-04-11:
> Sorry, I do have no capacities anymore to do frequent testing & communication. If you have done major work, than I can test for you. Regards! 

It's the purpose of the latest refactoring.

felixbecker2, 2021-04-17:
On Sat, 10 Apr 2021 15:49:44 -0700, CYRIL INGENIERIE ***@***.***> wrote about "Re: [cyring/CoreFreq] `modprobe corefreqk` fails with system stall, Segmentation fault and trace in kernel log/ `dmesg`. (#227)":

> Can you please get the latest commit from `develop` and try again ?
> `insmod corefreqk.ko`

`dmesg`:  
```
CoreFreq(3:-1): Processor [ 06_7A] Architecture [Atom/Gemini Lake] CPU [4/4]
```

> ## if no error happened then proceed with daemon and client
> `./corefreqd -d`

```
CoreFreq Daemon 1.84.5  Copyright (C) 2015-2021 CYRIL INGENIERIE

  Processor [Intel(R) Celeron(R) N4100 CPU @ 1.10GHz]
  Architecture [Atom/Gemini Lake] 4/4 CPU Online.
  SleepInterval(1000), SysGate(2000), 2326 tasks

    CPU #000 @ 1094.40 MHz
    CPU #001 @ 1094.40 MHz
    CPU #002 @ 1094.40 MHz
    CPU #003 @ 1094.40 MHz
        NTFY || ....
    Thread [7efda264c640] Init CYCLE 003
    Thread [7efda264c640] Init CHILD 003
    Thread [7efda2e4d640] Init CHILD 002
    Thread [7efda2e4d640] Init CYCLE 002
    Thread [7efda3e4f640] Init CHILD 000
    Thread [7efda3e4f640] Init CYCLE 000
    Thread [7efda364e640] Init CYCLE 001
    Thread [7efda364e640] Init CHILD 001

```

> `./corefreq-cli -s -m -R -c 1 -i 1 -g 1 -C 1`

```
Processor                              [Intel(R) Celeron(R) N4100 CPU @ 1.10GHz]
|- Architecture                                               [Atom/Gemini Lake]
|- Vendor ID                                                      [GenuineIntel]
|- Microcode                                                        [0x00000022]
|- Signature                                                           [  06_7A]
|- Stepping                                                            [      1]
|- Online CPU                                                          [  4/  4]
|- Base Clock                                                          [ 99.492]
|- Frequency            (MHz)                      Ratio                        
                 Min    795.94                    <   8 >                       
                 Max   1094.41                    <  11 >                       
|- Factory                                                             [100.000]
                       1100                       [  11 ]                       
|- Performance                                                                  
   |- P-State                                                                   
                 TGT    795.94                    <   8 >                       
|- Turbo Boost                                                         [   LOCK]
                  1C   2387.81                    <  24 >                       
                  2C   2288.32                    <  23 >                       
                  3C   2288.32                    <  23 >                       
                  4C   2288.32                    <  23 >                       
|- Uncore                                                              [   LOCK]
|- TDP                                                           Level [  0:0  ]
   |- Programmable                                                     [   LOCK]
   |- Configuration                                                    [   LOCK]
   |- Turbo Activation                                                 [   LOCK]
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [N]          AES [Y]  AVX/AVX2 [N/N] 
|- AVX512-F     [N]    AVX512-DQ [N]  AVX512-IFMA [N]   AVX512-PF [N] 
|- AVX512-ER    [N]    AVX512-CD [N]    AVX512-BW [N]   AVX512-VL [N] 
[... 210 lines cut ...]
000  111.06  6266  31.5750   54  000000000000000000    0.000000000   0.000000000
001   18.34  6266  31.5750   54  000000000000000000    0.000000000   0.000000000
002   69.38  6266  31.5750   54  000000000000000000    0.000000000   0.000000000
003   17.77  6266  31.5750   54  000000000000000000    0.000000000   0.000000000

              Package        Cores          Uncore         Memory
Energy(J):    0.002199463    0.000089172    0.000004578    0.000087830
Power(W) :    0.002199463    0.000089172    0.000004578    0.000087830
```

&rarr; no obvious problems.

-- 
Vernunft, der Erfahrung wie des Nachdenkens soll man nur mündlich
verhandeln. Das ausgesprochene Wort ist sogleich tot, wenn es nicht
durch ein folgendes, dem Hörer gemäßes, am Leben erhalten wird. Man
merke nur auf ein geselliges Gespräch! Gelangt das Wort nicht schon
tot zu dem Hörer, so ermordet er es alsogleich durch Widerspruch,
Bestimmen, Bedingen, Ablenken, Abspringen und wie die tausendfältigen
Unarten des Unterhaltens auch heißen mögen. Mit dem Geschriebenen ist
es noch schlimmer. Niemand mag lesen als das, woran er schon
einigermaßen gewöhnt ist; das Bekannte, das Gewohnte verlangt er
unter veränderter Form. Doch hat das Geschriebene den Vorteil, dass
es dauert und die Zeit abwarten kann, wo ihm zu wirken gegönnt ist.
                -- Goethe, Maximen und Reflektionen, Nr. 583

cyring, 2021-04-18:
Great! Apparently no more crash.
Other things could be improved but that will claim a lot of your time with testings.
Feel free to close the issue.

felixbecker2, 2021-04-24:
> Feel free to close the issue.

/close #227

-- 
Erleuchtung, die:
  Wenn Du plötzlich merkst, dass Du sie gar nicht brauchst.
  (Janosch.)

Enlightenment:
  When you suddenly realise that you do not need it.
  (Janosch.)

closed 2021-04-24 by cyring

cyring, 2021-05-09:
Hello,

If you have a chance to try the latest `develop` branch, please let me know if Geminy Lake is better supported  ?

Thank you